openwrt/openwrt #11481: simple-adblock: does not work with dnsmasq.nftset

mistepien, 2022-12-12:
dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack no-ipset nftset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile

dnsmasq-full - 2.87-6

luci-app-simple-adblock - 1.9.3-3
simple-adblock - 1.9.3-4

and in luci you can choose "dnsmasq nft set" but in logs you will see "Mon Dec 12 18:33:18 2022 user.notice simple-adblock [8372]: ERROR: simple-adblock config validation failed!
Mon Dec 12 18:33:18 2022 user.notice simple-adblock [8372]: Please check if the '/etc/config/simple-adblock' contains correct values for config options.
Mon Dec 12 18:33:18 2022 user.notice simple-adblock [8372]: ERROR: simple-adblock config validation failed!
Mon Dec 12 18:33:18 2022 user.notice simple-adblock [8372]: Please check if the '/etc/config/simple-adblock' contains correct values for config options."

`root@thorgal:/etc/init.d# ./simple-adblock start
simple-adblock.config.enabled=1 validates as bool with true
simple-adblock.config.force_dns=1 validates as bool with true
simple-adblock.config.force_dns_port[0]=53 validates as list(integer) with true
simple-adblock.config.force_dns_port[1]=853 validates as list(integer) with true
simple-adblock.config.parallel_downloads=1 validates as bool with true
simple-adblock.config.debug=0 validates as bool with true
simple-adblock.config.compressed_cache=0 validates as bool with true
simple-adblock.config.ipv6_enabled is unset and defaults to bool 0
simple-adblock.config.allow_non_ascii=0 validates as bool with true
simple-adblock.config.canary_domains_icloud=0 validates as bool with true
simple-adblock.config.canary_domains_mozilla=0 validates as bool with true
simple-adblock.config.config_update_enabled=1 validates as bool with true
simple-adblock.config.config_update_url=https://cdn.jsdelivr.net/gh/openwrt/packages/net/simple-adblock/files/simple-adblock.conf.update validates as string with true
simple-adblock.config.download_timeout=10 validates as range(1,60) with true
simple-adblock.config.curl_max_file_size is unset and defaults to uinteger (null)
simple-adblock.config.curl_retry=3 validates as range(0,30) with true
simple-adblock.config.verbosity=2 validates as range(0,2) with true
simple-adblock.config.procd_trigger_wan6=0 validates as bool with true
simple-adblock.config.led is unset and defaults to or("", "none", file, device, string) (null)
simple-adblock.config.dns=dnsmasq.nftset validates as or("dnsmasq.addnhosts", "dnsmasq.conf", "dnsmasq.ipset", "dnsmasq.servers", "unbound.adb_list") with false
simple-adblock.config.dns_instance=0 validates as or(list(integer, string)) with true
simple-adblock.config.allowed_domain[0]=cdn.jsdelivr.net validates as list(string) with true
simple-adblock.config.allowed_domains_url is unset and defaults to list(string) (null)
simple-adblock.config.blocked_domain is unset and defaults to list(string) (null)
simple-adblock.config.blocked_domains_url[0]=https://cdn.jsdelivr.net/gh/AdguardTeam/cname-trackers@master/combined_disguised_trackers_justdomains.txt validates as list(string) with true
simple-adblock.config.blocked_domains_url[1]=https://adguardteam.github.io/AdGuardSDNSFilter/Filters/filter.txt validates as list(string) with true
simple-adblock.config.blocked_hosts_url[0]=https://adaway.org/hosts.txt validates as list(string) with true
simple-adblock.config.blocked_hosts_url[1]=https://cdn.jsdelivr.net/gh/hoshsadiq/adblock-nocoin-list/hosts.txt validates as list(string) with true
simple-adblock.config.blocked_hosts_url[2]=https://pgl.yoyo.org/as/serverlist.php?hostformat=hosts&showintro=1&mimetype=plaintext validates as list(string) with true
simple-adblock.config.blocked_hosts_url[3]=https://winhelp2002.mvps.org/hosts.txt validates as list(string) with true
simple-adblock.config.blocked_hosts_url[4]=https://someonewhocares.org/hosts/hosts validates as list(string) with true
simple-adblock.config.blocked_hosts_url[5]=https://easylist-downloads.adblockplus.org/easylist.txt validates as list(string) with true
simple-adblock.config.blocked_hosts_url[6]=https://hblock.molinero.dev/hosts validates as list(string) with true
simple-adblock.config.dnsmasq_config_file_url is unset and defaults to string (null)
ERROR: simple-adblock config validation failed!
Please check if the '/etc/config/simple-adblock' contains correct values for config options.
simple-adblock.config.enabled=1 validates as bool with true
simple-adblock.config.force_dns=1 validates as bool with true
simple-adblock.config.force_dns_port[0]=53 validates as list(integer) with true
simple-adblock.config.force_dns_port[1]=853 validates as list(integer) with true
simple-adblock.config.parallel_downloads=1 validates as bool with true
simple-adblock.config.debug=0 validates as bool with true
simple-adblock.config.compressed_cache=0 validates as bool with true
simple-adblock.config.ipv6_enabled is unset and defaults to bool 0
simple-adblock.config.allow_non_ascii=0 validates as bool with true
simple-adblock.config.canary_domains_icloud=0 validates as bool with true
simple-adblock.config.canary_domains_mozilla=0 validates as bool with true
simple-adblock.config.config_update_enabled=1 validates as bool with true
simple-adblock.config.config_update_url=https://cdn.jsdelivr.net/gh/openwrt/packages/net/simple-adblock/files/simple-adblock.conf.update validates as string with true
simple-adblock.config.download_timeout=10 validates as range(1,60) with true
simple-adblock.config.curl_max_file_size is unset and defaults to uinteger (null)
simple-adblock.config.curl_retry=3 validates as range(0,30) with true
simple-adblock.config.verbosity=2 validates as range(0,2) with true
simple-adblock.config.procd_trigger_wan6=0 validates as bool with true
simple-adblock.config.led is unset and defaults to or("", "none", file, device, string) (null)
simple-adblock.config.dns=dnsmasq.nftset validates as or("dnsmasq.addnhosts", "dnsmasq.conf", "dnsmasq.ipset", "dnsmasq.servers", "unbound.adb_list") with false
simple-adblock.config.dns_instance=0 validates as or(list(integer, string)) with true
simple-adblock.config.allowed_domain[0]=cdn.jsdelivr.net validates as list(string) with true
simple-adblock.config.allowed_domains_url is unset and defaults to list(string) (null)
simple-adblock.config.blocked_domain is unset and defaults to list(string) (null)
simple-adblock.config.blocked_domains_url[0]=https://cdn.jsdelivr.net/gh/AdguardTeam/cname-trackers@master/combined_disguised_trackers_justdomains.txt validates as list(string) with true
simple-adblock.config.blocked_domains_url[1]=https://adguardteam.github.io/AdGuardSDNSFilter/Filters/filter.txt validates as list(string) with true
simple-adblock.config.blocked_hosts_url[0]=https://adaway.org/hosts.txt validates as list(string) with true
simple-adblock.config.blocked_hosts_url[1]=https://cdn.jsdelivr.net/gh/hoshsadiq/adblock-nocoin-list/hosts.txt validates as list(string) with true
simple-adblock.config.blocked_hosts_url[2]=https://pgl.yoyo.org/as/serverlist.php?hostformat=hosts&showintro=1&mimetype=plaintext validates as list(string) with true
simple-adblock.config.blocked_hosts_url[3]=https://winhelp2002.mvps.org/hosts.txt validates as list(string) with true
simple-adblock.config.blocked_hosts_url[4]=https://someonewhocares.org/hosts/hosts validates as list(string) with true
simple-adblock.config.blocked_hosts_url[5]=https://easylist-downloads.adblockplus.org/easylist.txt validates as list(string) with true
simple-adblock.config.blocked_hosts_url[6]=https://hblock.molinero.dev/hosts validates as list(string) with true
simple-adblock.config.dnsmasq_config_file_url is unset and defaults to string (null)
ERROR: simple-adblock config validation failed!
Please check if the '/etc/config/simple-adblock' contains correct values for config options.
`

As you see:
`simple-adblock.config.dns=dnsmasq.nftset validates as or("dnsmasq.addnhosts", "dnsmasq.conf", "dnsmasq.ipset", "dnsmasq.servers", "unbound.adb_list") with false`

If eg. "dnsmasq.addnhosts" or "dnsmasq.servers" are chosen then it works like a charm.

dave14305, 2022-12-12:
This is best reported at https://github.com/openwrt/packages/issues where @stangri can track it better. Sounds `uci_load_validate` in the init script needs an extra value now.

mistepien, 2022-12-12:
Thanks, I do not know very well internals of OpenWRT, I found the line 1513 and added "dnsmasq.nftset" and was going to pull request, but you were faster. You guys do a great job!

closed 2022-12-12 by mistepien