openwrt/openwrt #11498: Running 5.15 on the Askey RT4230W

dadogroove, 2022-12-13:
I have been trying to run the new 5.15 master on the Askey RT4230W (ipq806x) without success. It is just causing the router to get stuck in a boot loop. I was able to get it running by editing the dts file. I removed this line

 qcom,boot-partitions = <0x0 0x1180000 0x1340000 0x2400000>;

	nand@0 {
		reg = <0>;
		compatible = "qcom,nandcs";

		nand-ecc-strength = <4>;
		nand-bus-width = <8>;
		nand-ecc-step-size = <512>;

		qcom,boot-partitions = <0x0 0x1180000 0x1340000 0x2400000>;

		partitions {
			compatible = "fixed-partitions";
			#address-cells = <1>;
			#size-cells = <1>;

			partition@0 {
				label = "0:SBL1";
				reg = <0x0000000 0x0040000>;
				read-only;
			};`

After that it seems to be working just as it did with the older 5.10 master.

Ansuel, 2022-12-13:
@dadogroove do you have by chance a bootlog of the crash?

dadogroove, 2022-12-13:
Here is the crash log. I will try the new boot-partition line and report back.

`U-Boot 1.0.6 [spf.5.0.2_csu1.1] (Sep 26 2019 - 02:44:05)

smem ram ptable found: ver: 0 len: 5
DRAM:  1003 MiB
setting 0x177e as machine type from smem
NAND:  SF: Unsupported manufacturer 00
ipq_spi: SPI Flash not found (bus/cs/speed/mode) = (0/0/48000000/0)
256 MiB
MMC:
PCI0 Link Intialized
PCI1 Link Intialized
In:    serial
Out:   serial
Err:   serial
MMC Device 0 not found
cdp: get part failed for 0:HLOS
Net:   MAC0 addr:80:78:71:e1:5f:db
athrs17_reg_init: complete
athrs17_vlan_config ...done
S17c init  done
MAC1 addr:80:78:71:e1:5f:dc
eth0, eth1
Hit space key to stop autoboot:  0
Creating 1 MTD partitions on "nand0":
0x000002400000-0x000010000000 : "mtd=0"
UBI: attaching mtd1 to ubi0
UBI: physical eraseblock size:   131072 bytes (128 KiB)
UBI: logical eraseblock size:    126976 bytes
UBI: smallest flash I/O unit:    2048
UBI: VID header offset:          2048 (aligned 2048)
UBI: data offset:                4096
UBI: attached mtd1 to ubi0
UBI: MTD device name:            "mtd=0"
UBI: MTD device size:            220 MiB
UBI: number of good PEBs:        1756
UBI: number of bad PEBs:         4
UBI: max. allowed volumes:       128
UBI: wear-leveling threshold:    4096
UBI: number of internal volumes: 1
UBI: number of user volumes:     3
UBI: available PEBs:             19
UBI: total number of reserved PEBs: 1737
UBI: number of PEBs reserved for bad PEB handling: 17
UBI: max/mean erase counter: 13/7
Read 7208960 bytes from volume kernel to 44000000
   Loading Kernel Image ... OK
OK

device nand0 <nand0>, # parts = 1
 #: name                size            offset          mask_flags
 0: firmware            0x0dc00000      0x02400000      0

active partition: nand0,0 - (firmware) 0x0dc00000 @ 0x02400000

defaults:
mtdids  : none
mtdparts: none
Setting up atags for msm partition: firmware
Using machid 0x177e from environment

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.15.77 (user@mint-7010) (arm-openwrt-linux-muslgnueabi-gcc (OpenWrt GCC 11.3.0 r21215+43-10ba380ec3) 11.3.0, GNU ld (GNU Binutils) 2.37) #0 SMP Fri Nov 11 01:39:14 2022
[    0.000000] CPU: ARMv7 Processor [512f04d0] revision 0 (ARMv7), cr=10c5787d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Askey RT4230W REV6
[... 237 lines cut ...]
[    3.213932] 1f15              32 mtdblock21
[    3.213940]  (driver?)
[    3.220485] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
[    3.222665] CPU0: stopping
[    3.230904] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 5.15.77 #0
[    3.233596] Hardware name: Generic DT based system
[    3.239767] [<c030e5b4>] (unwind_backtrace) from [<c030a130>] (show_stack+0x14/0x20)
[    3.244366] [<c030a130>] (show_stack) from [<c064abdc>] (dump_stack_lvl+0x40/0x4c)
[    3.252266] [<c064abdc>] (dump_stack_lvl) from [<c030cfdc>] (do_handle_IPI+0x144/0x184)
[    3.259643] [<c030cfdc>] (do_handle_IPI) from [<c030d038>] (ipi_handler+0x1c/0x2c)
[    3.267540] [<c030d038>] (ipi_handler) from [<c0379e9c>] (handle_percpu_devid_irq+0x84/0x16c)
[    3.275183] [<c0379e9c>] (handle_percpu_devid_irq) from [<c0373760>] (handle_domain_irq+0x68/0x94)
[    3.283776] [<c0373760>] (handle_domain_irq) from [<c0664ccc>] (gic_handle_irq+0x8c/0xb4)
[    3.292631] [<c0664ccc>] (gic_handle_irq) from [<c0300b7c>] (__irq_svc+0x5c/0x78)
[    3.300873] Exception stack(0xc0d01ed0 to 0xc0d01f18)
[    3.308340] 1ec0:                                     00000000 00000000 2eaca000 ef710480
[    3.313381] 1ee0: 00000000 bf8ff060 c1d00040 00000000 ef70f6f0 00000000 00000000 00000000
[    3.321538] 1f00: 0005fe60 c0d01f20 c07ea528 c07ea548 60000013 ffffffff
[    3.329693] [<c0300b7c>] (__irq_svc) from [<c07ea548>] (cpuidle_enter_state+0x180/0x37c)
[    3.336120] [<c07ea548>] (cpuidle_enter_state) from [<c07ea794>] (cpuidle_enter+0x3c/0x5c)
[    3.344452] [<c07ea794>] (cpuidle_enter) from [<c0351360>] (do_idle+0x204/0x2a4)
[    3.352524] [<c0351360>] (do_idle) from [<c0351724>] (cpu_startup_entry+0x1c/0x20)
[    3.360074] [<c0351724>] (cpu_startup_entry) from [<c0c0118c>] (start_kernel+0x6ac/0x6c0)
[    3.367455] [<c0c0118c>] (start_kernel) from [<00000000>] (0x0)
[    3.375704] Rebooting in 1 seconds..`

dadogroove, 2022-12-13:
Sorry for the long post. Thought I formatted that correctly.

Also, quick question as I am still learning. Can I change the dts file and just make again or are there more steps involved to get it to use the updated dts file in the compile? Thanks.

Ansuel, 2022-12-13:
@dadogroove you should be able to change the dts in the target directory and kernel should rebuilt...

dadogroove, 2022-12-13:
@Ansuel It works fine with the suggested change to the boot-partitions line. Thanks for the help!

Ansuel, 2022-12-13:
Thanks! I'm preparing a patch is it ok to add a tested-by tag? (i need your real name and surname and email) It's totally optional but it's good to have some credits :D

dadogroove, 2022-12-13:
Sure. Glad to help out.
Matt Buczko
mbuczko@hotmail.com

closed 2022-12-14 by openwrt-bot

Ansuel, 2022-12-14:
@dadogroove sorry for the issue... fix merged in master, just wait for the new image to get build.

dadogroove, 2022-12-14:
@Ansuel No worries. Thanks for the quick fix of the issue!

commit 266b261d (2022-12-14): ipq806x: 5.15: fix wrong boot-partitions values for split partitions

The refreshed patch actually use the format of <start size start size>
instead of <start end start end>. This cause boot fail since the rootfs
can't be mounted with these wrong values.

Fix it to the correct format in each affected dts.

Fixes: #11498
Fixes: 6134ba4a34db ("ipq806x: 5.15: add boot-partitions binding to fix block warning")
Tested-by: Matt Buczko <mbuczko@hotmail.com> # Askey RT4230W