openwrt/openwrt #11514: [Zyxel nwa50ax] factory firmware 22.03.2 not flashing using web ui from native fw

filimonic, 2022-12-14:
Factory firmware provided on [wiki page](https://openwrt.org/toh/hwdata/zyxel/zyxel_nwa50ax) which is [Firmware OpenWrt Install URL](https://downloads.openwrt.org/releases/22.03.2/targets/ramips/mt7621/openwrt-22.03.2-ramips-mt7621-zyxel_nwa50ax-squashfs-factory.bin)` is not flashing using web firmware upgrade, but the one which is [Firmware OpenWrt snapshot Install URL](https://downloads.openwrt.org/snapshots/targets/ramips/mt7621/openwrt-ramips-mt7621-zyxel_nwa50ax-squashfs-factory.bin) - does.

Steps: 

1. Buy new nwa50ax  
2. boot and go to web ui
3. flash `6.29(ABYW.1)C0` firmware to switch active fw to `B` block
4. auto reboot and go to web ui
5. flash `openwrt-22.03.2-ramips-mt7621-zyxel_nwa50ax-squashfs-factory.bin`
6. fw reboots back to original fw (`B` block)

if in step 5 use snapshot fw, on step 6 router boots OpenWRT

AdamWeglarz, 2022-12-26:
Can you send me by pm snapshot you have used for first install?
One I have download today is not working.

filimonic, 2022-12-26:
@AdamWeglarz , I home I am correct, as I could find them in "downloads" bin. 
https://disk.yandex.ru/d/ybeAXcdxntZdsQ 
I flashed `-factory`, which is 12 Mb, and then connected via wire to device and upgrade through ssh to `sysupgrade` which is 6 mb:
```
cd /tmp
wget http://192.168.xx.yy/fw.bin 
sysupgrade -v fw.bin
```

AdamWeglarz, 2022-12-26:
I had to install older zyxel release and I have manage to upgrade.

filimonic, 2022-12-27:
> I had to install older zyxel release and I have manage to upgrade.

Could you please tell which one exactly worked ?

AdamWeglarz, 2022-12-28:
I have managed to upgrade when downgraded to NWA50AX_6.25(ABYW.8)C0 [https://download.zyxel.com/NWA50AX/firmware/NWA50AX_6.25(ABYW.8)C0.zip](url) and later I have used snapshot you have provided.

Adam

AdamWeglarz, 2022-12-30:
In fact I success only after i upgrade official software and just after boot I install openwrt snapshot factory

AdamWeglarz, 2023-01-08:
https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=a0b7fef0ffe4cd9cca39a652a37e4f3ce8f0a681 Here is explained why I had to install first official release.

closed 2023-08-24 by filimonic