openwrt/openwrt #11518: Router WE826-T2 16MB, after transferring overlay to USB, the modem and Wi-Fi disappeared

mickhaorex, 2022-12-15:
Firmware OpenWRT 21.02.5 r16688-fa9a932fdb. After the firmware, the router worked fine, WiFi and LTE/3G modem were active. There was also a built-in memory of 1.2 MB. This is very small for installing packages, I decided to increase the built-in memory due to an external USB drive (64 Gb). I did it according to the instructions https://openwrt.org/docs/guide-user/additional-software/extroot_configuration .
After all the manipulations described in the article, the memory increased, but the LTE/3G modem and WiFi disappeared.
[image]
[image]
[image]
[image]
[image]

mickhaorex, 2022-12-15:
I understand that this is probably nonsense, but maybe just copy the /etc folder to /mnt/sda2/upper/etc:

[image]

[image]

brada4, 2022-12-15:
Could you post `mount` and `df` outputs?

mickhaorex, 2022-12-15:
I re-flashed the router, even before your answer. At the moment, the commands you are requesting look like this:

[image]

mickhaorex, 2022-12-15:
And at the address http://192.168.1.1/cgi-bin/luci/admin/system/mounts so:
[image]

mickhaorex, 2022-12-15:
The markup of the USB Flash file system is as follows:
[image]

This is a screenshot from the MiniTool Partition Wizard program in windows

mickhaorex, 2022-12-16:
Thanks! I'll try to do it this way

mickhaorex, 2022-12-16:
I never mastered OpenWRT. Perhaps I will do it in the future, because this is a very interesting topic. It is a pity that there is no ready-made solution for working with my router.  So far I have stopped at Wifix firmware

vinniec2, 2025-09-17:
Sorry if I intervene, I was planning to buy this router and while looking for information I think I accidentally stumbled upon what the problem is due to in [this message](https://forum.openwrt.org/t/support-for-zbt-we826-q/50997/16) which I am reporting here:
(p.s. I think WE826-T2 is the same off -Q and diffferent to -T)

> He is asking about the WE826-Q and not the WE826-T which has OpenWrt
> support.
> 
> The WE826-Q is an AP147-010 clone and can run that image except that
> there are some problem areas.
> 
> First, it has a hardware watchdog monitor that requires a heartbeat on
> GPIO 2 or else the router will reboot after a short period.
> 
> Second, the 920-usb-chipidea-AR933x-platform-support.patch disables
> the USB because the hardware is missing what appears to be some pullups
> on pins of the SOC. That patch needs to be changed in order to make the
> USB and PCie slot work. ZBT uses Lede 17.01 as their firmware as that
> patch came as part of 18.06.