openwrt/openwrt #11545: Mode WIFI STA bridge with LAN does not work

annick96, 2022-12-17:
I cannot associate OpenWrt WIFI wiith my LiveBox 6 Orange used in France by more than ten million users

Run without problem with PandoraBox , Padavan , repeater Orange , AsRock G10 , and surely with many other software

look here

https://forum.openwrt.org/t/mode-wifi-sta-bridge-with-lan-does-not-work/145782

https://forum.openwrt.org/t/why-openwrt-is-bad-in-wifi-mt76x/145226/59

Hurricos, 2022-12-18:
You cannot bridge (create a switch between) a regular STA with ethernet, because WiFi switching uses >=3 addresses, while ethernet only uses two. This is a longstanding limitation with the protocol.

The other devices and implementations you're referring to use hacks like routed switching (OpenWrt's implementation for this is [relayd](https://openwrt.org/docs/guide-user/network/wifi/relay_configuration#using_relayd)), or use modified kernel modules.

The proper OpenWrt way to do this is to use 4-address mode protocols like 802.11s or WDS.

annick96, 2022-12-18:
**WHY run without problem with PandoraBox , Padavan , repeater Orange , AsRock G10 , and surely with many other software**

Hurricos, 2022-12-18:
Because:

> The other devices and implementations you're referring to use hacks like routed switching (OpenWrt's implementation for this is [relayd](https://openwrt.org/docs/guide-user/network/wifi/relay_configuration#using_relayd)), or use modified kernel modules.

Hurricos, 2022-12-18:
This issue is a request for support, not a technical issue, and needs to be closed.

closed 2022-12-19 by annick96

annick96, 2022-12-19:
Sorry but that is a MT76x driver problem

I have a build with MTK driver & run without problem