openwrt/openwrt #11557: ath79: asus-pl/rp*: mac-address and calibration data overlapping?

PolynomialDivision, 2022-12-18:
The targets `asus,pl-ac56`, `asus,rp-ac51`, `asus,rp-ac66` extract ath9j-calibration-data from `caldata_extract "art" 0x1000 0x440`(https://github.com/openwrt/openwrt/blob/eb7ffeafbfa78235b05abb4ae479376150d7a814/target/linux/ath79/generic/base-files/etc/hotplug.d/firmware/10-ath9k-eeprom#L18)

However, all targets also extract their mac-address from:
```
calibration_art_1000: calibration@1000 {
        reg = <0x1000 0x440>;
};
```
That is overlapping?

PolynomialDivision, 2022-12-19:
What do you think about ths @DragonBluep ?

john-tho, 2022-12-19:
Looks okay if it works: We use caldata offset 2 for `ath9k_patch_mac`
https://github.com/openwrt/openwrt/blob/6d1df35747d7f02f52973ba5604642dcc17ddf5f/package/base-files/files/lib/functions/caldata.sh#L167
https://elixir.bootlin.com/linux/latest/C/ident/macAddr
Commit message was explicit about this too: https://github.com/openwrt/openwrt/commit/e1dcaeb55c939365cdd27e469b63fca295bf1a2e

Would need a flash dump to be sure though?

981213, 2022-12-28:
That's expected. ath9k calibration data contains a field for mac address. Most vendors don't use that field though, they store a separated mac address elsewhere.

PolynomialDivision, 2022-12-28:
Thanks! I will close.

closed 2022-12-28 by PolynomialDivision