openwrt/openwrt #11563: /etc/config/network: option `tunlink` of WireGuard peer does not work

Livy-Inverse, 2022-12-19:
I have a router with multiple WAN connections, with one of them being a connection to a private corporate network which blocks virtually any connection to the Internet.

I create a WireGuard tunnel from that router to my home router, and by default, OpenWrt creates a route to my home router address via the corporate network gateway, and it blocks the WireGuard connection.

With the unrestricted PPPoE connection to my ISP being defined in **myisp** uci section, I tried to edit WireGuard peer with option tunlink, but it still binds my home router IP address to the corporate network gateway:
```
config device 'veth0'
	option name 'veth0'
	option type 'macvlan'
	option ifname 'eth0.<some_vid>'

config interface 'corporate'
	option device 'veth0'
	option proto 'static'
	option ipaddr '172.16.0.1'
	option netmask '255.255.255.0'
	option gateway '172.16.0.254'
	option metric '500'

config interface 'myisp'
	option device 'eth0.<another_vid>'
	option proto 'pppoe'
	option username '<user_name>'
	option password '<password>'

config interface 'vpn'
	option proto 'wireguard'
	option listen_port '51820'
	option private_key '<private_key>'
	list addresses '192.168.1.1/24'

config wireguard_vpn 'home'
	option description 'My home router'
	option public_key '<public_key>'
	option preshared_key '<pre_shared_key>'
	list allowed_ips '192.168.1.2/32'
	option route_allowed_ips '1'
	option endpoint_host '<home_router_ip_addr>'
	option endpoint_port '51820'
	option tunlink '@myisp'
```
I also tried **myisp** instead of **@myisp** but it does not work either. The result is always like this:
```
root@OpenWrt:~# ip route show
[...]
<home_router_ip_addr> via 172.16.0.254 dev veth0 proto static metric 500
[...]
```
I know I can remove the option gateway in the configuration of the **corporate** interface. But that gateway still allows some basic Internet services and I want to keep it as the last resort in case all WANs are down. Also, my configuration has a lot of WAN connections, and by removing that gateway, OpenWrt will fallback to a another random one for the WireGuard connection. I want it to use the specific myisp interface only.

I am using the latest version 22.03.2.

jow-, 2022-12-19:
The `tunlink` option is not defined for wireguard peers.

closed 2022-12-19 by jow-

Livy-Inverse, 2022-12-19:
[According to the Wiki](https://openwrt.org/docs/guide-user/network/tunneling_interface_protocols#wireguard_peers), it does have. Maybe it was someone's mistake?

I also find [this commit](https://github.com/openwrt/openwrt/commit/47b2ee2d9a9a1790f9bf8a528640c333af39e4ba). I don't know if it get merged or rejected. Sorry because I am not a programmer and not familiar with Git development system.

jow-, 2022-12-19:
The wiki is wrong in this case. The commit you mentioned implemented the `tunlink` option for the interface section.

Livy-Inverse, 2022-12-19:
Thanks for your correction.

One more thing though. In `option device` of `config interface` section, I always use at sign to refer to a uci section (e.g: **@wan**), and no at sign to refer to a kernel interface (e.g: **eth1**). Should the `option tunlink` be **myisp** or **@myisp**?

jow-, 2022-12-19:
The tunlink option should refer to a *logical interface* (the name of a `config interface ...` section) without an `@` prefix.