openwrt/openwrt #11566: xiaomi router 4a gigabit encounters significant wierless problem after updating openwrt

Nick24-Kobe, 2022-12-19:
I have update to openwrt according to the guide, but there seems to be a significant  wireless problem. There is no any wireless infomation in overview or network tab. log reports that eports that mt7603e 0000:01:00.0: Invalid MAC address, using random address  b2:4a:96:17:68:6b. Details as follows,

[    0.000000] Linux version 5.10.146 (builder@buildhost) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 r19803-9a599fee93) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 SMP Fri Oct 14 22:44:41 2022
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:4
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is Xiaomi Mi Router 4A Gigabit Edition
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] On node 0 totalpages: 32768
[    0.000000]   Normal zone: 288 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 32768 pages, LIFO batch:7
[    0.000000] percpu: Embedded 15 pages/cpu s30256 r8192 d22992 u61440
[    0.000000] pcpu-alloc: s30256 r8192 d22992 u61440 alloc=15*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32480
[    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Writing ErrCtl register=0003250f
[    0.000000] Readback ErrCtl register=0003250f
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 119472K/131072K available (7026K kernel code, 631K rwdata, 828K rodata, 1292K init, 243K bss, 11600K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 256
[    0.000000] CPU Clock: 880MHz
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
[    0.000013] sched_clock: 64 bits at 880MHz, resolution 1ns, wraps every 4398046511103ns
[    0.007941] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
[    0.016992] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
[    0.083144] pid_max: default: 32768 minimum: 301
[    0.087900] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.095101] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.105578] rcu: Hierarchical SRCU implementation.
[    0.110597] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
[    0.118660] smp: Bringing up secondary CPUs ...
[    0.123898] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.123908] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.123919] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.123994] CPU1 revision is: 0001992f (MIPS 1004Kc)
[    0.178456] Synchronize counters for CPU 1: done.
[    0.210436] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.210445] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.210454] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.210501] CPU2 revision is: 0001992f (MIPS 1004Kc)
[    0.269625] Synchronize counters for CPU 2: done.
[    0.300204] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.300212] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.300221] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.300272] CPU3 revision is: 0001992f (MIPS 1004Kc)
[    0.354809] Synchronize counters for CPU 3: done.
[    0.384675] smp: Brought up 1 node, 4 CPUs
[    0.392750] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.402567] futex hash table entries: 1024 (order: 3, 32768 bytes, linear)
[    0.409573] pinctrl core: initialized pinctrl subsystem
[    0.417127] NET: Registered protocol family 16
[... 330 lines cut ...]
[  260.935574] br-lan: port 2(lan2) entered forwarding state
[  261.413207] br-lan: port 2(lan2) entered disabled state
[  262.480545] mt7530 mdio-bus:1f lan2: Link is Up - 100Mbps/Full - flow control rx/tx
[  262.488380] br-lan: port 2(lan2) entered blocking state
[  262.494168] br-lan: port 2(lan2) entered forwarding state
[  316.004773] mt7530 mdio-bus:1f lan1: Link is Up - 1Gbps/Full - flow control off
[  316.012161] br-lan: port 1(lan1) entered blocking state
[  316.017704] br-lan: port 1(lan1) entered forwarding state

Last but not least, my mtd is different from that in example, is that the key reason rusulting in the wireless problem?
mtd details as follows:

dev:    size   erasesize  name
mtd0: 01000000 00010000 "ALL"
mtd1: 00030000 00010000 "Bootloader"
mtd2: 00010000 00010000 "KF"
mtd3: 00010000 00010000 "Bdata"
mtd4: 00010000 00010000 "Factory"
mtd5: 00010000 00010000 "crash"
mtd6: 00010000 00010000 "cfg_bak"
mtd7: 00100000 00010000 "overlay"
mtd8: 00e70000 00010000 "OS1"
mtd9: 001a0000 00010000 "kernel"
mtd10: 00cd0000 00010000 "rootfs"
mtd11: 00010000 00010000 "Config"

khanjui, 2022-12-20:
It's fine on my device using latest snapshot.

Nick24-Kobe, 2022-12-20:
@khanjui I don't know why the wifi is disabled, and the light is always yellow

khanjui, 2022-12-21:
Did you try upgrading to latest snapshot without keeping the settings?

Nick24-Kobe, 2022-12-21:
Yes, I tried. The light is yellow and can't connect to the router after upgrading to latest snapshot without keeping the settings. I have to power off the router and restart it, but it still has no wireless settings.

Nick24-Kobe, 2022-12-21:
@khanjui Yes, I tried. The light is yellow and can't connect to the router after upgrading to latest snapshot without keeping the settings. I have to power off the router and restart it, but it still has no wireless settings.

khanjui, 2022-12-21:
https://forum.openwrt.org/t/xiaomi-4a-3-gigaport-cannot-flash-openwrt-21-02/122878
Related?

Nick24-Kobe, 2023-01-14:
> It is printed in log. GD25Q128
Jan 14 11:49:52 kernel: SPI flash chip: GD25Q128 (c8 40180000) (16384 Kbytes)

kvanbiesen, 2023-02-14:
Have the same log as @Nick24-Kobe 

Same chip, same error, but networking and wireless is working for me however wlan drops very often...
`
[    0.000000] Linux version 5.10.161 (builder@buildhost) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 r20028-43d71ad93e) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 SMP Tue Jan 3 00:24:21 2023
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is Xiaomi Mi Router 4A Gigabit Edition
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] On node 0 totalpages: 32768
[    0.000000]   Normal zone: 288 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 32768 pages, LIFO batch:7
[    0.000000] percpu: Embedded 15 pages/cpu s30480 r8192 d22768 u61440
[    0.000000] pcpu-alloc: s30480 r8192 d22768 u61440 alloc=15*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32480
[    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Writing ErrCtl register=00040090
[    0.000000] Readback ErrCtl register=00040090
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 119472K/131072K available (7039K kernel code, 630K rwdata, 832K rodata, 1276K init, 243K bss, 11600K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 256
[    0.000000] CPU Clock: 880MHz
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
[    0.000013] sched_clock: 64 bits at 880MHz, resolution 1ns, wraps every 4398046511103ns
[    0.007942] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
[    0.017000] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
[    0.083163] pid_max: default: 32768 minimum: 301
[    0.087928] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.095131] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.105754] rcu: Hierarchical SRCU implementation.
[    0.110793] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
[    0.118897] smp: Bringing up secondary CPUs ...
[    0.124179] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.124190] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.124201] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.124278] CPU1 revision is: 0001992f (MIPS 1004Kc)
[    0.178692] Synchronize counters for CPU 1: done.
[    0.211150] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.211159] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.211168] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.211216] CPU2 revision is: 0001992f (MIPS 1004Kc)
[    0.270112] Synchronize counters for CPU 2: done.
[    0.300706] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.300715] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.300724] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.300774] CPU3 revision is: 0001992f (MIPS 1004Kc)
[    0.355298] Synchronize counters for CPU 3: done.
[    0.385168] smp: Brought up 1 node, 4 CPUs
[    0.393514] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.403335] futex hash table entries: 1024 (order: 3, 32768 bytes, linear)
[... 585 lines cut ...]
[205615.057885] br-lan: port 5(wlan1) entered blocking state
[205615.063294] br-lan: port 5(wlan1) entered forwarding state
`

Mtd looks completely different but throws errors. Wifi and networking tabs are working but badly. Dropping wifi mostly of the time.....
`
[    0.654219] spi-nor spi0.0: gd25q128 (16384 Kbytes)
[    0.659166] 8 fixed-partitions partitions found on MTD device spi0.0
[    0.665572] OF: Bad cell count for /palmbus@1e000000/spi@b00/flash@0/partitions
[    0.672871] OF: Bad cell count for /palmbus@1e000000/spi@b00/flash@0/partitions
[    0.680986] OF: Bad cell count for /palmbus@1e000000/spi@b00/flash@0/partitions
[    0.688330] OF: Bad cell count for /palmbus@1e000000/spi@b00/flash@0/partitions
[    0.695891] Creating 8 MTD partitions on "spi0.0":
[    0.700688] 0x000000000000-0x000000030000 : "u-boot"
[    0.706660] 0x000000030000-0x000000040000 : "u-boot-env"
[    0.712972] 0x000000040000-0x000000050000 : "Bdata"
[    0.718890] 0x000000050000-0x000000060000 : "factory"
[    0.725165] 0x000000060000-0x000000070000 : "crash"
[    0.730970] 0x000000070000-0x000000080000 : "cfg_bak"
[    0.737077] 0x000000080000-0x000000180000 : "overlay"
[    0.743100] 0x000000180000-0x000001000000 : "firmware"

`

same gd25q128 as @Nick24-Kobe

Matthaiks, 2023-02-23:
Netgear R6220
OpenWrt SNAPSHOT (r22127-595509cc78)

It also throws the "Invalid MAC address, using random address" message during startup.

OpenWrt 22.03 is fine.

FredyVia, 2023-04-09:
The same model has similar issues, with poor 5G signal, with a maximum of txpower only 3dBM.Also in the log: Invalid MAC address, using random address. It is somewhat similar to this problem.
https://github.com/openwrt/mt76/issues/657

Djfe, 2023-04-10:
@Matthaiks does it do that in snapshot still? some new stuff was fixed there :)

Djfe, 2023-04-10:
@FredyVia "the same model" which device? 😅
Overall it sounds like eeprom extraction is causing the issues here.

@Nick24-Kobe is this from stock rom? because your first logread shows a different list of mtd partitions

```
dev: size erasesize name
mtd0: 01000000 00010000 "ALL"
mtd1: 00030000 00010000 "Bootloader"
mtd2: 00010000 00010000 "KF"
mtd3: 00010000 00010000 "Bdata"
mtd4: 00010000 00010000 "Factory"
mtd5: 00010000 00010000 "crash"
mtd6: 00010000 00010000 "cfg_bak"
mtd7: 00100000 00010000 "overlay"
mtd8: 00e70000 00010000 "OS1"
mtd9: 001a0000 00010000 "kernel"
mtd10: 00cd0000 00010000 "rootfs"
mtd11: 00010000 00010000 "Config" 
```

Matthaiks, 2023-04-10:
@Djfe It was fixed.

Djfe, 2023-04-10:
great to hear 👍
Then we can focus on the xiaomi here

FredyVia, 2023-04-14:
error still exist
here is my system log
```
Sun Apr  9 12:27:52 2023 kern.notice kernel: [    0.000000] Linux version 5.10.176 (builder@buildhost) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 r20123-38ccc47687) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 SMP Sun Apr 9 12:27:46 2023
Sun Apr  9 12:27:52 2023 kern.info kernel: [    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
Sun Apr  9 12:27:52 2023 kern.info kernel: [    0.000000] printk: bootconsole [early0] enabled
Sun Apr  9 12:27:52 2023 kern.info kernel: [    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
Sun Apr  9 12:27:52 2023 kern.info kernel: [    0.000000] MIPS: machine is Xiaomi Mi Router 4A Gigabit Edition
Sun Apr  9 12:27:52 2023 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Sun Apr  9 12:27:52 2023 kern.info kernel: [    0.000000] VPE topology {2,2} total 4
Sun Apr  9 12:27:52 2023 kern.info kernel: [    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Sun Apr  9 12:27:52 2023 kern.info kernel: [    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Sun Apr  9 12:27:52 2023 kern.warn kernel: [    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Sun Apr  9 12:27:52 2023 kern.info kernel: [    0.000000] Zone ranges:
Sun Apr  9 12:27:52 2023 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
Sun Apr  9 12:27:52 2023 kern.info kernel: [    0.000000]   HighMem  empty
Sun Apr  9 12:27:52 2023 kern.info kernel: [    0.000000] Movable zone start for each node
Sun Apr  9 12:27:52 2023 kern.info kernel: [    0.000000] Early memory node ranges
Sun Apr  9 12:27:52 2023 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
Sun Apr  9 12:27:52 2023 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
Sun Apr  9 12:27:52 2023 kern.debug kernel: [    0.000000] On node 0 totalpages: 32768
Sun Apr  9 12:27:52 2023 kern.debug kernel: [    0.000000]   Normal zone: 288 pages used for memmap
Sun Apr  9 12:27:52 2023 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Sun Apr  9 12:27:52 2023 kern.debug kernel: [    0.000000]   Normal zone: 32768 pages, LIFO batch:7
Sun Apr  9 12:27:52 2023 kern.info kernel: [    0.000000] percpu: Embedded 15 pages/cpu s30480 r8192 d22768 u61440
Sun Apr  9 12:27:52 2023 kern.debug kernel: [    0.000000] pcpu-alloc: s30480 r8192 d22768 u61440 alloc=15*4096
Sun Apr  9 12:27:52 2023 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Sun Apr  9 12:27:52 2023 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32480
Sun Apr  9 12:27:52 2023 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
Sun Apr  9 12:27:52 2023 kern.info kernel: [    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
Sun Apr  9 12:27:52 2023 kern.info kernel: [    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
Sun Apr  9 12:27:52 2023 kern.info kernel: [    0.000000] Writing ErrCtl register=0001b100
Sun Apr  9 12:27:52 2023 kern.info kernel: [    0.000000] Readback ErrCtl register=0001b100
Sun Apr  9 12:27:52 2023 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Sun Apr  9 12:27:52 2023 kern.info kernel: [    0.000000] Memory: 119472K/131072K available (7064K kernel code, 631K rwdata, 832K rodata, 1252K init, 243K bss, 11600K reserved, 0K cma-reserved, 0K highmem)
Sun Apr  9 12:27:52 2023 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Sun Apr  9 12:27:52 2023 kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Sun Apr  9 12:27:52 2023 kern.info kernel: [    0.000000] 	Tracing variant of Tasks RCU enabled.
Sun Apr  9 12:27:52 2023 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
Sun Apr  9 12:27:52 2023 kern.info kernel: [    0.000000] NR_IRQS: 256
Sun Apr  9 12:27:52 2023 kern.info kernel: [    0.000000] CPU Clock: 880MHz
Sun Apr  9 12:27:52 2023 kern.info kernel: [    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
Sun Apr  9 12:27:52 2023 kern.info kernel: [    0.000012] sched_clock: 64 bits at 880MHz, resolution 1ns, wraps every 4398046511103ns
Sun Apr  9 12:27:52 2023 kern.info kernel: [    0.007941] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
Sun Apr  9 12:27:52 2023 kern.info kernel: [    0.016998] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
Sun Apr  9 12:27:52 2023 kern.info kernel: [    0.083159] pid_max: default: 32768 minimum: 301
Sun Apr  9 12:27:52 2023 kern.info kernel: [    0.087919] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Sun Apr  9 12:27:52 2023 kern.info kernel: [    0.095122] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Sun Apr  9 12:27:52 2023 kern.info kernel: [    0.105722] rcu: Hierarchical SRCU implementation.
Sun Apr  9 12:27:52 2023 kern.info kernel: [    0.110772] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
Sun Apr  9 12:27:52 2023 kern.info kernel: [    0.118880] smp: Bringing up secondary CPUs ...
[... cut ...]
Fri Apr 14 16:43:22 2023 daemon.notice hostapd: wlan1: STA 8a:99:15:d4:e5:51 IEEE 802.11: did not acknowledge authentication response
Fri Apr 14 16:43:22 2023 daemon.notice hostapd: wlan1: STA 8a:99:15:d4:e5:51 IEEE 802.11: did not acknowledge authentication response
Fri Apr 14 16:43:22 2023 daemon.notice hostapd: wlan1: STA 8a:99:15:d4:e5:51 IEEE 802.11: did not acknowledge authentication response
Fri Apr 14 16:43:22 2023 daemon.notice hostapd: wlan1: STA 8a:99:15:d4:e5:51 IEEE 802.11: did not acknowledge authentication response
Fri Apr 14 16:44:17 2023 auth.err passwd: password for root changed by root
```

FredyVia, 2023-04-14:
same as https://github.com/openwrt/mt76/issues/633

kvanbiesen, 2023-04-15:
@FredyVia 

, as mine doesnt seem to have to have this problem, its been up for couople of weeks and works fine, cant see ree issues, dont have the random error plus i also have the full speed it seems on the iwinfo (not capped at 3TX), mine is going at 23Tx.  we even have the same chip (en25qx128a )
> 
> Mon Apr  3 20:12:08 2023 kern.notice kernel: [    0.000000] Linux version 5.15.94 (builder@buildhost) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 12.2.0 r22131-218f425b61) 12.2.0, GNU ld (GNU Binutils) 2.39) #0 SMP Thu Feb 23 21:22:27 2023
> Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
> Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000] printk: bootconsole [early0] enabled
> Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
> Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000] MIPS: machine is Xiaomi Mi Router 4A Gigabit Edition
> Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
> Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000] VPE topology {2,2} total 4
> Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
> Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
> Mon Apr  3 20:12:08 2023 kern.warn kernel: [    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
> Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000] Zone ranges:
> Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
> Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000]   HighMem  empty
> Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000] Movable zone start for each node
> Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000] Early memory node ranges
> Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
> Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
> Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000] percpu: Embedded 11 pages/cpu s15856 r8192 d21008 u45056
> Mon Apr  3 20:12:08 2023 kern.debug kernel: [    0.000000] pcpu-alloc: s15856 r8192 d21008 u45056 alloc=11*4096
> Mon Apr  3 20:12:08 2023 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
> Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32480
> Mon Apr  3 20:12:08 2023 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
> Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
> Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
> Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000] Writing ErrCtl register=0001c010
> Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000] Readback ErrCtl register=0001c010
> Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
> Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000] Memory: 118496K/131072K available (7414K kernel code, 632K rwdata, 1496K rodata, 1260K init, 251K bss, 12576K reserved, 0K cma-reserved, 0K highmem)
> Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
> Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
> Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000] 	Tracing variant of Tasks RCU enabled.
> Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
> Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000] NR_IRQS: 256
> Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
> Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000005] sched_clock: 64 bits at 880MHz, resolution 1ns, wraps every 4398046511103ns
> Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.008075] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
> Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.066216] pid_max: default: 32768 minimum: 301
> Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.071035] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
> Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.078247] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
> Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.089166] rcu: Hierarchical SRCU implementation.
> Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.094202] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
> Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.102324] smp: Bringing up secondary CPUs ...
> Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.107601] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
> Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.107625] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[... cut ...]
>           Mode: Master  Channel: 36 (5.180 GHz)  HT Mode: VHT80
>           Center Channel 1: 42 2: unknown
>           Tx-Power: 23 dBm  Link Quality: 54/70
>           Signal: -56 dBm  Noise: unknown
>           Bit Rate: 676.0 MBit/s
>           Encryption: WPA2 PSK (CCMP)
>           Type: nl80211  HW Mode(s): 802.11ac/n
>           Hardware: 14C3:7662 14C3:7662 [MediaTek MT76x2E]
>           TX power offset: none
>           Frequency offset: none
>           Supports VAPs: yes  PHY name: phy1

FredyVia, 2023-04-15:
> @FredyVia
> 
> , as mine doesnt seem to have to have this problem, its been up for couople of weeks and works fine, cant see ree issues, dont have the random error plus i also have the full speed it seems on the iwinfo (not capped at 3TX), mine is going at 23Tx. we even have the same chip (en25qx128a )
> 
> > Mon Apr  3 20:12:08 2023 kern.notice kernel: [    0.000000] Linux version 5.15.94 (builder@buildhost) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 12.2.0 r22131-218f425b61) 12.2.0, GNU ld (GNU Binutils) 2.39) #0 SMP Thu Feb 23 21:22:27 2023
> > Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
> > Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000] printk: bootconsole [early0] enabled
> > Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
> > Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000] MIPS: machine is Xiaomi Mi Router 4A Gigabit Edition
> > Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
> > Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000] VPE topology {2,2} total 4
> > Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
> > Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
> > Mon Apr  3 20:12:08 2023 kern.warn kernel: [    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
> > Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000] Zone ranges:
> > Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
> > Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000]   HighMem  empty
> > Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000] Movable zone start for each node
> > Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000] Early memory node ranges
> > Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
> > Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
> > Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000] percpu: Embedded 11 pages/cpu s15856 r8192 d21008 u45056
> > Mon Apr  3 20:12:08 2023 kern.debug kernel: [    0.000000] pcpu-alloc: s15856 r8192 d21008 u45056 alloc=11_4096
> > Mon Apr  3 20:12:08 2023 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
> > Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32480
> > Mon Apr  3 20:12:08 2023 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
> > Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
> > Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
> > Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000] Writing ErrCtl register=0001c010
> > Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000] Readback ErrCtl register=0001c010
> > Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
> > Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000] Memory: 118496K/131072K available (7414K kernel code, 632K rwdata, 1496K rodata, 1260K init, 251K bss, 12576K reserved, 0K cma-reserved, 0K highmem)
> > Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
> > Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
> > Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000] 	Tracing variant of Tasks RCU enabled.
> > Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
> > Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000] NR_IRQS: 256
> > Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
> > Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.000005] sched_clock: 64 bits at 880MHz, resolution 1ns, wraps every 4398046511103ns
> > Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.008075] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
> > Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.066216] pid_max: default: 32768 minimum: 301
> > Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.071035] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
> > Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.078247] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
> > Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.089166] rcu: Hierarchical SRCU implementation.
> > Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.094202] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
> > Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.102324] smp: Bringing up secondary CPUs ...
> > Mon Apr  3 20:12:08 2023 kern.info kernel: [    0.107601] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[... cut ...]
> > Mode: Master  Channel: 36 (5.180 GHz)  HT Mode: VHT80
> > Center Channel 1: 42 2: unknown
> > Tx-Power: 23 dBm  Link Quality: 54/70
> > Signal: -56 dBm  Noise: unknown
> > Bit Rate: 676.0 MBit/s
> > Encryption: WPA2 PSK (CCMP)
> > Type: nl80211  HW Mode(s): 802.11ac/n
> > Hardware: 14C3:7662 14C3:7662 [MediaTek MT76x2E]
> > TX power offset: none
> > Frequency offset: none
> > Supports VAPs: yes  PHY name: phy1

I got the same error after I re-flash the official image 2.28.62 then flashed openwrt 22.03.4

FredyVia, 2023-04-16:
here is my /proc/mtd
```
mtd0: 01000000 00010000 "ALL"
mtd1: 00030000 00010000 "Bootloader"
mtd2: 00010000 00010000 "Config"
mtd3: 00010000 00010000 "Bdata"
mtd4: 00010000 00010000 "Factory"
mtd5: 00010000 00010000 "crash"
mtd6: 00010000 00010000 "cfg_bak"
mtd7: 00100000 00010000 "overlay"
mtd8: 00d00000 00010000 "OS1"
mtd9: 00b30000 00010000 "rootfs"
mtd10: 00180000 00010000 "disk"
```

FredyVia, 2023-04-17:
it's my fault, my factory partition is broken(maybe breeds write that partition before), everything is ok after flash the factory parition

Djfe, 2023-04-17:
@Nick24-Kobe can we close this? do you need to restore the factory partition from a backup as well maybe?

webysther, 2023-05-15:
> I have Mi 4 A Giga (DVB4224GL) and my mtd on 22.03.2 is a tad different @Nick24-Kobe :
> 
> ```
> dev:    size   erasesize  name
> mtd0: 00030000 00010000 "u-boot"
> mtd1: 00010000 00010000 "u-boot-env"
> mtd2: 00010000 00010000 "Bdata"
> mtd3: 00010000 00010000 "factory"
> mtd4: 00010000 00010000 "crash"
> mtd5: 00010000 00010000 "cfg_bak"
> mtd6: 00100000 00010000 "overlay"
> mtd7: 00e80000 00010000 "firmware"
> mtd8: 00290eb0 00010000 "kernel"
> mtd9: 00bef150 00010000 "rootfs"
> ```
> 
> Have you tried opening router and confirmed flash chip manufacturer? If I am not mistaken, support for new flash chips has landed recently (#11538). I have had zero issues with router in general however this is the supported model from 2020.

I have the same MTD:

```
dev:    size   erasesize  name
mtd0: 00030000 00010000 "u-boot"
mtd1: 00010000 00010000 "u-boot-env"
mtd2: 00010000 00010000 "Bdata"
mtd3: 00010000 00010000 "factory"
mtd4: 00010000 00010000 "crash"
mtd5: 00010000 00010000 "cfg_bak"
mtd6: 00100000 00010000 "overlay"
mtd7: 00e80000 00010000 "firmware"
mtd8: 002942d4 00010000 "kernel"
mtd9: 00bebd2c 00010000 "rootfs"
mtd10: 00820000 00010000 "rootfs_data"
```

and same problem, in my case 22.03.5, fresh or using the same settings, tried change a few things, but I only get disconnected (AP-STA-DISCONNECTED) and no error (log level error):

```
Mon May 15 07:02:29 2023 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED xx:xx:xx:xx:xx:xx
Mon May 15 07:02:29 2023 daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: authenticated
Mon May 15 07:02:29 2023 daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: associated (aid 5)
Mon May 15 07:02:29 2023 daemon.notice hostapd: wlan0: AP-STA-CONNECTED xx:xx:xx:xx:xx:xx
Mon May 15 07:02:29 2023 daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx WPA: pairwise key handshake completed (RSN)
Mon May 15 07:02:29 2023 daemon.notice hostapd: wlan0: EAPOL-4WAY-HS-COMPLETED xx:xx:xx:xx:xx:xx
Mon May 15 07:02:32 2023 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.1.67 xx:xx:xx:xx:xx:xx
Mon May 15 07:02:32 2023 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.1.67 xx:xx:xx:xx:xx:xx pwci01

get disconnected from (Intel AX201)

Mon May 15 07:02:59 2023 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED xx:xx:xx:xx:xx:xx
Mon May 15 07:02:59 2023 daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: authenticated
Mon May 15 07:02:59 2023 daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: associated (aid 5)
Mon May 15 07:03:00 2023 daemon.notice hostapd: wlan0: AP-STA-CONNECTED xx:xx:xx:xx:xx:xx
Mon May 15 07:03:00 2023 daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx WPA: pairwise key handshake completed (RSN)
Mon May 15 07:03:00 2023 daemon.notice hostapd: wlan0: EAPOL-4WAY-HS-COMPLETED xx:xx:xx:xx:xx:xx
Mon May 15 07:03:03 2023 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.1.67 xx:xx:xx:xx:xx:xx
Mon May 15 07:03:03 2023 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.1.67 xx:xx:xx:xx:xx:xx pwci01
```

The pattern is using more modern device (AX ones) I have problems, the old ones take more time to happen, the behavior is equal in 2Ghz and 5Ghz, I have tried keep only one, better:

- Change channel *not work*
- Keep only 2GHZ *not work*
- Keep only 5GHz *not work*
- Change mode between N and legacy *not work*
- Disable WMM *not work*
- Beacon interval 300 *not work*
- Distance optimization 10 meters *not work*
- Coverage cell density (every mode) *not work*
- Country code US/IT/Default *not work*
- Force 40MHz mode *not work*
- Between 20MHz and 40MHz *not work*
- Downgrade from WP3 to WP2 *not work*
- Isolate Clients *not work*
- Disable short preamble *not work*
- DTIM Interval 1 *not work*
- Disable Disassociate on low ack *not work*
- Reseted my wifi client settings *not work*

Signal: -50
Noise: -100
Phy: ~50Mbit/s
Load (1/5/15): 0.5-1.0

Any suggestions?

webysther, 2023-05-15:
`
wlan0     ESSID: "N/A"
          Access Point: N/A
          Mode: Master  Channel: 11 (2.462 GHz)  HT Mode: HT20
          Center Channel 1: 11 2: unknown
          Tx-Power: 23 dBm  Link Quality: 56/70
          Signal: -54 dBm  Noise: unknown
          Bit Rate: 127.0 MBit/s
          Encryption: mixed WPA/WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11b/g/n
          Hardware: 14C3:7603 14C3:7603 [MediaTek MT7603E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0
`

kokutoukiritsugu, 2023-06-29:
maybe try Xiaomi Mi Router 4A Gigabit v2 Edition

https://github.com/openwrt/openwrt/issues/11645#issuecomment-1613008631

kvanbiesen, 2023-06-29:
judging from the log out he provides, he uses an en25qx128a  chip which make me think that its an "old" device with the first chipset.

@webysther 

i have older snapshot running on my router, and i nly get this:
Thu Jun 29 12:44:38 2023 daemon.notice hostapd: phy1-ap0: AP-STA-DISCONNECTED c4:5d:83:3e:79:dd
Thu Jun 29 12:44:38 2023 daemon.info hostapd: phy1-ap0: STA c4:5d:83:3e:79:dd IEEE 802.11: disassociated due to inactivity
Thu Jun 29 12:44:39 2023 daemon.info hostapd: phy1-ap0: STA c4:5d:83:3e:79:dd IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)

But i think this as intented, as the device goes to sleep. (dont have AX devices tho)

zandatsu07, 2024-07-25:
I don't have a proper documentation, but i have the same problem in 5Ghz wifi since i upgraded my OpenWrt from david bentham's old [build](https://gitlab.com/db260179/xiaomi-m4a/-/releases) to 21.02, since then, my wifi can't even reach 400Mbps up&down anymore.