openwrt/openwrt #11577: sunxi: missing dependency on arm-trusted-firmware-sunxi-a64

stintel, 2022-12-20:
### Describe the bug

The olimex_a64-olinuxino and olimex_a64-olinuxino-emmc devices (and probably others) lack dependencies for building arm-trusted-firmware-sunxi-a64.

### OpenWrt version

r21563-9ed1830bdc

### OpenWrt target/subtarget

sunxi/cortexa53

### Device

Olinuxino A64 eMMC

### Image kind

Self-built image

### Steps to reproduce

```
printf "CONFIG_TARGET_sunxi=y\nCONFIG_TARGET_sunxi_cortexa53=y\nCONFIG_TARGET_sunxi_cortexa53_DEVICE_olimex_a64-olinuxino-emmc=y\n" > .config
make defconfig
grep arm-trusted-firmware-sunxi-a64 .config
```

### Actual behaviour

```
# CONFIG_PACKAGE_arm-trusted-firmware-sunxi-a64 is not set
```

### Expected behaviour

```
CONFIG_PACKAGE_arm-trusted-firmware-sunxi-a64=y
```

### Additional info

This is not a problem on the buildbots because they enable all packages. However, when building u-boot this throws the following non-critical warning:

```
WARNING: BL31 file /home/stijn/Development/OpenWrt/openwrt/staging_dir/target-aarch64_cortex-a53_musl/image/bl31_sun50i_a64.bin NOT found, resulting binary is non-functional
```

The resulting u-boot binary is indeed non-functional, as it lacks bl31 completely. This is observed in serial console as a freeze after the following lines:

```
U-Boot SPL 2020.04-OpenWrt-r21561-6675a9aaf4 (Dec 20 2022 - 13:57:34 +0000)
DRAM: 1024 MiB
Trying to boot from MMC1
```

Steps to improve the situation:
- [ ] add dependency
- [ ] ideally u-boot build fails on the above warning so we don't end up with an unbootable image

### Diffconfig

Not relevant with the steps to reproduce.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

john-tho, 2022-12-21:
I think this is because the depend u-boot package has the manufacturer name in front of the package name, so depends on a package that does not exist:
https://github.com/openwrt/openwrt/blob/9ed1830bdc1e58efb3e5b17c0e484e1a2655b550/package/boot/uboot-sunxi/Makefile#L37
https://github.com/openwrt/openwrt/blob/9ed1830bdc1e58efb3e5b17c0e484e1a2655b550/package/boot/uboot-sunxi/Makefile#L41

Try:
```Makefile
 DEPENDS:=+PACKAGE_u-boot-a64-olinuxino-emmc:arm-trusted-firmware-sunxi-a64 
```

Same with the other device.
These are the only sunxi a64 devices in tmp/.config-packages which have
```Makefile
select firmware if PACKAGE
```
the rest don't have `if PACKAGE` because they are defined by their own package

ynezz, 2022-12-21:
> ideally uboot-sunxi build fails on the above warning so we don't end up with an unbootable image
https://lists.denx.de/pipermail/u-boot/2020-March/403105.html

stintel, 2022-12-21:
> Try:
> 
> ```makefile
>  DEPENDS:=+PACKAGE_u-boot-a64-olinuxino-emmc:arm-trusted-firmware-sunxi-a64 
> ```

Thanks for the suggestion. Will probably have a look at this again next year.

stintel, 2022-12-21:
> https://lists.denx.de/pipermail/u-boot/2020-March/403105.html and https://lists.denx.de/pipermail/u-boot/2020-March/403293.html

Looks like in current versions that script is gone. Unfortunately there are too many patches to easily bump uboot-sunxi to something newer. Oh well. What about adding an explicit check for that file to exist in one of the OpenWrt Makefiles as a bandaid?