openwrt/openwrt #11578: WARNING: pip is configured with locations that require TLS/SSL, however the ssl module in Python is not available.

very20101, 2022-12-21:
ARMv8 
ubuntu 22.04
========================
 make[2] -C feeds/packages/lang/python/python-yaml compile
    ERROR: package/feeds/packages/python-yaml failed to build.
make -r package/compile: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
..............

work/Openwrt_retry/Openwrt_retry/workspace/openwrt/staging_dir/hostpkg/include/python3.10" LDFLAGS="-L/home/runner/work/Openwrt_retry/Openwrt_retry/workspace/openwrt/staging_dir/host/lib -L/home/runner/work/Openwrt_retry/Openwrt_retry/workspace/openwrt/staging_dir/hostpkg/lib -L/home/runner/work/Openwrt_retry/Openwrt_retry/workspace/openwrt/staging_dir/target-aarch64_cortex-a53_musl/host/lib -lpython3.10 -Wl,-rpath,/home/runner/work/Openwrt_retry/Openwrt_retry/workspace/openwrt/staging_dir/hostpkg/lib" PYTHONPATH="/home/runner/work/Openwrt_retry/Openwrt_retry/workspace/openwrt/staging_dir/hostpkg/lib/python3.10:/home/runner/work/Openwrt_retry/Openwrt_retry/workspace/openwrt/staging_dir/hostpkg/lib/python3.10/site-packages" PYTHONDONTWRITEBYTECODE=0 PYTHONOPTIMIZE="" /home/runner/work/Openwrt_retry/Openwrt_retry/workspace/openwrt/staging_dir/hostpkg/bin/pip3.10 --cache-dir "/home/runner/work/Openwrt_retry/Openwrt_retry/workspace/openwrt/dl/pip-cache" --disable-pip-version-check install --no-binary :all: --require-hashes  --requirement ../host-pip-requirements/ply.txt  ' 
WARNING: pip is configured with locations that require TLS/SSL, however the ssl module in Python is not available.
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/ply/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/ply/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/ply/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/ply/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/ply/
Could not fetch URL https://pypi.org/simple/ply/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='pypi.org', port=443): Max retries exceeded with url: /simple/ply/ (Caused by SSLError("Can't connect to HTTPS URL because the SSL module is not available.")) - skipping
ERROR: Could not find a version that satisfies the requirement ply==3.11 (from versions: none)
ERROR: No matching distribution found for ply==3.11
make[2]: *** [Makefile:48: /home/runner/work/Openwrt_retry/Openwrt_retry/workspace/openwrt/build_dir/target-aarch64_cortex-a53_musl/pypi/pycparser-2.21/.built] Error 1
make[2]: Leaving directory '/home/runner/work/Openwrt_retry/Openwrt_retry/workspace/openwrt/feeds/packages/lang/python/python-pycparser'
time: package/feeds/packages/python-pycparser/compile#0.69#0.14#8.63
    ERROR: package/feeds/packages/python-pycparser failed to build.
make[1]: *** [package/Makefile:116: package/feeds/packages/python-pycparser/compile] Error 1
make[1]: Leaving directory '/home/runner/work/Openwrt_retry/Openwrt_retry/workspace/openwrt'
make: *** [/home/runner/work/Openwrt_retry/Openwrt_retry/workspace/openwrt/include/toplevel.mk:231: package/compile] Error 2
Error: Process completed with exit code 2.

lovehackintosh, 2022-12-21:
Hi,see https://github.com/openwrt/packages/pull/20137

very20101, 2022-12-21:
Thank you! it works.

closed 2022-12-21 by very20101