openwrt/openwrt #11580: [22.03] CI: tools: fail on macos

Ansuel, 2022-12-21:
Hi, currently we have a problem with mkimage on our CI tools tests on macos for branch 22.03. I searched some missing commit but still I can't find the cause...

Anyone have a clue?

This is the related test https://github.com/openwrt/openwrt/actions/runs/3614116572/jobs/6090334003

---

@neheb Calling you wonder if you recognize that error?

neheb, 2022-12-21:
No idea

neheb, 2022-12-31:
@Ansuel maybe https://github.com/u-boot/u-boot/commit/3b142045e8a7f0ab17b6099e9226296af45967d0 needs backporting.

Ansuel, 2022-12-31:
@neheb the strange thing is that 21.02 works correctly.

for some reason the error is in missing sdl2-config ?

neheb, 2022-12-31:
sdl2-config… strange.

Spudz76, 2023-01-01:
I would personally debug the `logs/tools/mkimage/compile.txt-Makefile:697: pipe: Too many open files` errors, maybe disable parallel compile?

Ansuel, 2023-01-01:
You can download the full compile log

Spudz76, 2023-01-01:
Well yeah but it's all the same stuff it shows on the web view.

There is a rash of "Too many open files" right before outputs don't exist, which don't exist probably because it couldn't make any more pipes so the whole stack crashed (unrelated to any code bugs).  Something is using too many forks and should be set non-parallel to triage.

Or, more likely the MacOS environment is set too small (compared to others) and needs to be fixed.

Ansuel, 2023-01-06:
https://github.com/Ansuel/openwrt/actions/runs/3856007408/jobs/6571702208 

btw this is the complete log with -j1 any idea? @Spudz76 @neheb

robimarko, 2023-01-06:
Changing the `uLimit` from 256 should solve this.

Spudz76, 2023-01-06:
Fixing the too many open files thing by raising ulimit will not hurt.
```
$ grep 'Too.many.open' 1_build-macos-latest.txt  | wc -l
11208
```

But then there are a few things I filtered out:

`dosfstools` is a little angry
```
2023-01-06T15:40:41.1430990Z curl: (22) The requested URL returned error: 404
2023-01-06T15:40:41.6383540Z curl: (22) The requested URL returned error: 412
2023-01-06T15:40:44.1349650Z configure.ac:75: warning: macro 'AM_ICONV' not found in library
2023-01-06T15:40:45.3205950Z configure.ac:75: error: possibly undefined macro: AM_ICONV
2023-01-06T15:41:04.1968220Z time: tools/dosfstools/compile#10.00#6.66#23.31
```

`e2fsprogs` says `configure: WARNING: unrecognized options: --enable-threads` but that's just ignored

`fakeroot` some more 404's and autoconf is apparently too old (?)
```
2023-01-06T15:43:05.5736280Z curl: (22) The requested URL returned error: 404
2023-01-06T15:43:05.6376250Z curl: (22) The requested URL returned error: 404
2023-01-06T15:43:06.1126410Z curl: (22) The requested URL returned error: 404
2023-01-06T15:43:09.8896870Z configure.ac:3: error: Autoconf version 2.71 or higher is required
2023-01-06T15:43:09.8951510Z aclocal.real: error: echo failed with exit status: 63
2023-01-06T15:43:49.8337900Z time: tools/fakeroot/compile#19.57#14.34#44.48
```

`findutils` seems extra upset, for some reason it adds `-D__nonnull\(params\)=` which triggers tons of warnings:
```
2023-01-06T15:47:26.7270020Z <command line>:2:9: warning: '__nonnull' macro redefined [-Wmacro-redefined]
2023-01-06T15:48:06.9339750Z time: tools/findutils/compile#84.39#56.89#256.99
```

`mkimage` seems to be loading linux headers which is probably wrong (and redefining `__weak` which is probably bad), and then it crashes everything
```
2023-01-06T15:51:47.0383630Z include/linux/compiler_attributes.h:271:9: warning: '__weak' macro redefined [-Wmacro-redefined]
2023-01-06T16:07:30.8170270Z make[5]: *** [scripts/Makefile.host:95: tools/gen_eth_addr] Error 1
2023-01-06T16:07:30.8173920Z make[4]: *** [Makefile:2081: tools-only] Error 2
2023-01-06T16:07:30.8253360Z make[3]: *** [Makefile:555: __build_one_by_one] Error 2
2023-01-06T16:07:30.8256400Z make[2]: *** [Makefile:44: /Volumes/OpenWrt/openwrt/build_dir/host/u-boot-2022.01/.built] Error 2
2023-01-06T16:07:30.8262950Z time: tools/mkimage/compile#398.66#417.97#976.69
```

Ansuel, 2023-01-06:
I should run a similar test on master and we should compare the output...
Looks like something is missing in the config phase of these tools?

But why using Linux container works correctly? Are we missing something
specific for MacOS? Like some required include?

Spudz76, 2023-01-06:
Apple does like to make breaking changes, especially in XCode.  But it could also be something with the runner-image.

[the readme for the current github macos-12 runner](https://github.com/actions/runner-images/blob/macOS-12/20221215.1/images/macos/macos-12-Readme.md) says it's XCode 14.0.1 which to me means lots of things should break (recently bumped major from 13.x)

Does the 21.02 build use the `20221215.1` version of the macos-12 runner?  If it uses an older image, check the matching image-readme (like above) and see what XCode and such it uses.

Also IMO it should always be using `clang` because that's what comes with XCode, although the images include GCC (probably from homebrew).  The Clang version major matches XCode version so I guess you could consider them belonging together, and probably tested together.  Maybe some packages find gcc and prefer it (mv/rm it in early image setup / pre-build? can't use it if they can't find it...)

robimarko, 2023-01-07:
Tried the `uLimit` change and that did not fix anything.