openwrt/openwrt #11588: Image Builder Makefile ignores variable PACKAGES, USER_PACKAGES works instead

odon-noda, 2022-12-21:
### Describe the bug

## Issue 
Variable PACKAGES does not affect packages built into the image

## Workaround
Use variable USER_PACKAGES instead

## Root cause
[Makefile](https://github.com/openwrt/openwrt/blob/master/target/imagebuilder/files/Makefile) line 240 onwards

### OpenWrt version

r19803-9a599fee93

### OpenWrt target/subtarget

ath79/generic

### Device

TP-Link Archer C6 v2 EU

### Image kind

Official downloaded image

### Steps to reproduce

1. Download [openwrt-imagebuilder-22.03.2-ath79-generic.Linux-x86_64.tar.xz](https://downloads.openwrt.org/releases/22.03.2/targets/ath79/generic/) from Sat Oct 15 22:12:04 2022
2. Follow [Image Builder Usage](https://openwrt.org/docs/guide-user/additional-software/imagebuilder?rev=1671558902#usage) instructions
3. Build image
```
user@host:~/images/openwrt-imagebuilder-22.03.2-ath79-generic.Linux-x86_64$ make image PACKAGES="auc"
```
The package ```auc``` is missing in the generated image ```bin/targets/ath79/generic/openwrt-22.03.2-ath79-generic-tplink_archer-c6-v2-squashfs-sysupgrade.bin```

### Actual behaviour

```
# auc
-ash: auc: not found
#
```
Command ```auc``` does not exist after installing generating image into the device

### Additional info

I've found it using this debug 
```
user@host:~/images/openwrt-imagebuilder-22.03.2-ath79-generic.Linux-x86_64$ diff Makefile https://raw.githubusercontent.com/openwrt/openwrt/master/target/imagebuilder/files/Makefile
116d115
< # echo 'BUILD_PACKAGES = "$(BUILD_PACKAGES)"'
241,243d239
< #     $(info PACKAGES is $(value PACKAGES))
< #     $(info BUILD_PACKAGES is $(BUILD_PACKAGES))
<
253,256d248
<
<       $(info PACKAGES is $(value PACKAGES))
<       $(info BUILD_PACKAGES is $(BUILD_PACKAGES))
<       $(info USER_PACKAGES is $(USER_PACKAGES))
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ynezz, 2023-01-03:
I'm not able to reproduce this bug:

```
$ make image PROFILE="tplink_archer-c6-v2" PACKAGES="auc" 2>&1 | grep auc

Packages: auc ath10k-firmware-qca9888-ct base-files busybox ca-bundle dnsmasq dropbear firewall4 fstools kernel kmod-ath10k-ct kmod-ath9k kmod-gpio-button-hotplug kmod-nft-offload libc libgcc libustream-wolfssl logd mtd netifd nftables odhcp6c odhcpd-ipv6only opkg ppp ppp-mod-pppoe procd procd-seccomp procd-ujail swconfig uboot-envtools uci uclient-fetch urandom-seed urngd wpad-basic-wolfssl
Installing auc (0.3.1-1) to root...
Downloading https://downloads.openwrt.org/releases/22.03.2/packages/mips_24kc/packages/auc_0.3.1-1_mips_24kc.ipk
Configuring auc.

$ grep auc bin/targets/ath79/generic/*.manifest
auc - 0.3.1-1

$ binwalk -e bin/targets/ath79/generic/openwrt-22.03.2-ath79-generic-tplink_archer-c6-v2-squashfs-sysupgrade.bin
$ find _openwrt-22.03.2-ath79-generic-tplink_archer-c6-v2-squashfs-sysupgrade.bin.extracted -name auc
_openwrt-22.03.2-ath79-generic-tplink_archer-c6-v2-squashfs-sysupgrade.bin.extracted/squashfs-root/usr/sbin/auc
_openwrt-22.03.2-ath79-generic-tplink_archer-c6-v2-squashfs-sysupgrade.bin.extracted/auc
```

closed 2023-01-03 by ynezz

odon-noda, 2023-01-06:
Hi @ynezz , you are right

The root cause is that Makefile does not consume environment variable PACKAGES exported from -bash

```
$  # from environment 
$ export PACKAGES="auc"; export PROFILE="tplink_archer-c6-v2";  make image 2>&1 | grep auc ; echo $?
1

$ # inline
# make image PROFILE="tplink_archer-c6-v2" PACKAGES="auc" 2>&1 | grep auc ; echo $?
PACKAGES is auc
Packages: auc ath10k-firmware-qca9888-ct base-files busybox ca-bundle dnsmasq dropbear firewall4 fstools kernel kmod-ath10k-ct kmod-ath9k kmod-gpio-button-hotplug kmod-nft-offload libc libgcc libustream-wolfssl logd mtd netifd nftables odhcp6c odhcpd-ipv6only opkg ppp ppp-mod-pppoe procd procd-seccomp procd-ujail swconfig uboot-envtools uci uclient-fetch urandom-seed urngd wpad-basic-wolfssl
Installing auc (0.3.1-1) to root...
Copying /home/kaja/images/openwrt-imagebuilder-22.03.2-ath79-generic.Linux-x86_64/dl/auc_0.3.1-1_mips_24kc.ipk.
Configuring auc.
0
$
```

Thank you for teaching me about the command ```binwalk```.