openwrt/openwrt #11590: AVM Fritzbox 7520/7530 only 100baseT on two ports

Chris56, 2022-12-22:
### Describe the bug

openwrt-22.03.2 on 7520/7530:
Fresh install.
Before I have checked with the original OEM OS: all 4 LAN Ports had been switched to 1000baseT and all 4 ports did actually use that link speed.
After installing openwrt, two of the four port only operate on 100baseT !!
This behaviour would be the case with an OEM 7520, where the AVM firmware reduces two ports to 100tbaseT.
My guess is, that when powering on, this is the default and the 100baseT ports need to be switched to 1000baseT by the booting OS . But this is only a guess.
As AVM can by - choice switch - between 100 and 1000 on a running system, there must be a way to do it.
Maybe I only oversee a setting, but so far I haven't seen any.
Chris

### OpenWrt version

22.03.2

### OpenWrt target/subtarget

ipq40xx/generic

### Device

Fritz!box 7520/7530

### Image kind

Official downloaded image

### Steps to reproduce

Just install an see!

### Actual behaviour

2 ports only run on 100baseT link speed.

### Expected behaviour

All 4 port run on 1000baseT link speed.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

dhewg, 2022-12-22:
Yeah, this is an artificial limitation.
With DSA all ports are running at 1000baseT, so nobody bothered to fix it for outdated non-DSA drivers.

dhewg, 2022-12-22:
See also https://github.com/openwrt/openwrt/pull/4721#issuecomment-996504471
That's merged on the master branch, so there you get DSA which in return fixes this

Chris56, 2022-12-22:
The documentation: https://openwrt.org/toh/avm/avm_fritz_box_7530
says:
...  but its OEM software reduces the speed of 2x LAN ports and the USB port. Those limits do not apply to OpenWRT.
So this is a false statement. Is there anybody, who can fix it?
Alternatively: (I am not a driver coder). How can I make use of the named "DSA" driver (whatever this is ..)

dhewg, 2022-12-22:
Oh, and if you want to update to master and are interested in using the DSL line with OpenWrt: https://forum.openwrt.org/t/adding-support-for-vrx518-and-maybe-vrx320/55160/214

Chris56, 2022-12-22:
Is there a binary, which I can use with sysupgrade?

dhewg, 2022-12-22:
For the DSL part: no, because that's not merged due to firmware licensing issues.
You can download a master snapshot though, that'll give you DSA but no DSL

Chris56, 2022-12-22:
Thanks for the insights.
DSL too would be great. However I'm preparing this box for a production environment. Therefore, unfortunately, I cannot help debugging dsl.

closed 2022-12-22 by Chris56