openwrt/openwrt #11599: [odhcpd] IPv6 relay error: cannot discover LAN device, Bad file descriptor sending to wan@pppoe-wan

Alice-space, 2022-12-23:
### Describe the bug

## Device Version: 
```
OpenWrt 22.03.2
odhcpd-ipv6only 2022-03-22-860ca900-1
odhcp6c 2022-08-05-7d21e8d8-18 
```
## Description

After pppoe reconnection, the IPv6-PD prefix is no longer issued, only a /64 address is issued at `pppoe-wan`, and none of the LAN devices have IPv6 addresses.

After trying with the relay method (see config file below), I can get the address assigned, but the OpenWrt log keeps reporting errors and the LAN devices cannot access any ipv6 site.

Syslog:
```
odhcpd[1605]: setsockopt(PACKET_ADD_MEMBERSHIP): Invalid argument
odhcpd[1605]: Failed to send to 240e:xxx%wan@pppoe-wan (Bad file descriptor)
```
`/etc/config/dhcp`
```
config dhcp 'lan'
	option interface 'lan'
	option start '100'
	option limit '150'
	option leasetime '12h'
	option dhcpv4 'server'
	option ra 'relay'
	option ndp 'relay'
	option dhcpv6 'relay'

config dhcp 'wan'
	option interface 'wan'
	option ignore '1'
	option master '1'
	option ra 'relay'
	option ndp 'relay'
	option dhcpv6 'relay'
```

## Temporary Solution
I ping6 the router on a LAN device and I can access v6 sites. 
To Solve the problem automatically I added script
```
#!/bin/sh
[ "$1" = "pppoe-wan" ] || exit 0
ip -6 route add default dev br-lan table 6
ip -6 rule add iif $1 lookup 6
```

Is this a bug of odhcpd-ipv6only?

## VLAN

I set up my `pppoe-wan` interface on an VLAN device `wan.2`. I don't know if VLAN will affect odhcpd.
`/etc/config/network`
```
config interface 'wan'
	option proto 'pppoe'
	option ipv6 'auto'
	option username 'xxx'
	option password 'xxx'
	option device 'wan.2'
	option ip6assign '64'

config device
	option type '8021q'
	option ifname 'wan'
	option vid '2'
	option name 'wan.2'
```

Chinese Ver. https://www.v2ex.com/t/903292

### OpenWrt version

r19803-9a599fee93

### OpenWrt target/subtarget

ramips/mt7621

### Device

Newifi-D2

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

stintel, 2022-12-23:
I guess multicast is not supported on PPPoE interfaces. Not sure if this is actually fixable. @dedeckeh thoughts?

biltongza, 2026-08-21:
I was getting spurious bad file descriptors trying to send to pppoe-wan, it turns out my issue was that I had configured NDP relay on the wan interface. When I turned that off the log spam stopped.