openwrt/openwrt #11604: Cannot install package luci-app-vpn-policy-routing ZBT WE1326 BKC

amanjuman, 2022-12-24:
### Describe the bug

make -j4 image PROFILE=zbtlink_zbt-we1326 PACKAGES="luci -luci-theme-bootstrap -dnsmasq luci-theme-material luci-app-sqm luci-app-wireguard luci-app-vpn-policy-routing dnsmasq-full iperf3 wget curl ca-certificates ca-bundle nano"

Collected errors:
 * opkg_install_cmd: Cannot install package luci-app-vpn-policy-routing.
make[2]: *** [Makefile:169: package_install] Error 255
make[1]: *** [Makefile:124: _call_image] Error 2
make: *** [Makefile:242: image] Error 2

### OpenWrt version

22.03.2

### OpenWrt target/subtarget

ramips-mt7621.Linux-x86_64

### Device

zbtlink_zbt-we1326

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

anaelorlinski, 2023-01-07:
on 22.03 use luci-app-pbr instead

jow-, 2023-01-07:
As written above, the package has been superseded by `luci-app-pbr` in 22.03 and later.

closed 2023-01-07 by jow-