openwrt/openwrt #11608: Current bcm27xx Snapshots missing BCMGENET Ethernet driver

satmandu, 2022-12-25:
### Describe the bug

```
root@office:/lib/modules/5.15.84# zcat /proc/config.gz | grep GENET
CONFIG_BCMGENET=m
root@office:/lib/modules/5.15.84#
```

Am I missing which kmod package this driver has been moved to?

This provides the bcmgenet driver used for the stock CM4/RPI4 Ethernet driver.

I'm using the DFrobot router board with my CM4.

Luckily recent kernel updates add the other driver I need for the other Ethernet interface on this board by default!

But that doesn't help me get the main interface up...

(On raspberry pi kernels,`CONFIG_BCMGENET=y` is set.)

### OpenWrt version

r21584-b2bfea481e

### OpenWrt target/subtarget

bcm27xx/bcm2711

### Device

DFrobot CM4 routerboard

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

john-tho, 2022-12-25:
Looking at 5.15.85 (later than this snapshot image—buildbot not caught up yet), because it has a Kconfig fix for BCMGENET: https://lore.kernel.org/all/20221125115003.30308-1-yuehaibing@huawei.com/)

I think this is another issue with `ALL_KMODS` and `PTP_1588_CLOCK`. I think somehow packages can downgrade `PTP_1588_CLOCK_OPTIONAL` from `=y` to `=m`, then symbols which depend on `PTP_1588_CLOCK_OPTIONAL` can only be modules (see similar https://github.com/openwrt/openwrt/commit/97c77fff28cf001399f33c7bc1ec6687ba18450b & https://github.com/openwrt/openwrt/commit/8be6350f6646b37929f40a9a482343f22af2065b)

From `PTP_1588_CLOCK_OPTIONAL` introduction: https://lore.kernel.org/lkml/20210812183509.1362782-1-arnd@kernel.org/
> introduce a separate Kconfig symbol "CONFIG_PTP_1588_CLOCK_OPTIONAL" that any driver can depend on if it is able to use PTP support when available, but works fine without it. Whenever CONFIG_PTP_1588_CLOCK=m, those drivers are then prevented from being built-in

Quick & dirty fix (for 5.15.85), and will want doing on the other `PTP_1588_CLOCK_OPTIONAL=y, BCMGENET=y` OpenWrt kernel configs: set `PTP_1588_CLOCK` as built-in
```patch
commit 7756fdf2873698e4e8791e6fc27835f5c57dbd63
Author: John Thomson <git@johnthomson.fastmail.com.au>
Date:   Sun Dec 25 22:43:38 2022 +1000

    bcm27xx: bcm2711: build-in PTP_1588_CLOCK
    
    Signed-off-by: John Thomson <git@johnthomson.fastmail.com.au>

diff --git a/target/linux/bcm27xx/bcm2711/config-5.15 b/target/linux/bcm27xx/bcm2711/config-5.15
index 2f8f9dccb1b2..a721556d2b7d 100644
--- a/target/linux/bcm27xx/bcm2711/config-5.15
+++ b/target/linux/bcm27xx/bcm2711/config-5.15
@@ -313,6 +313,7 @@ CONFIG_MUTEX_SPIN_ON_OWNER=y
 CONFIG_NEED_DMA_MAP_STATE=y
 CONFIG_NEED_SG_DMA_LENGTH=y
 CONFIG_NET_FLOW_LIMIT=y
+CONFIG_NET_PTP_CLASSIFY=y
 CONFIG_NET_SELFTESTS=y
 CONFIG_NLS=y
 CONFIG_NLS_ASCII=y
@@ -360,7 +361,9 @@ CONFIG_PM_SLEEP=y
 CONFIG_PM_SLEEP_SMP=y
 CONFIG_POWER_RESET=y
 CONFIG_POWER_SUPPLY=y
+CONFIG_PPS=y
 CONFIG_PRINTK_TIME=y
+CONFIG_PTP_1588_CLOCK=y
 CONFIG_PTP_1588_CLOCK_OPTIONAL=y
 CONFIG_PWM=y
 CONFIG_PWM_BCM2835=y

```

---
The process of working this out…

5.15.85 https://elixir.bootlin.com/linux/v5.15.85/source/drivers/net/ethernet/broadcom/Kconfig#L71
```
config BCMGENET
	tristate "Broadcom GENET internal MAC support"
	depends on HAS_IOMEM
	depends on PTP_1588_CLOCK_OPTIONAL || !ARCH_BCM2835
	select MII
	select PHYLIB
	select FIXED_PHY
	select BCM7XXX_PHY
	select MDIO_BCM_UNIMAC
	select DIMLIB
	select BROADCOM_PHY if ARCH_BCM2835
	help
	  This driver supports the built-in Ethernet MACs found in the
	  Broadcom BCM7xxx Set Top Box family chipset.
```

```
git grep -E "CONFIG_(BCMGENET|HAS_IOMEM|PTP_1588_CLOCK)" -- package/ target/linux/bcm27xx
package/kernel/linux/modules/other.mk:  CONFIG_PTP_1588_CLOCK \
package/kernel/linux/modules/other.mk:  KCONFIG:=CONFIG_PTP_1588_CLOCK_QORIQ
target/linux/bcm27xx/bcm2708/config-5.15:CONFIG_HAS_IOMEM=y
target/linux/bcm27xx/bcm2708/config-5.15:CONFIG_PTP_1588_CLOCK_OPTIONAL=y
target/linux/bcm27xx/bcm2709/config-5.15:CONFIG_BCMGENET=y
target/linux/bcm27xx/bcm2709/config-5.15:CONFIG_HAS_IOMEM=y
target/linux/bcm27xx/bcm2709/config-5.15:CONFIG_PTP_1588_CLOCK_OPTIONAL=y
target/linux/bcm27xx/bcm2710/config-5.15:CONFIG_HAS_IOMEM=y
target/linux/bcm27xx/bcm2710/config-5.15:CONFIG_PTP_1588_CLOCK_OPTIONAL=y
target/linux/bcm27xx/bcm2711/config-5.15:CONFIG_BCMGENET=y
target/linux/bcm27xx/bcm2711/config-5.15:CONFIG_HAS_IOMEM=y
target/linux/bcm27xx/bcm2711/config-5.15:CONFIG_PTP_1588_CLOCK_OPTIONAL=y
```
A minimal config does have BCMGENET builtin:
```
cat <<EOF>.config
CONFIG_TARGET_bcm27xx=y
CONFIG_TARGET_bcm27xx_bcm2711=y
CONFIG_TARGET_bcm27xx_bcm2711_DEVICE_rpi-4=y
CONFIG_DEVEL=y
CONFIG_EXPERIMENTAL=y
EOF

make defconfig
make V=s target/linux/compile
# I did interrupt after kernel proper started building
grep -E "CONFIG_(BCMGENET|HAS_IOMEM|PTP_1588_CLOCK)" build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-5.15.85/.config

CONFIG_BCMGENET=y
# CONFIG_PTP_1588_CLOCK is not set
CONFIG_PTP_1588_CLOCK_OPTIONAL=y
CONFIG_HAS_IOMEM=y

```
With ALL_KMODS, `BCMGENET` get set `=m` only during compile
```
cat <<EOF>.config
CONFIG_TARGET_bcm27xx=y
CONFIG_TARGET_bcm27xx_bcm2711=y
CONFIG_TARGET_bcm27xx_bcm2711_DEVICE_rpi-4=y
CONFIG_DEVEL=y
CONFIG_EXPERIMENTAL=y
CONFIG_ALL_KMODS=y
EOF

make defconfig
make kernel_oldconfig
grep -E "CONFIG_(BCMGENET|HAS_IOMEM|PTP_1588_CLOCK)" build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-5.15.85/.config
# as for minimal build

make V=s target/linux/compile
# I did interrupt after kernel proper started building

grep -E "CONFIG_(BCMGENET|HAS_IOMEM|PTP_1588_CLOCK)" build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-5.15.85/.config

CONFIG_BCMGENET=m
CONFIG_PTP_1588_CLOCK=m
CONFIG_PTP_1588_CLOCK_OPTIONAL=m
# CONFIG_PTP_1588_CLOCK_KVM is not set
# CONFIG_PTP_1588_CLOCK_IDT82P33 is not set
# CONFIG_PTP_1588_CLOCK_IDTCM is not set
CONFIG_HAS_IOMEM=y
```

cut the kmods which depend on `kmod-ptp`, and `BCMGEMAC` stays builtin:
```
cat <<EOF>>.config
CONFIG_TARGET_bcm27xx=y
CONFIG_TARGET_bcm27xx_bcm2711=y
CONFIG_TARGET_bcm27xx_bcm2711_DEVICE_rpi-4=y
CONFIG_DEVEL=y
CONFIG_EXPERIMENTAL=y
EOF
git grep -B4 kmod-ptp -- package/ | sed -nE 's:.*KernelPackage/(.*):# CONFIG_PACKAGE_kmod-\1 is not set:p' >> .config
echo "# CONFIG_PACKAGE_kmod-ixgbevf is not set">>.config
echo "CONFIG_ALL_KMODS=y">>.config
make defconfig
grep kmod-ptp .config

# CONFIG_PACKAGE_kmod-ptp is not set

make V=s target/linux/compile
# I did interrupt after kernel proper started building

grep -E "CONFIG_(BCMGENET|HAS_IOMEM|PTP_1588_CLOCK)" build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-5.15.85/.config

CONFIG_BCMGENET=y
# CONFIG_PTP_1588_CLOCK is not set
CONFIG_PTP_1588_CLOCK_OPTIONAL=y
CONFIG_HAS_IOMEM=y

```

satmandu, 2022-12-26:
Is this fix in the 5.15.85 based snapshot? Or should I wait for a later snapshot for this fix?

satmandu, 2022-12-28:
@john-tho Who should I ask to try to get this fixed in the OpenWRT tree so that the snapshots work again? Or should this be addressed upstream on the kernel list?

dave14305, 2022-12-28:
Should be fixed by 5863363493565f4b651baeb83251769f2b6d8929

satmandu, 2022-12-29:
> Should be fixed by 5863363493565f4b651baeb83251769f2b6d8929

Thanks!

Any chance of asking for an earlier snapshot rebuild to confirm that this fixes the snapshot?

satmandu, 2022-12-29:
Ah I see there's a new snapshot build! I'll try that...

satmandu, 2022-12-29:
I can confirm that the latest snapshot fixes the issue!  Thanks all!

```
root@openwrt:~# opkg install kmod-ikconfig
Installing kmod-ikconfig (5.15.85-1) to root...
Downloading https://downloads.openwrt.org/snapshots/targets/bcm27xx/bcm2711/kmods/5.15.85-1-8f0c39ac4b8c4c04d21fcfb992948c34/kmod-ikconfig_5.15.85-1_aarch64_cortex-a72.ipk
Configuring kmod-ikconfig.
root@openwrt:~#  zcat /proc/config.gz | grep GENET
CONFIG_BCMGENET=y
```

closed 2022-12-29 by satmandu