openwrt/openwrt #11614: Firewall4 is not work for dockerd with bridge network mode ( maybe wireguard also )

0neday, 2022-12-26:
### Describe the bug

docker with bridge network mode
on router use 
```
root@OpenWrt:~#  curl --head 192.168.1.2:8123 -v
> HEAD / HTTP/1.1
> Host: 192.168.1.2:8123
> User-Agent: curl/7.86.0
> Accept: */*
>
* Empty reply from server
curl: (52) Empty reply from server
```
but on my win10 pc, not work

wireguard
on router 
```
root@OpenWrt:~# ifconfig   vpn-client
vpn-client Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00
          inet addr:10.0.1.100  P-t-P:10.0.1.100  Mask:255.255.255.255
          UP POINTOPOINT RUNNING NOARP  MTU:1420  Metric:1
          RX packets:10 errors:0 dropped:0 overruns:0 frame:0
          TX packets:54 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:992 (992.0 B)  TX bytes:2848 (2.7 KiB)

root@OpenWrt:~# ping 10.0.1.1
PING 10.0.1.1 (10.0.1.1): 56 data bytes
64 bytes from 10.0.1.1: seq=0 ttl=64 time=20.944 ms

```
on pc 
```
C:\Users\hongs>ping 10.0.1.1

正在 Ping 10.0.1.1 具有 32 字节的数据:
来自 192.168.1.2 的回复: 无法连到端口。
来自 192.168.1.2 的回复: 无法连到端口。
```

### OpenWrt version

OpenWrt 22.03.2 r19803-9a599fee93 / LuCI openwrt-22.03 branch git-22.288.45147-96ec0cd

### OpenWrt target/subtarget

OpenWrt 22.03.2 r19803-9a599fee93 / LuCI openwrt-22.03 branch git-22.288.45147-96ec0cd

### Device

x86

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Spudz76, 2022-12-26:
This is being tracked over in packages at https://github.com/openwrt/packages/issues/16818

`dockerd` has no checkmark so it has not been reworked yet

0neday, 2022-12-26:
thank you, reroll to 21.02.5, dockerd and wireguard work fine.

jow-, 2022-12-29:
Please follow up on openwrt/packages#16818

closed 2022-12-29 by jow-