openwrt/openwrt #11618: custom config

eslamsayed67, 2022-12-27:
### Describe the bug

When I add a file named files and inside it a file called etc and inside it a file named config and put the file to be modified for example the wireless file and reassemble the update file does not work on the device gives me this message mount_root: overlay filesystem has not been fully initialized yet

### OpenWrt version

snapshot

### OpenWrt target/subtarget

ath79

### Device

archer c7 v2

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Spudz76, 2022-12-27:
Better way is putting your settings into a high-numbered `uci-defaults` script instead (so it runs last).
`./files/etc/uci-defaults/99_wireless_local`
```
#!/bin/sh
uci batch <<EOC
    set wireless.default_radio0.ssid='ThisWIFI'
    set wireless.default_radio0.encryption='psk2'
    set wireless.default_radio0.key='bad_passw0rd'
    set wireless.default_radio0.wds='1'
    set wireless.default_radio1.ssid='ThisWIFI'
    set wireless.default_radio1.encryption='psk2'
    set wireless.default_radio1.key='bad_passw0rd'
    set wireless.default_radio1.wds='1'
    set wireless.radio0.beacon_int='150'
    set wireless.radio0.channel='auto'
    set wireless.radio0.country='US'
    set wireless.radio0.htmode='HT40'
    set wireless.radio0.txpower='auto'
    set wireless.radio1.channel='auto'
    set wireless.radio1.country='US'
    set wireless.radio1.htmode='VHT80'
    set wireless.radio1.txpower='auto'
    commit
EOC
```

eslamsayed67, 2022-12-28:
> 

thx pro for help me i will try

eslamsayed67, 2022-12-28:
> Better way is putting your settings into a high-numbered `uci-defaults` script instead (so it runs last). `./files/etc/uci-defaults/99_wireless_local`
>
> ```
> #!/bin/sh
> uci batch <<EOC
>     set [...]

it's work but default wifi is disable i need default is enable

jow-, 2022-12-29:
The quoted log messages do not hint at a bug, the rest is a general support question, please refer to the forum for that.

closed 2022-12-29 by jow-