openwrt/openwrt #11620: fw4: flushes entire nft ruleset when stopped, but only flushes fw4 table on restart

bluewavenet, 2022-12-27:
### Describe the bug

`service firewall restart` flushes the fw4 table (and all chains and rules within) as expected.

But:
`service firewall stop` flushes the entire nft ruleset, even tables not belonging to fw4.

So doing
`service firewall restart`
has a very different effect to doing
```
service firewall stop
service firewall start
```

This would go unnoticed if the only rules in the nft ruleset were those of fw4, but can be catastrophic if other packages have their own nft tables, possibly resulting in a soft brick of the router.

Tested on Gl-inet mt300mt-v2, and on numerous other devices/targets giving the same result.

### OpenWrt version

r19803-9a599fee93

### OpenWrt target/subtarget

ramips/mt76x8

### Device

Gl-inet mt300mt-v2, but also tested on numerous devices/targets giving the same result

### Image kind

Official downloaded image

### Steps to reproduce

First, list tables:
```
root@OpenWrt:~# nft list tables
table inet fw4
root@OpenWrt:~#
```

Now add a new table:
```
root@OpenWrt:~# nft add table inet test
root@OpenWrt:~# nft list tables
table inet fw4
table inet test
root@OpenWrt:~#
```

Restart fw4:
```
root@OpenWrt:~# service firewall restart
root@OpenWrt:~# nft list tables
table inet fw4
table inet test
root@OpenWrt:~#
```
Stop fw4:
```
root@OpenWrt:~# service firewall stop
root@OpenWrt:~# nft list tables
root@OpenWrt:~#
```
All tables have gone...

Start fw4:
```
root@OpenWrt:~# service firewall start
root@OpenWrt:~# nft list tables
table inet fw4
root@OpenWrt:~# 
```
Only the fw4 table remains, our test table has gone.

### Actual behaviour

As shown in Steps to Reproduce.

### Expected behaviour

fw4 should not flush and delete tables created by other packages or by manual creation
Restart and Stop/Start should have the same effect.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Spudz76, 2022-12-27:
Not sure this is a bug, I think it's intended behavior.  Pretty sure fw4 has various event hooks where packages can attach, to protect or recreate their custom tables.  I think `miniupnpd` has been converted and uses the hooks, as an example.

Stopping a firewall should result in blank everything, otherwise the firewall could still be (partially) active, right?

bluewavenet, 2022-12-28:
@Spudz76 

> Stopping a firewall should result in blank everything

It should blank everything **_that it added_**, then that firewall package will be totally inactive.

What if another package provides "firewalling" functionality with its own tables with chains at a higher priority (ie more negative) than those set by fw4?

An example of such a package is openNDS, a captive portal. openNDS has its own nft tables and chains for capturing, redirecting, access control, traffic monitoring, traffic control etc.. ie. it is a full functional firewall in its own right that expects to chain on to lower priority rules (eg. rules that can be set and managed by fw4).

fw4 has only a single nft table, aptly named "fw4" that contains all its chains.
Doing an fw4 daemon restart quite correctly flushes the `fw4` table, deletes it then recreates it again with all the chains defined in the `firewall` config file.

However, doing a stop/start clears the entire nftables ruleset. This is inconsistent behaviour and makes the incorrect assumption that "fw4" has full ownership of the nftables ruleset.

**_Fw4 manages only one single table in the bigger nftables world. Anything else should be outside its scope._**

As far as I am aware, there is no fw4 hook for a daemon stop or a daemon start - please correct me if I am wrong.
**There is a hook for daemon restart,** and indeed openNDS uses it for the purpose of controlling passthrough to the lower priority fw4 chains.

A quick look at `miniupnpd` makefile indicates it installs a script that will be executed when fw4 _restarts_, not when it starts after a stop.
My guess is that running:
```
service firewall stop
service firewall start
```

will break `miniupnpd`...?

dave14305, 2022-12-28:
> Fw4 manages only one single table in the bigger nftables world. Anything else should be outside its scope.

But additional tables can be added via fw4 includes (either ruleset-pre, ruleset-post, or by nftables scripts), so I think it gets complicated quickly.

bluewavenet, 2022-12-28:
@dave14305 
The fw4 ruleset resides entirely within the inet/fw4 table context.
No mechanism is provided by fw4 package to directly create new tables.

Sure you can do this indirectly with an include file, but that can also be done via any other package or on the command line.

With includes you are choosing to add something that fw4 does not understand, and if the something added is a table it will be outside that inet/fw4 table context.

**_This makes it very simple, rather than complicated_** - ie if it is not in the inet/fw4 table it is an addon.
An fw4 restart keeps it uncomplicated by not touching anything outside inet/fw4.

**BUT** an fw4 stop/start **MAKES** it complicated by willy-nilly deleting things it knows nothing about.

Spudz76, 2022-12-28:
The hook for stop or start are universal via dependency.  The firewall-using app should depend on fw4 so that when fw4 is stopped, they also stop themselves.  Then when fw4 is started again it will start the dependent-on-tables service(s) which will regenerate their special tables and link them in, just like at boot.

Interesting OpenNDS is not in the list of things being checked [in the firewall4 conversion](https://github.com/openwrt/packages/issues/16818).  Perhaps it is still using the old firewall methods which would explain why it doesn't interoperate quite right with fw4.

Blank flushed everything is the correct state for a stopped fw4.

bluewavenet, 2022-12-28:
@Spudz76 
fw4 is not nftables, it is a management utility for its own inet/fw4 subset of nftables.
As I said before:

> **_Fw4 manages only one single table in the bigger nftables world. Anything else should be outside its scope._**

fw4 is an excellent firewall package for OpenWrt, but this small issue can cause mayhem with other packages that have their own nft tables.
The problem is in the init.d file.
The section:
```
stop_service() {
	fw4 flush
}

```
should read:
```
stop_service() {
	fw4 stop
}

```

> Interesting OpenNDS [.....] doesn't interoperate quite right with fw4.

That is not the case, the latest v9.10.0 openNDS interoperates just fine with fw4.

It was mentioned here because it is an example of a "firewall" class package that has the ability to chain onwards to a **_lower priority firewall_** nft ruleset (as found in fw4).

This issue has been created here to indicate to others the potential problems caused by this behaviour.

Spudz76, 2022-12-28:
Cool.  Stopped state of fw4 is no tables / all flushed.  I don't know how your decree of something that isn't how it is, makes it true.

bluewavenet, 2022-12-28:
@Spudz76 

> I don't know how your decree of something that isn't how it is, makes it true.

Eh?

>  Cool. Stopped state of fw4 is no tables / all flushed

This is very wrong.

Scenario:
1. Start a firewall class package that adds its own nft ruleset and has the ability to chain onwards to tables with lower priority chains (for example fw4).
2. Start fw4, _it creates the inet/fw4 table_.
4. Stop fw4. This flushes/deletes **_the entire nftables ruleset_** breaking the first firewall class package.

How is this cool?

If you find you need to do a clean-up of random rules that you have lost track of, then sure, you can do an `nft flush ruleset` if all else fails.

Even better, why not have an additional config option:
`option enable_flush_ruleset_on_stop '1'`

and have the init.d script read the option and do the business.

Then everyone will be happy...

dave14305, 2022-12-28:
If I create an additional table via the documented fw4 include mechanism, I would expect fw4 to delete it on a stop/restart, otherwise there is no other method for it to be cleaned, since it isn’t part of a separate package.

I imagine there is a fundamental design principle that guides the implementation. firewall3 also did a flush on stop, but I don’t know if it traversed all iptables chains or not.

bluewavenet, 2022-12-28:
@dave14305 

> ....there is no other method for it to be cleaned

If you wrote the include you of course will know the name of the table it created. Cleanup is easy:

`nft delete inet mytable` for example.....

Spudz76, 2022-12-28:
How is it cool to have a stopped firewall where there are still any rules/tables existing?  Stop means put it back to where it would be at boot, ie no anything, all three main chains default accept.  As if there were no firewall.

Stopped firewall literally means "no rules exist".  Like, no firewall.  Firewall no.  Firewall stop.  No firewall.  Rules or tables remaining after a stop are incorrectly managed.

If other packages want their stuff restored on start, they should use the inclusion hooks provided, if they don't wanna then I guess their stupid hack tables get deleted, oh well.

jow-, 2022-12-28:
> If you wrote the include you of course will know the name of the table it created. Cleanup is easy:

Not exactly. There's support for explicit and implicit `*.nft` file includes as well as script based includes. The fw4 utility has no knowledge of the tables, chains, rules included in `*.nft` files or created by executed scripts.

dave14305, 2022-12-28:
I do think the wiki needs clarification on the `stop` and `flush` parameters to the `firewall` init script. Passing `stop` to `firewall` triggers `fw4 flush` internally.

https://openwrt.org/docs/guide-user/firewall/overview#process_control

jow-, 2022-12-29:
I'm fine with flushing on restart as well. Would actually safe some code if we treat `restart` as `stop; start`

dave14305, 2022-12-29:
> just not sure if this would violate POLA or break existing setups due to a behavioral change.

If a `ruleset-post` include adds an additional table, `fw4 stop` won’t remove it, but `fw4 start` will try to create it again.

kien-truong, 2022-12-29:
```
fw4 should not flush and delete tables created by other packages or by manual creation
```
I'm not sure if using multiple user-space utilities to manage firewall should be supported, this can get hairy real quick. AFAIK, there's no other firewall utilities that support this behavior either: `iptables`, `ufw`, `firewalld` also take exclusive control of the firewall and disregards all external rules when they reload their rule set.

champtar, 2022-12-29:
> ```
> fw4 should not flush and delete tables created by other packages or by manual creation
> ```
>
> I'm not sure if using multiple user-space utilities to manage firewall should be supported, this can [...]

When using nftables backend, firewalld only modify firewalld table and also call iptables-nft -F for legacy reasons.

nftables introduced the concept of multiple tables so multiple software can manage independently their own rules without complex coordination. For example a software like fail2ban should be able to run with ufw / firewalld / fw4 without having to know about them.

champtar, 2022-12-29:
> ~I'm fine with flushing on restart as well. Would actually safe some code if we treat `restart` as `stop; start`~
> 
> Edit: ah, turns out that `fw4 stop; fw4 start` behaves like `fw4 restart`. Issue seems to be the init script only which maps `stop` to `fw4 flush` indeed.
> 
> The fix will be simple:
> 
> ```diff
> diff --git a/root/etc/init.d/firewall b/root/etc/init.d/firewall
> index b555a63..0076e55 100755
> --- a/root/etc/init.d/firewall
> +++ b/root/etc/init.d/firewall
> @@ -17,7 +17,7 @@ start_service() {
>  }
>  
>  stop_service() {
> -       fw4 flush
> +       fw4 stop
>  }
>  
>  reload_service() {
> ```
> 
> ... just not sure if this would violate POLA or break existing setups due to a behavioral change.

`fw4 flush` should only flush fw4 table, this is the expected behavior with nftables IMO (each program handle his table and for simple use cases it works)
Anyone that want to flush everything can call `nft flush ruleset`

Also `/etc/init.d/firewall stop` should not flush, this open a tiny insecure window on shutdown/reboot, so maybe just call `fw4 stop` (this patch) and print a message to say how to flush

Shine-, 2022-12-29:
Perhaps I have wrong expectations here, but I would expect....

When I stop the firewall, it stops and unloads all rules that might be loaded.
When I start the previously stopped firewall, it starts from a definied initial state and adds all (its) rules.
When I restart the firewall, it does nothing else than a "stop" followed by "start" does.

If other packages sideload rule sets that the firewall doesn't know about, it's these other packages' design problem and should not be taken care of by the firewall.

The firewall should **definitely not** leave rogue rule sets from other packages lingering around. This would create setups that are impossible to troubleshoot, or worse.

I'm fully in agreement with @kien-truong here.

IMO, the firewall should even prevent any sideloading, so there can never be rule sets that it's not aware of. It's a security product, after all.

As for the "tiny insecure window on shutdown/reboot" mentioned by @champtar, this is what a firewall product usually has an "initial rule set" for, which is loaded whenever the product is stopped or before it starts (e.g. at boot time). Details differ from product to product, but usually it's something like "block all and allow management from a predefined trusted port".

This is what I'd personally expect from a firewall product. If it did something different, I'd consider it a bug.

bluewavenet, 2022-12-29:
@champtar @jow- 

> nftables introduced the concept of multiple tables so multiple software can manage independently their own rules without complex coordination.

This is exactly the point of bringing up this issue.

As I said earlier:

> If you find you need to do a clean-up of random rules that you have lost track of, then sure, you can do an nft flush ruleset if all else fails.
>
> Even better, why not have an additional config [...]

Shine-, 2022-12-29:
> If you find you need to do a clean-up of random rules that you have lost track of, then sure, you can do an nft flush ruleset if all else fails.

`service firewall stop` is supposed to clean up, and for that matter, `restart` as well since, in my eyes, it's `stop; start`.

Any software that wants to add own nft sets or rules can add these via implicit/explicit fw4 includes. That's what this include mechanism was created for, after all. Then, `service firewall start` will load them as expected.

jow-, 2022-12-29:
> nftables introduced the concept of multiple tables so multiple software can manage independently their own rules without complex coordination.

This was my first thought too when learning about nftables, however in practice, the multiple table support is not as useful as it seems on a first sight. The fact that `ACCEPT` verdicts from one table are not carried over into another makes it largely unusable for many scenarios. Taking miniupnpd for example; initially we thought that we could just let it manage it's own upnp table and not have it interact with fw4 at all, would make for a clean and robust design in theory. Just that it does not work this way; traffic ACCEPTed by upnp managed rules in the upnp table end up being dropped in the fw4 table because ACCEPT verdicts are not final, only DROP ones are.

In practice this means that the "just have multiple independent nft tables" does not hold water. Yes, it will work for incrementally tightening drop policies where each table "filters away" a portion of the overall traffic, but it will fail completely for "hole punching" scenarios as each subsequent table needs to be made aware of the accept verdicts of the preceeding one, either by duplicating rules, relying on shared sets or shared connection marks, reintroducing the need for a global coordination which multiple tables ought to eliminate.

This all could be mitigated if upstream nftables either introduces new meta maches to inspect prio verdicts (e.g. `meta nft action != accept drop`) or a new "soft drop" target that only drops traffic not being accepted previously.

Anyhow, the point I want to make is that it is not as easy as "just use a different table for every app" to end up with a conclusive, deterministic firewall ruleset.

bluewavenet, 2022-12-29:
@jow- @champtar 

> Anyhow, the point I want to make is that it is not as easy as "just use a different table for every app" to end up with a conclusive, deterministic firewall ruleset.

I would certainly agree with that!

> The fact that `ACCEPT` verdicts from one table are not carried over 

It is more the fact that `ACCEPT` verdicts do not terminate processing in nftables, whereas they do in iptables, but yes that is a problem.

From what we have now, the option to select the behaviour of the init.d script `stop_service()` function, would, in my opinion, be a good step.

It could default to enabled (thus keeping things as they currently are), yet be compatible with packages that require dynamic control of their own table/chain structure without requiring a restart just because fw4 has been stopped then started.

Captive portals are an example of such a package type, with continuous monitoring of authenticated and pre-authenticated traffic. A restart will result in loss of tracking of authenticated traffic and require re-authentication.

There are few **if any** circumstances where stopping the fw4 service is required but it can happen, usually, in my experience, inadvertently.
Restarting fw4 does not create any problems, so adding additional rules to fw4 is not an issue, in fact it adds a great deal of flexibility to the use of the package types that "require dynamic control of their own table/chain structure".

Finally, porting of such packages to OpenWrt is currently  quite straightforward. Enforcing the use of fw4 include scripts would make porting a non-trivial task in many cases.

jow-, 2022-12-29:
I am against options affecting init script behavior. This introduces too much variance and untested code paths. We should decide on one specific behavior and stick with it.

bluewavenet, 2022-12-29:
@jow- 

> variance and untested code paths

Indeed, always a possibility, however unlikely. Personally, I would make the change, but I respect your point of view.

Then, given the likely rare occurrence of a service stop, I suggest we do nothing but perhaps add a short comment to the init.d script with regard to the issue. Such a comment could be very helpful to someone struggling with the issue, as well as a marker for future developers.

Spudz76, 2022-12-29:
So then summarizing, we should leave weird holes and options so that people can be lazy.

Lazy in a security portion of the system.

Makes sense,

champtar, 2022-12-30:
> This was my first thought too when learning about nftables, however in practice, the multiple table support is not as useful as it seems on a first sight. The fact that `ACCEPT` verdicts from one [...]

I know ... sorry for not phrasing it explicitly. My point is for all the packages that just drop / reject, why should fw4 flush those rules ?

Spudz76, 2022-12-30:
> for all the packages that just drop / reject, why should fw4 flush those rules ?

Because a stopped firewall has no rules at all.  Otherwise, it's not stopped, right?

Shine-, 2022-12-30:
Also, restarting the firewall should get me back to a known, defined state. **Not** back to the same unknown state that it already had before restarting it (or even worse, with include-chains loaded twice because the restart didn't unload them).

bluewavenet, 2022-12-30:
> Because a stopped firewall has no rules at all

The firewall functionality in OpenWrt is provided by nftables (not the firewall4 package).

The package "firewall4" is a user interface to enable setting of nftables rules in the inet/fw4 table. It runs as a service providing a uci interface for manipulating nftables via a standard uci config file. By extension, additional packages provide integration of Firewall4 into the Luci GUI. 

Stopping the Firewall4 service should flush and delete the inet/fw4 table it created when it started.

The ACTUAL firewall is nftables. So to stop the firewall just do `nft flush ruleset`.
Although I cannot think of a reason why you would want to do this as it will leave you with no firewall functionality - and a "none routing" router.

For compatibility with Firewall3, and user convenience, Firewall4 include scripts were added. This is where:

> Lazy in a security portion of the system

comes into play. Firewall4 has no means of knowing what an include script has added.. so it takes the line of least resistance and flushes everything if you stop the service.

But then, if you think about it, what possible scenario is there for stopping the firewall4 service?

It gets restarted all the time, by hotplug events, by Luci etc, but does not flush the entire nftables ruleset then, so why should it when it is stopped.

> with include-chains loaded twice because the restart didn't unload them

That is the problem with include scripts, they were only added to firewall4 to give some compatibility with firewall3.
If "everyone" managed their own tables then it would not be an issue (notwithstanding an nftables ACCEPT not terminating chain processing - can be a problem, but is a very different problem depending upon what you want to achieve).

Spudz76, 2022-12-30:
Everyone understands what you're saying, but you're still incorrect.

champtar, 2022-12-30:
> Everyone understands what you're saying, but you're still incorrect.

"everyone that disagree with me is wrong" ... or maybe people just have different views on what fw4 responsibilities should be.

You can't unflush, having `fw4 flush` flush the whole firewall means every software needs to be adapted, even simple ones like fail2ban that could have worked without modification.

> with include-chains loaded twice because the restart didn't unload them

If you use fw4 include to create new tables, just write them as follow:
```
table inet test
flush table inet test
table inet test {
...
}
```
fw4 can continue to flush the fw4 table as needed

> Because a stopped firewall has no rules at all. Otherwise, it's not stopped, right?

Maybe I just want to stop fw4 from modifying the rules while I do some tests.
Also without rules you don't have NAT which for IPv4 is required most of the time.
Having `/etc/init.d/firewall stop` just stop fw4 AND print how to flush allow all use cases.

Shine-, 2022-12-30:
> You can't unflush, having `fw4 flush` flush the whole firewall means every software needs to be adapted, even simple ones like fail2ban that could have worked without modification.

I start to understand. It's the old "lazy software developer vs. IT security" fight here. Security should always win. Or rather, security **will** win sooner or later. The question is just, will security win before or after something bad has happened :-) 

> If you use fw4 include to create new tables, just write them as follow:

Understood. I (the user) have to jump through hoops and deal with unexpected behavior of the firewall, just so the lazy developer (see above) won't need to adapt his packages to fw4 :-)

> Maybe I just want to stop fw4 from modifying the rules while I do some tests. Also without rules you don't have NAT which for IPv4 is required most of the time. Having `/etc/init.d/firewall stop` just stop fw4 AND print how to flush allow all use cases.

I think you don't understand. If the firewall is stopped, then it's stopped. Nothing works, nada, no "tests" possible because no rules, no NAT, no routing, no nothing (except the tiny loophole that allows device administration while the initial "block all" state is active.) "Stop", for a firewall, does **not** mean "permit all" or "freeze ruleset". "Stop" means "everything stops working, and that was the purpose of stopping the firewall".

**That** is the concept of a firewall.

What you probably want is a new feature, "freeze ruleset" or similar, to stop fw4 from listening to hotplug events et al.

@jow- mentioned POLA above (thanks for that).

POLA dictates that:
* "restart" is expected to do the same as "stop" followed by "start", just without locking me out because my connection may terminate after "stop".
* "restart" resp. "start/stop" will **always** return me to a known, defined state. It must **not** keep any settings lingering around that I may not even be aware of.
* For "the user", the service "firewall" **is** the firewall. He doesn't know and doesn't have to know which product is behind it, be it ipt, nft, or something completely different, and how to use that particular product. It's the package developer's responsibility to be compatible with whatever is behind "the firewall", not the user's.

Spudz76, 2022-12-31:
> maybe people just have different views on what fw4 responsibilities should be

There is one view, which is `fw4` controls everything firewall related, and you integrate your package with it or suffer broken stuff.  This is what literally everyone here has been repeating to you.

This is the standard for firewall management.  Standards and best practices are not subject to opinion, just like building codes.  If you want to slap together a rickety shed, don't be surprised when the city comes by and tears it down for you.

bluewavenet, 2022-12-31:
Last word trolls? ~Lets see...

cedric-dufour, 2025-01-21:
Having a look at both `/sbin/fw4` and `/etc/init.d/firewall`, I'd kinda say that:
- `/sbin/fw4` restart behavior is expected, assuming its purpose is to deal specifically with OpenWrt stock `fw4` table (but then it should _ignore_ `/usr/share/nftables.d/ruleset-*`; I haven't checked whether that's the case)
- `/etc/init.d/firewall restart` should be updated to `fw4 flush; fw4 start`, assuming its purpose is to deal with "firewall rules" at large (including `/usr/share/nftables.d/ruleset-*`)

My particular issue was having custom `netdev` table definitions in `/usr/share/nftables.d/ruleset-pre`, with `/etc/init.d/firewall restart` _not_ deleting/flushing that table (like `fw4 flush` would) and its definitions being added/_duplicated_ at each restart.

Spudz76, 2025-01-21:
I approve of that take although I still consider `fw4` to be synonymous with `firewall` because `fw4` is the chosen "firewall" manager.

However I could see how using raw `nftables.d` stuff outside/above the `fw4` manager makes a lot of sense.  But I also think all of that could be implemented through `fw4` hooks "outside/above itself", and prefer single scopes of configuration (digging around for rules or hooks in fw4 and never find them because they are nftables rules).

Sort of like you could configure network interfaces with any of like three methods in Debian/Ubuntu (legacy ifupdown, networkd files, netplan files...) but not being consistent and putting all network config in a single method (probably netplan) would be annoying scope stretching.

bluewavenet, 2025-01-22:
@Spudz76 @cedric-dufour 
I'm not wanting to kick off old arguments, but I will state a few facts:
 1. fw4 is NOT a service daemon. It runs then terminates.
 2. Procd is used to start fw4 on booting and on request.
 3. fw4 sets up a static set of firewall rules in a single table.
 4. fw4 does not manage the full nft rule set, in fact, it does not manage anything. It only, by default, sets up a single table for its own use.
 5. fw4 provides an excellent means for users to set up the rules within the fw4 table without needing any knowledge of nftables.
 6. fw4 does not support dynamic rules - it cannot because it is not a service daemon.
 7. For the vast majority of situations, a static ruleset is just fine and fw4 will do everything you want.
 8. Anyone developing a package that needs dynamic rules must bare in mind that fw4 is just a dumb setup utility that blindly reads its own config, sets up its table then goes away (hence the references to lazy software developers a couple of years ago eg someone who ports a package to OpenWrt without checking what goes on in the nft ruleset).

> I approve of that take although I still consider fw4 to be synonymous with firewall because fw4 is the chosen "firewall" manager.

Yes fw4 is indeed equivalent in connotation to "firewall". But it was developed very specifically to be _**OpenWrt's firewall configuration utility.**_ The actual firewall is nftables.

reinerotto, 2026-09-27:
This open bug caused quite some time and effort of mine, when testing mwan3-nft. [https://forum.openwrt.org/t/lets-talk-about-firewall4-default-nftables-firewall/231246/196](url)  In a few words, on certain (wireless) interface events, mwan3s private table is flushed, without notice. Which happens in /sbin/fw4, function flush. 
What has fw4 to mess around with other tables ????
In the dawn of computer programming, we also had the principle of "Modular Programming", which means, no side effects. A developer should not know, how a "module" works. Both principles are forgotten, obviously.

brada4, 2026-10-03:
@Spudz76 > This is the standard for firewall management.

This is standard arrogant BS. firewalld or vyos does not mess with external tables.

Kind of spotted here 
https://github.com/openwrt/openwrt/issues/14449
Ie MAP with multiple ranges is UNSUPPORTABLE with rules or includes. Now your take. 

https://github.com/openwrt/firewall4/pull/96
My attempt (probably will rot away in few years)