openwrt/openwrt #11625: x86 build fail after Kernel_menuconfig

tapper82, 2022-12-28:
### Describe the bug

So steps to trigger this bug.
1. Use make menuconfig to set up a X86 build.
2. let it build 
3. After the build is dun open Kernel_menuconfig and close it with out changeing any settings.
4. Try and run make again
The build will fail.

This build was set to build with Kernel 5.15 and GCC12,. Even after doing a make dirclean and switching back to build with GCC11 the build still fails for me. 
This is on WSL windows 11 and building from the master branch.

### OpenWrt version

Master from dec 28 2022

### OpenWrt target/subtarget

X86/64

### Device

Sophos XG-85r1

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

tapper82, 2022-12-28:
Logs say:

.config:1167:warning: override: CPU_FREQ_DEFAULT_GOV_SCHEDUTIL changes choice state
.config:3802:warning: override: M686 changes choice state
.config:7135:warning: override: SPARSEMEM_MANUAL changes choice state
*
* Restart config...
*
*
* Mitigations for speculative execution vulnerabilities
*
Mitigations for speculative execution vulnerabilities (SPECULATION_MITIGATIONS) [Y/n/?] y
  Remove the kernel mapping in user mode (PAGE_TABLE_ISOLATION) [Y/n/?] y
  Avoid speculative indirect branches in kernel (RETPOLINE) [Y/n/?] y
    Enable return-thunks (RETHUNK) [Y/n/?] y
      Enable UNRET on kernel entry (CPU_UNRET_ENTRY) [Y/n/?] (NEW) make[7]: *** [scripts/kconfig/Makefile:77: syncconfig] Error 1
make[6]: *** [Makefile:620: syncconfig] Error 2
make[5]: *** [Makefile:740: include/config/auto.conf.cmd] Error 2
make[4]: *** [Makefile:25: /home/tapper/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.85/.modules] Error 2
make[3]: *** [Makefile:11: compile] Error 2
time: target/linux/compile#0.89#0.12#0.96

robimarko, 2022-12-28:
What can happen is that despite you not saving when asked the new symbols from the generic config will get added or dropped from the target config, so can you check with `git diff` if the target config changed?

tapper82, 2022-12-28:
diff --git a/target/linux/ramips/mt7621/config-5.10 b/target/linux/ramips/mt7621/config-5.10
index 8c077d211d..b443fc09be 100644
--- a/target/linux/ramips/mt7621/config-5.10
+++ b/target/linux/ramips/mt7621/config-5.10
@@ -25,7 +25,6 @@ CONFIG_CPU_HAS_PREFETCH=y
 CONFIG_CPU_HAS_RIXI=y
 CONFIG_CPU_HAS_SYNC=y
 CONFIG_CPU_IDLE=y
-# CONFIG_CPU_IDLE_GOV_LADDER is not set
 CONFIG_CPU_IDLE_GOV_TEO=y
 CONFIG_CPU_LITTLE_ENDIAN=y
 CONFIG_CPU_MIPS32=y
@@ -42,7 +41,6 @@ CONFIG_CPU_SUPPORTS_32BIT_KERNEL=y
 CONFIG_CPU_SUPPORTS_HIGHMEM=y
 CONFIG_CPU_SUPPORTS_MSA=y
 CONFIG_CRC16=y
-CONFIG_CRYPTO_ACOMP2=y
 CONFIG_CRYPTO_BLAKE2S=y
 CONFIG_CRYPTO_DEFLATE=y
 CONFIG_CRYPTO_HASH_INFO=y
@@ -50,6 +48,7 @@ CONFIG_CRYPTO_LIB_BLAKE2S_GENERIC=y
 CONFIG_CRYPTO_LIB_POLY1305_RSIZE=2
 CONFIG_CRYPTO_LZO=y
 CONFIG_CRYPTO_RNG2=y
+CONFIG_CRYPTO_ZSTD=y
 CONFIG_CSRC_R4K=y
 CONFIG_DEBUG_PINCTRL=y
 CONFIG_DIMLIB=y
@@ -97,6 +96,7 @@ CONFIG_HAS_IOMEM=y
 CONFIG_HAS_IOPORT_MAP=y
 CONFIG_HIGHMEM=y
 CONFIG_I2C=y
+CONFIG_I2C_ALGOBIT=y
 CONFIG_I2C_BOARDINFO=y
 CONFIG_I2C_CHARDEV=y
 CONFIG_I2C_GPIO=y
@@ -109,7 +109,6 @@ CONFIG_IRQ_DOMAIN_HIERARCHY=y
 CONFIG_IRQ_FORCED_THREADING=y
 CONFIG_IRQ_MIPS_CPU=y
 CONFIG_IRQ_WORK=y
-# CONFIG_KERNEL_ZSTD is not set
 CONFIG_LED_TRIGGER_PHY=y
 CONFIG_LIBFDT=y
 CONFIG_LOCK_DEBUGGING_SUPPORT=y
@@ -117,6 +116,7 @@ CONFIG_LZO_COMPRESS=y
 CONFIG_LZO_DECOMPRESS=y
 CONFIG_MDIO_BUS=y
 CONFIG_MDIO_DEVICE=y
+CONFIG_MDIO_DEVRES=y
 CONFIG_MEDIATEK_GE_PHY=y
 CONFIG_MEMFD_CREATE=y
 CONFIG_MFD_SYSCON=y
@@ -126,7 +126,6 @@ CONFIG_MIKROTIK_RB_SYSFS=y
 CONFIG_MIPS=y
 CONFIG_MIPS_ASID_BITS=8
 CONFIG_MIPS_ASID_SHIFT=0
-CONFIG_MIPS_CBPF_JIT=y
 CONFIG_MIPS_CLOCK_VSYSCALL=y
 CONFIG_MIPS_CM=y
 # CONFIG_MIPS_CMDLINE_BUILTIN_EXTEND is not set
@@ -139,6 +138,7 @@ CONFIG_MIPS_CPS_CPUIDLE=y
 # CONFIG_MIPS_CPS_NS16550_BOOL is not set
 CONFIG_MIPS_CPS_PM=y
 CONFIG_MIPS_CPU_SCACHE=y
+CONFIG_MIPS_EBPF_JIT=y
 # CONFIG_MIPS_ELF_APPENDED_DTB is not set
 CONFIG_MIPS_GIC=y
 CONFIG_MIPS_L1_CACHE_SHIFT=5
@@ -229,6 +229,7 @@ CONFIG_RALINK=y
 # CONFIG_RALINK_WDT is not set
[... 1129 lines cut ...]
 CONFIG_HAS_DMA=y
@@ -264,6 +260,7 @@ CONFIG_PAGE_OFFSET=0xC0000000
 CONFIG_PC104=y
 # CONFIG_PC8736x_GPIO is not set
 # CONFIG_PC87413_WDT is not set
+# CONFIG_PCENGINES_APU2 is not set
 CONFIG_PCI=y
 CONFIG_PCI_ATS=y
 CONFIG_PCI_BIOS=y
@@ -300,6 +297,7 @@ CONFIG_PTP_1588_CLOCK_OPTIONAL=y
 CONFIG_RATIONAL=y
 CONFIG_RD_BZIP2=y
 CONFIG_RD_GZIP=y
+CONFIG_RETHUNK=y
 CONFIG_RETPOLINE=y
 CONFIG_RTC_CLASS=y
 CONFIG_RTC_MC146818_LIB=y
@@ -317,6 +315,7 @@ CONFIG_SCx200HR_TIMER=y
 # CONFIG_SCx200_WDT is not set
 CONFIG_SERIAL_8250_PCI=y
 # CONFIG_SERIAL_LANTIQ is not set
+CONFIG_SERIAL_MCTRL_GPIO=y
 CONFIG_SERIO=y
 CONFIG_SERIO_I8042=y
 CONFIG_SERIO_LIBPS2=y

tapper82, 2022-12-28:
I use ./scripts/env to build for more than one router. I have envs set up for 3 routers. x86, ipq806x and ramips. After opening and closing Kernel_menuconfig it is just the x86 build that is broken.

tapper82, 2022-12-28:
I did "git status" and got this:
modified:   target/linux/ramips/mt7621/config-5.10                                                              
 
        modified:   target/linux/x86/64/config-5.15                                                                     
 
        modified:   target/linux/x86/config-5.15                                                                        

 I did not make any changes to these files. Opening Kernel_menuconfig should not make anychanges with out your say so.
After doing "git reset --hard" my build works again.
Kernel_menuconfig is still broken.

robimarko, 2022-12-28:
You can try to use git restore, but the issue is that generic and target config have diverted a lot.
I assume this is 22.03 or other stable release that see very few refreshed

tapper82, 2022-12-28:
@robimarko No mate this is on master.

robimarko, 2022-12-28:
Damn, thats a huge diff for master

a2505, 2025-01-19:
使用`git checkout .`之后重新编译就可以