openwrt/openwrt #11630: Intel WiFi AX201 Wireless card recognized like Detected Intel(R) Wireless-AC 9560 160MHz

WhiteWolf84, 2022-12-28:
### Describe the bug

Hi,

the wireless card Intel AX201 is recognized as "Intel(R) Wireless-AC 9560 160MHz" this is generating many problems during the AP mode, like frequent disconnections.

There is already some patched into the wild, like those:

https://bbs.archlinux.org/viewtopic.php?id=268094
https://bugzilla.redhat.com/show_bug.cgi?id=1977295

Seems that there is a wrong association into the code itself.

### OpenWrt version

r19803-9a599fee93

### OpenWrt target/subtarget

x86/64

### Device

KVM Virtual Machine with PCIE Passtrough

### Image kind

Official downloaded image

### Steps to reproduce

Install the kmod-iwlwifi, and using the card with the required and wrong (iwlwifi-QuZ-a0-jf-b0-66.ucode) firmware downloaded from linux kernel git.

### Actual behaviour

Wireless instability

### Expected behaviour

Wireless stability

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

981213, 2022-12-28:
Could you update your snapshot version and try again? The patch for the issues you mentioned is accepted on 2021.09.02 and our wireless drivers are updated to linux 6.1-rc8 recently.

WhiteWolf84, 2022-12-28:
Hi,

Thanks for the fast reply!

here the results:

[    4.405118] Intel(R) Wireless WiFi driver for Linux
[    4.444699] iwlwifi 0000:01:00.0: api flags index 2 larger than supported by driver
[    4.447324] iwlwifi 0000:01:00.0: TLV_FW_FSEQ_VERSION: FSEQ Version: 89.3.35.37
[    4.455675] PPP generic driver version 2.4.2
[    4.456938] NET: Registered PF_PPPOX protocol family
[    4.458622] iwlwifi 0000:01:00.0: loaded firmware version 72.daa05125.0 QuZ-a0-jf-b0-72.ucode op_mode iwlmvm
[    4.462335] iwlwifi 0000:01:00.0: Detected Intel(R) Wireless-AC 9560 160MHz, REV=0x351
[    4.468516] thermal thermal_zone0: failed to read out thermal zone (-61)
[    4.584632] iwlwifi 0000:01:00.0: Detected RF JF, rfid=0x105110
[    4.643741] iwlwifi 0000:01:00.0: base HW address: 64:6e:e0:9e:f3:7c

I think that the problem is still here.

OpenWrt Version r21603-fb15cb4ce9

Best Regards

WhiteWolf84, 2022-12-28:
A small update, the wireless seems to be more stable anyway. So I'm going to test it and give you some feedback.

WhiteWolf84, 2022-12-29:
Hi,

the problem still there :-(

WhiteWolf84, 2023-01-05:
I'm still with the problem, and I'm constantly upgrading the OpenWrt with snapshots.

Someone has acknowledged the problem?

981213, 2023-01-06:
> I don't think that the patch is in place.

The patches in the link you mentioned are already part of the iwlwifi used in OpenWrt, so you are probably facing a different issue than those people.

This seems to be an issue in the upstream kernel driver, not an OpenWrt specific problem. To confirm this, you can try booting other Linux distros (e.g. Ubuntu/Fedora/...) in your VM, check the kernel log and see if this incorrect detection still happens.
If that's true, there aren't much we can do from OpenWrt side. You may get more help by reporting this issue to linux kernel.

Spudz76, 2023-01-06:
Can you post output of `lspci -vns 01:00.0`

WhiteWolf84, 2023-01-06:
Hi guys,

you are right, I'm so deadly sorry,  it seem I was wrongs. I was sure to have seen the right device name into OpenWRT KERNEL Firmware loading log (dmesg), but it seem that the right name is only showed via LSPCI command. So my wireless stability problems are coming from another cause... :-(

Thanks for your patience and help!

WhiteWolf84, 2023-01-06:
Hi!

maybe I've found the problem! I was using only hostapd in order to make wireless AP working, but I've found another package wpad-openssl that seems to work a lot better and with WPA3 Security!

WhiteWolf84, 2023-01-06:
> Can you post output of `lspci -vns 01:00.0`

root@OpenWrt:~# lspci -nnk
00:00.0 Host bridge [0600]: Intel Corporation 82G33/G31/P35/P31 Express DRAM Controller [8086:29c0]
        Subsystem: Red Hat, Inc. QEMU Virtual Machine [1af4:1100]
lspci: Unable to load libkmod resources: error -12
00:01.0 VGA compatible controller [0300]: Red Hat, Inc. QXL paravirtual graphic card [1b36:0100] (rev 05)
        Subsystem: Red Hat, Inc. QEMU Virtual Machine [1af4:1100]
00:1a.0 USB controller [0c03]: Intel Corporation 82801I (ICH9 Family) USB UHCI Controller #4 [8086:2937] (rev 03)
        Subsystem: Red Hat, Inc. QEMU Virtual Machine [1af4:1100]
        Kernel driver in use: uhci_hcd
00:1a.1 USB controller [0c03]: Intel Corporation 82801I (ICH9 Family) USB UHCI Controller #5 [8086:2938] (rev 03)
        Subsystem: Red Hat, Inc. QEMU Virtual Machine [1af4:1100]
        Kernel driver in use: uhci_hcd
00:1a.2 USB controller [0c03]: Intel Corporation 82801I (ICH9 Family) USB UHCI Controller #6 [8086:2939] (rev 03)
        Subsystem: Red Hat, Inc. QEMU Virtual Machine [1af4:1100]
        Kernel driver in use: uhci_hcd
00:1a.7 USB controller [0c03]: Intel Corporation 82801I (ICH9 Family) USB2 EHCI Controller #2 [8086:293c] (rev 03)
        Subsystem: Red Hat, Inc. QEMU Virtual Machine [1af4:1100]
        Kernel driver in use: ehci-pci
00:1b.0 Audio device [0403]: Intel Corporation 82801I (ICH9 Family) HD Audio Controller [8086:293e] (rev 03)
        Subsystem: Red Hat, Inc. QEMU Virtual Machine [1af4:1100]
00:1c.0 PCI bridge [0604]: Red Hat, Inc. QEMU PCIe Root port [1b36:000c]
        Subsystem: Red Hat, Inc. Device [1b36:0000]
        Kernel driver in use: pcieport
00:1c.1 PCI bridge [0604]: Red Hat, Inc. QEMU PCIe Root port [1b36:000c]
        Subsystem: Red Hat, Inc. Device [1b36:0000]
        Kernel driver in use: pcieport
00:1c.2 PCI bridge [0604]: Red Hat, Inc. QEMU PCIe Root port [1b36:000c]
        Subsystem: Red Hat, Inc. Device [1b36:0000]
        Kernel driver in use: pcieport
00:1c.3 PCI bridge [0604]: Red Hat, Inc. QEMU PCIe Root port [1b36:000c]
        Subsystem: Red Hat, Inc. Device [1b36:0000]
        Kernel driver in use: pcieport
00:1d.0 USB controller [0c03]: Intel Corporation 82801I (ICH9 Family) USB UHCI Controller #1 [8086:2934] (rev 03)
        Subsystem: Red Hat, Inc. QEMU Virtual Machine [1af4:1100]
        Kernel driver in use: uhci_hcd
00:1d.1 USB controller [0c03]: Intel Corporation 82801I (ICH9 Family) USB UHCI Controller #2 [8086:2935] (rev 03)
        Subsystem: Red Hat, Inc. QEMU Virtual Machine [1af4:1100]
        Kernel driver in use: uhci_hcd
00:1d.2 USB controller [0c03]: Intel Corporation 82801I (ICH9 Family) USB UHCI Controller #3 [8086:2936] (rev 03)
        Subsystem: Red Hat, Inc. QEMU Virtual Machine [1af4:1100]
        Kernel driver in use: uhci_hcd
00:1d.7 USB controller [0c03]: Intel Corporation 82801I (ICH9 Family) USB2 EHCI Controller #1 [8086:293a] (rev 03)
        Subsystem: Red Hat, Inc. QEMU Virtual Machine [1af4:1100]
        Kernel driver in use: ehci-pci
00:1e.0 PCI bridge [0604]: Intel Corporation 82801 PCI Bridge [8086:244e] (rev 92)
00:1f.0 ISA bridge [0601]: Intel Corporation 82801IB (ICH9) LPC Interface Controller [8086:2918] (rev 02)
        Subsystem: Red Hat, Inc. QEMU Virtual Machine [1af4:1100]
        Kernel driver in use: lpc_ich
00:1f.2 SATA controller [0106]: Intel Corporation 82801IR/IO/IH (ICH9R/DO/DH) 6 port SATA Controller [AHCI mode] [8086:2922] (rev 02)
        Subsystem: Red Hat, Inc. QEMU Virtual Machine [1af4:1100]
        Kernel driver in use: ahci
00:1f.3 SMBus [0c05]: Intel Corporation 82801I (ICH9 Family) SMBus Controller [8086:2930] (rev 02)
        Subsystem: Red Hat, Inc. QEMU Virtual Machine [1af4:1100]
05:01.0 PCI bridge [0604]: Red Hat, Inc. QEMU PCI-PCI bridge [1b36:0001]
05:02.0 PCI bridge [0604]: Red Hat, Inc. QEMU PCI-PCI bridge [1b36:0001]
05:03.0 PCI bridge [0604]: Red Hat, Inc. QEMU PCI-PCI bridge [1b36:0001]
05:04.0 PCI bridge [0604]: Red Hat, Inc. QEMU PCI-PCI bridge [1b36:0001]
06:07.0 SATA controller [0106]: Intel Corporation 82801IR/IO/IH (ICH9R/DO/DH) 6 port SATA Controller [AHCI mode] [8086:2922] (rev 02)
        Subsystem: Red Hat, Inc. QEMU Virtual Machine [1af4:1100]
        Kernel driver in use: ahci
06:08.0 Communication controller [0780]: Red Hat, Inc. Virtio console [1af4:1003]
        Subsystem: Red Hat, Inc. Device [1af4:0003]
        Kernel driver in use: virtio-pci
06:09.0 Communication controller [0780]: Red Hat, Inc. Virtio console [1af4:1003]
        Subsystem: Red Hat, Inc. Device [1af4:0003]
        Kernel driver in use: virtio-pci
06:10.0 Network controller [0280]: Intel Corporation Wi-Fi 6 AX201 160MHz [8086:4df0] (rev 01)
        Subsystem: Intel Corporation Device [8086:0034]
        Kernel driver in use: iwlwifi
06:12.0 Ethernet controller [0200]: Red Hat, Inc. Virtio network device [1af4:1000]
        Subsystem: Red Hat, Inc. Device [1af4:0001]
        Kernel driver in use: virtio-pci
06:13.0 Ethernet controller [0200]: Red Hat, Inc. Virtio network device [1af4:1000]
        Subsystem: Red Hat, Inc. Device [1af4:0001]
        Kernel driver in use: virtio-pci
root@OpenWrt:~# lspci -vns 06:10.0
06:10.0 0280: 8086:4df0 (rev 01)
        Subsystem: 8086:0034
        Physical Slot: 16-1
        Flags: bus master, fast devsel, latency 0, IRQ 21
        Memory at c8600000 (64-bit, non-prefetchable) [size=16K]
        Capabilities: [c8] Power Management version 3
        Capabilities: [d0] MSI: Enable- Count=1/1 Maskable- 64bit+
        Capabilities: [80] MSI-X: Enable+ Count=16 Masked-
        Kernel driver in use: iwlwifi
lspci: Unable to load libkmod resources: error -12

Spudz76, 2023-01-06:
Yes the default wpad doesn't work well for me either, I use the wpad-wolfssl flavor for a similar and much better result (on totally different hardware).

wirl, 2025-01-23:
My AX101 is also detected as Wireless-AC 9560 which results in it trying to load the wrong firmware.

Dmesg:
```
[    7.585987] Intel(R) Wireless WiFi driver for Linux
[    7.586703] iwlwifi 0000:00:14.3: enabling device (0000 -> 0002)
[    7.588564] iwlwifi 0000:00:14.3: Detected crf-id 0x1300202, cnv-id 0x80400 wfpm id 0x80000030
[    7.589430] iwlwifi 0000:00:14.3: PCI dev 54f0/0034, rev=0x370, rfid=0x105110
[    7.590178] iwlwifi 0000:00:14.3: Detected Intel(R) Wireless-AC 9560 160MHz
[    7.591147] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-so-a0-jf-b0-89.ucode failed with error -2
[    7.592431] iwlwifi 0000:00:14.3: Falling back to sysfs fallback for: iwlwifi-so-a0-jf-b0-89.ucode
[    7.595010] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-so-a0-jf-b0-88.ucode failed with error -2
[    7.596200] iwlwifi 0000:00:14.3: Falling back to sysfs fallback for: iwlwifi-so-a0-jf-b0-88.ucode
[    7.598718] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-so-a0-jf-b0-87.ucode failed with error -2
[    7.599868] iwlwifi 0000:00:14.3: Falling back to sysfs fallback for: iwlwifi-so-a0-jf-b0-87.ucode
[    7.602340] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-so-a0-jf-b0-86.ucode failed with error -2
[    7.603484] iwlwifi 0000:00:14.3: Falling back to sysfs fallback for: iwlwifi-so-a0-jf-b0-86.ucode
[    7.605927] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-so-a0-jf-b0-85.ucode failed with error -2
[    7.607135] iwlwifi 0000:00:14.3: Falling back to sysfs fallback for: iwlwifi-so-a0-jf-b0-85.ucode
[    7.609611] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-so-a0-jf-b0-84.ucode failed with error -2
[    7.610882] iwlwifi 0000:00:14.3: Falling back to sysfs fallback for: iwlwifi-so-a0-jf-b0-84.ucode
[    7.613388] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-so-a0-jf-b0-83.ucode failed with error -2
[    7.614664] iwlwifi 0000:00:14.3: Falling back to sysfs fallback for: iwlwifi-so-a0-jf-b0-83.ucode
[    7.617258] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-so-a0-jf-b0-82.ucode failed with error -2
[    7.618562] iwlwifi 0000:00:14.3: Falling back to sysfs fallback for: iwlwifi-so-a0-jf-b0-82.ucode
[    7.621192] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-so-a0-jf-b0-81.ucode failed with error -2
[    7.622439] iwlwifi 0000:00:14.3: Falling back to sysfs fallback for: iwlwifi-so-a0-jf-b0-81.ucode
[    7.625290] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-so-a0-jf-b0-80.ucode failed with error -2
[    7.626696] iwlwifi 0000:00:14.3: Falling back to sysfs fallback for: iwlwifi-so-a0-jf-b0-80.ucode
[    7.629607] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-so-a0-jf-b0-79.ucode failed with error -2
[    7.631036] iwlwifi 0000:00:14.3: Falling back to sysfs fallback for: iwlwifi-so-a0-jf-b0-79.ucode
[    7.635974] PPP generic driver version 2.4.2
[    7.636712] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-so-a0-jf-b0-78.ucode failed with error -2
[    7.638066] iwlwifi 0000:00:14.3: Falling back to sysfs fallback for: iwlwifi-so-a0-jf-b0-78.ucode
[    7.639501] NET: Registered PF_PPPOX protocol family
[    7.640755] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-so-a0-jf-b0-77.ucode failed with error -2
[    7.642093] iwlwifi 0000:00:14.3: Falling back to sysfs fallback for: iwlwifi-so-a0-jf-b0-77.ucode
[    7.644693] iwlwifi 0000:00:14.3: no suitable firmware found!
[    7.645434] iwlwifi 0000:00:14.3: minimum version required: iwlwifi-so-a0-jf-b0-77
[    7.646245] iwlwifi 0000:00:14.3: maximum version supported: iwlwifi-so-a0-jf-b0-89
[    7.647107] iwlwifi 0000:00:14.3: check git://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git
```

lspci:
```
00:14.3 Network controller [0280]: Intel Corporation CNVi: Wi-Fi [8086:54f0]
	Subsystem: Intel Corporation Device [8086:0034]

00:14.3 0280: 8086:54f0
	Subsystem: 8086:0034
	Flags: fast devsel, IRQ 16
	Memory at 6001114000 (64-bit, non-prefetchable) [size=16K]
	Capabilities: [c8] Power Management version 3
	Capabilities: [d0] MSI: Enable- Count=1/1 Maskable- 64bit+
	Capabilities: [40] Express Root Complex Integrated Endpoint, IntMsgNum 0
	Capabilities: [80] MSI-X: Enable- Count=16 Masked-
```

Package ```iwlwifi-firmware-ax101``` has been installed. This chipset is part of the N100 platform.

How can I make this work in OpenWrt? I am using the latest snapshots.

wirl, 2025-01-24:
I guess with 2500 open issues reporting anything is pointless. This problem is 3 years old already.

Spudz76, 2025-01-24:
`build_dir/target-*/linux-*/linux-*/drivers/net/wireless/intel/iwlwifi/pcie/drv.c` does not contain device:subsys id `54F0:0034` although it does have references to `iwl_ax101_name` but that gets selected via some other identification method than the id's (looks like more of a "looks like a duck" method based on capabilities).  Most "`54F0:`" devices are the 'iwl9560'/`iwlax211`/`iwlax411` variety but those should not get selected because they specify non-`0034` subsys ids.  Unknown why the AX101 is not selected as normal with the ids, or why any of the ones that do not match the subsys did get selected.

Next step is to download the Ubuntu kernel package source and half-build it and inspect the same file(s) in that tree (maybe they have patches we don't?).  Generally OpenWRT just uses whatever is upstream so a bug like this would be upstream's problem.  If Arch/Debian/Ubuntu have patched it and not pushed those fixes upstream (so everyone else can benefit) then that's their fault for not doing opensource correctly.  Or it's iwl driver's fault for not accepting those changes if they were submitted but not merged.

Spudz76, 2025-01-24:
Of note in Archlinux's AUR patch set [0001-ax101.patch](https://aur.archlinux.org/cgit/aur.git/tree/0001-ax101.patch?h=iwlwifi-ax101-dkms) the "looks like a duck" bit is adjusted to use `IWL_CFG_ANY` instead of the more specific (and apparently wrong) `IWL_CFG_160`.

Then to make extra functions work it needs:
```
cd /lib/firmware/intel/ && (
ln -s ibt-1040-4150.ddc ibt-0040-1050.ddc
ln -s ibt-1040-4150.sfi ibt-0040-1050.sfi
)
```

Intel apparently completely dropped the ball despite the report AND the fixes being in that thread of theirs.

wirl, 2025-01-24:
Thank you so much for all the information! A very interesting read and disappointed Intel didn't sort this out. Specially considering many n100 boards have this chip soldered on *and* are sold by Intel partners as Linux compatible. Guess technically correct, they are compatible just not very plug 'n play haha. I have to add that I haven't had any issues with Intel nics and OpenWrt before.