openwrt/openwrt #11635: when will openwrt 21.02 integrate the latest curl with release 7.87.0

chinaraul, 2022-12-29:
### Describe the bug

dear experts:
From curl website, the latest release 7.87.0 has been released, and when will openwrt 21.02 integrate the latest curl with release 7.87.0? Thanks!

### OpenWrt version

955462af7+r49254

### OpenWrt target/subtarget

ipq/ipq807x_64

### Device

nokia 5g cpe

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Mijzelf, 2022-12-29:
Unless curl 7.87.0 fixes a major security bug, I wouldn't hold my breath. 21.02 is the old stable release, and I wouldn't expect any updates except for security updates.

jow-, 2022-12-29:
As written above, stable (and in particular old stable) release usually only fix security issues.

closed 2022-12-29 by jow-