openwrt/openwrt #11645: Xiaomi Mi Router 4A Gigabit Edition: Bootloop unable to enable mt7603e

chris-heo, 2022-12-30:
### Describe the bug

After a crash:

```
[23090.245075] ------------[ cut here ]------------
[23090.249757] WARNING: CPU: 3 PID: 1005 at target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-09-06-d7054646/mt7603/mac.c:208 mt7603_filter_tx+0x178/0x180 [mt7603e]
[23090.264539] Modules linked in: iptable_nat batman_adv xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_CT wireguard smsc95xx rndis_host pl2303 nf_nat nf_conntrack mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mcs7830 mac80211 libchacha20poly1305 ipt_REJECT ftdi_sio ebtable_nat ebtable_filter ebtable_broute dm9601 cfg80211 cdc_subset cdc_ncm cdc_ether cdc_eem ax88179_178a asix xt_time xt_tcpudp xt_quota xt_pkttype xt_owner xt_multiport xt_mark xt_mac xt_limit xt_comment xt_addrtype xt_TCPMSS xt_LOG usbserial usbnet ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_kmp ts_fsm ts_bm smsc r8152 poly1305_mips pegasus nf_reject_ipv4 nf_log_ipv6 nf_log_ipv4 nf_log_common nf_defrag_ipv6 nf_defrag_ipv4 libcurve25519_generic iptable_mangle iptable_filter ipheth ip_tables hso ebtables ebt_vlan ebt_stp ebt_snat ebt_redirect ebt_pkttype ebt_mark_m ebt_mark ebt_limit ebt_ip6 ebt_ip ebt_dnat ebt_arpreply ebt_arp ebt_among
[23090.265849]  ebt_802_3 compat chacha_mips ax88796b sch_teql sch_sfq sch_red sch_prio sch_pie sch_multiq sch_gred sch_fq sch_dsmark sch_codel em_text em_nbyte em_meta em_cmp act_simple act_pedit act_ipt act_csum libcrc32c sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow act_skbedit act_mirred act_gact ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 dummy ip_tunnel veth vfat fat ntfs nls_utf8 nls_iso8859_2 nls_iso8859_15 nls_iso8859_13 nls_iso8859_1 nls_cp852 nls_cp850 nls_cp437 nls_cp1250 vxlan udp_tunnel ip6_udp_tunnel sha256_generic libsha256 seqiv jitterentropy_rng drbg kpp hmac cmac uas usb_storage leds_gpio sd_mod scsi_mod act_police cls_basic sch_tbf sch_ingress gpio_button_hotplug ext4 mbcache jbd2 usbcore nls_base usb_common mii crc32c_generic
[23090.426049] CPU: 3 PID: 1005 Comm: napi/phy0-7 Not tainted 5.10.146 #0
[23090.432941] Stack : 809a0000 80000000 00000003 8007c4e8 80830000 80729294 00000000 00000000
[23090.441318]         82c11c6c 80980000 806f978c 81815db8 807c2dc7 00000001 82c11c10 78a3de2c
[23090.449690]         00000000 00000000 806f978c 82c11ab0 ffffefff 00000000 ffffffea 00000000
[23090.458060]         82c11abc 00000310 807c79e0 ffffffff 00000000 00000001 00000000 80700000
[23090.466434]         00000009 00000003 00003a98 80000000 00000018 803a3814 0000000c 8098000c
[23090.474808]         ...
[23090.477254] Call Trace:
[23090.479723] [<80007fc0>] show_stack+0x30/0x100
[23090.484179] [<8031f0e8>] dump_stack+0x9c/0xcc
[23090.488550] [<8002fe64>] __warn+0xc0/0x12c
[23090.492638] [<8002ff2c>] warn_slowpath_fmt+0x5c/0xac
[23090.497620] [<82ed4864>] mt7603_filter_tx+0x178/0x180 [mt7603e]
[23090.503536] [<82ed49f8>] mt7603_wtbl_set_ps+0x130/0x138 [mt7603e]
[23090.509630] [<82ed1b68>] mt7603_sta_ps+0x38/0x42c [mt7603e]
[23090.515220] [<82ee50b0>] mt76_rx_poll_complete+0x2e4/0x5f8 [mt76]
[23090.521313] [<82ee1acc>] mt76_dma_rx_poll+0x3b0/0x7b4 [mt76]
[23090.527001] [<804a9ac8>] __napi_poll+0x3c/0x15c
[23090.531527] [<804a9ca8>] napi_threaded_poll+0xc0/0x128
[23090.536658] [<800504ac>] kthread+0x12c/0x134
[23090.540918] [<80002ff8>] ret_from_kernel_thread+0x14/0x1c
[23090.546301] 
[23090.547937] ---[ end trace 9ef1a3287edf0354 ]---
```

the device gets caught in a boot loop:

```
[   16.637607] usbserial: USB Serial support registered for FTDI USB Serial Device
[   16.948065] usbcore: registered new interface driver MOSCHIP usb-ethernet driver
[   16.968201] mt7621-pci 1e140000.pcie: bus=2 slot=1 irq=23
[   16.973680] pci 0000:00:01.0: enabling device (0006 -> 0007)
[   16.979362] mt7603e 0000:02:00.0: enabling device (0000 -> 0002)

===================================================================
     		MT7621   stage1 code Oct 28 2018 20:39:32 (ASIC)
     		CPU=500000000 HZ BUS=166666666 HZ
==================================================================

Change MPLL source from XTAL to CR...
```
(empty lines removed)

after entering failsafe mode and reboot, the device goes back to the bootloop. Only a full power cycle revives it.

For more details of the traces, please see the attached files:

* [2022-12-27_09.00.log](https://github.com/openwrt/openwrt/files/10324782/2022-12-27_09.00.log) - appearance of the error
* [2022-12-28_23.00.log](https://github.com/openwrt/openwrt/files/10324783/2022-12-28_23.00.log) - incl. reboot with higher debug level

The ´::SysTime::´ log messages were added via cronjob for easier bug hunting.
When the crash occurred, 5 clients were connected via WiFi (can't tell whether 2.4G or 5G).

### OpenWrt version

r19800+8-a7fb589e8a

### OpenWrt target/subtarget

ramips/mt7621

### Device

Xiaomi Mi Router 4A Gigabit Edition

### Image kind

Self-built image

### Steps to reproduce

No special interaction needed with the device

### Actual behaviour

Device is in boot loop and not accessible via Ethernet/WiFi, full power cycle is needed to get it back to work

### Expected behaviour

Device should do a clean boot

### Additional info

At the risk of this ticket being abandoned: The build FW is from https://github.com/freifunkMUC/gluon

### Diffconfig

```text
n/a
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

khanjui, 2022-12-30:
It was fixed some time ago, compile with latest master.

grische, 2022-12-30:
@khanjui do you know the commit? Do you know if this was or is going to be backported to 22.03?

khanjui, 2022-12-31:
@grische I'm just a user, not a developer. But I IIRC the commit was one of those back in 16 Dec. https://github.com/openwrt/openwrt/commits?author=nbd168

grische, 2023-01-02:
@nbd168 can you help with the questions maybe?

Djfe, 2023-01-16:
maybe test `22.03.03`
https://firmware-selector.openwrt.org/?version=22.03.3&target=ramips%2Fmt7621&id=xiaomi_mi-router-4a-gigabit

If the issue is fixed there, then you need to ask ffmuc to update their code the release upstream.
If it isn't, then you need to wait, I guess.
I tried finding the commit in the MT76 repo but failed to do so. Else ffmuc could've cherry-picked that commit.

Neustradamus, 2023-05-22:
@chris-heo, @grische: Have you tried the last?

chris-heo, 2023-06-11:
sorry for the late reply - currently the APs run gluon based on OpenWrt 22.03-SNAPSHOT, r20044+9-ddeeb35007 and stability is still so-so. 
Sometimes the devices run up to a week then they crash multiple times per day.
It's annoying enough that I decided to build an external [hardware watchdog](https://hobbyelektronik.org/w/index.php/Freifunk-Rebooter) (linked if anybody's interested, German only) which works for me but is still a dirty workaround.
Unfortunately, currently no logs are available and the devices are (physically) placed a bit out of reach for me (ssh is available).
If there's anything I can provide in advance, just let me know.

kokutoukiritsugu, 2023-06-29:
same problem
but i found is Xiaomi Mi Router 4A Gigabit v2 Edition
hardware is MT7621DTA+MT7603EN+MT7613BEN 

try use:

https://downloads.openwrt.org/snapshots/targets/ramips/mt7621/openwrt-ramips-mt7621-xiaomi_mi-router-4a-gigabit-v2-squashfs-sysupgrade.bin