openwrt/openwrt #11648: odhcpd segmentation fault on receipt of DHCPV4_MSG_INFORM

clutch2sft, 2022-12-30:
### Describe the bug

Standard log messages:

```
Dec 29 20:08:27 172.24.0.11 odhcpd[9508]: Received DHCPV4_MSG_INFORM from e0:91:f5:52:54:85 on HS
Dec 29 20:08:36 172.24.0.11 odhcpd[9520]: Received 0 Bytes from 172.16.5.184%HS@Dot1QSwitch.5
Dec 29 20:08:38 172.24.0.11 odhcpd[9520]: Received 0 Bytes from 172.16.5.184%HS@Dot1QSwitch.5
Dec 29 20:08:42 172.24.0.11 odhcpd[9520]: Received 0 Bytes from 172.16.5.184%HS@Dot1QSwitch.5
Dec 29 20:08:45 172.24.0.11 odhcpd[9520]: Received 0 Bytes from 172.16.5.184%HS@Dot1QSwitch.5
Dec 29 20:08:48 172.24.0.11 odhcpd[9520]: Received 0 Bytes from 172.16.5.184%HS@Dot1QSwitch.5
Dec 29 20:09:48 172.24.0.11 odhcpd[9520]: Received 300 Bytes from 172.16.5.184%HS@Dot1QSwitch.5
Dec 29 20:09:48 172.24.0.11 odhcpd[9520]: Got DHCPv4 request on IdCleanseHS
Dec 29 20:09:48 172.24.0.11 odhcpd[9520]: Received DHCPV4_MSG_INFORM from e0:91:f5:52:54:85 on HS
Dec 29 20:09:48 172.24.0.11 procd: Instance odhcpd::instance1 s in a crash loop 6 crashes, 76 seconds since last crash
```

When stopped with

```
sysctl odhcpd stop
```
and started with 
```
/usr/sbin/odhcpd
```
Throws segmenation fault and dies on receipt of  DHCPV4_MSG_INFORM.

### OpenWrt version

r19803-9a599fee93

### OpenWrt target/subtarget

mediatek/mt7622

### Device

Belkin RT3200 (as in Linksys E8450)

### Image kind

Self-built image

### Steps to reproduce

You have to be running odhcpd as the ipv4 dhcp server and have a client that sends the DHCPV4_MSG_INFORM message.

'''
config odhcpd 'odhcpd'
        option maindhcp '1'
        option leasefile '/tmp/hosts/odhcpd/dhcp.leases'
        option leasetrigger '/usr/lib/unbound/odhcpd.sh'
        option loglevel '7'
```

### Actual behaviour

odhcpd enters crash loops and dies. -OR- throws segmentation fault when run at the command line.

### Expected behaviour

odhcpd takes appropriate action for DHCPV4_MSG_INFORM message.

### Additional info

This system is remote.  I will get a packet capture of the inform message as soon as I can but wanted to get the initial report entered.

### Diffconfig

```text
CONFIG_TARGET_mediatek=y
CONFIG_TARGET_mediatek_mt7622=y
CONFIG_TARGET_mediatek_mt7622_DEVICE_linksys_e8450-ubi=y
CONFIG_ALL=y
CONFIG_ALL_KMODS=y
CONFIG_ALL_NONSHARED=y
CONFIG_DEVEL=y
CONFIG_AUTOREMOVE=y
CONFIG_IB=y
CONFIG_IB_STANDALONE=y
CONFIG_IMAGEOPT=y
CONFIG_INITOPT=y
CONFIG_IPK_FILES_CHECKSUMS=y
CONFIG_MAKE_TOOLCHAIN=y
CONFIG_PREINITOPT=y
CONFIG_SDK=y
```
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2022-12-31:
The new odhcpd just crashes in a loop because you never stopped original.
Please use /etc/init.d scripts

clutch2sft, 2022-12-31:
@brada4 - Thanks for your input.  

odhcpd is not still running causing a crash loop.  Because I did stop the original.  The crash loop happens first when run from init.d.  Then it dies after being in a crash loop.  As wittnessed by 

```
ps | grep odhcp
```
Which shows no running instance of odhcpd after the crash loop is reported in the log.

Even so I issue a stop via sysctl before running at the command line ... and running at the command line was only to see if additional output was available that wasn't getting logged.

brada4, 2022-12-31:
sysctl changes lowlevel system parameters, nothing to do with services. If you run systemd like on ubuntu, there is sys**tem**ctl to juggle them, but on openwrt you need to use those scripts to generate default format from files in etc/config aka uci database

clutch2sft, 2022-12-31:
@brada4  Thanks for the clarification.  So we can safely ignore my test where I ran odhcp from the command line.  

However, the issue is that when running odhcp normally it enters a crash loop upon receipt of the noted packet.... and dies eventually.

brada4, 2022-12-31:
`tcpdump -i eth0 -n port 67 and port 68 -w inform.pcap`
`logread > logread.out`

open first in wireshark and try  to cross-reference by timestamps

clutch2sft, 2022-12-31:
@brada4 Perfect thanks you and -

How lucky right after you wrote that the router came online and I was able to capture the requested information.  (The logs are sent to a syslog-ng server so I grabbed it from there)
```
 /sbin/logread -f -h R122022336947 -r 10.200.200.10 514 -p /var/run/logread.2.pid
```
The Log
```
Dec 31 13:57:09 172.24.0.11 kernel: [ 1927.825194] device Dot1QSwitch.5 entered promiscuous mode
Dec 31 13:57:09 172.24.0.11 kernel: [ 1927.830657] device Dot1QSwitch entered promiscuous mode
Dec 31 13:57:43 172.24.0.11 odhcpd[8695]: Received 300 Bytes from 172.16.5.184%HS@Dot1QSwitch.5
Dec 31 13:57:43 172.24.0.11 odhcpd[8695]: Got DHCPv4 request on HS
Dec 31 13:57:43 172.24.0.11 odhcpd[8695]: Received DHCPV4_MSG_INFORM from e0:91:f5:52:54:85 on HS
Dec 31 13:59:19 172.24.0.11 hostapd: wlan1: STA 4a:09:30:d3:71:a3 RADIUS: updated TX/RX stats: rx_bytes=14544432 [0:0] tx_bytes=349501303 [0:0] bytes_64bit=1
Dec 31 13:59:26 172.24.0.11 odhcpd[8715]: Received 300 Bytes from 172.16.5.184%HS@Dot1QSwitch.5
Dec 31 13:59:26 172.24.0.11 odhcpd[8715]: Got DHCPv4 request on HS
Dec 31 13:59:26 172.24.0.11 odhcpd[8715]: Received DHCPV4_MSG_INFORM from e0:91:f5:52:54:85 on HS
Dec 31 14:00:06 172.24.0.11 odhcpd[8792]: Received 330 Bytes from 172.16.5.184%HS@Dot1QSwitch.5
Dec 31 14:00:06 172.24.0.11 odhcpd[8792]: Got DHCPv4 request on HS
Dec 31 14:00:06 172.24.0.11 odhcpd[8792]: Received DHCPV4_MSG_INFORM from e0:91:f5:52:54:85 on HS
Dec 31 14:00:12 172.24.0.11 odhcpd[8796]: Received 330 Bytes from 172.16.5.184%HS@Dot1QSwitch.5
Dec 31 14:00:12 172.24.0.11 odhcpd[8796]: Got DHCPv4 request on HS
Dec 31 14:00:12 172.24.0.11 odhcpd[8796]: Received DHCPV4_MSG_INFORM from e0:91:f5:52:54:85 on HS
Dec 31 14:00:12 172.24.0.11 procd: Instance odhcpd::instance1 s in a crash loop 6 crashes, 1 seconds since last crash
Dec 31 14:00:37 172.24.0.11 dropbear[8830]: Child connection from 172.24.0.5:55109
Dec 31 14:00:48 172.24.0.11 dropbear[8830]: Pubkey auth succeeded for 'root' with ssh-rsa key SHA256:0WZPy6kmwXk9MYgpjmAB5q80O9edSfQV79OrjGDpo1M from 172.24.0.5:55109
Dec 31 14:01:49 172.24.0.11 kernel: [ 2207.520991] device Dot1QSwitch.5 left promiscuous mode
Dec 31 14:01:49 172.24.0.11 kernel: [ 2207.526134] device Dot1QSwitch left promiscuous mode
Dec 31 14:02:07 172.24.0.11 logread[1503]: Logread connected to 10.200.200.10:514 via tcp
```

ps output take right after the crash loop:

```
root@R122022336947:~# ps | grep dhcp
 6580 root      1004 S    odhcp6c -s /lib/netifd/dhcpv6.script -P0 -t120 wan
 6581 root      1248 S    udhcpc -p /var/run/udhcpc-wan.pid -s /lib/netifd/dhcp.script -f -t 0 -i wan -x hostname:R122
 8840 root      1248 S    grep dhcp
root@R122022336947:~# /etc/init.d/odhcp stop
-ash: /etc/init.d/odhcp: not found
root@R122022336947:~# /etc/init.d/odhcpd stop
root@R122022336947:~# /etc/init.d/odhcpd start
root@R122022336947:~# ps | grep dhcp
 6580 root      1004 S    odhcp6c -s /lib/netifd/dhcpv6.script -P0 -t120 wan
 6581 root      1248 S    udhcpc -p /var/run/udhcpc-wan.pid -s /lib/netifd/dhcp.script -f -t 0 -i wan -x hostname:R122
 8895 root      1448 S    /usr/sbin/odhcpd
 8921 root      1248 S    grep dhcp
root@R122022336947:~#
```
And the pcap file is attached.
[dhcppcap.zip](https://github.com/openwrt/openwrt/files/10327885/dhcppcap.zip)

dave14305, 2023-01-01:
This update to odhcpd (already in master/snapshots) needs to be backported to 22.03.

63db906516013b9593b59d09b0dc49149a323da7

clutch2sft, 2023-01-01:
@dave14305  I *think* I can accomplish this on my local build.  BUT - I am not able to, no smart enough to?, be able to write the PR for the 22.03 release.  However, THANK YOU! for your most valuable insights.  Sure will make my life about 1000X better.

Also not sure I should close this?  Probably let whomever can do the backport (or say whether or not they are willing to do a backport for this) comment and close as appropriate.

ggg

dave14305, 2023-01-01:
Might need to ping @dedeckeh.

Happy New Year!

dave14305, 2023-01-02:
Should now be fixed by https://github.com/openwrt/openwrt/commit/558aa1f4821b31bf9e9a97319fccb99df57e9045

closed 2023-01-02 by clutch2sft