openwrt/openwrt #11650: Regression: DSA breaks roaming to WLAN bridged to VLAN

dahopem, 2022-12-30:
### Describe the bug

Subject: Regression: DSA breaks roaming to WLAN bridged to VLAN

# Summary

When bridging a WLAN with a VLAN on DSA-enabled OpenWRT versions, an erroneous FDB entry creates 100% packet loss for 300 seconds.

# Summary analysis

When a VLAN-tagged packet is received by the device from the an Ethernet (LAN) port side, 2 FDB entries for the source MAC address of the packet are created in the forwarding database (FDB) of the bridge:

1. one VLAN-tagged FDB entry and
2. one VLAN-untagged FDB entry

Both FDB entries tell the switch to forward packets whose destination is that MAC address to that Ethernet port.

When then a packet from the same source is received by the device from the WiFi (WLAN) side (e.g. after roaming a client device to that WiFi) and forwarded to the Ethernet (LAN) side (with a VLAN tag), only 1 of these 2 FDB entries are updated (namely the VLAN-tagged FDB entry). The other VLAN-untagged FDB entry remains unchanged, erroneously.

When then another VLAN-tagged packet (e.g. an ARP reply packet or a DHCP reply packet) is received by the device from the Ethernet (LAN) side and that packet's destination is the client device's MAC address (=the VLAN-untagged FDB entry's MAC address), then the switch believes that this packet ought to be forwarded to the Ethernet port (even although it is received from exactly this port), while it actually should be forwarded to the CPU of the router (and then further to the WLAN). This packet then gets dropped.

This packet loss continues until, after about 300 seconds (5 minutes), the erroneous FDB entry expires.

# Impact of this bug

This bug affects potentially all DSA-enabled platforms, but at least a sizable subset of all DSA-enabled platforms.

It has been verified to exist on:

1. OpenWRT 22.03 on BT HomeHub 5a (hh5a).
2. Turris OS 4.0 on [Turris Omnia](https://forum.turris.cz/t/omnia-vlan-on-dsa-port-breaks-arp-responses-tos-4-0-5/12584/37) already since the year 2020.

# How to reproduce

## Steps

1. Ensure that your OpenWRT device A and software supports Distributed Switch Architecture (DSA).
2. Setup your OpenWRT device A to have at least one WiFi interface and at least one ethernet port with VLAN support:
   1. Create bridge device "br-switch", with the list of bridge ports only consisting of port "lan2".
   2. Enable "VLAN filtering" for that bridge-device and create a VLAN with VLAN ID 31, local enabled and Egress tagged for port "lan2".
   3. Create OpenWRT interface "users" with device "br-switch.31".
   4. Create a OpenWRT WiFi interface with SSID "test". Ensure (under Interface Configuration/General Setup/Network) that it is connected to network "users".
   5. Verify by running "brctl show" on the command line (e.g. using SSH) that there is a bridge called "br-switch" with at least 2 members (one member being "lan2" and the other member being a WiFi interface, for example "wlan1-1").
   6. Reboot the OpenWRT device A.
3. Connect your OpenWRT device A to another WiFi router B
   1. Ensure that WiFi router B also support VLANs
   2. Connect an Ethernet cable to port "lan2" of OpenWRT device A and a suitable Ethernet port of WiFi router B.
   3. Setup WiFi router B such that it has a VLAN with VLAN ID 31 which is available (tagged) at the Ethernet port of WiFi router B.
   4. Setup a WiFi interface on WiFi router B with the same SSID "test". Similarly to OpenWRT device A, ensure that that this WiFi interface is bridged to VLAN ID 31.
4. Connect a client device C to OpenWRT device A.
   1. Run "ping" from the client device C to the IP address of WiFi router B.
   2. Observe that client device C receives ping replies.
5. Roam to WiFi router B. 
   1. Change the physical position of client device C to be close to WiFi router B and away from OpenWRT device A. (You may need to reduce the output power of OpenWRT device A if the devices are close to each other.)
   2. Wait 15 seconds.
   3. Verify that client device C is now associated with WiFi router B. (You may verify this looking at the UI of WiFi router B or looking at the output of running "iw dev wlan0 link" on client device C.)
   4. Observe that client device C still receives ping replies although its WLAN association has changed.
6. Roam to OpenWRT device A again. 
   1. Change the physical position of client device C to be close to OpenWRT device A and away from WiFi router B . (You may need to reduce the output power of WiFi router B if the devices are close to each other.)
   2. Wait 15 seconds.
   3. Verify that client device C is now associated with OpenWRT device A. (You may verify this looking at the UI of WiFi router B or looking at the output of running "iw dev wlan0 link" on client device C.)
  

## Expected result

It is expected that client device C still receives ping replies although its WLAN association has changed again.

## Observed result

It can be observed that client device C does not receive ping replies once its WLAN association has changed again.
[... 124 lines cut ...]
   4. Setup a WiFi interface on WiFi router B with the same SSID "test". Similarly to OpenWRT device A, ensure that that this WiFi interface is bridged to VLAN ID 31.
4. Connect a client device C to OpenWRT device A.
   1. Run "ping" from the client device C to the IP address of WiFi router B.
   2. Observe that client device C receives ping replies.
5. Roam to WiFi router B. 
   1. Change the physical position of client device C to be close to WiFi router B and away from OpenWRT device A. (You may need to reduce the output power of OpenWRT device A if the devices are close to each other.)
   2. Wait 15 seconds.
   3. Verify that client device C is now associated with WiFi router B. (You may verify this looking at the UI of WiFi router B or looking at the output of running "iw dev wlan0 link" on client device C.)
   4. Observe that client device C still receives ping replies although its WLAN association has changed.
6. Roam to OpenWRT device A again. 
   1. Change the physical position of client device C to be close to OpenWRT device A and away from WiFi router B . (You may need to reduce the output power of WiFi router B if the devices are close to each other.)
   2. Wait 15 seconds.
   3. Verify that client device C is now associated with OpenWRT device A. (You may verify this looking at the UI of WiFi router B or looking at the output of running "iw dev wlan0 link" on client device C.)

### Actual behaviour

It can be observed that client device C does not receive ping replies once its WLAN association has changed again.

### Expected behaviour

It is expected that client device C still receives ping replies although its WLAN association has changed again.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

f00b4r0, 2023-01-11:
Possibly related: #11218 #11277 #11029

Brain2000, 2023-03-31:
I have experienced the exact same issue on the Belkin RT3200.
I worked around it with the following:
```
brctl setfd br-lan 0.01
brctl setageing br-lan 10

brctl setfd br-vlan8 0.01
brctl setageing br-vlan8 10
...
repeat for every vlan
```
This doesn't solve the initial 5 minute wait when you first boot up, but it does resolve the roaming issue once it is past the initial 5 minutes.

However, this issue goes further. Or perhaps me setting the lower ageing causes this secondary problem.

**After a couple of days, the bridge will stop learning newly connected wifi devices altogether.**

The only traffic that is passed from a connecting station is the 802.11X to the radius server. But that's generated by the local router anyways. Once the station has been authenticated, nothing can pass through from the station's mac address.

I have recently added the ip-bridging utility (I couldn't find it, thank  you for mentioning the package @dahopem so hopefully that might help me find a workaround.... (I would have never found it since "ip-bridging" hints at layer 3)

In the meantime my workaround for this secondary fdb blockade is a reboot.

Flole998, 2023-04-23:
Just for completeness: I am seeing these issues on the Netgear RBR50 and Cisco Meraki MR33, both are also IPQ40xx based. On the Forum @Ansuel suggested that further debugging is needed and this can definitely be solved, but unfortunately I'm not sure where to start.

spolack, 2023-05-08:
I'm also affected by that issue

Znevna, 2023-06-14:
Can anyone test if this commit fixes this issue? https://github.com/openwrt/openwrt/commit/244328b19c3943bb145b72f0d85062f535e56fbd

thebub, 2023-06-20:
I'm also encountering the issue on MediaTek devices (ramips/mt7621; TP-Link EAP235-Wall & TP-Link EAP615-Wall). However I haven't had the chance to validate with any of the recent build, whether the issue is fixed.

jtkohl, 2023-06-21:
I'm seeing this work OK for me, perhaps it's an accidental configuration tweak?
When the mobile device is associated with OpenWRT access point A (Linksys MR-8300, 23.05.0-rc1), the bridge ethernet info shows:
```
01:02:03:04:05:06 dev eth0 self permanent
01:02:03:04:05:06 dev phy1-ap0 vlan 5 offload master br-vlan 
```
When the mobile device roams to the other OpenWRT access point B (Netgear R7800, 22.03.5), the bridge ethernet info on access point A changes to:
```
01:02:03:04:05:06 dev lan1 vlan 5 offload master br-vlan 
01:02:03:04:05:06 dev lan1 vlan 5 self 
```

I'm watching the system logs on both OpenWRT systems, and I see the device changing its association between the access points.  Momentary packet loss (1 or 2 seconds at most), but no 5-minute delay.

`brctl show` shows that STP is disabled.
 
```
root@linksysAP:~# brctl show
bridge name	bridge id		STP enabled	interfaces
br-vlan		7fff.e89f8015759c	no		guest-hi
							phy2-ap0
							rep-only6
							lan4
							lan2
							lan-24g-radio1
							rep-local6
							rep-iot
							wan
							iot-2.4G
							lan-hi
							guest-24g
							lan-radio1
							rep-24ghz
							guest-low
							lan-5g-hi
							lan3
							phy1-ap0
							lan-5g-low
							lan1
							lan-low
```
I think that matters...when I turn on STP, roaming behavior is inconsistent I think and similar to that described in earlier reports above.

Flole998, 2023-06-21:
You only have a single vlan bridge configured. I think that might be important. In my case I had a bridge without vlan tagging and another one with vlan tagging.

krokodilerian, 2023-07-29:
FWIW. TP-Link EAP615 (mt7621), no STP, running 2 ESSIDs, each in a separate VLAN. Had exactly this problem with 22.03.5, 23.05r2 works.  Haven't tried to bisect it.

thebub, 2023-07-29:
I will try one of the more recent builds next week to see whether this is solved.

Flole998, 2023-07-29:
For mt76xx there has been a commit that supposedly fixes it, that doesn't help on ipq40xx and other affected architectures though.

dseven, 2023-07-30:
> For mt76xx there has been a commit that supposedly fixes it, that doesn't help on ipq40xx and other affected architectures though.

A link to the commit or/and discussion about how it supposedly fixes this issue would be helpful, so we can see if there's an commonality....

Flole998, 2023-07-30:
Try searching for something like "mt76 dsa roaming".

thebub, 2023-07-30:
I think he might be referring to this commit, as listed kn the 23.05-rc2 changelog:
[09322f3 kernel: remove bridge offload hack (-846)](https://git.openwrt.org/?p=openwrt/openwrt.git;a=commitdiff;h=09322f3c0ce0750f7983bfe55ec035fdd0ac9580)

Flole998, 2023-07-30:
I'm referring to https://github.com/openwrt/openwrt/issues/9706 which contains a patch file aswell.

schuettecarsten, 2023-07-30:
> I'm referring to #9706 which contains a patch file aswell.

Please let me know if the patch fixes your issue.

Flole998, 2023-07-30:
For obvious reasons the patch won't fix any issue on non-mt76 hardware. So I don't even need to test it on ipq40xx.

dseven, 2023-07-30:
> I'm referring to #9706 which contains a patch file aswell.

#9706 is still open. There is a patch that seems to make the problem go away for some people. A patch file is not a commit - it's just a file that someone posted with some code changes - there is no indication that it has been committed to any git repo. It also sounds like it's dodging the issue by configuring things in a way that the OpenWRT developers don't want to (allowing switch learning). This would not explain why the problem has apparently gone away for @krokodilerian anyway (since, AFAICT, it has not been committed / merged)....

Flole998, 2023-07-30:
That patch file is apparently a ported version of some other commit. There is a commit somewhere aswell, but I couldn't find it when doing a quick search

quarkysg, 2023-08-01:
@Flole998 if you are referring to the patch that @schuettecarsten [linked ](https://github.com/openwrt/openwrt/issues/9706#issuecomment-1326281353) to, it's not ported from anywhere.  I just commented out the codes that disabled mt7530/1 switch MAC autolearn.

My analysis of the issue suggested that the DSA framework is not updating switch ARL tables at all and with switch's autolearn disabled, switch has no way of sending the network packets it receives correctly.  Turning on the switch's autolearn feature seems to solve the issue.

rbubley, 2023-09-17:
> > Can anyone test if this commit fixes this issue? [244328b](https://github.com/openwrt/openwrt/commit/244328b19c3943bb145b72f0d85062f535e56fbd)
> 
> On a BT HomeHub 5a with build from master today (git log shows above commit), I still see the issue:
> 
> ```
> root@hh5a-1:~# bridge fdb | grep xx:xx:xx:xx:xx:xx
> xx:xx:xx:xx:xx:xx dev lan3 self
> xx:xx:xx:xx:xx:xx dev phy1-ap0 vlan 1 master br-lan
> root@hh5a-1:~#
> ```
> 
> Hope I'm not doing something stupid, but it doesn't seem to have made a difference.
> 
> As a side-note; I've never been able to get `bridge fdb del` to work...
> 
> ```
> root@hh5a-1:~# bridge fdb del xx:xx:xx:xx:xx:xx dev lan3
> RTNETLINK answers: No such file or directory
> root@hh5a-1:~#
> ```

FWIW, a workaround that works for me on HH5A (based on various hints and pointers above and elsewhere):

A file, `/etc/init.d/fixfdbservice` with execute permission containing:

```sh
#!/bin/sh /etc/rc.common

USE_PROCD=1
START=99
STOP=01

start_service() {
    procd_open_instance
    procd_set_param command /bin/sh "/var/fixfdb.sh"
    procd_set_param stdout 1
    procd_set_param stderr 1
    procd_close_instance
}
```

Together with a file, `/var/fixfdb.sh`, again with execute permission, containing:

```sh
#!/bin/sh

/usr/sbin/bridge monitor fdb | while read -r mac d dev rest; do
     if [ "$mac" != Deleted ]; then
       echo Found "$mac" on "$dev";
       /usr/sbin/bridge fdb show br br-lan | while read -r omac d odev rest; do
         if [ "$omac" = "$mac" ] && [ "$odev" != "$dev" ]; then
           echo Removing old entry from "$odev";
           /usr/sbin/bridge fdb del "$mac" dev "$odev" "$rest";
         fi;
       done;
     fi;
   done
```

rondoval, 2023-09-28:
Redmi AX6000 running 23.05.0-rc3 and 23.05 snapshots is affected too.
Problem is, the workaround does not work either.
Given e.g. two entries:
xx:xx:xx:xx:xx:xx dev wan vlan 206 master br-lan
xx:xx:xx:xx:xx:xx dev wan vlan 206 self

I can do: 
`bridge fdb del xx:xx:xx:xx:xx:xx dev wan vlan 206 master`
but
```
bridge fdb del xx:xx:xx:xx:xx:xx dev wan vlan 206 self
RTNETLINK answers: No such file or directory
```

dseven, 2023-09-29:
@rondoval Do you have an actual problem, where traffic is not passing (for some period)? Your example above shows two entries, but they both point to the `wan` device, so there doesn't appear to be a conflict, where in the problem case we have entries pointing to two different devices. I'm not sure that this is a manifestation of the issue that we're dealing with (or not dealing with!) here.

rondoval, 2023-10-02:
> @rondoval Do you have an actual problem, where traffic is not passing (for some period)? Your example above shows two entries, but they both point to the `wan` device, so there doesn't appear to be [...]

Yes, I do have a problem where I end up having (after roaming between APs) entries pointing to both a local wifi device (e.g. phy0-ap1) and LAN interface (in my case the wan device - just a name of the port on Redmi AX6000).

This was an example meant to show that the workaround mentioned in posts above does not work on Redmi AX6000, because it is not possible to remove one of the entries pointing towards LAN. One has to wait till it expires.

schuettecarsten, 2023-10-07:
Looks like other devices have similar issues with HW MAC autolearning:
https://github.com/openwrt/openwrt/commit/b0048b314620cc99f155c8c5cd4541af5e99e578

micw, 2023-10-16:
@rbubley Your script is amazing. I had `brctl setageing br-lan 3` as a workaround running which worked but still had a lag of 3 s. With your script, the wrong entry is removed instantly. Now my seamless roaming finally works :-)

Thank you a lot.

ccdunder, 2023-10-31:
I have two Belkin RT3200 (mt7621) with identical configs and observe this issue with 23.05.0 firmware, but not 22.03.03.

Config is essentially a dumb AP:
* Two bridges (one for VLAN filtering and one for #10219).
* All roaming options are enabled except 802.11r.

Let me know what I can do to help debug & test.

ccdunder, 2023-10-31:
Disabling MAC address learning (`learning`) on my trunked port device (`wan`) seems to have solved the issue.

Behavior:
FDB correctly registers only the wireless interface (no 2nd device) when I'm on the AP.
FDB correctly registers only the VLAN interface (no 2nd device) when I'm on the other AP.

schuettecarsten, 2023-11-02:
> ```shell
> #!/bin/sh
> 
> /usr/sbin/bridge monitor fdb | while read -r mac d dev rest; do
>      if [ "$mac" != Deleted ]; then
>        echo Found "$mac" on "$dev";
>        /usr/sbin/bridge fdb show br br-lan | while read -r omac d odev rest; do
>          if [ "$omac" = "$mac" ] && [ "$odev" != "$dev" ]; then
>            echo Removing old entry from "$odev";
>            /usr/sbin/bridge fdb del "$mac" dev "$odev" "$rest";
>          fi;
>        done;
>      fi;
>    done
> ```

The script shows the following errors on my Belkin 3200:

```
Found 68:37:e9:38:12:85 on lan1
Removing old entry from br-eth.1
Error: either "to" is duplicate, or "master br-lan" is a garbage.
```

rbubley, 2023-11-02:
> > ```shell
> > #!/bin/sh
> > 
> > /usr/sbin/bridge monitor fdb | while read -r mac d dev rest; do
> >      if [ "$mac" != Deleted ]; then
> >        echo Found "$mac" on "$dev";
> >        /usr/sbin/bridge fdb show br br-lan | while read -r omac d odev rest; do
> >          if [ "$omac" = "$mac" ] && [ "$odev" != "$dev" ]; then
> >            echo Removing old entry from "$odev";
> >            /usr/sbin/bridge fdb del "$mac" dev "$odev" "$rest";
> >          fi;
> >        done;
> >      fi;
> >    done
> > ```
> 
> @rbubley The script shows the following errors on my Belkin 3200 whenever an entry needs to be removed:
> 
> ```
> Found 68:37:e9:38:12:85 on lan1
> Removing old entry from br-eth.1
> Error: either "to" is duplicate, or "master br-lan" is a garbage.
> ```

At a guess, your bridge device is not called `br-lan` -- try replacing the `br-lan` on line 4 with your bridge device name.

gavindekock, 2023-11-03:
+1 Thank you very much for sharing this script.

gavindekock, 2023-11-03:
> > > Can anyone test if this commit fixes this issue? [244328b](https://github.com/openwrt/openwrt/commit/244328b19c3943bb145b72f0d85062f535e56fbd)
> > 
> > 
> > On a BT HomeHub 5a with build from master today (git log shows above commit), I still see the issue:
> > ```
> > root@hh5a-1:~# bridge fdb | grep xx:xx:xx:xx:xx:xx
> > xx:xx:xx:xx:xx:xx dev lan3 self
> > xx:xx:xx:xx:xx:xx dev phy1-ap0 vlan 1 master br-lan
> > root@hh5a-1:~#
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > Hope I'm not doing something stupid, but it doesn't seem to have made a difference.
> > As a side-note; I've never been able to get `bridge fdb del` to work...
> > ```
> > root@hh5a-1:~# bridge fdb del xx:xx:xx:xx:xx:xx dev lan3
> > RTNETLINK answers: No such file or directory
> > root@hh5a-1:~#
> > ```
> 
> FWIW, a workaround that works for me on HH5A (based on various hints and pointers above and elsewhere):
> 
> A file, `/etc/init.d/fixfdbservice` with execute permission containing:
> 
> ```shell
> #!/bin/sh /etc/rc.common
> 
> USE_PROCD=1
> START=99
> STOP=01
> 
> start_service() {
>     procd_open_instance
>     procd_set_param command /var/fixfdb.sh
>     procd_set_param stdout 1
>     procd_set_param stderr 1
>     procd_close_instance
> }
> ```
> 
> Together with a file, `/var/fixfdb.sh`, again with execute permission, containing:
> 
> ```shell
> #!/bin/sh
> 
> /usr/sbin/bridge monitor fdb | while read -r mac d dev rest; do
>      if [ "$mac" != Deleted ]; then
>        echo Found "$mac" on "$dev";
>        /usr/sbin/bridge fdb show br br-lan | while read -r omac d odev rest; do
>          if [ "$omac" = "$mac" ] && [ "$odev" != "$dev" ]; then
>            echo Removing old entry from "$odev";
>            /usr/sbin/bridge fdb del "$mac" dev "$odev" "$rest";
>          fi;
>        done;
>      fi;
>    done
> ```
> 
> (This relies on the ip-bridge package, (`opkg install ip-bridge`))

+1 Thank you for sharing this script.

mm2293, 2023-11-24:
I'm also getting "RTNETLINK answers: No such file or directory" when I try to delete, that's frustrating -.-

DervilRus, 2023-11-29:
> Together with a file, `/var/fixfdb.sh`, again with execute permission, containing:

I don't understand one thing. Does the /var/fixfdb.sh file disappear after a reboot?

gavindekock, 2023-12-08:
> > Together with a file, `/var/fixfdb.sh`, again with execute permission, containing:
> 
> I don't understand one thing. Does the /var/fixfdb.sh file disappear after a reboot?

I've noticed that same and placed `/var/fixfdb.sh` in `/bin/fixfdb.sh` instead and updated `/etc/init.d/fixfdbservice` to refer to it instead.

rbubley, 2023-12-09:
@DervilRus  and @gavindekock  - You are quite right, and I have amended the script and instructions above accordingly.

DervilRus, 2023-12-09:
could you show me your working network and wireless configs? I want to make sure that my network topology with VLAN is configured correctly.

I can’t figure out what is correct to add to the VLAN interface, bridge or software VLAN.

rbubley, 2023-12-09:
> could you show me your working "network" and "wireless" configs? I want to make sure that my network topology with VLAN is configured correctly.
> 
> I can’t figure out what is correct to add to the VLAN interface, bridge or software VLAN? is it necessary to create a separate bridge for each vlan network?

It might be a bit off-topic, but sure:

/etc/config/network (the ports settings are because I want certain physical ports to be connected to particular VLANs)
```
config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'lan1'
	list ports 'lan2'
	list ports 'lan3'
	list ports 'lan4'
	list ports 'wan'

config bridge-vlan
	option device 'br-lan'
	option vlan '1'
	option description 'Main LAN'
	list ports 'lan3:u*'
	list ports 'lan4:u*'
	list ports 'wan:t'

config bridge-vlan
	option device 'br-lan'
	option vlan '2'
	list ports 'wan:t'

config bridge-vlan
	option device 'br-lan'
	option vlan '3'
	list ports 'wan:t'

config bridge-vlan
	option device 'br-lan'
	option vlan '4'
	option description 'Guest LAN'
	list ports 'lan1:u*'
	list ports 'lan2:u*'
	list ports 'wan:t'

config interface 'main'
	option broadcast '192.168.4.255'
	option device 'br-lan.1'
	option dns '192.168.4.1'
	option gateway '192.168.4.1'
	option ipaddr '192.168.4.5'
	option ipv6 '0'
	option netmask '255.255.255.0'
	option proto 'static'

config interface 'guest'
	option auto '1'
	option broadcast '192.168.3.255'
	option device 'br-lan.4'
	option ipv6 '0'
	option proto 'none'

config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config device
	option name 'br-lan.4'
	option type '8021q'
	option ifname 'br-lan'
	option vid '4'
	option ipv6 '0'

config device
	option name 'br-lan.1'
	option type '8021q'
	option ifname 'br-lan'
	option vid '1'
	option ipv6 '0'
```

/etc/config/wireless
```
config wifi-device 'radio0'
	option cell_density '0'
	option channel '140'
	option country 'GB'
	option distance '15'
	option htmode 'VHT40'
	option hwmode '11a'
	option path 'pci0000:01/0000:01:00.0/0000:02:00.0'
	option type 'mac80211'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option dtim_period '3'
	option encryption 'psk2+ccmp'
	option ft_over_ds '0'
	option ft_psk_generate_local '1'
	option ieee80211r '1'
	option key '<REDACTED>'
	option mobility_domain 'ca51'
	option mode 'ap'
	option network 'main'
	option reassociation_deadline '20000'
	option ssid '<REDACTED>'

config wifi-device 'radio1'
	option cell_density '0'
	option channel '6'
	option country 'GB'
	option distance '15'
	option htmode 'HT20'
	option hwmode '11g'
	option path 'pci0000:00/0000:00:0e.0'
	option type 'mac80211'

config wifi-iface 'default_radio1'
	option device 'radio1'
	option dtim_period '3'
	option encryption 'psk2+ccmp'
	option ft_over_ds '0'
	option ft_psk_generate_local '1'
	option ieee80211r '1'
	option key '<REDACTED>'
	option mobility_domain 'ca50'
	option mode 'ap'
	option network 'main'
	option reassociation_deadline '20000'
	option ssid '<REDACTED>'

config wifi-iface 'wifinet0'
	option device 'radio1'
	option dtim_period '3'
	option encryption 'psk2+ccmp'
	option ft_over_ds '0'
	option ft_psk_generate_local '1'
	option ieee80211r '1'
	option key '<REDACTED>'
	option mobility_domain 'ca60'
	option mode 'ap'
	option network 'guest'
	option reassociation_deadline '20000'
	option ssid '<REDACTED>'
	option skip_inactivity_poll '1'
	option disassoc_low_ack '0'

config wifi-iface 'wifinet1'
	option device 'radio0'
	option dtim_period '3'
	option encryption 'psk2+ccmp'
	option ft_over_ds '0'
	option ft_psk_generate_local '1'
	option ieee80211r '1'
	option key '<REDACTED>'
	option mobility_domain 'ca61'
	option mode 'ap'
	option network 'guest'
	option reassociation_deadline '20000'
	option ssid '<REDACTED>'
	option skip_inactivity_poll '1'
	option disassoc_low_ack '0'
```

raenye, 2023-12-09:
> It might be a bit off-topic, but sure:

Why have VLANS 2 and 3, if I may ask? (they're not connected to anything AFAICT)

rbubley, 2023-12-09:
> > It might be a bit off-topic, but sure:
> 
> Why have VLANS 2 and 3, if I may ask? (they're not connected to anything AFAICT)

They are used elsewhere on the network, and it makes the network configs easier to maintain when they are all similar  -- that way if I decide to repurpose one of the physical ports to access VLAN 2 or 3 it's a one line change rather than adding the whole block. (But I agree it is redundant).

DervilRus, 2023-12-09:
> It might be a bit off-topic, but sure:
> 
> /etc/config/network (the ports settings are because I want certain physical ports to be connected to particular VLANs)

Thanks a lot. I went to edit my configs :) I would also like to find working configs with Dynamic VLAN configured (WPA2-SAE)

DervilRus, 2023-12-12:
```
Tue Dec 12 10:44:44 2023 daemon.info fixfdb.sh[3189]: Found 48:74:12:61:a0:13 on phy1-ap2
Tue Dec 12 10:44:44 2023 daemon.info fixfdb.sh[3189]: Removing old entry from lan0
Tue Dec 12 10:44:44 2023 daemon.err fixfdb.sh[3189]: Error: either "to" is duplicate, or "vlan 20 master br-lan" is a garbage.
Tue Dec 12 10:44:44 2023 daemon.info fixfdb.sh[3189]: Removing old entry from lan0
Tue Dec 12 10:44:44 2023 daemon.err fixfdb.sh[3189]: Error: either "to" is duplicate, or "vlan 20 self" is a garbage.
```
Errors like this appear all the time. I don't understand if this is right or wrong. There is only one bridge in the system: br-lan.

rbubley, 2023-12-12:
> ```
> Tue Dec 12 10:39:54 2023 daemon.info fixfdb.sh[3189]: Found 6c:5a:b0:9a:4c:f2 on lan0
> Tue Dec 12 10:40:24 2023 daemon.info fixfdb.sh[3189]: Found 14:eb:b6:6f:4e:ce on lan0
> Tue Dec 12 10:41:50 2023 daemon.info fixfdb.sh[3189]: Found 4a:27:f5:00:4e:81 on eth1
> Tue Dec 12 10:41:59 2023 daemon.info fixfdb.sh[3189]: Found 48:74:12:61:a0:13 on lan0
> Tue Dec 12 10:42:00 2023 daemon.info fixfdb.sh[3189]: Found 48:74:12:61:a0:13 on lan0
> Tue Dec 12 10:43:03 2023 daemon.info fixfdb.sh[3189]: Found 48:74:12:61:a0:13 on phy1-ap2
> Tue Dec 12 10:43:03 2023 daemon.info fixfdb.sh[3189]: Removing old entry from lan0
> Tue Dec 12 10:43:03 2023 daemon.err fixfdb.sh[3189]: Error: either "to" is duplicate, or "vlan 20 master br-lan" is a garbage.
> Tue Dec 12 10:43:03 2023 daemon.info fixfdb.sh[3189]: Removing old entry from lan0
> Tue Dec 12 10:43:03 2023 daemon.err fixfdb.sh[3189]: Error: either "to" is duplicate, or "vlan 20 self" is a garbage.
> Tue Dec 12 10:43:56 2023 daemon.info fixfdb.sh[3189]: Found 48:74:12:61:a0:13 on lan0
> Tue Dec 12 10:44:19 2023 daemon.info fixfdb.sh[3189]: Found b6:7e:e9:66:16:ab on lan0
> Tue Dec 12 10:44:19 2023 daemon.info fixfdb.sh[3189]: Found b6:7e:e9:66:16:ab on lan0
> Tue Dec 12 10:44:44 2023 daemon.info fixfdb.sh[3189]: Found 48:74:12:61:a0:13 on phy1-ap2
> Tue Dec 12 10:44:44 2023 daemon.info fixfdb.sh[3189]: Removing old entry from lan0
> Tue Dec 12 10:44:44 2023 daemon.err fixfdb.sh[3189]: Error: either "to" is duplicate, or "vlan 20 master br-lan" is a garbage.
> Tue Dec 12 10:44:44 2023 daemon.info fixfdb.sh[3189]: Removing old entry from lan0
> Tue Dec 12 10:44:44 2023 daemon.err fixfdb.sh[3189]: Error: either "to" is duplicate, or "vlan 20 self" is a garbage.
> Tue Dec 12 10:46:16 2023 daemon.info fixfdb.sh[3189]: Found b4:b0:24:66:3a:6e on lan0
> Tue Dec 12 10:47:23 2023 daemon.info fixfdb.sh[3189]: Found 6c:5a:b0:9a:4c:f2 on lan0
> Tue Dec 12 10:50:00 2023 daemon.info fixfdb.sh[3189]: Found be:16:4d:07:87:7b on eth1
> Tue Dec 12 10:50:06 2023 daemon.info fixfdb.sh[3189]: Found 4a:27:f5:00:4e:81 on eth1
> ```
> 
> Errors like this appear all the time. I don't understand if this is right or wrong. There is only one bridge in the system: br-lan.

That's not right. I'll take a look in a bit.

rbubley, 2023-12-16:
I have revised the script slightly to have more informative logging, which should help diagnose issues from @DervilRus ; 

`/bin/fixfdb.sh`
```Shell
#!/bin/sh

bridge_device="br-lan"

# monitor all new entries to the "forwarding database"; read the output line by
# line into variables 'mac', 'd', 'dev', and 'rest'
/usr/sbin/bridge monitor fdb | while read -r mac d dev rest; do 
  if [ "$mac" != Deleted ]; then
    echo Found "$mac" on "$dev";
    /usr/sbin/bridge fdb show br "$bridge_device" | while read -r omac d odev rest; do
# Check whether the mac address already exists on the bridge device, but 
# attached to a different network device -- if found, remove it
      if [ "$omac" = "$mac" ] && [ "$odev" != "$dev" ]; then
        echo Attempting to remove old entry for "$mac" from "$odev" as it should now be on "$dev";
# run bridge fdb del to remove the old entry; if error message is "RTNETLINK answers: No such file or directory", then the entry was already removed. Echo an appropriate message in that case.
        /usr/sbin/bridge fdb del "$mac" dev "$odev" "$rest" 2>&1 | grep -q "No such file or directory" && echo "Entry for $mac already removed from $odev" || echo "Error running /usr/sbin/bridge fdb del $mac dev $odev $rest";    
      fi;
    done;
 fi;
done
```

For me, what it revealed is that I no longer seem to have this issue in 23.05.02...

bof90, 2023-12-24:
Hi @rbubley - thanks for the script, I'm trying it on my HH5A's which are seeing this same issue, running openwrt v22.03.5.

When I start the service, either using luci or on the command-line, 'ps |grep fix' is showing 2 of the /bin/fixfdb.sh processes running, do you also see this? I'm assuming this is not supposed to be the case. If I stop the service, one of the processes is left running, I have to kill it manually.

 # ps |grep fix
  7217 root      1380 S    {fixfdb.sh} /bin/sh /bin/fixfdb.sh
  7220 root      1380 S    {fixfdb.sh} /bin/sh /bin/fixfdb.sh
  7354 root      1308 S    grep fix

 # /etc/init.d/fixfdbservice info
 {
         "fixfdbservice": {
                 "instances": {
                         "instance1": {
                                 "running": true,
                                 "pid": 7217,
                                 "command": [
                                         "/bin/fixfdb.sh"
                                 ],
                                 "term_timeout": 5
                         }
                 }
         }
 }

rbubley, 2023-12-25:
> Hi @rbubley - thanks for the script, I'm trying it on my HH5A's which are seeing this same issue, running openwrt v22.03.5.
>
>
>
> When I start the service, either using luci or on the command-line, 'ps [...]

Yes, I can confirm it appears twice for me too. No, I don't know why. It doesn't seem to stop it working, or indeed to give repeated lines in the log file. I also note that various other scripts started in the same way appear twice in the process list  too.

bof90, 2023-12-25:
> Yes, I can confirm it appears twice for me too. No, I don't know why. It doesn't seem to stop it working, or indeed to give repeated lines in the log file. I also note that various other scripts [...]

Working perfectly to fix the roaming issue. And yes the 2 processes look normal... If I kill the 'extra' process I get no logging from the script. Many thanks for your efforts and sharing!

rany2, 2024-02-23:
@corvin84 that offload parameter is obviously not right, maybe try to fix the script so it's not sent?

corvin84, 2024-02-23:
> /usr/sbin/bridge fdb del ac:80:fb:bc:aa:d1 dev lan1 vlan 11 master br-lan

That's was the first thing i tried but i get:

```
root@router:/etc/config# /usr/sbin/bridge fdb del ac:80:fb:bc:aa:d1 dev lan1 vlan 11 master br-lan
Error: either "to" is duplicate, or "br-lan" is a garbage.
```

Although it is still there:
```
root@router:/etc/config#  /usr/sbin/bridge fdb show br br-lan |grep ac:80:fb:bc:aa:d1
ac:80:fb:bc:aa:d1 dev phy1-ap0 vlan 11 offload master br-lan
```

rany2, 2024-02-23:
@corvin84 I don't know but maybe try without `master br-lan` as well? So `/usr/sbin/bridge fdb del ac:80:fb:bc:aa:d1 dev lan1 vlan 11 master`

corvin84, 2024-02-23:
> @corvin84 I don't know but maybe try without `br-lan` as well? So `/usr/sbin/bridge fdb del ac:80:fb:bc:aa:d1 dev lan1 vlan 11 master`

Seems like the fix is applied because I no longer experience the problem and when i check the log messages indeed the incorrect entry gets removed correctly. However I don't think the below check is correct in the script. When you remove an entry the bridge command does not give you anything. If you run the command on a dev/port that does not exists then indeed it gives the "RTNETLINK answers: No such file or directory" output but in here this command is run executed once so if it removes the device/port successfully it will always give the Error output. 

` /usr/sbin/bridge fdb del "$mac" dev "$odev" "$rest" 2>&1 | grep -q "No such file or directory" && echo "Entry for $mac already removed from $odev" || echo "Error running /usr/sbin/bridge fdb del $mac dev $odev $rest";    `

corvin84, 2024-02-23:
> @corvin84 I don't know but maybe try without `br-lan` as well? So `/usr/sbin/bridge fdb del ac:80:fb:bc:aa:d1 dev lan1 vlan 11 master`

Script works as expected but there is an error the way the command output is checked. The entry is indeed removed correctly. Thanks all. Is there any positive sign that it will be fixed in one of the upcoming version?

rbubley, 2024-02-24:
@corvin84 do you have a fix for the script? Happy to amend...

corvin84, 2024-02-24:
> @corvin84 do you have a fix for the script? Happy to amend...

Hi,

So i added `sed` to remove `offload` and `br-lan` from the end of the lines because these should not be part of the `fdb del` command: 

` /usr/sbin/bridge fdb show br "$bridge_device" | sed -e 's/offload\s//g' -e 's/br-lan\s$//g' | while read -r omac d odev rest; do`

However it still does not work but i have no idea why:

```
# Log
Sat Feb 24 20:27:13 2024 daemon.info fixfdb.sh[4089]: Attempting to remove old entry for ac:80:fb:bc:aa:d1 from lan1 as it should now be on lan4
Sat Feb 24 20:27:13 2024 daemon.info fixfdb.sh[4089]: Error running /usr/sbin/bridge fdb del ac:80:fb:bc:aa:d1 dev lan1 vlan 11 self
Sat Feb 24 20:27:13 2024 daemon.info fixfdb.sh[4089]: Found ac:80:fb:bc:aa:d1 on lan1
Sat Feb 24 20:27:14 2024 daemon.info fixfdb.sh[4089]: Found ac:80:fb:bc:aa:d1 on lan4

# So it is still on lan1
root@router:~# bridge fdb show br br-lan dev lan4 vlan 11 | grep ac:80
root@router:~# bridge fdb show br br-lan dev lan1 vlan 11 | grep ac:80
ac:80:fb:bc:aa:d1 vlan 11 master br-lan
ac:80:fb:bc:aa:d1 vlan 11 self

# I can run the command without any problem:
root@bzs_router:~# /usr/sbin/bridge fdb del ac:80:fb:bc:aa:d1 dev lan1 vlan 11 master
root@bzs_router:~# echo $?
0

# To confirm: 
root@bzs_router:~# bridge fdb show br br-lan dev lan4 vlan 11 | grep ac:80
root@bzs_router:~# bridge fdb show br br-lan dev lan1 vlan 11 | grep ac:80
ac:80:fb:bc:aa:d1 vlan 11 self
```

corvin84, 2024-02-26:
So I tried various things, for some reason if the fdb entry contains `self` then I am unable to delete the entry thus I filtered those and removed the br-lan from the end of the lines and also offload from the lines. Even though I do see the script is doing its job I am still having this roaming problem. Any other suggestions? 22.03 was trouble free for me I started to see this issue when I upgraded to 23. I used DSA on 22.03 before...

rany2, 2024-02-26:
@corvin84 you could avoid this issue all together by just using the dsa interfaces (lan1, lan2, etc.) directly instead of doing vlan on the bridge itself containing those dsa interfaces.

So lan1.\<vlanid> instead of br-vlan.\<vlanid>. That's what I've done and it works fine. IMHO i think its the least hacky solution to all this, but you will need to have your entire network be tagged; won't work if you need untagged traffic.

rany2, 2024-02-26:
> but you will need to have your entire network be tagged; won't work if you need untagged traffic.

To clarify, you could still have other ports be untagged but you can't have a single port, say lan1, be both tagged and untagged. It won't work reliably, behaves very weirdly.

rany2, 2024-02-27:
> is it only required to be set like that on the router or on the dump APs as well?

I've done so on all my routers/dumb APs to be safe.

> Can I setup multiple VLANs per LAN port without a bridge?

Yes, that is fine. 

> Do you know if it is ever going to be fixed?

No idea.

corvin84, 2024-02-27:
> > is it only required to be set like that on the router or on the dump APs as well?
> 
> I've done so on all my routers/dumb APs to be safe. I don't know if you need to do it on both.
> 
> > Can I setup multiple VLANs per LAN port without a bridge?
> 
> Yes, that is fine.
> 
> > Do you know if it is ever going to be fixed?
> 
> No idea.

Okay, I will do the same. I guess it is not an issue if I have end devices connected to one of the AP and I use PVID?

fascinatedcow, 2024-03-14:
> @corvin84 you could avoid this issue all together by just using the dsa interfaces (lan1, lan2, etc.) directly instead of doing vlan on the bridge itself containing those dsa interfaces.
>
> So [...]

Hello. Did you have to do anything particularly special to get this to work? I tried like so:
1. Removed `lan2` and `lan3` from my existing bridge (`br-lan`)
2. Created `lan2.4` and `lan3.4` (vlan 4 was not previously in use anywhere)
3. Created a new bridge (`expvlan4`) and added `lan2.4` and `lan3.4` to it
4. Assigned an IP to `exp4vlan`

But I was not able to get any traffic through it. Connecting a computer to one of the ports and sending tagged frames I wasn't even able to ping the address I'd assigned to expvlan4. Did I miss something?

I agree, this solution looks the least hacky and is more similar to how things used to look with swconfig.

[108 more replies]