openwrt/openwrt #11654: [21.02-SNAPSHOT] cannot build: Reference to non-existent node or label "macaddr_factory_e000"

dfateyev, 2022-12-30:
### Describe the bug

Tried to build a latest 21.02-SNAPSHOT from the repo, and received a compile error:
```
make -C image compile install TARGET_BUILD=
make[5]: Entering directory '/home/builder/openwrt/openwrt-21.02-mt7621/target/linux/ramips/image'
make[5]: Nothing to be done for 'compile'.
mipsel-openwrt-linux-musl-cpp -nostdinc -x assembler-with-cpp -I/home/builder/openwrt/openwrt-21.02-mt7621/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.225/arch/mips/boot/dts -I/home/builder/openwrt/openwrt-21.02-mt7621/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.225/arch/mips/boot/dts/include -I/home/builder/openwrt/openwrt-21.02-mt7621/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.225/include/ -undef -D__DTS__  -o /home/builder/openwrt/openwrt-21.02-mt7621/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_adslr_g7.dtb.tmp ../dts/mt7621_adslr_g7.dts
/home/builder/openwrt/openwrt-21.02-mt7621/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.225/scripts/dtc/dtc -O dtb -i../dts/ -Wno-unit_address_vs_reg -Wno-simple_bus_reg -Wno-unit_address_format -Wno-pci_bridge -Wno-pci_device_bus_num -Wno-pci_device_reg -Wno-avoid_unnecessary_addr_size -Wno-alias_paths -Wno-graph_child_address -Wno-graph_port -Wno-unique_unit_address  -o /home/builder/openwrt/openwrt-21.02-mt7621/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_adslr_g7.dtb /home/builder/openwrt/openwrt-21.02-mt7621/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_adslr_g7.dtb.tmp
rm -f /home/builder/openwrt/openwrt-21.02-mt7621/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_adslr_g7.dtb.tmp
...
 /home/builder/openwrt/openwrt-21.02-mt7621/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_glinet_gl-mt1300.dtb.tmp ../dts/mt7621_glinet_gl-mt1300.dts
/home/builder/openwrt/openwrt-21.02-mt7621/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.225/scripts/dtc/dtc -O dtb -i../dts/ -Wno-unit_address_vs_reg -Wno-simple_bus_reg -Wno-unit_address_format -Wno-pci_bridge -Wno-pci_device_bus_num -Wno-pci_device_reg -Wno-avoid_unnecessary_addr_size -Wno-alias_paths -Wno-graph_child_address -Wno-graph_port -Wno-unique_unit_address  -o /home/builder/openwrt/openwrt-21.02-mt7621/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_glinet_gl-mt1300.dtb /home/builder/openwrt/openwrt-21.02-mt7621/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_glinet_gl-mt1300.dtb.tmp
rm -f /home/builder/openwrt/openwrt-21.02-mt7621/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_glinet_gl-mt1300.dtb.tmp
mipsel-openwrt-linux-musl-cpp -nostdinc -x assembler-with-cpp -I/home/builder/openwrt/openwrt-21.02-mt7621/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.225/arch/mips/boot/dts -I/home/builder/openwrt/openwrt-21.02-mt7621/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.225/arch/mips/boot/dts/include -I/home/builder/openwrt/openwrt-21.02-mt7621/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.225/include/ -undef -D__DTS__  -o /home/builder/openwrt/openwrt-21.02-mt7621/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_gnubee_gb-pc1.dtb.tmp ../dts/mt7621_gnubee_gb-pc1.dts
/home/builder/openwrt/openwrt-21.02-mt7621/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.225/scripts/dtc/dtc -O dtb -i../dts/ -Wno-unit_address_vs_reg -Wno-simple_bus_reg -Wno-unit_address_format -Wno-pci_bridge -Wno-pci_device_bus_num -Wno-pci_device_reg -Wno-avoid_unnecessary_addr_size -Wno-alias_paths -Wno-graph_child_address -Wno-graph_port -Wno-unique_unit_address  -o /home/builder/openwrt/openwrt-21.02-mt7621/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_gnubee_gb-pc1.dtb /home/builder/openwrt/openwrt-21.02-mt7621/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_gnubee_gb-pc1.dtb.tmp
rm -f /home/builder/openwrt/openwrt-21.02-mt7621/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_gnubee_gb-pc1.dtb.tmp
mipsel-openwrt-linux-musl-cpp -nostdinc -x assembler-with-cpp -I/home/builder/openwrt/openwrt-21.02-mt7621/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.225/arch/mips/boot/dts -I/home/builder/openwrt/openwrt-21.02-mt7621/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.225/arch/mips/boot/dts/include -I/home/builder/openwrt/openwrt-21.02-mt7621/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.225/include/ -undef -D__DTS__  -o /home/builder/openwrt/openwrt-21.02-mt7621/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_gnubee_gb-pc2.dtb.tmp ../dts/mt7621_gnubee_gb-pc2.dts
/home/builder/openwrt/openwrt-21.02-mt7621/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.225/scripts/dtc/dtc -O dtb -i../dts/ -Wno-unit_address_vs_reg -Wno-simple_bus_reg -Wno-unit_address_format -Wno-pci_bridge -Wno-pci_device_bus_num -Wno-pci_device_reg -Wno-avoid_unnecessary_addr_size -Wno-alias_paths -Wno-graph_child_address -Wno-graph_port -Wno-unique_unit_address  -o /home/builder/openwrt/openwrt-21.02-mt7621/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_gnubee_gb-pc2.dtb /home/builder/openwrt/openwrt-21.02-mt7621/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_gnubee_gb-pc2.dtb.tmp
../dts/mt7621.dtsi:478.16-483.5: ERROR (phandle_references): /ethernet@1e100000/mac@1: Reference to non-existent node or label "macaddr_factory_e000"

  also defined at ../dts/mt7621_gnubee_gb-pc2.dts:117.8-124.3
ERROR: Input tree has errors, aborting (use -f to force output)
make[5]: *** [Makefile:207: /home/builder/openwrt/openwrt-21.02-mt7621/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_gnubee_gb-pc2.dtb] Error 2
make[5]: Leaving directory '/home/builder/openwrt/openwrt-21.02-mt7621/target/linux/ramips/image'
make[4]: *** [Makefile:23: install] Error 2
make[4]: Leaving directory '/home/builder/openwrt/openwrt-21.02-mt7621/target/linux/ramips'
make[3]: *** [Makefile:11: install] Error 2
make[3]: Leaving directory '/home/builder/openwrt/openwrt-21.02-mt7621/target/linux'
time: target/linux/install#33.03#13.94#45.19
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/install] Error 1
make[2]: Leaving directory '/home/builder/openwrt/openwrt-21.02-mt7621'
make[1]: *** [target/Makefile:19: /home/builder/openwrt/openwrt-21.02-mt7621/staging_dir/target-mipsel_24kc_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/builder/openwrt/openwrt-21.02-mt7621'
make: *** [/home/builder/openwrt/openwrt-21.02-mt7621/include/toplevel.mk:230: world] Error 2
```

### OpenWrt version

21.02-SNAPSHOT

### OpenWrt target/subtarget

ramips/mt7621

### Device

ZBT-WE1326

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Ansuel, 2022-12-31:
@dfateyev it's strange that on buildbot this is not present... let me check this.
You said also master is affected?

Ansuel, 2022-12-31:
I can see from buildbot image gets create correctly for release 21.02

Ansuel, 2022-12-31:
Ok it's caused by this! https://github.com/openwrt/openwrt/commit/d604032c2a50fc6228bcaae78baee1048ff9af16

Ansuel, 2022-12-31:
@dfateyev can you test the linked pr?

Ansuel, 2022-12-31:
Fixed!

closed 2022-12-31 by Ansuel

dfateyev, 2022-12-31:
@Ansuel thanks for the prompt fix, I can confirm now ramips build is working

commit 76c9c2b1 (2022-12-31): rampis: fix Reference to non-existent node for GB-PC2

Fix cannot build: Reference to non-existent node or label
"macaddr_factory_e000" dtb compilation error.

The cherry-pick had to be reworked to use the old mtd-mac-address way as
openwrt-21.02 still wasn't migrated to nvmem implementation.

Fixes: d604032c2a50 ("ramips: fix GB-PC1 and GB-PC2 device support")
Fixes: #11654
Fixes: #11385

[ rework commit message, add more fixes tag ]