openwrt/openwrt #11657: ipv6 dns bug

f4nff, 2022-12-31:
### Describe the bug

```
config dhcp 'lan'
	option interface 'lan'
	option start '100'
	option limit '150'
	option dhcpv4 'server'
	option leasetime '24h'
	option ra 'server'
	option ra_default '2'
	list ra_flags 'managed-config'
	list ra_flags 'other-config'
	list ra_flags 'home-agent'
	option force '1'
	list dns '2001:4860:4860::8888'
```

```
config dhcp 'lan'
	option interface 'lan'
	option start '100'
	option limit '150'
	option dhcpv4 'server'
	option leasetime '24h'
	option ra 'server'
	option ra_default '2'
	list ra_flags 'managed-config'
	list ra_flags 'other-config'
	list ra_flags 'home-agent'
	option force '1'
	list dns '2001:4860:4860::8888'
	list dhcp_option '6,8.8.8.8'
```

如果不设置	list dhcp_option '6,8.8.8.8' ，	list dns '2001:4860:4860::8888' 是生效的，
如果设置 list dhcp_option '6,8.8.8.8'  ，	list dns '2001:4860:4860::8888' 不生效，
两个配置不能共存。

### OpenWrt version

r19803-9a599fee93

### OpenWrt target/subtarget

x86/64

### Device

	BROUNION R86S

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

f4nff, 2022-12-31:
```
	list dns '2001:4860:4860::8888'
	list dhcp_option '6,8.8.8.8'
```
两者不能共存，只能生效其中一个。

f4nff, 2022-12-31:
**odhcp6c | 2022-08-05-7d21e8d8-18
odhcpd | 2022-03-22-860ca900-1
odhcpd-ipv6only | 2022-03-22-860ca900-1**

f4nff, 2023-01-01:
```
config dhcp 'lan'
	option interface 'lan'
	option start '100'
	option limit '150'
	option dhcpv4 'server'
	option leasetime '24h'
	option ra 'server'
	option ra_default '2'
	list ra_flags 'managed-config'
	list ra_flags 'other-config'
	list ra_flags 'home-agent'
	option force '1'
	list dns '2001:4860:4860::8888'
	list dhcp_option '6,8.8.8.8'
```

如果这样配置，windows 获取不到ipv6 dns
[image]

Spudz76, 2023-01-01:
My guess is providing the `dhcp_option 6` replaces any previous convenience setting like `list dns` (which just defines option 6, but a different way).

Either use always `dns` or always `dhcp_option 6`, do not mix.

```
	list dns '2001:4860:4860::8888'
	list dns '8.8.8.8'
```
or
```
	list dhcp_option '6,2001:4860:4860::8888'
	list dhcp_option '6,8.8.8.8'
```

jow-, 2023-01-01:
Configuration issue

closed 2023-01-01 by jow-

f4nff, 2023-01-02:
[image]

f4nff, 2023-01-02:
```
config dhcp 'lan'
	option interface 'lan'
	option start '100'
	option limit '150'
	option dhcpv4 'server'
	option leasetime '24h'
	option ra 'server'
	option ra_default '2'
	list ra_flags 'managed-config'
	list ra_flags 'other-config'
	list ra_flags 'home-agent'
	option force '1'
	list dhcp_option '6,8.8.8.8'
	list dhcp_option '6,2001:4860:4860::8888'
```
这样设置之后，依然获取不到dns ipv6

[image]

f4nff, 2023-01-02:
```
config dhcp 'lan'
	option interface 'lan'
	option start '100'
	option limit '150'
	option dhcpv4 'server'
	option leasetime '24h'
	option ra 'server'
	option ra_default '2'
	list ra_flags 'managed-config'
	list ra_flags 'other-config'
	list ra_flags 'home-agent'
	option force '1'
	list dns '2001:4860:4860::8888'
	list dns '8.8.8.8'
```

我这样设置之后，lan口直接挂掉，

你应该压根就没有去测试，

f4nff, 2023-01-02:
@jow-   bug依然存在，并没有得到解决。

Spudz76, 2023-01-02:
I thought additional DNS servers all went here:
[image]

f4nff, 2023-01-03:
[Spudz76](https://github.com/Spudz76)

```
	list dns '2001:4860:4860::8888'
	list dns '8.8.8.8'
```
or

```
	list dhcp_option '6,2001:4860:4860::8888'
	list dhcp_option '6,8.8.8.8'
```

两种方法都不可以，

Spudz76, 2023-01-03:
Yes, but try putting them both in luci under network->interfaces->lan->advanced where shown, instead of any of that

f4nff, 2023-01-03:
[image]

[image]

放在这里一样不可以，

f4nff, 2023-01-03:
你有测试过ipv6 dns ？