openwrt/openwrt #11658: Unable to install wireguard-tools. kmod-wireguard Is missing

naltabra, 2022-12-31:
### Describe the bug

Same problem in shell or Luci
[image]

### OpenWrt version

r0-9594846

### OpenWrt target/subtarget

ipq807x/generic

### Device

Ax3600

### Image kind

Official downloaded image

### Steps to reproduce

Image https://openwrt.org/inbox/toh/xiaomi/xiaomi_ax3600

in Luci install package wireguard-tools. kmod-wireguard Is missing

### Actual behaviour

Install stops

### Expected behaviour

. kmod-wireguard should have been installed

### Additional info

This is image created by Robimarko and present at openwrt.com

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

obsy, 2022-12-31:
No, ipq807x is not supported by OpenWrt. Someone posted links to Robimarko's image.

brada4, 2023-01-02:
OP got help via proper channel
https://github.com/robimarko/openwrt/issues/91

closed 2023-01-02 by 981213