openwrt/openwrt #11661: openwrt 22.03.x broken on mvebu (does not boot)

damnms, 2022-12-31:
### Describe the bug

i have 22.03.02 running on a sd. i was using 21.xx and updated with luci to 22.03. i usually create my own images.
i now wanted to create a new image but flash that with dd and not with luci to the sd. that image did not boot.
the error i got was from BootRom complaining about bad headers:

```
BootROM - 1.73
 
Booting from MMC
BootROM: Bad header at offset 00000200
BootROM: Bad header at offset 00004400
BootROM: Bad header at offset 00200000
BootROM: Bad header at offset 00400000
BootROM: Bad header at offset 00600000
BootROM: Bad header at offset 00800000
BootROM: Bad header at offset 00A00000
BootROM: Bad header at offset 00C00000
BootROM: Bad header at offset 00E00000
BootROM: Bad header at offset 01000000
BootROM: Bad header at offset 01200000
BootROM: Bad header at offset 01400000
BootROM: Bad header at offset 01600000
BootROM: Bad header at offset 01800000
BootROM: Bad header at offset 01A00000
BootROM: Bad header at offset 01C00000
BootROM: Bad header at offset 01E00000
BootROM: Bad header at offset 02000000
BootROM: Bad header at offset 02200000
BootROM: Bad header at offset 02400000
BootROM: Bad header at offset 02600000
BootROM: Bad header at offset 02800000
BootROM: Bad header at offset 02A00000
BootROM: Bad header at offset 02C00000
BootROM: Bad header at offset 02E00000
BootROM: Bad h
Trying Uart 
```

that confused me so it took a couple of hours to figure out whats going on. so after making sure its no hardware problem, i switched back to 19.x (official image) to see if it boots. worked. then i went to 21.x. - worked too. then i went to 22.03.0 and got the same error again.

this one works: https://downloads.openwrt.org/releases/21.02.5/targets/mvebu/cortexa9/openwrt-21.02.5-mvebu-cortexa9-solidrun_clearfog-pro-a1-squashfs-sdcard.img.gz
this one does not: https://downloads.openwrt.org/releases/22.03.0/targets/mvebu/cortexa9/openwrt-22.03.0-mvebu-cortexa9-solidrun_clearfog-pro-a1-squashfs-sdcard.img.gz

i also tested the 22.03 branch (self built image) to see if someone maybe noticed that and fixed it, but nope, the latest .img i created did also not boot. 

### OpenWrt version

22.03.0 and above

### OpenWrt target/subtarget

mvebu/cortexa9

### Device

SolidRun Clearfog Pro

### Image kind

Official downloaded image

### Steps to reproduce

build the 22.03.0 (or above) openwrt release for mvebu (solidrun clearfog) 
write it with dd to any sd
try to boot that sd 
the bad header error appears (usb<->uart connection)

### Actual behaviour
[... 225 lines cut ...]
CONFIG_PACKAGE_snort=y
CONFIG_PACKAGE_swconfig=y
CONFIG_PACKAGE_tc-mod-iptables=y
CONFIG_PACKAGE_tc-tiny=y
CONFIG_PACKAGE_terminfo=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_unbound-daemon=y
CONFIG_PACKAGE_usbids=m
CONFIG_PACKAGE_usbip=m
CONFIG_PACKAGE_usbip-client=m
CONFIG_PACKAGE_usbip-server=m
CONFIG_PACKAGE_vim=y
CONFIG_PACKAGE_wireless-regdb=y
CONFIG_PACKAGE_xtables-nft=y
CONFIG_PACKAGE_zlib=y
CONFIG_PCRE_JIT_ENABLED=y
# CONFIG_TARGET_ROOTFS_INITRAMFS is not set
CONFIG_TARGET_ROOTFS_PARTSIZE=8096
# CONFIG_TARGET_ROOTFS_TARGZ is not set
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

anomeome, 2022-12-31:
Due to disabled target? _master_ image work?

damnms, 2022-12-31:
> Due to disabled target?

Not sure whats meant with that.

> master image work?
No, downloaded https://downloads.openwrt.org/snapshots/targets/mvebu/cortexa9/openwrt-mvebu-cortexa9-solidrun_clearfog-pro-a1-squashfs-sdcard.img.gz, flashed to sd, looks like same behaviour (only the power led turns on of the device). but i did not have my laptop connected with usb-uart, so not sure what the output was, but yeah, looks like the same problem.

tmn505, 2023-01-02:
The messages from console indicate that BootROM can't find any valid SPL (it loads U-Boot in 2nd stage) on printed offsets. Workaround would be writing `u-boot-spl.kwb` from 21.02.x "Supplementary Files" beginning at offset `0x200` of SD card. What strange is that there's SPL at `0x200` offset in 22.03.x image but the bytes differ, guess that's from upstream U-Boot changes. Would be great if You could bisect the problem.

damnms, 2023-01-02:
if you can guide me a little bit, i would love to help to fix that :)
i also asked here for a bit of guidance or hints how to figure out what is broken

https://forum.openwrt.org/t/22-03-x-broken-on-mvebu-cortexa9-how-to-debug/147144/2

damnms, 2023-01-02:
i found these:
[oli@fedora openwrt]$ find . -name u-boot-spl.kwb
./build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/u-boot-clearfog/u-boot-2021.01/u-boot-spl.kwb
./build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/u-boot-clearfog/u-boot-2021.01/.pkgdir/u-boot-clearfog/u-boot-spl.kwb
./bin/targets/mvebu/cortexa9/u-boot-clearfog/u-boot-spl.kwb
so i guess i could just go with the bin/targets/...u-boot-spl.kwb

please correct me if my steps are wrong, never did something 
1) create a master openwrt image
2) flash it to sd
3) dd if=u-boot-spl.kwb of=/dev/sdc skip=0x200

just afraid if the sizes of the .kwb change between 22.03 and 21.02 then i would probably overwrite something

tmn505, 2023-01-02:
To test workaround, lets assume that the kernel and rootfs is OK on 22.03 image so we won't have to rebuild it. So use that image and write it to SD card. Now overwrite SPL+U-Boot with `u-boot-spl.kwb` from here: https://downloads.openwrt.org/releases/21.02.5/targets/mvebu/cortexa9 (scroll to bottom) with `dd if=u-boot-spl.kwb of=/dev/sdX bs=512 seek=1 conv=notrunc` (conv=notrunc shouldn't be necessary). You don't have to worry about overwriting some data, the filesystem starts at `0x100000`. Now put the card in the ClearFog slot and check if it boots.

Bisecting in OpenWrt buildroot is nightmare for me and I guess it would be bigger nightmare for You.
We can simply build U-Boot outside from OpenWrt tree and also test it with same card by overwriting only the SPL+U-Boot. Clone the upstream U-Boot sources, set ARCH=arm and CROSS_COMPILE=/path/to/toolchain environment variables and start bisecting.
````
git bisect start
git bisect good 2021.01
git bisect bad 2022.07
````
Now compile ClearFog profile by `make clearfog_defconfig && echo "CONFIG_CMD_SETEXPR=y" >> .config && make -j$(nproc)` and write `u-boot-spl.kwb` to SD card. Check if board boots. If it boots invoke `git bisect good` otherwise `git bisect bad`.
Compile > write to SD card > check if boots > mark if it's good/bad till the last step. If it went properly in the end git will show what was the firs bad commit which will likely be the culprit. Keep somewhere each `u-boot-spl.kwb` that will be compiled. This bisection shouldn't take longer than an hour. At the end post `git bisect log` in this issue. Good luck.

damnms, 2023-01-02:
> check if it boots.

i can confirm that, works like a charm, yay :)
i'll try now the other stuff you mentioned with u-boot building

and thanks for your help, i really really appreciate that!

tmn505, 2023-01-02:
The upstream U-Boot source is here: https://source.denx.de/u-boot/u-boot.git
I already specified which U-Boot version was used for 21.02 (2021.01) and 22.03 (2022.07).
I forgot that U-Boot tags are prepended with `v` so instead of version `git bisect good 2021.01` use `git bisect good v2021.01`, the same with `git bisect bad 2022.07` use `git bisect bad v2022.07`.

damnms, 2023-01-02:
i installed now a lot of crosscompile-stuff from gcc for arm, but not sure what i need to enter exactly in the CROSS_COMPILE variable. when i check one of those compiler outputs from openwrt 

```
if arm-openwrt-linux-muslgnueabi-gcc -D_GNU_SOURCE -DHAVE_CONFIG_H -DLOCALEDIR='"/usr/share/locale"' -I. -I.. -I. -I. -I../lib -I.. -I../libelf -I../libebl -I../libdw -I../libdwelf -I/home/oli/openwrt/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/usr/include -I/home/oli/openwrt/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/include/fortify -I/home/oli/openwrt/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/include -I/home/oli/openwrt/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/usr/lib/libiconv-stub/include -I/home/oli/openwrt/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/usr/lib/libintl-stub/include -std=gnu99 -Wall -Wshadow -Wformat=2 -Wold-style-definition -Wstrict-prototypes -Wtrampolines -Wlogical-op -Wduplicated-cond -Wnull-dereference -Wimplicit-fallthrough=5 -Werror -Wunused -Wextra -Wstack-usage=262144 -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -fmacro-prefix-map=/home/oli/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/elfutils-0.180=elfutils-0.180 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/oli/openwrt/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/usr/lib/libiconv-stub/include -I/home/oli/openwrt/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/usr/lib/libintl-stub/include -D_GNU_SOURCE -Wno-unused-result -Wno-format-nonliteral -c -o asm_adduint8.os -fPIC -DPIC -DSHARED -DSYMBOL_VERSIONING -MT asm_adduint8.os -MD -MP \
  -MF ".deps/asm_adduint8.Tpo" `test -f 'asm_adduint8.c' || echo './'`asm_adduint8.c; \
then cat ".deps/asm_adduint8.Tpo" >> ".deps/asm_adduint8.Po"; \
     rm -f ".deps/asm_adduint8.Tpo"; \
else rm -f ".deps/asm_adduint8.Tpo"; exit 1; \
fi
```

unfortunately i dont have a musl-eabi, what i have is:
[oli@fedora bin]$ ls *gcc*
arm-linux-gnu-gcc         arm-none-eabi-gcc-ar      gcc-ranlib
arm-linux-gnu-gcc-ar      arm-none-eabi-gcc-nm      winegcc
arm-linux-gnu-gcc-nm      arm-none-eabi-gcc-ranlib  x86_64-redhat-linux-gcc
arm-linux-gnu-gcc-ranlib  gcc                       x86_64-redhat-linux-gcc-12
arm-none-eabi-gcc         gcc-ar
arm-none-eabi-gcc-12.2.0  gcc-nm

can i pick arm-none-eabi? i am a bit lost :P 

```
...
  AR      spl/drivers/built-in.o
  LD      spl/u-boot-spl
arm-none-eabi-ld.bfd: warning: u-boot-spl has a LOAD segment with RWX permissions
  OBJCOPY spl/u-boot-spl-nodtb.bin
  SYM     spl/u-boot-spl.sym
  CAT     spl/u-boot-spl-dtb.bin
  COPY    spl/u-boot-spl.bin
  MKIMAGE u-boot-with-spl.kwb
  CFGCHK  u-boot.cfg
  OFCHK   .config

```
it built something but not sure if thats the correct arch

tmn505, 2023-01-02:
The bare-metal compiler should suffice so if compiler is in PATH:
CROSS_COMPILE=arm-none-eabi-
or if bare-metal is not enough:
CROSS_COMPILE=arm-linux-gnu-
If compiler is not in PATH You should specify full path, something like this:
CROSS_COMPILE=/usr/bin/arm-linux-gnu-

tmn505, 2023-01-02:
Unfortunately You didn't finish properly the bisection. After specifying the bad revision/commit/tag it outputted this:
> Binäre Suche: danach noch 812 Commits zum Testen übrig (ungefähr 10 Schritte)
[10189cb700114b81413934e63fac9c3e6b336887] configs: synquacer: Ignore OsIndications on DeveloperBox

As You can see the are 10 steps which means 10 revisions to test. The tree HEAD is at 10189cb700114b81413934e63fac9c3e6b336887. Now You should compile the U-Boot, write it to SD card and check if it boots. If it boots invoke `git bisect good` otherwise `git bisect bad`. Git will move tree HEAD to different revision and again You should compile the U-Boot, write it to SD card, check if it boots and mark the result. Each time You'll see numbers of remaining steps(revisions) decreasing, until git will tell which was the first bad commit.

Edit after Your edit. 
You did "manual" bisection, that's also good. Now testing 464 builds is really overkill, that is where `git bisect` comes to help. Now set `git bisect good v2021.10-rc1` and `git bisect bad v2021.10-rc2`. The number of steps will be the number of builds You'll need to test.

damnms, 2023-01-02:
thats awesome, thanks! i was not aware that git has such a feature :)
now i also understand it, first i thought bisect does something different

```
2226ca17348663b37f84ea1dde1363e2212860e4 is the first bad commit
commit 2226ca17348663b37f84ea1dde1363e2212860e4
Author: Pali Rohár <pali@kernel.org>
Date:   Fri Jul 23 11:14:29 2021 +0200

    arm: mvebu: Load U-Boot proper binary in SPL code based on kwbimage header

```

```
[oli@fedora u-boot]$ git bisect log
git bisect start
# Status: warte auf guten und schlechten Commit
# good: [b70b9b07463db2f6937c7ea6d7fb5122feb7ba1b] Prepare v2021.10-rc1
git bisect good b70b9b07463db2f6937c7ea6d7fb5122feb7ba1b
# Status: warte auf schlechten Commit, 1 guter Commit bekannt
# bad: [a0da2dda4ed9d0aee5265e9cd8876734f9f80e09] Prepare v2021.10-rc2
git bisect bad a0da2dda4ed9d0aee5265e9cd8876734f9f80e09
# bad: [a0953b34d9d8d9309c3eabbb75746fef66b15ffe] Merge https://source.denx.de/u-boot/custodians/u-boot-spi
git bisect bad a0953b34d9d8d9309c3eabbb75746fef66b15ffe
# good: [6c7f152eefd760ceb71aabc36c0025814d645a83] tools: kwbimage: Change maximum number of arguments in binary header to 256
git bisect good 6c7f152eefd760ceb71aabc36c0025814d645a83
# bad: [ea40b20431c0c5de1856c56f965757a40df8d3d7] moveconfig: Update to newer kconfiglib
git bisect bad ea40b20431c0c5de1856c56f965757a40df8d3d7
# bad: [5534fb4f4833eda4e1a2df1c89a75db72e5e2008] arm64: a37xx: pinctrl: Correct PWM pins definitions
git bisect bad 5534fb4f4833eda4e1a2df1c89a75db72e5e2008
# good: [825a2ca023202f73a09660db7a49ac1109917ce8] tools: kwboot: Check for v1 header size
git bisect good 825a2ca023202f73a09660db7a49ac1109917ce8
# good: [9baab60b805472c0911ceabfcd991d59c92b3ddd] SPL: Add support for parsing board / BootROM specific image types
git bisect good 9baab60b805472c0911ceabfcd991d59c92b3ddd
# bad: [2a85fdad3e61ff71b1c3acc97e045f390a7263b0] arm: mvebu: Remove unused macro CONFIG_SYS_U_BOOT_OFFS
git bisect bad 2a85fdad3e61ff71b1c3acc97e045f390a7263b0
# bad: [628fced8fb4b0300b881eedf2dddcfea75a12dc8] arm: mvebu: Remove legacy U-Boot header from kwbimage v1 files
git bisect bad 628fced8fb4b0300b881eedf2dddcfea75a12dc8
# bad: [2226ca17348663b37f84ea1dde1363e2212860e4] arm: mvebu: Load U-Boot proper binary in SPL code based on kwbimage header
git bisect bad 2226ca17348663b37f84ea1dde1363e2212860e4
# first bad commit: [2226ca17348663b37f84ea1dde1363e2212860e4] arm: mvebu: Load U-Boot proper binary in SPL code based on kwbimage header

```

tmn505, 2023-01-02:
Thanks for keeping it up and not being discouraged with the amount of steps involved. Now we can tag the person which made the change, fortunately I know that person GH handle.

@pali Hi Pali. Hope I'm not bothering by tagging You here. There's issue with commit in U-Boot:
```
2226ca17348663b37f84ea1dde1363e2212860e4 is the first bad commit
commit 2226ca17348663b37f84ea1dde1363e2212860e4
Author: Pali Rohár <pali@kernel.org>
Date:   Fri Jul 23 11:14:29 2021 +0200

    arm: mvebu: Load U-Boot proper binary in SPL code based on kwbimage header
```
which causes boot issue on ClearFog Pro, where BootROM doesn't recognize SPL. Please visit the bug report for details.

pali, 2023-01-02:
@tmn505 Sorry but here you have to ask @dhewg because he said that nobody cares about my advice.

damnms, 2023-01-02:
@pali oh i can tell you he is wrong, i care about your advice :)

tmn505, 2023-01-02:
@pali sorry to hear that and it's sad from my POV, I hope You'll change Your mind somewhere in the future. I won't pry much in that matter, will ask @dhewg what led to this.

@damnms I think we'll need to prepare bug report and send it to U-Boot mailing list. But before that I'll take a look in upcoming days if we missed something or miss-configured. BTW someone reported similar issue on SolidRun forum: https://community.solid-run.com/t/clearfog-pro-bootrom-bad-headers-noob-needs-help/458/1

damnms, 2023-01-02:
thanks, if i can help in any way, just let me know! 
i will inform that poor soul in the solidrun forum :)

ynezz, 2023-01-02:
> offset in 22.03.x image but the bytes differ

```diff
--- openwrt-21.02.5-mvebu-cortexa9-solidrun_clearfog-pro-a1-squashfs-sdcard.img.txt	2023-01-02 21:17:36.546556269 +0100
+++ openwrt-22.03.0-mvebu-cortexa9-solidrun_clearfog-pro-a1-squashfs-sdcard.img.txt	2023-01-02 21:14:41.720860820 +0100
@@ -1,12 +1,12 @@
   blockid = 0xae,
-  flags = 0x0,
+  flags = 0x1,
   nandpagesize = 0x0,
-  blocksize = 0x68744,
+  blocksize = 0x6ecd0,
   version = 0x1,
-  headersz_msb = 0x2,
-  headersz_lsb = 0x8000,
-  srcaddr = 0x28000,
-  destaddr = 0x7fffc0,
+  headersz_msb = 0x1,
+  headersz_lsb = 0xe3ac,
+  srcaddr = 0xf2,
+  destaddr = 0x800000,
   execaddr = 0x800000,
   options = 0x0,
   nandblocksize = 0x0,
@@ -14,4 +14,12 @@
   reserved4 = 0x0,
   reserved5 = 0x0,
   ext = 0x1,
-  checksum = 0xf
+  checksum = 0xeb
```

tmn505, 2023-01-03:
Thanks for the diff. I assume most, if not all values are correct, I don't know enough to argue about it (the "srcaddr" looks very different). I'll focus on "flags" value. Apparently it's debug flag (https://source.denx.de/u-boot/u-boot/-/blame/master/arch/arm/mach-mvebu/Makefile#L82). The publicly available datasheet for A370 (https://web.archive.org/web/20140617183701/https://www.marvell.com/embedded-processors/armada-300/assets/ARMADA370-FunctionalSpec-datasheet.pdf) states that the flag should be set to `0x0`, that's different SoC but doesn't hurt to try. Maybe ClearFog Boot ROM can't parse the KWB header if this flag is set to `0x1`.

@damnms Please post bootlog (output till kernel load will suffice) of working bootloader. Additionally does Your board have built-in eMMC. Next compile the latest U-Boot source, maybe it has been fixed. And for last, compile U-Boot with this patch applied: [a38x-nodebug.patch.gz](https://github.com/openwrt/openwrt/files/10339519/a38x-nodebug.patch.gz), we'll check if it's really a problem with debug flag set.

damnms, 2023-01-03:
ok, i will do that this evening. could you tell me how i can apply that patch? is git apply a38x-nodebug.patch enough?

tmn505, 2023-01-03:
Should work, if not use `zcat a38x-nodebug.patch.gz | patch -p1`.

damnms, 2023-01-03:
```
U-Boot SPL 2021.10-rc1-00137-g9baab60b80 (Jan 03 2023 - 19:04:59 +0100)
High speed PHY - Version: 2.0
EEPROM TLV detection failed: Using static config for Clearfog Pro.
Detected Device ID 6828
board SerDes lanes topology details:
 | Lane # | Speed |  Type       |
 --------------------------------
 |   0    |   3   | SATA0	|
 |   1    |   0   | SGMII1	|
 |   2    |   5   | PCIe1	|
 |   3    |   5   | USB3 HOST1	|
 |   4    |   5   | PCIe2	|
 |   5    |   0   | SGMII2	|
 --------------------------------
PCIe, Idx 1: detected no link
PCIe, Idx 2: detected no link
High speed PHY - Ended Successfully
mv_ddr: 14.0.0 
DDR3 Training Sequence - Switching XBAR Window to FastPath Window
mv_ddr: completed successfully
Trying to boot from MMC1
 
 
U-Boot 2021.10-rc1-00137-g9baab60b80 (Jan 03 2023 - 19:04:59 +0100)
 
SoC:   MV88F6828-A0 at 1600 MHz
DRAM:  1 GiB (800 MHz, 32-bit, ECC not enabled)
MMC:   mv_sdh: 0
Loading Environment from MMC... *** Warning - bad CRC, using default environment
 
Model: SolidRun Clearfog A1
Board: SolidRun Clearfog Pro
Net:   
Warning: ethernet@70000 (eth1) using random MAC address - 0a:11:b5:4d:b6:ca
eth1: ethernet@70000
Warning: ethernet@30000 (eth2) using random MAC address - ca:d9:a7:4d:23:b9
, eth2: ethernet@30000
Warning: ethernet@34000 (eth3) using random MAC address - b6:ca:1b:65:91:05
, eth3: ethernet@34000
Hit any key to stop autoboot:  3  2  1  0 
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot.scr
940 bytes read in 10 ms (91.8 KiB/s)
## Executing script at 00200000
Booting Linux with root=PARTUUID=1af593fa-02 rootfstype=auto rootwait console=ttyS0,115200
19745 bytes read in 13 ms (1.4 MiB/s)
3531184 bytes read in 166 ms (20.3 MiB/s)
Kernel image @ 0x800000 [ 0x000000 - 0x35e1b0 ]
## Flattened Device Tree blob at 00100000
   Booting using the fdt blob at 0x100000
   Loading Device Tree to 0fff8000, end 0ffffd20 ... OK
 
Starting kernel ...
```

tmn505, 2023-01-03:
Ok. I'll try to look around a bit more.

tmn505, 2023-01-03:
The header has size of `0x20`, checksum is at `0x1f`. You could could overwrite whole header, I don't know if it'll work but doesn't hurt to try.

damnms, 2023-01-03:
thought i could change the spl_image->entry_point = mhdr->execaddr; to something like spl_image->entry_point = 0x28000; but if its required to do some calculation for a checksum to work, guess i can't help :/

ynezz, 2023-01-03:
> AFAIK i have a board without emmc, but not 100% sure

Whats the output of `mmc list` command in u-boot?

damnms, 2023-01-03:
mv_sdh: 0 (SD)

ynezz, 2023-01-04:
> 2226ca17348663b37f84ea1dde1363e2212860e4 is the first bad commit
> commit 2226ca17348663b37f84ea1dde1363e2212860e4
> Author: Pali Rohár <pali@kernel.org>
> Date:   Fri Jul 23 11:14:29 2021 +0200
> 
>     arm: mvebu: Load U-Boot proper binary in SPL code based on kwbimage header

```diff
diff --git a/configs/clearfog_defconfig b/configs/clearfog_defconfig
index 8f28eb8f9868..d8edc453fff3 100644
--- a/configs/clearfog_defconfig
+++ b/configs/clearfog_defconfig
@@ -24,7 +24,6 @@ CONFIG_USE_PREBOOT=y
 CONFIG_SYS_CONSOLE_INFO_QUIET=y
 # CONFIG_DISPLAY_BOARDINFO is not set
 CONFIG_DISPLAY_BOARDINFO_LATE=y
-CONFIG_SYS_MMCSD_RAW_MODE_U_BOOT_DATA_PART_OFFSET=0x1
 CONFIG_SPL_I2C=y
 CONFIG_CMD_TLV_EEPROM=y
 CONFIG_SPL_CMD_TLV_EEPROM=y
```

This part looks quite suspicious as commit [14f14911f47c](https://github.com/u-boot/u-boot/commit/14f14911f47c) `("arm: mvebu: clearfog: set uboot image SD card offset")` states following:

> Armada 38x ROM skips the first SD card offset when loading SPL

So thats probably why we're using 0x200 offset in those SD card images? On the other hand @pali claims following in that bisected commit:

> When loading U-Boot via SPL from eMMC (in Marvell terminology SDIO), the kwbimage main header is stored at sector 0.

That likely breaks `spl_parse_image_header()` as it now gets garbage/zeroes from the SD card sector 0, but it should get kwb header from sector 1 instead?

@damnms can you please try following another fix attempt for v2022.07?

```diff
diff --git a/arch/arm/mach-mvebu/spl.c b/arch/arm/mach-mvebu/spl.c
index fa9a1d7ab65e..0375a22b3a66 100644
--- a/arch/arm/mach-mvebu/spl.c
+++ b/arch/arm/mach-mvebu/spl.c
@@ -47,10 +47,6 @@
 #if defined(CONFIG_SYS_MMCSD_RAW_MODE_U_BOOT_SECTOR) && CONFIG_SYS_MMCSD_RAW_MODE_U_BOOT_SECTOR != 0
 #error CONFIG_SYS_MMCSD_RAW_MODE_U_BOOT_SECTOR must be set to 0
 #endif
-#if defined(CONFIG_SYS_MMCSD_RAW_MODE_U_BOOT_DATA_PART_OFFSET) && \
-    CONFIG_SYS_MMCSD_RAW_MODE_U_BOOT_DATA_PART_OFFSET != 0
-#error CONFIG_SYS_MMCSD_RAW_MODE_U_BOOT_DATA_PART_OFFSET must be set to 0
-#endif
 #endif
 
 /*
diff --git a/configs/clearfog_defconfig b/configs/clearfog_defconfig
index 9f744d0caf96..4afe6caaeccc 100644
--- a/configs/clearfog_defconfig
+++ b/configs/clearfog_defconfig
@@ -24,6 +24,7 @@ CONFIG_USE_PREBOOT=y
 CONFIG_SYS_CONSOLE_INFO_QUIET=y
 # CONFIG_DISPLAY_BOARDINFO is not set
 CONFIG_DISPLAY_BOARDINFO_LATE=y
+CONFIG_SYS_MMCSD_RAW_MODE_U_BOOT_DATA_PART_OFFSET=0x1
 CONFIG_SPL_I2C=y
 CONFIG_CMD_TLV_EEPROM=y
 CONFIG_SPL_CMD_TLV_EEPROM=y
```

tmn505, 2023-01-04:
Yeah, I would assume that's the case, since there's SATA boot, which also skips first sector (MBR) where we deal with block devices.

@damnms Anyway of the result, is that Your email address in the top commit in this repo https://github.com/damnms/packages? It will be needed for bug report or patch sent to U-Boot ML.

damnms, 2023-01-04:
@ynezz that fix did unfortunately not solve it, same behaviour (no boot)

@tmn505 
> is that Your email address in the top commit in this repo

yes

ynezz, 2023-01-04:
> that fix did unfortunately not solve it, same behaviour (no boot)

Just double checking, are you using `dd if=u-boot-spl.kwb of=/dev/sdX bs=512 seek=1 conv=notrunc` command to transfer the bootloader to the SD card? Is there any difference in the console log output? Can you apply following debugging adjustments on top of that previous fix:

```
diff --git a/arch/arm/mach-mvebu/spl.c b/arch/arm/mach-mvebu/spl.c
index 3b6bc3897090..346f8c4fddc6 100644
--- a/arch/arm/mach-mvebu/spl.c
+++ b/arch/arm/mach-mvebu/spl.c
@@ -2,6 +2,7 @@
 /*
  * Copyright (C) 2014-2016 Stefan Roese <sr@denx.de>
  */
+#define DEBUG
 
 #include <common.h>
 #include <dm.h>
diff --git a/common/spl/spl.c b/common/spl/spl.c
index d55d3c284851..eb2d84c7924a 100644
--- a/common/spl/spl.c
+++ b/common/spl/spl.c
@@ -5,6 +5,7 @@
  *
  * Aneesh V <aneesh@ti.com>
  */
+#define DEBUG
 
 #include <common.h>
 #include <bloblist.h>
diff --git a/common/spl/spl_mmc.c b/common/spl/spl_mmc.c
index 212a2b099289..99bf9263ac5c 100644
--- a/common/spl/spl_mmc.c
+++ b/common/spl/spl_mmc.c
@@ -5,6 +5,8 @@
  *
  * Aneesh V <aneesh@ti.com>
  */
+#define DEBUG
+
 #include <common.h>
 #include <dm.h>
 #include <log.h>
```

and provide complete, unredacted serial console log? I hope it's going to provide more insight into the boot flow.

damnms, 2023-01-04:
is that "hunk" ok?

```
[oli@fedora ~]$ cd u-boot/
[oli@fedora u-boot]$ git status
HEAD losgelöst bei v2022.07
Sie sind gerade bei einer binären Suche, gestartet von Branch '840658b093'.
  (benutzen Sie "git bisect reset", um zum ursprünglichen Branch zurückzukehren)

Änderungen, die nicht zum Commit vorgemerkt sind:
  (benutzen Sie "git add <Datei>...", um die Änderungen zum Commit vorzumerken)
  (benutzen Sie "git restore <Datei>...", um die Änderungen im Arbeitsverzeichnis zu verwerfen)
	geändert:       arch/arm/mach-mvebu/spl.c
	geändert:       configs/clearfog_defconfig

keine Änderungen zum Commit vorgemerkt (benutzen Sie "git add" und/oder "git commit -a")
[oli@fedora u-boot]$ git branch
* (kein Branch, binäre Suche begonnen bei 840658b093)
  master
[oli@fedora u-boot]$ cat ../second.patch | patch -p1
patching file arch/arm/mach-mvebu/spl.c
Hunk #1 succeeded at 2 with fuzz 1.
patching file common/spl/spl.c
patching file common/spl/spl_mmc.c
```

ynezz, 2023-01-04:
@damnms what about following fix for v2022.07? It reverts commit 2975bba6dfdd ("arm: mvebu: Enable CONFIG_SPL_SYS_NO_VECTOR_TABLE for 32-bit mvebu")

```diff
diff --git a/arch/arm/mach-mvebu/Kconfig b/arch/arm/mach-mvebu/Kconfig
index a81b8e2b0df2..44334419f1e6 100644
--- a/arch/arm/mach-mvebu/Kconfig
+++ b/arch/arm/mach-mvebu/Kconfig
@@ -15,7 +15,6 @@ config ARMADA_32BIT
        select SPL_SIMPLE_BUS if SPL
        select SUPPORT_SPL
        select TRANSLATION_OFFSET
-       select SPL_SYS_NO_VECTOR_TABLE if SPL
        select ARCH_VERY_EARLY_INIT
 
 # ARMv7 SoCs...

```

damnms, 2023-01-05:
```
[oli@fedora ~]$ cd u-boot/
[oli@fedora u-boot]$ vim ../3rd_patch.patch
[oli@fedora u-boot]$ cat ../3rd_patch.patch | patch -p1
patching file arch/arm/mach-mvebu/Kconfig
Hunk #1 FAILED at 15.
1 out of 1 hunk FAILED -- saving rejects to file arch/arm/mach-mvebu/Kconfig.rej
[oli@fedora u-boot]$ git branch
* (HEAD losgelöst bei v2022.07)
  master
[oli@fedora u-boot]$ git status
HEAD losgelöst bei v2022.07
Änderungen, die nicht zum Commit vorgemerkt sind:
  (benutzen Sie "git add <Datei>...", um die Änderungen zum Commit vorzumerken)
  (benutzen Sie "git restore <Datei>...", um die Änderungen im Arbeitsverzeichnis zu verwerfen)
	geändert:       arch/arm/mach-mvebu/spl.c
	geändert:       common/spl/spl.c
	geändert:       common/spl/spl_mmc.c
	geändert:       configs/clearfog_defconfig

Unversionierte Dateien:
  (benutzen Sie "git add <Datei>...", um die Änderungen zum Commit vorzumerken)
	arch/arm/mach-mvebu/Kconfig.rej

keine Änderungen zum Commit vorgemerkt (benutzen Sie "git add" und/oder "git commit -a")

```
i'll do that manually

ynezz, 2023-01-05:
> the commit that breaks uboot for mvebu is 2226ca1734

Well, bisects are sometimes tricky, so can you double verify, that it really breaks after that commit? I would do that with following:

1. `git reset --hard 2226ca1734` build, flash, boot, report complete console U-Boot log
2. `git revert HEAD` build, flash, boot, report compete console U-Boot log (this reverts changes 2226ca1734, so should boot if that bisecting was correct)

If 1. doesn't boot and 2. boots, then it's clearly breaking on some change done in that commit.

> so not sure if another reverted commit might also fix that

That bootrom is a blackbox, I've no prior experience with this crap. Those bootrom error messages are not helpful either, so basically shooting blindly at some suspicious commits. I was looking at the changes in the binary header payload and in this 2975bba6dfdd ("arm: mvebu: Enable CONFIG_SPL_SYS_NO_VECTOR_TABLE for 32-bit mvebu") case I've noticed significant changes:

```diff
--- spl-v2021.01	2023-01-05 09:28:47.228226163 +0100
+++ spl-v2022.07	2023-01-05 09:29:10.743594603 +0100
@@ -1,2 +1,2 @@
-00000000: 0000 0000 0000 0000 0000 0000 0000 0000  ................
-00000010: 0f00 00ea 14f0 9fe5 14f0 9fe5 14f0 9fe5  ................
+00000000: ffff ffea dc00 00ea 0000 0fe1 1f10 00e2  ................
+00000010: 1a00 31e3 1f00 c013 1300 8013 c000 80e3  ..1.............
```

So I've simply tried to give it a go, but it was apparently just another dead end.

damnms, 2023-01-05:
> Well, bisects are sometimes tricky

yes, seems like its not that commit, i just tested it with git reset --hard 9baab60b80, when i do a git revert HEAD it wants to commit. that built did also not boot.
unfortunately i am really sick now (fever etc.), so not sure how much i can test the next days

damnms, 2023-01-06:
ok, so should i checkout 5029d7bf34ec test, and then 4375598c5274 and test?

ynezz, 2023-01-06:
> ok, so should i checkout 5029d7bf34ec test, and then 4375598c5274 and test?

Yep, thanks.

damnms, 2023-01-06:
5029d7bf34ec and 4375598c5274 worked

johnnyslee, 2023-01-06:
FYI, I always dd the SD image (built by ImageBuilder) to my clearfog-pro's built-in eMMC, and have never encountered this issue at all.

damnms, 2023-01-06:
dd to sd image and then to built-in emmc? like an installer from sd to emmc?
i have another board, i dont have emmc, and probably a much older board revision

johnnyslee, 2023-01-06:
Usually scp img.gz to /tmp/ and `zcat /tmp/img.gz | dd ...` in ssh.
But the very first installation was booting a linux zImage via tftp and then dd openwrt image from usb storage to emmc.

damnms, 2023-01-09:
it seems like git bisect does something, but not what it should. or i use it wrong.
i did now this bisect again:

```
[oli@fedora u-boot]$ git bisect log
git bisect start
# Status: warte auf guten und schlechten Commit
# good: [b70b9b07463db2f6937c7ea6d7fb5122feb7ba1b] Prepare v2021.10-rc1
git bisect good b70b9b07463db2f6937c7ea6d7fb5122feb7ba1b
# Status: warte auf schlechten Commit, 1 guter Commit bekannt
# bad: [a0da2dda4ed9d0aee5265e9cd8876734f9f80e09] Prepare v2021.10-rc2
git bisect bad a0da2dda4ed9d0aee5265e9cd8876734f9f80e09
# bad: [a0953b34d9d8d9309c3eabbb75746fef66b15ffe] Merge https://source.denx.de/u-boot/custodians/u-boot-spi
git bisect bad a0953b34d9d8d9309c3eabbb75746fef66b15ffe
# good: [6c7f152eefd760ceb71aabc36c0025814d645a83] tools: kwbimage: Change maximum number of arguments in binary header to 256
git bisect good 6c7f152eefd760ceb71aabc36c0025814d645a83
# bad: [ea40b20431c0c5de1856c56f965757a40df8d3d7] moveconfig: Update to newer kconfiglib
git bisect bad ea40b20431c0c5de1856c56f965757a40df8d3d7
# bad: [5534fb4f4833eda4e1a2df1c89a75db72e5e2008] arm64: a37xx: pinctrl: Correct PWM pins definitions
git bisect bad 5534fb4f4833eda4e1a2df1c89a75db72e5e2008
# good: [825a2ca023202f73a09660db7a49ac1109917ce8] tools: kwboot: Check for v1 header size
git bisect good 825a2ca023202f73a09660db7a49ac1109917ce8
# good: [9baab60b805472c0911ceabfcd991d59c92b3ddd] SPL: Add support for parsing board / BootROM specific image types
git bisect good 9baab60b805472c0911ceabfcd991d59c92b3ddd
# good: [2a85fdad3e61ff71b1c3acc97e045f390a7263b0] arm: mvebu: Remove unused macro CONFIG_SYS_U_BOOT_OFFS
git bisect good 2a85fdad3e61ff71b1c3acc97e045f390a7263b0
# good: [aa6943ca3122db7ea0063684f94b5dbdf9c5cf51] kwbimage: Add support for extracting images via dumpimage tool
git bisect good aa6943ca3122db7ea0063684f94b5dbdf9c5cf51
# good: [3d7b93d51c26092010626708ba9adaf9a65a764f] kwbimage: Update help message about how to extract from an existing image
git bisect good 3d7b93d51c26092010626708ba9adaf9a65a764f
# first bad commit: [5534fb4f4833eda4e1a2df1c89a75db72e5e2008] arm64: a37xx: pinctrl: Correct PWM pins definitions

```

in between each "git bisect good" or "git bisect bad" i did a make distclean, built the kwb and flashed it to sd.
when it worked, i did a "git bisect good", when it did not i did a "git bisect bad". 
i checked out the "bad commit" and that commit worked, the sd booted. so no idea wth is going on with that but guess it is better i'll do that manually.

ynezz, 2023-01-09:
FYI https://lore.kernel.org/u-boot/20230109213402.GA12418@meh.true.cz/

damnms, 2023-01-14:
i gave it a try today again, now disabled the write cache for all usb devices with an udev rule:
```
[oli@fedora u-boot]$ cat /etc/udev/rules.d/99-udisks2-usb_mount.rules 
SUBSYSTEMS=="usb", SUBSYSTEM=="block", ENV{ID_FS_USAGE}=="filesystem", ENV{UDISKS_MOUNT_OPTIONS_DEFAULTS}+="sync", ENV{UDISKS_MOUNT_OPTIONS_ALLOW}+="sync"
```
but it still behaves the same very nasty way, even flashing the 21.02.05 kwb still produces: U-Boot SPL 2021.10-rc1-00232-ga0953b34d9 (Jan 14 2023 - 09:54:49 +0100) - in the boot log

wondering if this is a fedora problem or if the adapter sucks. i could switch to opensuse tumbleweed if that helps?

damnms, 2023-01-15:
i am so sorry guys, i cant tell you... i guess i messed up with the error messages and messed that totally up :(
i changed my complete hardware, changed the laptop, the distribution, the sd reader, and now i am able to reproduce everything and correct my wrong postings. again, i am so sorry if my postings lead to confusion.

the commit 2226ca leads to this error:
```
U-Boot SPL 2021.10-rc1-00138-g2226ca1734 (Jan 15 2023 - 11:59:36 +0100)
High speed PHY - Version: 2.0
EEPROM TLV detection failed: Using static config for Clearfog Pro.
Detected Device ID 6828
board SerDes lanes topology details:
 | Lane # | Speed |  Type       |
 --------------------------------
 |   0    |   3   | SATA0	|
 |   1    |   0   | SGMII1	|
 |   2    |   5   | PCIe1	|
 |   3    |   5   | USB3 HOST1	|
 |   4    |   5   | PCIe2	|
 |   5    |   0   | SGMII2	|
 --------------------------------
PCIe, Idx 1: detected no link
PCIe, Idx 2: detected no link
High speed PHY - Ended Successfully
mv_ddr: 14.0.0 
DDR3 Training Sequence - Switching XBAR Window to FastPath Window
mv_ddr: completed successfully
Trying to boot from MMC1
mmc_load_image_raw_sector: mmc block read error
SPL: failed to boot from all boot devices
### ERROR ### Please RESET the board ###
```

this is very different than the offset error, and with that hopefully reliable hardware, i was able to figure out where the other (newer) error comes from:
commit 5cad2e6ccee1e5 introduces that behaviour:
```
BootROM - 1.73

Booting from MMC

BootROM: Bad header at offset 00000200

BootROM: Bad header at offset 00004400

BootROM: Bad header at offset 00200000

BootROM: Bad header at offset 00400000

BootROM: Bad header at offset 00600000

BootROM: Bad header at offset 00800000

BootROM: Bad header at offset 00A00000

BootROM: Bad header at offset 00C00000

BootROM: Bad header at offset 00E00000

BootROM: Bad header at offset 01000000

BootROM: Bad header at offset 01200000

BootROM: Bad header at offset 01400000

BootROM: Bad header at offset 01600000

BootROM: Bad header at offset 01800000

BootROM: Bad header at offset 01A00000

BootROM: Bad header at offset 01C00000

BootROM: Bad header at offset 01E00000

BootROM: Bad header at offset 02000000

BootROM: Bad header at offset 02200000

BootROM: Bad header at offset 02400000

BootROM: Bad header at offset 02600000

BootROM: Bad header at offset 02800000

BootROM: Bad header at offset 02A00000

BootROM: Bad header at offset 02C00000

BootROM: Bad header at offset 02E00000

BootROM: Bad h

Trying Uart 
```

again, i am really sorry if that lead to confusion and wasted time :(

so for me this looks like one error was added to another error. if i should test anything, feel free to tell me what i should change or what patch i should apply. another thing i noticed is that i always have to hit enter when i build, it is asking for cmd_setexpr_fmt where i hit enter (default = N)

damnms, 2023-01-17:
unfortunately i have to give this laptop back on thursday evening, so time is a bit running out if i should test something. after thursday i can only test again with the usb<->sd-adapter

ynezz, 2023-01-17:
> this is very different than the offset error

Indeed, IIRC then here U-Boot simply looks for stuff at sector 0, can you retry without `bs=512 seek=1` so you start writing at sector 0? 

>  commit 5cad2e6ccee1e5 introduces that behaviour:

Can you please double check, that e23ad5d55cbc16 (commit before 5cad2e6ccee1e5) is the last version accepted by BootROM?

damnms, 2023-01-17:
> Can you please double check, that e23ad5d55cbc16 (commit before 5cad2e6ccee1e5) is the last version accepted by BootROM?

Confirm.
Output of e23ad5...:

```
U-Boot SPL 2022.01-rc1-00092-ge23ad5d55c (Jan 17 2023 - 15:54:41 +0100)
High speed PHY - Version: 2.0
EEPROM TLV detection failed: Using static config for Clearfog Pro.
Detected Device ID 6828
board SerDes lanes topology details:
 | Lane # | Speed |  Type       |
 --------------------------------
 |   0    |   3   | SATA0	|
 |   1    |   0   | SGMII1	|
 |   2    |   5   | PCIe1	|
 |   3    |   5   | USB3 HOST1	|
 |   4    |   5   | PCIe2	|
 |   5    |   0   | SGMII2	|
 --------------------------------
High speed PHY - Ended Successfully
mv_ddr: 14.0.0 
DDR3 Training Sequence - Switching XBAR Window to FastPath Window
mv_ddr: completed successfully
Trying to boot from MMC1
ERROR: Not valid SPI/NAND/SATA/SDIO kwbimage v1
```

Output of 5cad...:
Bad Header at offset...

> Indeed, IIRC then here U-Boot simply looks for stuff at sector 0, can you retry without bs=512 seek=1 so you start writing at sector 0?

which commit should i test with that?

ynezz, 2023-01-17:
> which commit should i test with that?

2226ca should be good enough

damnms, 2023-01-17:
Did that, got the bad header error again, it looks like as if the broken part (from a newer uboot) is not overwritten without bs and seek and therefore tried to boot that

i will test it now with overwriting with bs=512 seek=1 of 2226ca and see if i get the old error again

ynezz, 2023-01-17:
> Did that, got the bad header error again, it looks like as if the broken part (from a newer uboot) is not overwritten without bs and seek and therefore tried to boot that

IMO it just means, that BootROM for SD source expects header in sector 1, not in sector 0.

damnms, 2023-01-17:
I have no eMMC, i can not comment on that.

ynezz, 2023-01-17:
> If you need that information, i can bisect to find the commit that leads to "Not valid SPI..." - just noticed this is another error.

You're probably not able to bisect that properly, IMO this is SPL output and it simply expects kwbimagev1 header in other offset.

damnms, 2023-01-17:
can i do anything else to get more information so fixing is possible for "someone" ? 
e.g. turn on some trace/debug-logging or so?

svfrd, 2023-01-18:
Clearfog Pro Rev 2.1, 8GB eMMC, owner here.

uboot releases based on openwrt 21.02.5 and 22.03.3 are broken. Sorry did not saved the serial console output.
Switched back to 19.07.10 by uart recovery.

SPL reset and unknown header offset errors.

ynezz, 2023-01-19:
> uboot releases based on openwrt 21.02.5 and 22.03.3 are broken

Thats interesting, so boot from eMMC is broken even in 21.02, but boot from SD card works.

> can i do anything else to get more information so fixing is possible for "someone" ?

Its highly unlikely, that anyone without access to the device with same/similar SoC family is going to fix that, it would take a lot of time. Unless someone else beats me to it, I plan to prepare proper upstream bug report about [5cad2e6ccee1](https://github.com/openwrt/openwrt/issues/11661#issuecomment-1385589140) and send it u-boot mailing list, Cc: involved folks (author of the commit and committer) and see how it goes.

damnms, 2023-01-19:
would it make sense to involve the vendor (clearfog)? i guess they are interested that their device is supported by openwrt :)
maybe they can ship some pieces to dev's

damnms, 2023-01-26:
> I plan to prepare proper upstream bug report 

any news about that?

damnms, 2023-01-27:
i informed SolidRun about this. They answered they will work on it next month and attempt to submit a solution to openwrt. Hope dies last :)

[36 more replies]