openwrt/openwrt #11668: [openwrt-21.02] rtl83xx driver failed to build due to backported dsa patches

1715173329, 2023-01-02:
### Describe the bug

Rafał Miłecki backported some dsa patches a month ago[^1], and that causes rtl83xx driver no longer build.
The rtl83xx driver is used by realtek target, so the target fails[^2] too.

[^1]: https://github.com/openwrt/openwrt/commit/1f5024aa73fcf3b67bba7785401f9cd7067c88f2
[^2]: https://buildbot.openwrt.org/openwrt-21.02/images/#/builders/44/builds/53

### OpenWrt version

openwrt-21.02 HEAD

### OpenWrt target/subtarget

realtek/generic

### Device

any devices under realtek/generic

### Image kind

Self-built image

### Actual behaviour

```
drivers/net/dsa/rtl83xx/common.c:22:13: warning: 'dump_fdb' defined but not used [-Wunused-function]
 static void dump_fdb(struct rtl838x_switch_priv *priv)
             ^~~~~~~~
  CC      drivers/net/dsa/rtl83xx/dsa.o
  CC      net/core/dev_addr_lists.o
drivers/net/dsa/rtl83xx/dsa.c:1274:22: error: initialization of 'enum dsa_tag_protocol (*)(struct dsa_switch *, int,  enum dsa_tag_protocol)' from incompatible pointer type 'enum dsa_tag_protocol (*)(struct dsa_switch *, int)' [-Werror=incompatible-pointer-types]
  .get_tag_protocol = rtl83xx_get_tag_protocol,
                      ^~~~~~~~~~~~~~~~~~~~~~~~
drivers/net/dsa/rtl83xx/dsa.c:1274:22: note: (near initialization for 'rtl83xx_switch_ops.get_tag_protocol')
drivers/net/dsa/rtl83xx/dsa.c:1316:22: error: initialization of 'enum dsa_tag_protocol (*)(struct dsa_switch *, int,  enum dsa_tag_protocol)' from incompatible pointer type 'enum dsa_tag_protocol (*)(struct dsa_switch *, int)' [-Werror=incompatible-pointer-types]
  .get_tag_protocol = rtl83xx_get_tag_protocol,
                      ^~~~~~~~~~~~~~~~~~~~~~~~
drivers/net/dsa/rtl83xx/dsa.c:1316:22: note: (near initialization for 'rtl930x_switch_ops.get_tag_protocol')
cc1: some warnings being treated as errors
make[8]: *** [scripts/Makefile.build:262: drivers/net/dsa/rtl83xx/dsa.o] Error 1
make[7]: *** [scripts/Makefile.build:497: drivers/net/dsa/rtl83xx] Error 2
make[6]: *** [scripts/Makefile.build:497: drivers/net/dsa] Error 2
make[5]: *** [scripts/Makefile.build:497: drivers/net] Error 2
make[4]: *** [Makefile:1739: drivers] Error 2
make[4]: *** Waiting for unfinished jobs....
```

### Additional info

I applied this patch and made it build, but I'm not sure if it's correct:
```patch
diff --git a/target/linux/realtek/files-5.4/drivers/net/dsa/rtl83xx/dsa.c b/target/linux/realtek/files-5.4/drivers/net/dsa/rtl83xx/dsa.c
index 987b47dc8f..17928e7650 100644
--- a/target/linux/realtek/files-5.4/drivers/net/dsa/rtl83xx/dsa.c
+++ b/target/linux/realtek/files-5.4/drivers/net/dsa/rtl83xx/dsa.c
@@ -145,7 +145,9 @@ const struct rtl83xx_mib_desc rtl83xx_mib[] = {
 /* DSA callbacks */
 
 
-static enum dsa_tag_protocol rtl83xx_get_tag_protocol(struct dsa_switch *ds, int port)
+static enum dsa_tag_protocol rtl83xx_get_tag_protocol(struct dsa_switch *ds,
+						     int port,
+						     enum dsa_tag_protocol mprot)
 {
 	/* The switch does not tag the frames, instead internally the header
 	 * structure for each packet is tagged accordingly.
```

layerscape's felix driver seems unused on 21.02 branch, but I think it should be fixed too:
```patch
diff --git a/target/linux/layerscape/patches-5.4/701-net-0263-net-dsa-ocelot-add-driver-for-Felix-switch-family.patch b/target/linux/layerscape/patches-5.4/701-net-0263-net-dsa-ocelot-add-driver-for-Felix-switch-family.patch
index 5dc50c5ee6..3d22fc42c5 100644
--- a/target/linux/layerscape/patches-5.4/701-net-0263-net-dsa-ocelot-add-driver-for-Felix-switch-family.patch
+++ b/target/linux/layerscape/patches-5.4/701-net-0263-net-dsa-ocelot-add-driver-for-Felix-switch-family.patch
@@ -109,7 +109,7 @@ Signed-off-by: David S. Miller <davem@davemloft.net>
 +	felix_vsc9959.o
 --- /dev/null
 +++ b/drivers/net/dsa/ocelot/felix.c
-@@ -0,0 +1,441 @@
+@@ -0,0 +1,442 @@
 +// SPDX-License-Identifier: GPL-2.0
 +/* Copyright 2019 NXP Semiconductors
 + */
@@ -122,7 +122,8 @@ Signed-off-by: David S. Miller <davem@davemloft.net>
 +#include "felix.h"
 +
 +static enum dsa_tag_protocol felix_get_tag_protocol(struct dsa_switch *ds,
-+						    int port)
++						    int port,
++						    enum dsa_tag_protocol mp)
 +{
 +	return DSA_TAG_PROTO_OCELOT;
 +}
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ynezz, 2023-01-02:
@rmilecki can you please take a look?

rmilecki, 2023-01-02:
Absolutely, checking it

rmilecki, 2023-01-02:
f5646ae0b903 ("realtek: update rtl83xx switch driver to the updated DSA API")

closed 2023-01-02 by rmilecki

rmilecki, 2023-01-03:
ee1eda7c7d51 ("layerscape: fix felix DSA driver compilation")

rmilecki, 2023-01-03:
Thanks for reporting this.