openwrt/openwrt #11674: FRITZ!Box 4020 generates new MAC on eth0 on each boot

grische, 2023-01-02:
### Describe the bug

Initial boot:
```
root@OpenWrt:~# cat /sys/class/net/eth0/address
e6:e5:91:1a:f6:1d
```

After power Cycle
```
root@OpenWrt:~# cat /sys/class/net/eth0/address
ca:0f:51:81:9d:85
```

### OpenWrt version

r19803-9a599fee93

### OpenWrt target/subtarget

ath79/generic

### Device

AVM FRITZ!Box 4020

### Image kind

Official downloaded image

### Steps to reproduce

1. Install OpenWRT
2. Run `for device in /sys/class/net/*/address; do cat "$device" | tr -d $'\n' && echo "$device" | sed 's/.*\/\([^\/]*\)\/address/ \1/'; done`
3. Power Cycle device
4. Run same command again.
5. Observe difference.

### Expected behaviour

Having the same MAC address one each boot

### Additional info

Potentially related to #9010

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

blocktrron, 2023-01-03:
Please share the output of 
`fritz_tffs -i /dev/mtd5 -a`

tachtler, 2023-01-03:
Hi blocktrron,

```
HWRevision=219
HWSubRevision=1
ProductID=Fritz_Box_HW219
SerialNumber=0000000000000000
annex=Ohne
autoload=yes
bootloaderVersion=1.2779
bootserport=tty0
cpufrequency=775000000
firstfreeaddress=0x8113DB4C
firmware_info=147.06.83
firmware_version=avm
flashsize=nor_size=0 sflash_size=16MB nand_size=0MB
maca=C8:0E:14:B5:BB:B7
macb=C8:0E:14:B5:BB:B8
macwlan=C8:0E:14:B5:BB:B9
macwlan2=C8:0E:14:B5:BB:BA
macdsl=C8:0E:14:B5:BB:B4
memsize=0x08000000
modetty0=38400,n,8,1,hw
modetty1=38400,n,8,1,hw
mtd0=0x9F000000,0x9F000000
mtd1=0x9F020000,0x9FF00000
mtd2=0x9F000000,0x9F020000
mtd3=0x9FF00000,0x9FF80000
mtd4=0x9FF80000,0xA0000000
my_ipaddress=192.168.178.1
prompt=Eva_AVM
req_fullrate_freq=216666666
sysfrequency=216666666
tr069_passphrase=[REDACTED]
tr069_serial=00040E-C80E14B5BBB7
urlader-version=3779
usb_board_mac=C8:0E:14:B5:BB:B5
usb_device_id=0x0000
usb_device_name=USB DSL Device
usb_manufacturer_name=AVM
usb_revision_id=0x0000
usb_rndis_mac=C8:0E:14:B5:BB:B6
webgui_pass=[REDACTED]
wlan_key=[REDACTED]
wlan_ssid=FRITZ!Box#4020#W
```
Thank you!
Klaus.

Djfe, 2023-01-08:
I can confirm this. eth0.1 stays the same after a reboot, but eth0 and eth1 are picked at random it seems.

The device is currently running Gluon 2022 and I don't have physical access to it atm, just ssh.
openwrt-22.03
openwrt commit 6c3e16055ae5d79205812e8436110fd862266c94

grische, 2023-03-04:
@blocktrron If `eth0` is not meant to have a fixed MAC, but only `eth0.1` (part of  `br-lan`) and `eth1`, then I think we can close the issue here.

Using the MAC addresses of the above TFFS dump, OpenWRT sets the following MACs after boot:
```
root@OpenWrt:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP qlen 1000
    link/ether e2:14:e7:0f:f6:6c brd ff:ff:ff:ff:ff:ff
    inet6 fe80::e014:e7ff:fe0f:f66c/64 scope link
       valid_lft forever preferred_lft forever
3: eth1: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc fq_codel state DOWN qlen 1000
    link/ether c8:0e:14:b5:bb:b8 brd ff:ff:ff:ff:ff:ff
5: wlan0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN qlen 1000
    link/ether c8:0e:14:b5:bb:b9 brd ff:ff:ff:ff:ff:ff
6: br-lan: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP qlen 1000
    link/ether c8:0e:14:b5:bb:b7 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.1/24 brd 192.168.1.255 scope global br-lan
       valid_lft forever preferred_lft forever
    inet6 fd84:ba58:ed3c::1/60 scope global noprefixroute
       valid_lft forever preferred_lft forever
    inet6 fe80::ca0e:14ff:feb5:bbb7/64 scope link
       valid_lft forever preferred_lft forever
7: eth0.1@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-lan state UP qlen 1000
    link/ether c8:0e:14:b5:bb:b7 brd ff:ff:ff:ff:ff:ff
```

I can confirm the issue mentioned by Djfe, but only for Gluon, not OpenWRT.

closed 2023-03-04 by grische