openwrt/openwrt #11677: ipq40xx, EA8300, DSA (master), sysupgrade does not restore settings

jeffsf, 2023-01-03:
### Describe the bug

After installing a local build off

commit c9a7c00f80 (master)
Author: Olliver Schinagl <oliver@schinagl.nl>
Date:   Tue Dec 13 16:35:37 2022 +0100

using the process described at https://github.com/openwrt/openwrt/pull/11405#issuecomment-1336614898 the system boots into the new firmware.

After modifying the config and confirming `option compat_version '2.0'`, subsequent attempts to `sysupgrade` to the new firmware on the other partition flash the firmware, but do not preserve the config. This occurs with either
```
sysupgrade -u /tmp/OpenWrt-2023-01-02_1321-0800-ipq40xx-generic-linksys_ea8300-squashfs-sysupgrade.bin
sysupgrade /tmp/OpenWrt-2023-01-02_1321-0800-ipq40xx-generic-linksys_ea8300-squashfs-sysupgrade.bin
```

### OpenWrt version

r21655+2-c9a7c00f80

### OpenWrt target/subtarget

ipq40xx/generic

### Device

Linksys EA8300

### Image kind

Self-built image

### Steps to reproduce

```
root@devel:/# sysupgrade /tmp/Open
root@devel:/# sysupgrade /tmp/OpenWrt-2023-01-02_1321-0800-ipq40xx-generic-links
ys_ea8300-squashfs-sysupgrade.bin [J
Sun Jan  1 22:56:23 PST 2023 upgrade: Saving config files...
Sun Jan  1 22:56:24 PST 2023 upgrade: Commencing upgrade. Closing all shell sessions.
Watchdog handover: fd=3
- watchdog -
Watchdog did not previously reset the system
[  210.720844] ath10k_pci 0000:01:00.0: peer-unmap-event: unknown peer id 2
[  210.989890] ath10k_pci 0000:01:00.0: peer-unmap-event: unknown peer id 1
Sun Jan  1 22:56:25 PST 2023 upgrade: Sending TERM to remaining processes ...
Sun Jan  1 22:56:25 PST 2023 upgrade: Sending signal TERM to wpa_supplicant (1035)
Sun Jan  1 22:56:25 PST 2023 upgrade: Sending signal TERM to wpa_supplicant (1058)
Sun Jan  1 22:56:25 PST 2023 upgrade: Sending signal TERM to uwsgi (1318)
[  211.229850] ath10k_pci 0000:01:00.0: peer-unmap-event: unknown peer id 3
Sun Jan  1 22:56:29 PST 2023 upgrade: Sending KILL to remaining processes ...
[  221.540212] stage2 (2654): drop_caches: 3
Sun Jan  1 22:56:36 PST 2023 upgrade: Switching to ramdisk...
[  225.903154] UBIFS (ubi0:1): background thread "ubifs_bgt0_1" stops
[  225.926741] UBIFS (ubi0:1): un-mount UBI device 0
Mon Jan  2 06:56:40 UTC 2023 upgrade: Performing system upgrade...
[  226.323527] ubi1: attaching mtd11
[  226.929591] ubi1: scanning is finished
[  226.939160] ubi1: attached mtd11 (name "rootfs", size 83 MiB)
[  226.939223] ubi1: PEB size: 131072 bytes (128 KiB), LEB size: 126976 bytes
[  226.943978] ubi1: min./max. I/O unit sizes: 2048/2048, sub-page size 2048
[  226.950719] ubi1: VID header offset: 2048 (aligned 2048), data offset: 4096
[  226.957694] ubi1: good PEBs: 664, bad PEBs: 0, corrupted PEBs: 0
[  226.964353] ubi1: user volume: 2, internal volumes: 1, max. volumes count: 128
[  226.970627] ubi1: max/mean erase counter: 6/3, WL threshold: 4096, image sequence number: 1651709940
[  226.977602] ubi1: available PEBs: 0, total reserved PEBs: 664, PEBs reserved for bad PEB handling: 40
[  226.986953] ubi1: background thread "ubi_bgt1d" started, PID 3370
UBI device number 1, total 664 LEBs (84312064 bytes, 80.4 MiB), available 0 LEBs (0 bytes), LEB size 126976 bytes (124.0 KiB)
Unlocking kernel ...

[... 573 lines cut ...]
CONFIG_PACKAGE_tcpdump-mini=y
CONFIG_PACKAGE_terminfo=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-lua=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uwsgi=y
CONFIG_PACKAGE_uwsgi-cgi-plugin=y
CONFIG_PACKAGE_uwsgi-luci-support=y
CONFIG_PACKAGE_uwsgi-syslog-plugin=y
CONFIG_PACKAGE_wpad-basic-wolfssl=m
CONFIG_PACKAGE_wpad-mesh-openssl=y
CONFIG_PACKAGE_xz=y
CONFIG_PACKAGE_xz-utils=y
CONFIG_PACKAGE_zlib=y
CONFIG_PCRE2_JIT_ENABLED=y
CONFIG_PCRE_JIT_ENABLED=y
CONFIG_RSYNC_acl=y
CONFIG_RSYNC_xattr=y
# CONFIG_WOLFSSL_HAS_OPENVPN is not set
```
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

jeffsf, 2023-01-03:
Can be replicated with "official" SNAPSHOT builds of January 2nd

https://downloads.openwrt.org/snapshots/targets/ipq40xx/generic/openwrt-ipq40xx-generic-linksys_ea8300-squashfs-factory.bin
https://downloads.openwrt.org/snapshots/targets/ipq40xx/generic/openwrt-ipq40xx-generic-linksys_ea8300-squashfs-sysupgrade.bin

(Local time here is UTC+8, US Pacific Time)

[screenlog.devel.2023-01-02_1919.log](https://github.com/openwrt/openwrt/files/10334209/screenlog.devel.2023-01-02_1919.log)

jeffsf, 2023-01-04:
Tracing things here, it looks like `do_upgrade_success` (which would call `nand_do_restore_config`) is not being called after `nand_upgrade_tar` returns. 

It's been a long time since I've looked at this code and there have been several refactors.

jow-, 2023-01-04:
@dangowrt

commit 8634c108 (2023-03-26): ipq40xx: Fix Linksys upgrade, restore config step

It appears that the refactor of the upgrade process for NAND devices
resulted in the nand_do_upgrade_success step not being called for
devices using the linksys.sh script. As a result, configuration
was not preserved over sysupgrade steps.

This was restored for some devices in
commit 84ff6c90dda1 ("base-files: bring back nand_do_upgrade_success").
This restored preservation of config for ipq40xx devices using the
linksys.sh script. Other devices and targets have not been examined.

Closes: #11677
Fixes: e25e6d8e54 ("base-files: fix and clean  [... cut]
closed 2023-03-26 by openwrt-bot