openwrt/openwrt #11681: [Xiaomi Mi Router CR6608(CR660x)]

ypj0202, 2023-01-03:
### Describe the bug

The ROM of this device is 128MB and 256MB RAM. I noticed that I am only able to use 64MB (58.6MB) for installing the software in '/overlay' partition. In the storage section, there are two partition: Disk and temp space(Which is 1/2 of the RAM). Is there a way to make use of all 128 MB instead of 64MB in the ROM or this how the system is designed?
### Output of `df -h`
```
root@G:~# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 4.0M      4.0M         0 100% /rom
tmpfs                   122.4M   1004.0K    121.4M   1% /tmp
/dev/ubi0_2              58.6M      4.8M     50.7M   9% /overlay
overlayfs:/overlay       58.6M      4.8M     50.7M   9% /
tmpfs                   512.0K         0    512.0K   0% /dev
```
### LuCI
[image]
 ### Partition
```
root@G:~# cat /proc/mtd
dev:    size   erasesize  name
mtd0: 00080000 00020000 "Bootloader"
mtd1: 00040000 00020000 "Nvram"
mtd2: 00040000 00020000 "Bdata"
mtd3: 00080000 00020000 "Factory"
mtd4: 00040000 00020000 "crash"
mtd5: 00040000 00020000 "crash_log"
mtd6: 00400000 00020000 "kernel"
mtd7: 07980000 00020000 "ubi"
```

### OpenWrt version

r19777-2853b6d652

### OpenWrt target/subtarget

ramips/mt7621

### Device

CR6608

### Image kind

Official downloaded image

### Actual behaviour

Only 64MB usable in `/overlay` partition

### Expected behaviour

128MB in total usable

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

infiWang, 2023-01-03:
I presume the ROM you're using is rather chunky, leaving less storage available?

ypj0202, 2023-01-03:
> chunky

I am using the ROM from Openwrt's website, with only default packages loaded. The size of the ROM is 6681KB(6.52MB). Even deducting that, there is still no way that it occupies half of the ROM. Is there anything I can provide to assist identifying/solving the problem?

jow-, 2023-01-04:
Please provide the complete `dmesg` output directly after boot in order to see the effective MTD partition table.

ypj0202, 2023-01-04:
```
[    0.000000] Linux version 5.10.146 (builder@buildhost) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 r19777-2853b6d652) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 SMP Fri Oct 7 23:34:56 2022
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is Xiaomi Mi Router CR6608
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] On node 0 totalpages: 65536
[    0.000000]   Normal zone: 576 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 65536 pages, LIFO batch:15
[    0.000000] percpu: Embedded 15 pages/cpu s30256 r8192 d22992 u61440
[    0.000000] pcpu-alloc: s30256 r8192 d22992 u61440 alloc=15*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64960
[    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Writing ErrCtl register=000204a0
[    0.000000] Readback ErrCtl register=000204a0
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 249360K/262144K available (7005K kernel code, 625K rwdata, 824K rodata, 1260K init, 243K bss, 12784K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 256
[    0.000000] CPU Clock: 880MHz
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
[    0.000012] sched_clock: 64 bits at 880MHz, resolution 1ns, wraps every 4398046511103ns
[    0.007948] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
[    0.017003] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
[    0.083165] pid_max: default: 32768 minimum: 301
[    0.087930] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.095133] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.105787] rcu: Hierarchical SRCU implementation.
[    0.110831] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
[    0.118935] smp: Bringing up secondary CPUs ...
[    0.124218] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.124230] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.124242] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.124321] CPU1 revision is: 0001992f (MIPS 1004Kc)
[    0.178740] Synchronize counters for CPU 1: done.
[    0.211272] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.211281] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.211290] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.211338] CPU2 revision is: 0001992f (MIPS 1004Kc)
[    0.270197] Synchronize counters for CPU 2: done.
[    0.300790] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.300798] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.300806] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.300858] CPU3 revision is: 0001992f (MIPS 1004Kc)
[    0.355382] Synchronize counters for CPU 3: done.
[    0.385252] smp: Brought up 1 node, 4 CPUs
[    0.393463] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.403285] futex hash table entries: 1024 (order: 3, 32768 bytes, linear)
[    0.410304] pinctrl core: initialized pinctrl subsystem
[    0.417966] NET: Registered protocol family 16
[    0.423654] thermal_sys: Registered thermal governor 'step_wise'
[... 258 lines cut ...]
[   31.757358] br-lan: port 3(lan3) entered blocking state
[   31.762742] br-lan: port 3(lan3) entered forwarding state
[   31.997840] br-lan: port 5(wlan1) entered blocking state
[   32.003219] br-lan: port 5(wlan1) entered disabled state
[   32.009189] device wlan1 entered promiscuous mode
[   32.014791] br-lan: port 5(wlan1) entered blocking state
[   32.020372] br-lan: port 5(wlan1) entered forwarding state
[   32.032634] mt7530 mdio-bus:1f lan2: Link is Up - 1Gbps/Full - flow control rx/tx
[   32.040388] br-lan: port 2(lan2) entered blocking state
[   32.045776] br-lan: port 2(lan2) entered forwarding state
[   32.721349] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[... cut ...]
[   33.085170] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1-1: link becomes ready
[   33.092407] br-lan: port 6(wlan1-1) entered blocking state
[   33.097956] br-lan: port 6(wlan1-1) entered forwarding state
```
When it boots up, it can recognize the flash size which is 128MB \
`[    0.670083] nand: 128 MiB, SLC, erase size: 128 KiB, page size: 2048, OOB size: 64`
However, when it mounts the partition, it becomes 64MB \
`[   11.714229] UBIFS (ubi0:2): FS size: 67170304 bytes (64 MiB, 529 LEBs), journal size 3301376 bytes (3 MiB, 26 LEBs)`

DragonBluep, 2023-01-05:
It should be caused by bad block. Please try to upgrade OpenWrt in LuCI and do not retain the configuration.

981213, 2023-01-06:
Hi! Could you ssh into the router, run the following command:
```
ubinfo -a
```
and paste the output here?

congbui137, 2024-01-23:
It was fine with me.

Output of `df -h`:
```
root@OpenWrt:~# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 5.0M      5.0M         0 100% /rom
tmpfs                   118.8M    504.0K    118.3M   0% /tmp
/dev/ubi0_1              98.9M    176.0K     94.0M   0% /overlay
overlayfs:/overlay       98.9M    176.0K     94.0M   0% /
tmpfs                   512.0K         0    512.0K   0% /dev
```

Output of `ubinfo -a`:
```
root@OpenWrt:~# ubinfo -a
UBI version:                    1
Count of UBI devices:           1
UBI control device major/minor: 10:62
Present UBI devices:            ubi0

ubi0
Volumes count:                           2
Logical eraseblock size:                 126976 bytes, 124.0 KiB
Total amount of logical eraseblocks:     964 (122404864 bytes, 116.7 MiB)
Amount of available logical eraseblocks: 0 (0 bytes)
Maximum count of volumes                 128
Count of bad physical eraseblocks:       0
Count of reserved physical eraseblocks:  20
Current maximum erase counter value:     2
Minimum input/output unit size:          2048 bytes
Character device major/minor:            251:0
Present volumes:                         0, 1

Volume ID:   0 (on ubi0)
Type:        dynamic
Alignment:   1
Size:        40 LEBs (5079040 bytes, 4.8 MiB)
State:       OK
Name:        rootfs
Character device major/minor: 251:1
-----------------------------------
Volume ID:   1 (on ubi0)
Type:        dynamic
Alignment:   1
Size:        900 LEBs (114278400 bytes, 108.9 MiB)
State:       OK
Name:        rootfs_data
Character device major/minor: 251:2
```

[image]