openwrt/openwrt #11687: ipv4 dhcp leases not shown in overview, cannot make static ipv4 lease

gzivdo, 2023-01-04:
### Describe the bug

In trunk whole last year (may be more) ipv4 leases not shown.
Also there is no way to simply completly disable ipv6 by one check box. Even if i disable it completly, ipv4 leases not displayed, not able make static ipv4 dhcp lease.

### OpenWrt version

r21657-9260027535

### OpenWrt target/subtarget

mediatek/mt7622

### Device

ax6s

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

jow-, 2023-01-04:
Please provide the output of the following commands:
 - `cat /etc/openwrt_release`
 - `cat /etc/config/dhcp`
 - `cat /tmp/dhcp.leases`
 - `ubus call system board`
 - `ubus call luci-rpc getDHCPLeases`

gzivdo, 2023-01-04:
cat /etc/openwrt_release
```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r21657-9260027535'
DISTRIB_TARGET='mediatek/mt7622'
DISTRIB_ARCH='aarch64_cortex-a53'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r21657-9260027535'
DISTRIB_TAINTS=''
```

cat /etc/config/dhcp
```
config dnsmasq
        option domainneeded '1'
        option localise_queries '1'
        option rebind_protection '1'
        option rebind_localhost '1'
        option local '/lan/'
        option domain 'lan'
        option expandhosts '1'
        option authoritative '1'
        option readethers '1'
        option leasefile '/tmp/dhcp.leases'
        option resolvfile '/tmp/resolv.conf.d/resolv.conf.auto'
        option localservice '1'
        option ednspacket_max '1232'

config dhcp 'lan'
        option interface 'lan'
        option start '100'
        option limit '150'
        option leasetime '12h'
        option dhcpv4 'server'
        list ra_flags 'managed-config'
        list ra_flags 'other-config'

config dhcp 'wan'
        option interface 'wan'
        option ignore '1'

config odhcpd 'odhcpd'
        option maindhcp '0'
        option leasefile '/tmp/hosts/odhcpd'
        option leasetrigger '/usr/sbin/odhcpd-update'
        option loglevel '4'

config host
        option name 'minipc1'
        option dns '1'
        option mac '4C:1D:96:-:-:-'
        option ip '192.168.1.101'

```
cat /tmp/dhcp.leases - empty

ubus call system board
```
{
        "kernel": "5.15.85",
        "hostname": "OpenWrt",
        "system": "ARMv8 Processor rev 4",
        "model": "Xiaomi Redmi Router AX6S",
        "board_name": "xiaomi,redmi-router-ax6s",
        "rootfs_type": "squashfs",
        "release": {
                "distribution": "OpenWrt",
                "version": "SNAPSHOT",
                "revision": "r21657-9260027535",
                "target": "mediatek/mt7622",
                "description": "OpenWrt SNAPSHOT r21657-9260027535"
        }
}
```
ubus call luci-rpc getDHCPLeases
```
{
        "dhcp_leases": [

        ],
        "dhcp6_leases": [

        ]
}
```

But devices getting ipv4 addresses via dhcp.

jow-, 2023-01-04:
Please provide the complete output of `logread` right after boot.

gzivdo, 2023-01-04:
I really sorry, i just found another dhcp server in my lan.
But reporting this issue is triggered me to make investigation.

jow-, 2023-01-04:
No problem, glad you found the culprit

closed 2023-01-04 by jow-