openwrt/openwrt #11688: [22.03]"No rule to make target" at compile time

scmplie, 2023-01-04:
### Describe the bug

make[4]: Entering directory '/home/scmplie/Desktop/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.161'
make[5]: *** No rule to make target '/home/scmplie/Desktop/openwrt/build_dir/target-x86_64_musl/linux-x86_64/button-hotplug/button-hotplug.o', needed by '/home/scmplie/Desktop/openwrt/build_dir/target-x86_64_musl/linux-x86_64/button-hotplug/button-hotplug.mod'.  Stop.
make[4]: *** [Makefile:1837: /home/scmplie/Desktop/openwrt/build_dir/target-x86_64_musl/linux-x86_64/button-hotplug] Error 2
make[4]: Leaving directory '/home/scmplie/Desktop/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.161'
make[3]: *** [Makefile:53: /home/scmplie/Desktop/openwrt/build_dir/target-x86_64_musl/linux-x86_64/button-hotplug/.built] Error 2
make[3]: Leaving directory '/home/scmplie/Desktop/openwrt/package/kernel/button-hotplug'
time: package/kernel/button-hotplug/compile#0.99#0.28#1.27
    ERROR: package/kernel/button-hotplug failed to build.
make[2]: *** [package/Makefile:116: package/kernel/button-hotplug/compile] Error 1
make[2]: Leaving directory '/home/scmplie/Desktop/openwrt'
make[1]: *** [package/Makefile:110: /home/scmplie/Desktop/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/scmplie/Desktop/openwrt'
make: *** [/home/scmplie/Desktop/openwrt/include/toplevel.mk:230: world] Error 2

I have compiled this firmware twice, and this error is reported every time.I even reinstalled the system on the second try, all the steps followed the instructions, but it still reported the same error

### OpenWrt version

openwrt-22.03

### OpenWrt target/subtarget

X86/X86_64

### Device

ubuntu22.04

### Image kind

Self-built image

### Additional info

scmplie@scmplie-virtual-machine:~/Desktop/openwrt/build_dir/target-x86_64_musl/linux-x86_64/button-hotplug$ gcc -c button-hotplug.c 
button-hotplug.c:18:10: fatal error: linux/kmod.h: No such file or directory
   18 | #include <linux/kmod.h>
      |          ^~~~~~~~~~~~~~
compilation terminated.

I tried to compile the single file button-hotplug.c, and it said the lack of kmod.h. I really don't konw how to finish this.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

scmplie, 2023-01-08:
i finally konw that this problem had been fixed in the snapshot，but not in stable.

lingfeng0303, 2023-06-08:
最好找到解决方案了么？

scmplie, 2023-06-08:
没有，要么用快照版，要么不加额外的插件可以编译成功