openwrt/openwrt #11695: Belkin E8450 Wireless Performance Problem

clutch2sft, 2023-01-04:
### Describe the bug

Admittedly this isn’t exactly a bug definition.

However, I have done extensive testing and can say without a doubt that there is an issue at a minimum with the performance on this platform.  Most notably and reproducibly older .11n clients suffer severe performance decline on the 2.4Gig radio.  To a lesser degree modern clients do have similar issues as seen in the wireshark trace files but it is less noticable in the test results and in the user experience.

Here is a typical “bad” result:

https://www.waveform.com/tools/bufferbloat?test-id=75c6201f-0bc1-4b5b-af29-4a3f8c8b74ba

And here is a typical result in the case of the same client connected to many different things.  Such as – a WAX202 running OpenWrt, a Belkin E8450 running Belkin firmware, some old .11N AP I had laying around…. Etc.

https://www.waveform.com/tools/bufferbloat?test-id=b4076163-4548-40b2-ac6c-9939353f75e8

To be clear I am strictly focused on the bandwidth performance not latency.

I have run at least 200 tests in each configuration.

The most notable thing I can find in the Wireshark capture of the “bad” caputre are thousands of dup acks that do not exist in the more performant captures.

Generally this testing has been limited to HT20 and with the following adapter/windows client.

```
    Name                   : Wi-Fi
    Description            : Dell Wireless 1705 802.11b/g/n (2.4GHZ)
    GUID                   : 5b74f3db-76a5-4762-ba7b-7b3bd1f428f7
    Physical address       : 4c:bb:58:9b:f9:9f
    State                  : connected
    SSID                   : Belkin05582
    BSSID                  : e8:9f:80:e7:f8:7e
    Network type           : Infrastructure
    Radio type             : 802.11n
    Authentication         : Open
    Cipher                 : None
    Connection mode        : Profile
    Channel                : 3
    Receive rate (Mbps)    : 72.2
    Transmit rate (Mbps)   : 72.2
    Signal                 : 100%
    Profile                : Belkin05582
'''
Although I have done some testing in HT40 and can say that this particular client shows modest improvement when the TX/RX rate is 150Mbps on the E8450 running OpenWrt.  By modest I mean sometimes being able to achieve 16-18Mbps of download throughput.

Interestingly upload throughput rarely suffers.  Most of my tests were conducted with a 350Mbps/30Mbps circuit.  At the same time I did conduct multiple tests at another location on a 80Mbps/12Mbps circuit.

### OpenWrt version

r21663-b069d4ac6d

### OpenWrt target/subtarget

mediatek/mt7622

### Device

E8450

### Image kind

Official downloaded image

### Steps to reproduce

First I have run many tests on many images for this platform.  However, to file this bug I did the following:

1) firstboot && reload now
2) upload latest snapshot from the repository
3) turn on the default SSID (OpenWrt) MediaTek MT7622 802.11bgn
4) connected the .11n client noted above 
5) ran wireshark on the client and conducted the performance test.

NOTE:  I also did a lot of work up to now with the iperf client and here are some results of that to the router and to the WAN interface from a ver similar client.  To make more clear the WAN interface of the E8450 is 10.0.0.186.

```
C:\Users\George\Downloads>iperf.exe -c 172.16.5.1 -i 1
------------------------------------------------------------
Client connecting to 172.16.5.1, TCP port 5001
TCP window size: 64.0 KByte (default)
------------------------------------------------------------
[ 1] local 172.16.5.164 port 62184 connected with 172.16.5.1 port 5001
[ ID] Interval Transfer Bandwidth
[ 1] 0.00-1.00 sec 1.88 MBytes 15.7 Mbits/sec
[ 1] 1.00-2.00 sec 768 KBytes 6.29 Mbits/sec
[ 1] 2.00-3.00 sec 0.000 Bytes 0.000 bits/sec
[ 1] 3.00-4.00 sec 0.000 Bytes 0.000 bits/sec
[ 1] 4.00-5.00 sec 5.13 MBytes 43.0 Mbits/sec
[ 1] 5.00-6.00 sec 5.75 MBytes 48.2 Mbits/sec
[ 1] 6.00-7.00 sec 5.63 MBytes 47.2 Mbits/sec
[ 1] 7.00-8.00 sec 5.88 MBytes 49.3 Mbits/sec
[ 1] 8.00-9.00 sec 5.50 MBytes 46.1 Mbits/sec
[ 1] 9.00-10.00 sec 5.75 MBytes 48.2 Mbits/sec
[ 1] 0.00-10.22 sec 36.4 MBytes 29.9 Mbits/sec

C:\Users\George\Downloads>iperf.exe -c 10.0.0.186 -i 1 -p 5002
------------------------------------------------------------
Client connecting to 10.0.0.186, TCP port 5002
TCP window size: 64.0 KByte (default)
------------------------------------------------------------
[ 1] local 172.16.5.164 port 62188 connected with 10.0.0.186 port 5002
[ ID] Interval Transfer Bandwidth
[ 1] 0.00-1.00 sec 6.00 MBytes 50.3 Mbits/sec
[ 1] 1.00-2.00 sec 5.25 MBytes 44.0 Mbits/sec
[ 1] 2.00-3.00 sec 5.63 MBytes 47.2 Mbits/sec
[ 1] 3.00-4.00 sec 1.25 MBytes 10.5 Mbits/sec
[ 1] 4.00-5.00 sec 128 KBytes 1.05 Mbits/sec
[ 1] 5.00-6.00 sec 128 KBytes 1.05 Mbits/sec
[ 1] 6.00-7.00 sec 128 KBytes 1.05 Mbits/sec
[ 1] 7.00-8.00 sec 0.000 Bytes 0.000 bits/sec
[ 1] 8.00-9.00 sec 0.000 Bytes 0.000 bits/sec
[ 1] 9.00-10.00 sec 0.000 Bytes 0.000 bits/sec
[ 1] 0.00-11.90 sec 18.5 MBytes 13.0 Mbits/sec

```

Which is an indicator to me there is something up in the path inside the router.  

### Actual behaviour

Almost always speeds are below 10Mbps with an occassional outlier which I can not explain.  Outlier = 1 in 50 tests manages to perform more or less as expected.

### Expected behaviour

In testing this same client to other devices I see the performance I would roughly expect for a client that connects at 72.2.  Specifically most often in the 48-50Mbps downloads and occassionaly as high as 63-65Mbps.

### Additional info

The RF environment at 1 test site is completely clean but in all test cases the noise readings were below -80 and often way below.

I'm happy to keep all these test things around to help in any way that I can with data, traces, etc.  Please let me know how I can help.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

clutch2sft, 2023-01-04:
Here are three tests I just completed back to back to back.  I did forget to mention above that the situation is exacerbated by having anything connected to a LAN port.  In this test a Linux PC was connected to LAN1 but it was mostly idle.
[testing.txt](https://github.com/openwrt/openwrt/files/10347644/testing.txt)

clutch2sft, 2023-01-04:
Struggled to attached large pcaps for reference.  

Also something I missed in my notes above - Having a LAN port, such as LAN1 connected causes more dramatic impact on performance.  Even if LAN1 isn't passing traffic.

Caputers and notes:

[https://github.com/clutch2sft/pcap.git](url)

clutch2sft, 2023-01-04:
Struggled to attached large pcaps for reference.  

Also something I missed in my notes above - Having a LAN port, such as LAN1 connected causes more dramatic impact on performance.  Even if LAN1 isn't passing traffic.

[image]
[image]
[image]

ThiloteE, 2023-01-05:
I have not looked at your wireshark data, but please have a look at https://forum.openwrt.org/t/best-practices-and-tools-for-measuring-wifi-performance/146923/13?u=thilotee

> `iperf.exe -c 172.16.5.1 -i 1`
> `iperf.exe -c 10.0.0.186 -i 1 -p 5002`

both commands capture uplink traffic from your client towards the router, instead of traffic that is sent from router to your client. My interpretation therefore would be, that this more likely seems to be a problem with the client or distance/hindrances inbetween the devices.

Recommendations:

1. Check out [iperf2](https://sourceforge.net/projects/iperf2/) or [iperf3](https://iperf.fr/). Iperf is old.
2. You do not want to run the iperf server on the router, but on one of the clients connected to it (iperf needs cpu power).
3. Try to do some tests with a or multiple different clients. Your client [dell wireless 1705](https://wikidevi.wi-cat.ru/Dell_Wireless_1705_(DW1705)) is a very old device from 2012, so drivers/firmware might be outdated and compatibility might be low, although that would be a pity, after all 802.11n is a global standard.
4. Most importantly: Run iperf with the `-R` command. (--reverse = run in reverse mode (server sends, client receives)

clutch2sft, 2023-01-06:
@ThiloteE 

Thank you for taking an interest.  Indeed the issue with the traffic is from the AP to the client.

Yes I had some confusion with the iperf stuff initially and appreciate the link.  I will review that for sure.

At the same time, the exact same test setup with the exact same client and AP but with the Belkin Firmware instead of the OpenWrt firmware gives expected results.  So IMHO there is something wrong in the way that the Openwrt Firmware is handling the phy (wifi) medium.

I did another full round of tests today to verify my thought process around this and the data is just clear in my head that the issue is not the client.  This issue I first first noticed this with an ACER laptop and then confirmed it with a DELL laptop so also two different WiFi clients (both windows) suffering the exact same issue.

I couldn't get my wireshark files uploaded as they are too large.  I started to learn to use LFS but ran out of time.  I hope to resolve that soon.

But I would be happy to share them directly to someone who has more experience looking at them than I.

My test setup for today was:

WiFi Client <---> AP/Router <----> switch <----> Provider Modem/Router <------>My Cloud DC Iperf
                                     |                         |
                         Iperf in Router             |
                                                        Iperf Server

I will edit this when I consolidate some results.

ThiloteE, 2023-01-07:
I like your network diagramm, but where exactly is your iperf client? Where do you start your measurements from? At "Wifi client"?
Where is "172.16.5.1"?

clutch2sft, 2023-01-07:
> I like your network diagramm, but where exactly is your iperf client? Where do you start your measurements from? From "Wifi client"? Where is "172.16.5.1"?

@ThiloteE - I debated that in the drawing.  Here is why.  I found this issue when using the router in my "production" and in that case I do use the 172.16 network for all things as I find less conflict with providers private network choice.  I also use a private image (based on OpenWrt respository not my own) in that scenerio.  However, to make the test more suitable for this post *MOST* of what I reported is from a device that was "factory reset" and loaded with a standard OpenWrt image.  Ergo that 172.16 network went away.  All test data I post here today will follow that methodology and I will no longer post anything relating to the "production" setup I use.

Yes - all testing will be initiated from the WiFi Client and the AP/Router will always be 192.168.1.1 going forward unless stated otherwise.  It took a lot of work but I have this standard test setup now and cleaned the RF up.  IE: I have a spectrum analyser showing zero anything running in the 2.4 Channel  1 band.  Ch6 and Ch11 have some noise from loud neighbors but they are in the dirt (-90ish or worse).  So any 40Mhz data I post going forward will be with a base starting at Chan 1.  All packet captures (if I can figure out how to share them) are coming from another Belkin in monitor mode.

clutch2sft, 2023-01-09:
I'm closing this because I no longer know how to isolate the problem nor prove the problem is with the AP and not the client.  In the last two days I have done some additional and exhaustive testing and the following truths have bubbled up.

1) With Belkin firmware my legacy 2.4Ghz 802.11n clients never showed less than an average of 46Mbps DL/UL to my iperf server in the cloud.  No matter how I positioned the laptops under test this held true....over hundreds of test runs.

2) With every version of OpenWrt that I tested (22.03.2, 22.03.3, Snapshot Latest) I could get similar results to the Belkin firmware *sometimes* ....but mostly I saw in the 30Mpbs range and if the positioning was just right it would drop below 5Mbps/sec. 

This is something I'd be willing to invest more time in if anyone out there with the skills to debug the RF side wants data to help them solve the mystery.  The test setup work is done and I won't tear it down because I'll wan to test other clients anyway.  But as for now I have run out of talent on my end to get to a resolution.

closed 2023-01-09 by clutch2sft

ThiloteE, 2023-01-09:
I think your results already proof one thing: Original Belkin software seems to be more performant than OpenWRT on your device.

Why? - We don't know.

Could be:

- configuration
- driver/firmware

If you want to debug this further, with the results we have, I would suggest to focus on ONE setup with fixed hardware, then try various configurations with that hardware. You provide proof that your local wifi has problems (https://github.com/openwrt/openwrt/issues/11695#issue-1519442569), so limit the testing setup to your local network. Keep internet  (testing on the cloud) out of this, as you cannot control what your internet provider will do with your traffic.

e.g. do this: wifi client (iperf client) <--wifi 802.11n --> AP/Router <-- lan --> Computer (Iperf server).

then provide various measurements for this simple setup:

e.g.

```
iperf3 -c 192.168.178.39 -p 5201 -t 60 -i 10
iperf3 -c 192.168.178.39 -p 5201 -P 4 -t 60 -i 10
iperf3 -c 192.168.178.39 -p 5201 -P 8 -t 60 -i 10
iperf3 -c 192.168.178.39 -p 5201 -t 60 -i 10 --bidir
iperf3 -c 192.168.178.39 -p 5201 -P 4 -t 60 -i 10 --bidir
iperf3 -c 192.168.178.39 -p 5201 -P 8 -t 60 -i 10 --bidir
iperf3 -c 192.168.178.39 -p 5201 -t 60 -i 10 -R
iperf3 -c 192.168.178.39 -p 5201 -P 4 -t 60 -i 10 -R
iperf3 -c 192.168.178.39 -p 5201 -P 8 -t 60 -i 10 -R
iperf3 -c 192.168.178.39 -p 5201 -t 20 -i 2 -u -b 1000M
iperf3 -c 192.168.178.39 -p 5201 -t 20 -i 2 -u -b 1000M -R 
```
replace 192.168.178.39 with your iperf servers ip adress.

In general, as a feedback, you did many tests, but I was having a hard time following you, because in one measurement you run iperf on the router, then you run it on the cloud and you also talk about results from other measurements that you have not shared here, so it is confusing. I don't think you have to close it. We have no solution, but you seem to experience problems, so why not keep it open for now?

ThiloteE, 2023-01-09:
Are you sure original Belkin OEM uses 802.11n? Is it not a wifi 6 device? maybe they automatically set __ax__ in the 2.4 GHz band?

clutch2sft, 2023-01-10:
> By the way, are you sure original Belkin OEM uses 802.11n? Is it not a wifi 6 device? maybe they automatically set **ax** in the 2.4 GHz band?

The chipset won't do ax in that band.  The client won't do ax.  

I agree with you that my testing was a bit of a mess.  Also I think that this is a place to report bugs and not a forum for discussion and my thread was looking more like the former than the latter.  So ... I closed it.  I may get smarter and find the problem and open a real bug.  Thanks for your help!