openwrt/openwrt #11697: mt7613be mt7663 driver

bricco1981, 2023-01-05:
### Describe the bug

this radio is really unusable under mt76, please fix it, no need to share any log, it is just a mess

### OpenWrt version

any

### OpenWrt target/subtarget

mt7621

### Device

any

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

rodsmar, 2023-01-25:
I'm using MT7613 in my Archer C6 V3, it's very stable with latest mt76 driver.

Tip: I helped solve a signal range issue by fixing the EEPROM size in around past November. @nbd168 merged this.

bricco1981, 2023-02-15:
are you sure is very stable?

bricco1981, 2023-02-18:
ok, thanks, but leave that alone

rodsmar, 2023-02-21:
Yes, works pretty for me. I’m reach 400 ~ 500Mbps in Wi-Fi speedtest. Otherwise, the signal covers my whole house (about 35m2). Latency about 1 ~ 3ms (without charge). I use an Archer C6 V3 in AP more only.

bricco1981, 2023-02-22:
are you on kernel 5.15? or 5.10?

rodsmar, 2023-02-22:
I’ve tested in both kernel versions with open source driver. I Will test 5.15 with mt7603e proprietary driver.

tsipa, 2023-04-17:
+1 from me.

I think it's about problem with mt7613be doesn't support DFS, it just craps itself when trying to do a channel availability check

More info is here
https://openwrt.org/unsupported/dfs
and here https://forum.openwrt.org/t/5-ghz-wlan-craps-out-upon-radar-detection-start-dfs-cac-failed/71018/7

Without DFS majority of regions are just too noisy and it's useless unless you set your region to Panama which is fairly loose in term of regulations or manually patch regdb.

The topic starter with "no need to share any log" is not reasonable and helpful, although the comment "it works for me" without details e.g. version, VHT, freq, power, region is useless as well.

ravichaudhary133, 2024-01-01:
@rodsmar Hello dev, could you help me fix this issue for my device tp link mr600v2. It uses mt7613be and 7603 drivers. I think it has the same eeprom problem.

chirantan300, 2024-10-04:
> Yes, works pretty for me. I’m reach 400 ~ 500Mbps in Wi-Fi speedtest. The 5ghz Wi-Fi signal covers my whole house (about 35m2). Latency about 1 ~ 3ms (without charge). I use an Archer C6 V3 in AP mode only.
> 
> For me, the issue is on 2.4Ghz (MT7603e), which is not supported well in mt76 driver. In this case, the latency sometimes goes to 100ms+ in my IoT devices and takes a long time to echo my requests. When i compile the mt7603 proprietary driver, then issues not occurs with me.
> 
> Since mt7615 (where has MediaTek developers contribution) the mt76 driver works flawless in most devices with MediaTek Wireless chips (mt7613+).
Do you compile the standalone drivers or compile with the openwrt image, and can you please link me to the proprietary drivers that you used