openwrt/openwrt #11701: Netgear DGN 3500(B) image 22.03.x bricks router

VikingGer, 2023-01-05:
There's a LzmaDcode issue with version 22 upwards with the Netgear DGN 3500(B). Flashing those bricks the router. The Problem was described here:
https://forum.openwrt.org/t/dgn3500-brick-after-22-03-update/137332/13
A warning in the wiki has been placed:
https://openwrt.org/toh/netgear/dgn3500b

This problem is likely due to a too small kernel partition. I'm not too deep in understanding all that firmware-stuff in detail but willing to learn and offer to try newly build images for testing.
I suggest to enlarge that partition in the image and shrink the rootfs accordingly. I guess that should be possible with a 16MB flash.

Here some (cropped) bootlogs:
ROM VER: 1.1.3
CFG 05

ROM VER: 1.1.3
CFG 05

U-Boot 1.1.5-2.4 (Oct  4 2010 - 11:56:12)

**checkboard **
Boot from SPI
AR9 BOARD
CLOCK CPU 333M RAM 166M
DRAM:  64 MB
spi_init_f finsihed
         SFLASH_BAUDRATE = 30000000
         CONFIG_SPI_FLASH_SLOW_READ yes

 relocate_code start
 relocate_code finish.
Flash: 16 MB
  bi_flashstart=00000000
  bi_flashsize=01000000
  bi_flashoffset=00000000
spi_init_f finsihed
         SFLASH_BAUDRATE = 30000000
         CONFIG_SPI_FLASH_SLOW_READ yes
In:    serial
Out:   serial
Err:   serial
Net:   
*** amazon_s_sw_chip_init

switch chip id=0000ffff
init 8366 rb
 -The DATA=0x0
The DATA=0x1C
The DATA=0xC961
The DATA=0x1140
smi_init = 0
initChip verid = 0x3
rtl8366rb_initChip= 0
The 0x0=0x830    0
The 0x509=0x5937    0
rtl8368s_setAsicMacForceLink = 0   
rtl8366rb_setLedMode(2, null)=0
amazon_s Switch
board_init_r:
         PID=YP5724
         ethaddr=e0:91:f5:4d:35:5d
         serialnumber=23330CB9000D7
         pin=15408372

Type "run flash_nfs" to mount root filesystem over NFS

Hit any key to stop autoboot:  0 
AMAZON_S # 

Stock:
sercomm_trap: done
check_checksum start

[... 334 lines cut ...]
image=27, image+1=5, image+2=19

image-2=4D, image-1=6D, image=42

PID_Offset=0x639A33, checksum_in_file=0xBF9B, checksum_calculat=0xBF9B, checksum_data_string=BF9B
## Application terminated, rc = 0x0
## Booting image at 00050000 ...
*** eeprom_read
*** eeprom_read

addr = 50000
CFG_LOAD_ADDR = 80100000
.....................................addr = 80500000

   Image Name:   MIPS OpenWrt Linux-5.10.146
   Created:      2022-10-14  22:44:41 UTC
   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Data Size:    2486141 Bytes =  2.4 MB
   Load Address: 80002000
   Entry Point:  80002000
   Verifying Checksum ... OK
   Uncompressing Kernel Image ... ERROR: LzmaDecode.c, 561

Decoding error = 1
LZMA ERROR 1 - must RESET board to recover

DragonBluep, 2023-01-07:
A well-known problem. This patch should help.
```
diff --git a/target/linux/lantiq/image/Makefile b/target/linux/lantiq/image/Makefile
index facc3fbe09..9fdb949392 100644
--- a/target/linux/lantiq/image/Makefile
+++ b/target/linux/lantiq/image/Makefile
@@ -49,22 +49,31 @@ define Build/fullimage
 	rm $@.tmp
 endef
 
-define Build/loader-okli-compile
+define Build/loader-common
 	rm -rf $@.src
 	$(MAKE) -C lzma-loader \
 		PKG_BUILD_DIR="$@.src" \
 		TARGET_DIR="$(dir $@)" LOADER_NAME="$(notdir $@)" \
-		PLATFORM="lantiq" \
+		BOARD="$(BOARDNAME)" PLATFORM="lantiq" \
 		LZMA_TEXT_START=0x82000000 \
 		LOADADDR=0x80002000 \
-		FLASH_START=0x10000000 \
-		FLASH_OFFS=$(LOADER_FLASH_OFFS) \
-		FLASH_MAX=0x0 \
 		$(1) compile loader.bin
 	mv "$@.bin" "$@"
 	rm -rf $@.src
 endef
 
+define Build/loader-kernel
+	$(call Build/loader-common,LOADER_DATA="$@")
+endef
+
+define Build/loader-okli-compile
+	$(call Build/loader-common, \
+		FLASH_START=0x10000000 \
+		FLASH_OFFS=$(LOADER_FLASH_OFFS) \
+		FLASH_MAX=0x0 \
+	)
+endef
+
 define Build/prepend-loader-okli
 	cat "$(KDIR)/loader-$(word 1,$(1)).bin" "$@" >> "$@.new"
 	mv "$@.new" "$@"
diff --git a/target/linux/lantiq/image/ar9.mk b/target/linux/lantiq/image/ar9.mk
index 434dacb121..1921ddb5a8 100644
--- a/target/linux/lantiq/image/ar9.mk
+++ b/target/linux/lantiq/image/ar9.mk
@@ -119,6 +119,9 @@ define Device/netgear_dgn3500b
   DEVICE_MODEL := DGN3500B
   SOC := ar9
   IMAGE_SIZE := 16000k
+  KERNEL := kernel-bin | append-dtb | lzma | loader-kernel | uImage none
+#   not sure if this is necessary
+  KERNEL_INITRAMFS := $$(KERNEL)
   IMAGES += factory.img
   IMAGE/sysupgrade.bin := \
 	append-kernel | append-rootfs | dgn3500-sercom-footer 0x0 "DE" | \

```

VikingGer, 2023-01-08:
> A well-known problem. This patch should help.

Thank you, Sir. You made my day! The fix works and pull requests have been placed.

commit 4d9c38d6 (2023-04-02): lantiq: fix lzma-loader for Netgear DGN 3500(B)

Fixes Uncompressing Kernel Image ... ERROR: LzmaDecode.c, 561

Fixes: #11701 for both 3500(B)
commit 50d707e3 (2023-04-02): lantiq: fix lzma-loader for Netgear DGN 3500(B)

Fixes Uncompressing Kernel Image ... ERROR: LzmaDecode.c, 561

Fixes: #11701 for both 3500(B)

(cherry picked from commit 4d9c38d654fdfb72ef641573bbf5b0f73f638f2e)
closed 2023-01-08 by VikingGer