openwrt/openwrt #11702: xiaomi-4a-gigabit-edition has a new flash which is GigaDevice GD25Q128@44Mhz cause a endless reboot

yaotthaha, 2023-01-05:
### Describe the bug

[image]

### OpenWrt version

22.03

### OpenWrt target/subtarget

Null

### Device

Xiaomi Mi 4A Gigabit Edition

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

DragonBluep, 2023-01-05:
It is not a bug and has nothing to do with OpenWrt. GD25Q128 has been fully supported in the 5.10 kernel. Please try the stock u-boot and follow the wiki installation guide..
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/drivers/mtd/spi-nor/gigadevice.c?h=v5.10.162

closed 2023-01-06 by yaotthaha

wangz-code, 2023-11-06:
老哥  我的4A内存型号和你的一样, 刷完也是无限重启, 你后面是怎么解决的 ? 
Brother, my 4A memory model is the same as yours. After flashing, it restarts indefinitely. How did you solve it later?

@0xffffharry

0xffffharry, 2023-11-06:
> 老哥 我的4A内存型号和你的一样, 刷完也是无限重启, 你后面是怎么解决的 ? Brother, my 4A memory model is the same as yours. After flashing, it restarts indefinitely. How did you solve it later?
> 
> @0xffffharry

如果是刷了breed，可以试试这个，我就是按照这个刷入解决的：https://www.right.com.cn/forum/forum.php?mod=viewthread&tid=8298740

wangz-code, 2023-11-06:
> > 老哥 我的4A内存型号和你的一样, 刷完也是无限重启, 你后面是怎么解决的 ? Brother, my 4A memory model is the same as yours. After flashing, it restarts indefinitely. How did you solve it later?
> > @0xffffharry
> 
> 如果是刷了breed，可以试试这个，我就是按照这个刷入解决的：https://www.right.com.cn/forum/forum.php?mod=viewthread&tid=8298740

谢谢  我看了这个文章,  尝试了下, 发现我现在好像什么都进不去了,  路由器没任何反应  ping也没任何回应
 
原因是先刷了breed  在breed刷op后无限重启,   谷歌后 找到尝试在breed修改变量,  但是我心急了只修改变量忘记保存字段了 囧囧, 现在就什么都没反应,  官方的救砖工具也识别不了, 估计只能上编程器了,  拼夕夕刚刚下单了个土豪金的  🤕  🤕  🤕