openwrt/openwrt #11706: opkg upgrade of firewall4 did not seem to fix gotcha?

systemcrash, 2023-01-05:
### Describe the bug

On `22.03.2`, after upgrading `firewall4` to version `2022-10-18-7ae5e14b-1` as [recommend here](https://openwrt.org/releases/22.03/notes-22.03.2#known_issues), and using `ipsets`, I restart the firewall with an ipset lacking a name in `/etc/config/firewall` and I get

```
# /etc/init.d/firewall restart
Reference error: left-hand side expression is null
In [anonymous function](), file /usr/share/ucode/fw4.uc, line 2308, byte 72:
  called from function [arrow function] (/usr/share/ucode/fw4.uc:703:65)
  called from function foreach ([C])
  called from function [anonymous function] (/usr/share/ucode/fw4.uc:703:66)
  called from function render_ruleset (/usr/share/firewall4/main.uc:56:24)
  called from anonymous function (/usr/share/firewall4/main.uc:143:29)

 `            ipset = filter(this.state.ipsets, s => (s.name == rule.ipset.name))[0];`
  Near here ----------------------------------------------------------------------^

The rendered ruleset contains errors, not doing firewall restart.
```

I checked `/usr/share/ucode/fw4.uc` and line 2308 does not have the fix [suggested here](https://git.openwrt.org/?p=project/firewall4.git;a=blobdiff;f=root/usr/share/ucode/fw4.uc;h=5dce90dd09c2657f8fcc6dda0dfddb4f4ad13f91;hp=47e86cd7dd2a0f87caeccde51710330199905fd3;hb=700a925fd9c1f1ff404e6b125cd5347ad7c45668;hpb=6443ec7805295de07f6051662065a16b4a194f19). 

Changing line 2308 as at this commit fixes the problem. (add `?.`)

### OpenWrt version

22.03.2

### OpenWrt target/subtarget

ath79

### Device

archer_c7

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

systemcrash, 2023-01-06:
FAO @jow-

hnyman, 2023-01-06:
> upgrading firewall4 to version 2022-10-18-7ae5e14b-1
> ...
> does not have the fix [suggested here](https://git.openwrt.org/?p=project/firewall4.git;a=blobdiff;f=root/usr/share/ucode/fw4.uc;h=5dce90dd09c2657f8fcc6dda0dfddb4f4ad13f91;hp=47e86cd7dd2a0f87caeccde51710330199905fd3;hb=700a925fd9c1f1ff404e6b125cd5347ad7c45668;hpb=6443ec7805295de07f6051662065a16b4a194f19).

Out of curiosity, why would you expect a firewall version from October to include change authored in November?
```
author	Jo-Philipp Wich <jo@mein.io>	
	Tue, 29 Nov 2022 12:44:15 +0200 (11:44 +0100)
```

Note that 22.03 branch and master each have their own version of the firewall. 
Apparently the 22.03 firewall4 Makefile has not yet been updated to use the newest commit. 

Ps. 22.03.3 was tagged a few days ago and will be missing this fix.

systemcrash, 2023-01-07:
> Ps. 22.03.3 was tagged a few days ago and will be missing this fix.

That's....  bad 

> Out of curiosity, why would you expect a firewall version from October to include change authored in November?

Good point. Altho I thought it was put in the repo around December time...? Not sure how it works, or whether it's coincidence what I read on the openwrt website.

closed 2023-01-12 by systemcrash