openwrt/openwrt #11713: Wrong IPv6 address selection.

naruto522ru, 2023-01-06:
### Describe the bug

The problem is that the ping,traceroute,opkg (update wget return code 4 error) system **sometimes selects** an IPv6 address. Although my ISP does not provide it to me.
The most interesting thing is that if the domain does not have AAAA-record. For example ping will write bad address.

**P.S** Excuse me for the translator, but English is not my native language and I cannot write it fluently.

### OpenWrt version

r20028-43d71ad93e

### OpenWrt target/subtarget

ramips/mt7620

### Device

Asus RT-AC51U

### Image kind

Official downloaded image

### Steps to reproduce

[image]
[image]
[image]
[image]

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

jow-, 2023-01-07:
Please provide the output of `ip -6 route show table all` as well as `ifstatus wan6; ifstatus wan_6; ifstatus lan`

naruto522ru, 2023-01-07:
> Please provide the output of `ip -6 route show table all` as well as `ifstatus wan6; ifstatus wan_6; ifstatus lan`

[image]
[image]

But the first two commands did not. I have to put my interface name, but I don't know what my interface name is.

P.S Sorry, but I am new to OpenWRT specifically. That's why I don't know a lot of things.

naruto522ru, 2023-01-07:
There is an important observation. I also noticed especially it is visible from the phone from the browser. That clients are also given an IPv6 address.
As I have established first by the above-mentioned feature.
And now I decided to connect the second router (by wire) to the above mentioned.
That is, the connection scheme is as follows
_WAN->Router1 (192.168.0.1)->Router2(192.168.10.1)-PC\Phone e.g_
Connecting to the second router via ssh
I enter
`nslookup google.com 192.168.0.1`
[image]

naruto522ru, 2024-07-01:
Fixed in OpenWrt 23.05.3 r23809-234f1a2efa

closed 2024-07-01 by naruto522ru