openwrt/openwrt #11725: Disabling wifi in LuCI fails in most cases

Djfe, 2023-01-08:
### Describe the bug

I have issues disabling my WiFi radio using LuCI:
```
Sun Jan  8 03:02:52 2023 kern.info kernel: [  397.354591] device phy0-ap0 left promiscuous mode
Sun Jan  8 03:02:52 2023 kern.info kernel: [  397.359704] br-lan: port 4(phy0-ap0) entered disabled state
Sun Jan  8 03:02:53 2023 daemon.notice netifd: radio0 (4666): WARNING: Variable 'data' does not exist or is not an array/object
Sun Jan  8 03:02:53 2023 daemon.notice netifd: radio0 (4666): Bug: PHY is undefined for device 'radio0'
Sun Jan  8 03:02:53 2023 daemon.notice netifd: Wireless device 'radio0' is now down
```

This appears to affect other devices as well. Maybe it's limited to the radios of the MT76 driver though.
I'm planning on doing some bisection to find the version or maybe even the commit that introduced this bug. In-case it even is a regression.

### OpenWrt version

r21727-3c31f6b521

### OpenWrt target/subtarget

ramips/mt7621

### Device

ASUS RT-AX53U

### Image kind

Self-built image

### Steps to reproduce

Boot
Use LuCI to config WiFi and then enable 2.4GHz radio.
Disable said radio.

### Actual behaviour

Radio stays online, I can still see the SSID but connecting to the WiFi fails.
WiFi LED also keeps blinking (indicating that WiFi is in fact still turned on).

### Expected behaviour

The radio should go offline instead of broadcasting the SSID.

### Additional info

This bug specifically doesn't affect `wifi down`, that command works flawlessly everytime.

If I boot with WiFi already enabled, then disabling WiFi using LuCI works for the first time. The LEDs turn off etc.
Turning it on and then off again the second and all further attempts fail. You can see the error log in the bug description above.

**A successful attempt looks like this:**
```
Wed Jan  4 03:21:20 2023 kern.info kernel: [12797.346347] device phy0-ap0 left promiscuous mode
Wed Jan  4 03:21:20 2023 kern.info kernel: [12797.351418] br-lan: port 4(phy0-ap0) entered disabled state
Wed Jan  4 03:21:20 2023 daemon.notice hostapd: Remove interface 'phy0-ap0'
Wed Jan  4 03:21:20 2023 daemon.notice hostapd: phy0-ap0: interface state ENABLED->DISABLED
Wed Jan  4 03:21:20 2023 daemon.notice hostapd: phy0-ap0: AP-DISABLED
Wed Jan  4 03:21:20 2023 daemon.notice hostapd: phy0-ap0: CTRL-EVENT-TERMINATING
Wed Jan  4 03:21:20 2023 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Wed Jan  4 03:21:20 2023 daemon.notice hostapd: nl80211: deinit ifname=phy0-ap0 disabled_11b_rates=0
Wed Jan  4 03:21:20 2023 daemon.notice hostapd: nl80211: Failed to remove interface phy0-ap0 from bridge br-lan: Invalid argument
Wed Jan  4 03:21:20 2023 daemon.notice netifd: Wireless device 'radio0' is now down
```

I'm wondering whether the following two lines from a successful attempt play a role in this. The errors are obviously not intentional.
```
Wed Jan  4 03:21:20 2023 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Wed Jan  4 03:21:20 2023 daemon.notice hostapd: nl80211: Failed to remove interface phy0-ap0 from bridge br-lan: Invalid argument
```

Related Issues:
https://github.com/openwrt/openwrt/issues/11068#issuecomment-1371156201
https://github.com/openwrt/openwrt/pull/11220#issuecomment-1336263591
https://github.com/openwrt/openwrt/issues/7485 (peculiar: in my case LEDs always light up on radio activity unlike for the reporter, for me it's just that the led/radio stays alive when it should be offline)

Related Code (that introduced the error message and might tell us where to look):
https://github.com/openwrt/openwrt/pull/4314
https://github.com/openwrt/openwrt/commit/6603748e0ca697f484adfc16a0d49d35ed36954d (maybe?)

Feels like the 'data' variable wherever it is, wasn't initialized/set correctly and therefore the default 'radio0' is used which obviously doesn't exist. Though I have no clue which code files to look through 😅
What does LuCI do differently compared to `wifi down`?

### Diffconfig

```text
tmp/.config-package.in:56656:error: recursive dependency detected!
tmp/.config-package.in:56656:   symbol PACKAGE_luci-app-pbr depends on PACKAGE_luci-app-pbr
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

tmp/.config-package.in:56656:error: recursive dependency detected!
tmp/.config-package.in:56656:   symbol PACKAGE_luci-app-pbr depends on PACKAGE_luci-app-pbr
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

CONFIG_TARGET_ramips=y
CONFIG_TARGET_ramips_mt7621=y
CONFIG_TARGET_ramips_mt7621_DEVICE_asus_rt-ax53u=y
CONFIG_PACKAGE_attendedsysupgrade-common=y
CONFIG_PACKAGE_cgi-io=y
CONFIG_PACKAGE_liblucihttp=y
CONFIG_PACKAGE_liblucihttp-ucode=y
CONFIG_PACKAGE_luci=y
CONFIG_PACKAGE_luci-app-attendedsysupgrade=y
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-app-opkg=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-network=y
CONFIG_PACKAGE_luci-mod-status=y
CONFIG_PACKAGE_luci-mod-system=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-proto-ppp=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-rpcsys=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
# CONFIG_TARGET_ROOTFS_INITRAMFS is not set
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Djfe, 2023-01-08:
I almost thought this might be related
https://github.com/openwrt/mt76/issues/135#issuecomment-365038040

But it doesn't explain the errors in the logs and why it works once and why `wifi down` isn't affected.
Also there are no interrupts happening even though the SSID (advertisements) keep showing up on my phone.

Djfe, 2023-01-12:
ok so I tested `22.03.3` on a D-Link DIR-860L and disabling wifi over luci works just fine. Even repeatedly.
```
Thu Jan 12 07:20:37 2023 kern.info kernel: [  892.746245] device wlan0 left promiscuous mode
Thu Jan 12 07:20:37 2023 kern.info kernel: [  892.755737] br-lan: port 5(wlan0) entered disabled state
Thu Jan 12 07:20:37 2023 daemon.notice hostapd: Remove interface 'wlan0'
Thu Jan 12 07:20:37 2023 daemon.notice hostapd: wlan0: interface state ENABLED->DISABLED
Thu Jan 12 07:20:37 2023 daemon.notice hostapd: wlan0: AP-DISABLED
Thu Jan 12 07:20:37 2023 daemon.notice hostapd: wlan0: CTRL-EVENT-TERMINATING
Thu Jan 12 07:20:37 2023 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Thu Jan 12 07:20:37 2023 daemon.notice hostapd: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Thu Jan 12 07:20:37 2023 daemon.notice hostapd: nl80211: Failed to remove interface wlan0 from bridge br-lan: Invalid argument
Thu Jan 12 07:20:38 2023 daemon.notice netifd: Wireless device 'radio0' is now down
Thu Jan 12 07:20:40 2023 daemon.info dnsmasq[1]: read /etc/hosts - 4 addresses
Thu Jan 12 07:20:40 2023 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Thu Jan 12 07:20:40 2023 daemon.info dnsmasq[1]: read /tmp/hosts/odhcpd - 1 addresses
Thu Jan 12 07:20:40 2023 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Thu Jan 12 07:20:41 2023 daemon.warn odhcpd[1838]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
Thu Jan 12 07:20:46 2023 kern.info kernel: [  901.905476] device wlan1 left promiscuous mode
Thu Jan 12 07:20:46 2023 kern.info kernel: [  901.914692] br-lan: port 6(wlan1) entered disabled state
Thu Jan 12 07:20:46 2023 daemon.notice hostapd: Remove interface 'wlan1'
Thu Jan 12 07:20:46 2023 daemon.notice hostapd: wlan1: interface state ENABLED->DISABLED
Thu Jan 12 07:20:46 2023 daemon.notice hostapd: wlan1: AP-DISABLED
Thu Jan 12 07:20:46 2023 daemon.notice hostapd: wlan1: CTRL-EVENT-TERMINATING
Thu Jan 12 07:20:46 2023 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Thu Jan 12 07:20:46 2023 daemon.notice hostapd: nl80211: deinit ifname=wlan1 disabled_11b_rates=0
Thu Jan 12 07:20:46 2023 daemon.notice hostapd: nl80211: Failed to remove interface wlan1 from bridge br-lan: Invalid argument
Thu Jan 12 07:20:47 2023 daemon.notice netifd: Wireless device 'radio1' is now down
Thu Jan 12 07:20:49 2023 daemon.info dnsmasq[1]: read /etc/hosts - 4 addresses
Thu Jan 12 07:20:49 2023 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Thu Jan 12 07:20:49 2023 daemon.info dnsmasq[1]: read /tmp/hosts/odhcpd - 1 addresses
Thu Jan 12 07:20:49 2023 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Thu Jan 12 07:20:50 2023 daemon.warn odhcpd[1838]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
```

The difference between this and my asus router is (atleast), that the dlink is using a way older wifi chip from mediatek and that the interface is named different.
Here it is `wlan0` and `wlan1`.
While my asus uses `phy0-ap0` and `phy1-ap0`.

Djfe, 2023-01-12:
I also tested a Xiaomi Mi 4a Gigabit Edition 22.03.3 and also had no issues.
The WiFi interface were called wlan0 and wlan1 as well.

Next device to test will be a Netgear WAX202 I will get access to next week. It also has the same wifi chip. Hopefully it also has the same wifi interface name, so I can reproduce the bug :)

Djfe, 2023-01-23:
Ok so I just realized that this was introduced in master and probably was never a bug in release, yet, ups 😅
Better late than never.
Just tested on the WAX202 and the bug exists there as well.

You can even see my smartphone trying to connect to the wifi after disabling it via LuCI.

```
Mon Jan 23 03:55:25 2023 kern.info kernel: [  461.469528] device phy0-ap0 left promiscuous mode
Mon Jan 23 03:55:25 2023 kern.info kernel: [  461.474637] br-lan: port 4(phy0-ap0) entered disabled state
Mon Jan 23 03:55:26 2023 daemon.notice netifd: radio0 (5837): WARNING: Variable 'data' does not exist or is not an array/object
Mon Jan 23 03:55:26 2023 daemon.notice netifd: radio0 (5837): Bug: PHY is undefined for device 'radio0'
Mon Jan 23 03:55:26 2023 daemon.notice netifd: Wireless device 'radio0' is now down
Mon Jan 23 03:55:28 2023 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Mon Jan 23 03:55:28 2023 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 4 names
Mon Jan 23 03:55:28 2023 daemon.info dnsmasq[1]: read /tmp/hosts/odhcpd - 2 names
Mon Jan 23 03:55:28 2023 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Mon Jan 23 03:55:29 2023 daemon.info hostapd: phy0-ap0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: authenticated
Mon Jan 23 03:55:29 2023 daemon.info hostapd: phy0-ap0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: associated (aid 1)
Mon Jan 23 03:55:29 2023 daemon.notice hostapd: phy0-ap0: AP-STA-CONNECTED xx:xx:xx:xx:xx:xx auth_alg=open
Mon Jan 23 03:55:29 2023 daemon.info hostapd: phy0-ap0: STA xx:xx:xx:xx:xx:xx WPA: pairwise key handshake completed (RSN)
Mon Jan 23 03:55:29 2023 daemon.notice hostapd: phy0-ap0: EAPOL-4WAY-HS-COMPLETED xx:xx:xx:xx:xx:xx
Mon Jan 23 03:55:29 2023 daemon.warn odhcpd[1755]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
Mon Jan 23 03:55:47 2023 daemon.notice hostapd: phy0-ap0: AP-STA-DISCONNECTED xx:xx:xx:xx:xx:xx
Mon Jan 23 03:55:47 2023 daemon.info hostapd: phy0-ap0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: disassociated
Mon Jan 23 03:55:48 2023 daemon.info hostapd: phy0-ap0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Mon Jan 23 03:55:50 2023 daemon.info hostapd: phy0-ap0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: authenticated
Mon Jan 23 03:55:50 2023 daemon.info hostapd: phy0-ap0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: associated (aid 1)
Mon Jan 23 03:55:50 2023 daemon.notice hostapd: phy0-ap0: AP-STA-CONNECTED xx:xx:xx:xx:xx:xx auth_alg=open
Mon Jan 23 03:55:50 2023 daemon.info hostapd: phy0-ap0: STA xx:xx:xx:xx:xx:xx WPA: pairwise key handshake completed (RSN)
Mon Jan 23 03:55:50 2023 daemon.notice hostapd: phy0-ap0: EAPOL-4WAY-HS-COMPLETED xx:xx:xx:xx:xx:xx
Mon Jan 23 03:56:08 2023 daemon.notice hostapd: phy0-ap0: AP-STA-DISCONNECTED xx:xx:xx:xx:xx:xx
Mon Jan 23 03:56:08 2023 daemon.info hostapd: phy0-ap0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: disassociated
Mon Jan 23 03:56:09 2023 daemon.info hostapd: phy0-ap0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Mon Jan 23 03:56:11 2023 daemon.info hostapd: phy0-ap0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: authenticated
Mon Jan 23 03:56:11 2023 daemon.info hostapd: phy0-ap0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: associated (aid 1)
Mon Jan 23 03:56:11 2023 daemon.notice hostapd: phy0-ap0: AP-STA-CONNECTED xx:xx:xx:xx:xx:xx auth_alg=open
Mon Jan 23 03:56:11 2023 daemon.info hostapd: phy0-ap0: STA xx:xx:xx:xx:xx:xx WPA: pairwise key handshake completed (RSN)
Mon Jan 23 03:56:11 2023 daemon.notice hostapd: phy0-ap0: EAPOL-4WAY-HS-COMPLETED xx:xx:xx:xx:xx:xx
Mon Jan 23 03:56:29 2023 daemon.notice hostapd: phy0-ap0: AP-STA-DISCONNECTED xx:xx:xx:xx:xx:xx
Mon Jan 23 03:56:29 2023 daemon.info hostapd: phy0-ap0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: disassociated
Mon Jan 23 03:56:30 2023 daemon.info hostapd: phy0-ap0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)

Mon Jan 23 03:57:19 2023 kern.info kernel: [  575.549791] device phy1-ap0 left promiscuous mode
Mon Jan 23 03:57:19 2023 kern.info kernel: [  575.554885] br-lan: port 4(phy1-ap0) entered disabled state
Mon Jan 23 03:57:20 2023 daemon.notice netifd: radio1 (6550): WARNING: Variable 'data' does not exist or is not an array/object
Mon Jan 23 03:57:20 2023 daemon.notice netifd: radio1 (6550): Bug: PHY is undefined for device 'radio1'
Mon Jan 23 03:57:20 2023 daemon.notice netifd: Wireless device 'radio1' is now down
Mon Jan 23 03:57:22 2023 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Mon Jan 23 03:57:22 2023 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 4 names
Mon Jan 23 03:57:22 2023 daemon.info dnsmasq[1]: read /tmp/hosts/odhcpd - 2 names
Mon Jan 23 03:57:22 2023 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Mon Jan 23 03:57:24 2023 daemon.warn odhcpd[1755]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
Mon Jan 23 03:57:49 2023 daemon.notice hostapd: phy1-ap0: AP-STA-DISCONNECTED xx:xx:xx:xx:xx:xx
Mon Jan 23 03:57:49 2023 daemon.info hostapd: phy1-ap0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: disassociated
Mon Jan 23 03:57:51 2023 daemon.info hostapd: phy1-ap0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Mon Jan 23 03:57:52 2023 daemon.info hostapd: phy1-ap0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: authenticated
Mon Jan 23 03:57:52 2023 daemon.info hostapd: phy1-ap0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: associated (aid 1)
Mon Jan 23 03:57:52 2023 daemon.notice hostapd: phy1-ap0: AP-STA-CONNECTED xx:xx:xx:xx:xx:xx auth_alg=open
Mon Jan 23 03:57:52 2023 daemon.info hostapd: phy1-ap0: STA xx:xx:xx:xx:xx:xx WPA: pairwise key handshake completed (RSN)
Mon Jan 23 03:57:52 2023 daemon.notice hostapd: phy1-ap0: EAPOL-4WAY-HS-COMPLETED xx:xx:xx:xx:xx:xx
```

Djfe, 2023-01-25:
### tl;dr
@nbd168 Hi,
sorry to interrupt. Could you maybe help out in fixing this, please? :)

Summary:
One of your commits back in October appears to have introduced a small bug into master that prevents me from disabling WiFi in LuCI on devices with the WiFi Chipset MT7905. Release 22 is not affected, yet.

How this error manifests in logread:
```
Mon Jan 23 03:55:26 2023 daemon.notice netifd: radio0 (5837): WARNING: Variable 'data' does not exist or is not an array/object
Mon Jan 23 03:55:26 2023 daemon.notice netifd: radio0 (5837): Bug: PHY is undefined for device 'radio0'
```

___________________________

I spend a few hours on git bisect and compiling to find the commit that introduced the regression: 4d323303e7e5743f541e3b41dfb2ac1627e8d96d
```
git bisect good
4d323303e7e5743f541e3b41dfb2ac1627e8d96d is the first bad commit
commit 4d323303e7e5743f541e3b41dfb2ac1627e8d96d
Author: Felix Fietkau <nbd@nbd.name>
Date:   Fri Sep 30 20:14:05 2022 +0200

    mac80211: rename phy according to board.json entries on bringup

    This allows phy names specified in board.json to be used directly instead of
    the path option

    Signed-off-by: Felix Fietkau <nbd@nbd.name>

 .../mac80211/files/lib/netifd/wireless/mac80211.sh | 68 +++++++++++++++++++++-
 1 file changed, 65 insertions(+), 3 deletions(-)
```

I would've expected the interface's name-change itself to be reason for the issue.
But I can still deactivate WiFi successfully in LuCI in 6603748e0ca697f484adfc16a0d49d35ed36954d (one commit earlier):
```
Fri Oct 14 11:15:19 2022 kern.info kernel: [  175.906628] device phy0-ap0 left promiscuous mode
Fri Oct 14 11:15:19 2022 kern.info kernel: [  175.911701] br-lan: port 4(phy0-ap0) entered disabled state
Fri Oct 14 11:15:19 2022 daemon.notice hostapd: Remove interface 'phy0-ap0'
Fri Oct 14 11:15:19 2022 daemon.notice hostapd: phy0-ap0: interface state ENABLED->DISABLED
Fri Oct 14 11:15:19 2022 daemon.notice hostapd: phy0-ap0: AP-DISABLED
Fri Oct 14 11:15:19 2022 daemon.notice hostapd: phy0-ap0: CTRL-EVENT-TERMINATING
Fri Oct 14 11:15:19 2022 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Fri Oct 14 11:15:19 2022 daemon.notice hostapd: nl80211: deinit ifname=phy0-ap0 disabled_11b_rates=0
Fri Oct 14 11:15:19 2022 daemon.notice hostapd: nl80211: Failed to remove interface phy0-ap0 from bridge br-lan: Invalid argument
Fri Oct 14 11:15:19 2022 daemon.notice netifd: Wireless device 'radio0' is now down
```

Thanks for taking a look.

Djfe, 2023-02-02:
Hello @hauke,
sorry to bother you with this directly, I feel like this bug should be fixed before https://github.com/openwrt/openwrt/commit/4d323303e7e5743f541e3b41dfb2ac1627e8d96d gets pushed into a Release branch. So I pinged you to bring this to the attention of the maintainers'.
I don't want there to be people not being able to disable WiFi in LuCI (without a reboot) for months.

Djfe, 2023-06-09:
Regression happening on 23.03-rc1
https://pastebin.com/fr9dEFjT

I can't disable WiFi via LuCI
affected devices include atleast
- mt7621 with WiFi chip mt7905 (WiFi6)

Ansuel, 2023-06-09:
@jow- any help with this? seems we have this from a long time now

jow-, 2023-06-09:
@Ansuel - don't have time or insight to help with this, sorry. What LuCI does upon clicking `Disable` is equivalent to the following:

```
uci set wireless.@wifi-iface[0].disabled=1
uci commit network
/sbin/reload_config
```

Ansuel, 2023-06-09:
@Djfe can you confirm that using the following command the bug is reproduced?

Djfe, 2023-06-09:
It still shows the error message in logread, but it disables the wifi chip successfully
```
Fri Jun  9 21:34:47 2023 kern.info kernel: [  351.779559] device phy0-ap0 left promiscuous mode
Fri Jun  9 21:34:47 2023 kern.info kernel: [  351.784564] br-lan: port 4(phy0-ap0) entered disabled state
Fri Jun  9 21:34:47 2023 daemon.notice netifd: radio0 (4483): WARNING: Variable 'data' does not exist or is not an array/object
Fri Jun  9 21:34:47 2023 daemon.notice netifd: radio0 (4483): Bug: PHY is undefined for device 'radio0'
Fri Jun  9 21:34:47 2023 daemon.notice netifd: Wireless device 'radio0' is now down
Fri Jun  9 21:34:48 2023 daemon.notice hostapd: Remove interface 'phy0-ap0'
Fri Jun  9 21:34:48 2023 daemon.notice hostapd: phy0-ap0: interface state ENABLED->DISABLED
Fri Jun  9 21:34:48 2023 daemon.notice hostapd: phy0-ap0: AP-DISABLED
Fri Jun  9 21:34:48 2023 daemon.notice hostapd: phy0-ap0: CTRL-EVENT-TERMINATING
Fri Jun  9 21:34:48 2023 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Fri Jun  9 21:34:48 2023 daemon.notice hostapd: nl80211: deinit ifname=phy0-ap0 disabled_11b_rates=0
Fri Jun  9 21:34:48 2023 daemon.notice hostapd: nl80211: Failed to remove interface phy0-ap0 from bridge br-lan: Invalid argument
Fri Jun  9 21:34:48 2023 daemon.notice netifd: Wireless device 'radio0' is now up
Fri Jun  9 21:34:50 2023 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Fri Jun  9 21:34:50 2023 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 4 names
Fri Jun  9 21:34:50 2023 daemon.info dnsmasq[1]: read /tmp/hosts/odhcpd - 2 names
Fri Jun  9 21:34:50 2023 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Fri Jun  9 21:34:51 2023 daemon.warn odhcpd[1577]: A default route is present but there is no public prefix on lan thus we don't announce a default route by overriding ra_lifetime!
```

interestingly enough: LuCI has issues detecting the chip (as disabled) after doing this
[image]
It still shows the disable button instead of enable

@nbd168 probably knows what is causing this, but I don't know whether he has the time to fix it.
and I don't think he looks at github pings, so we might need to contact him on irc or via mail.

jow-, 2023-06-09:
What's in your `/etc/board.json` ?

Djfe, 2023-06-09:
```
root@OpenWrt:~# cat /etc/board.json 
{
	"model": {
		"id": "asus,rt-ax53u",
		"name": "ASUS RT-AX53U"
	},
	"led": {
		"usb": {
			"name": "USB",
			"sysfs": "blue:usb",
			"type": "usbport",
			"ports": [
				"usb1-port2"
			]
		},
		"wlan2g": {
			"name": "WiFi 2.4GHz",
			"sysfs": "mt76-phy0",
			"trigger": "phy0tpt"
		},
		"wlan5g": {
			"name": "WiFi 5GHz",
			"sysfs": "mt76-phy1",
			"trigger": "phy1tpt"
		}
	},
	"network": {
		"lan": {
			"ports": [
				"lan1",
				"lan2",
				"lan3"
			],
			"protocol": "static"
		},
		"wan": {
			"device": "wan",
			"protocol": "dhcp"
		}
	},
	"system": {
		"compat_version": "1.1"
	}
}
```

Djfe, 2023-06-09:
disabling via ssh and your commands
```
root@OpenWrt:~# logread -f
Fri Jun  9 23:24:04 2023 kern.info kernel: [  368.465923] device phy0-ap0 left promiscuous mode
Fri Jun  9 23:24:04 2023 kern.info kernel: [  368.470874] br-lan: port 4(phy0-ap0) entered disabled state
Fri Jun  9 23:24:04 2023 daemon.notice netifd: radio0 (4958): WARNING: Variable 'data' does not exist or is not an array/object
Fri Jun  9 23:24:04 2023 daemon.notice netifd: radio0 (4958): Bug: PHY is undefined for device 'radio0'
Fri Jun  9 23:24:04 2023 daemon.notice netifd: Wireless device 'radio0' is now down
Fri Jun  9 23:24:05 2023 daemon.notice hostapd: Remove interface 'phy0-ap0'
Fri Jun  9 23:24:05 2023 daemon.notice hostapd: phy0-ap0: interface state ENABLED->DISABLED
Fri Jun  9 23:24:05 2023 daemon.notice hostapd: phy0-ap0: AP-DISABLED
Fri Jun  9 23:24:05 2023 daemon.notice hostapd: phy0-ap0: CTRL-EVENT-TERMINATING
Fri Jun  9 23:24:05 2023 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Fri Jun  9 23:24:05 2023 daemon.notice hostapd: nl80211: deinit ifname=phy0-ap0 disabled_11b_rates=0
Fri Jun  9 23:24:05 2023 daemon.notice hostapd: nl80211: Failed to remove interface phy0-ap0 from bridge br-lan: Invalid argument
Fri Jun  9 23:24:05 2023 daemon.notice netifd: Wireless device 'radio0' is now up
Fri Jun  9 23:24:07 2023 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Fri Jun  9 23:24:07 2023 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 4 names
Fri Jun  9 23:24:07 2023 daemon.info dnsmasq[1]: read /tmp/hosts/odhcpd - 2 names
Fri Jun  9 23:24:07 2023 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Fri Jun  9 23:24:08 2023 daemon.warn odhcpd[1859]: A default route is present but there is no public prefix on lan thus we don't announce a default route by overriding ra_lifetime!
```
enabling via ssh and your commands
```
root@OpenWrt:~# logread -f
Fri Jun  9 23:25:16 2023 daemon.notice netifd: radio0 (5393): WARNING: Variable 'data' does not exist or is not an array/object
Fri Jun  9 23:25:16 2023 daemon.notice netifd: radio0 (5393): Bug: PHY is undefined for device 'radio0'
Fri Jun  9 23:25:16 2023 daemon.notice netifd: Wireless device 'radio0' is now down
Fri Jun  9 23:25:17 2023 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy0.conf (phy phy0-ap0) --> new PHY
Fri Jun  9 23:25:17 2023 kern.info kernel: [  441.120836] br-lan: port 4(phy0-ap0) entered blocking state
Fri Jun  9 23:25:17 2023 kern.info kernel: [  441.126538] br-lan: port 4(phy0-ap0) entered disabled state
Fri Jun  9 23:25:17 2023 kern.info kernel: [  441.132708] device phy0-ap0 entered promiscuous mode
Fri Jun  9 23:25:17 2023 daemon.notice hostapd: phy0-ap0: interface state UNINITIALIZED->COUNTRY_UPDATE
Fri Jun  9 23:25:17 2023 kern.info kernel: [  441.270870] IPv6: ADDRCONF(NETDEV_CHANGE): phy0-ap0: link becomes ready
Fri Jun  9 23:25:17 2023 kern.info kernel: [  441.277953] br-lan: port 4(phy0-ap0) entered blocking state
Fri Jun  9 23:25:17 2023 kern.info kernel: [  441.283561] br-lan: port 4(phy0-ap0) entered forwarding state
Fri Jun  9 23:25:17 2023 daemon.notice hostapd: phy0-ap0: interface state COUNTRY_UPDATE->ENABLED
Fri Jun  9 23:25:17 2023 daemon.notice hostapd: phy0-ap0: AP-ENABLED
Fri Jun  9 23:25:18 2023 daemon.notice netifd: Wireless device 'radio0' is now up
Fri Jun  9 23:25:18 2023 daemon.notice netifd: Network device 'phy0-ap0' link is up
Fri Jun  9 23:25:18 2023 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Fri Jun  9 23:25:18 2023 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 4 names
Fri Jun  9 23:25:18 2023 daemon.info dnsmasq[1]: read /tmp/hosts/odhcpd - 2 names
Fri Jun  9 23:25:18 2023 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Fri Jun  9 23:25:20 2023 daemon.warn odhcpd[1859]: A default route is present but there is no public prefix on lan thus we don't announce a default route by overriding ra_lifetime!
```

> You said the bug only manifests after one Disable/Enable cycle in LuCI, so repeat the above enable/disable command sequences a few times, with some seconds wait time in between.

That's not exactly how I meant it.

scenario 1:
WIFi disabled on boot -> I go into LuCI and enable wifi, then I disable wifi > triggers the bug (wifi chip keeps broadcasting ssid)

scenario 2:
WiFi enabled on boot -> I go into LuCI and disable wifi > doesn't trigger the bug
I then enable wifi and disable wifi > triggers the bug

This behavior is consistent and happens every time.
I'm not sure whether it affects all targets though. x86 might not be affected

Regardless: I ran your commands 3 times in a row while waiting in between and the wifi chip was disabled successfully.

```
root@OpenWrt:~# cat /etc/config/wireless 

config wifi-device 'radio0'
	option type 'mac80211'
	option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0'
	option channel '1'
	option band '2g'
	option htmode 'HE20'
	option country 'DE'
	option cell_density '0'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option ssid 'OpenWrt'
	option encryption 'psk2+ccmp'
	option key 'xx'

config wifi-device 'radio1'
	option type 'mac80211'
	option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0+1'
	option channel '36'
	option band '5g'
	option htmode 'HE80'
	option country 'DE'
	option cell_density '0'

config wifi-iface 'default_radio1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option ssid 'OpenWrt'
	option encryption 'psk2+ccmp'
	option key 'xx'
```
(I replaced the password)

jow-, 2023-06-09:
A more directly translation of LuCI actions to cli commands:

Disabling:
```
ubus call uci delete '{ "config": "wireless", "section": "radio0", "option": "disabled" }'
ubus call uci set '{ "config": "wireless", "section": "default_radio0", "values": { "disabled": "1" } }'
ubus call uci commit '{ "config": "wireless" }'
ubus call uci apply '{ "rollback": false }'
```

Enabling:
```
ubus call uci delete '{ "config": "wireless", "section": "radio0", "option": "disabled" }'
ubus call uci delete '{ "config": "wireless", "section": "default_radio0", "option": "disabled" }'
ubus call uci commit '{ "config": "wireless" }'
ubus call uci apply '{ "rollback": "false" }'
```

Djfe, 2023-06-10:
Two things:
first
`uci del wireless.radio0.disabled` always returns `uci: Entry not found`
`ubus call uci delete '{ "config": "wireless", "section": "radio0", "option": "disabled" }'` always returns `Command failed: Not found`
My guess is this is expected behavior

second
`ubus call uci apply '{ "rollback": "false" }'` always returns `Command failed: Invalid argument`
which doesn't feel like you'd expect it.

Regardless of that, those commands still run and enable/disable wifi just fine

```
root@OpenWrt:~# logread -f
Fri Jun  9 23:58:17 2023 kern.info kernel: [ 2421.825119] device phy0-ap0 left promiscuous mode
Fri Jun  9 23:58:17 2023 kern.info kernel: [ 2421.830076] br-lan: port 4(phy0-ap0) entered disabled state
Fri Jun  9 23:58:18 2023 daemon.notice netifd: radio0 (10236): WARNING: Variable 'data' does not exist or is not an array/object
Fri Jun  9 23:58:18 2023 daemon.notice netifd: radio0 (10236): Bug: PHY is undefined for device 'radio0'
Fri Jun  9 23:58:18 2023 daemon.notice netifd: Wireless device 'radio0' is now down
Fri Jun  9 23:58:18 2023 daemon.notice hostapd: Remove interface 'phy0-ap0'
Fri Jun  9 23:58:18 2023 daemon.notice hostapd: phy0-ap0: interface state ENABLED->DISABLED
Fri Jun  9 23:58:18 2023 daemon.notice hostapd: phy0-ap0: AP-DISABLED
Fri Jun  9 23:58:18 2023 daemon.notice hostapd: phy0-ap0: CTRL-EVENT-TERMINATING
Fri Jun  9 23:58:18 2023 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Fri Jun  9 23:58:18 2023 daemon.notice hostapd: nl80211: deinit ifname=phy0-ap0 disabled_11b_rates=0
Fri Jun  9 23:58:18 2023 daemon.notice hostapd: nl80211: Failed to remove interface phy0-ap0 from bridge br-lan: Invalid argument
Fri Jun  9 23:58:18 2023 daemon.notice netifd: Wireless device 'radio0' is now up
Fri Jun  9 23:58:20 2023 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Fri Jun  9 23:58:20 2023 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 4 names
Fri Jun  9 23:58:20 2023 daemon.info dnsmasq[1]: read /tmp/hosts/odhcpd - 2 names
Fri Jun  9 23:58:20 2023 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Fri Jun  9 23:58:21 2023 daemon.warn odhcpd[1859]: A default route is present but there is no public prefix on lan thus we don't announce a default route by overriding ra_lifetime!
Fri Jun  9 23:59:26 2023 daemon.notice netifd: radio0 (10651): WARNING: Variable 'data' does not exist or is not an array/object
Fri Jun  9 23:59:26 2023 daemon.notice netifd: radio0 (10651): Bug: PHY is undefined for device 'radio0'
Fri Jun  9 23:59:26 2023 daemon.notice netifd: Wireless device 'radio0' is now down
Fri Jun  9 23:59:27 2023 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy0.conf (phy phy0-ap0) --> new PHY
Fri Jun  9 23:59:27 2023 kern.info kernel: [ 2491.068839] br-lan: port 4(phy0-ap0) entered blocking state
Fri Jun  9 23:59:27 2023 kern.info kernel: [ 2491.074477] br-lan: port 4(phy0-ap0) entered disabled state
Fri Jun  9 23:59:27 2023 kern.info kernel: [ 2491.080916] device phy0-ap0 entered promiscuous mode
Fri Jun  9 23:59:27 2023 daemon.notice hostapd: phy0-ap0: interface state UNINITIALIZED->COUNTRY_UPDATE
Fri Jun  9 23:59:27 2023 kern.info kernel: [ 2491.235886] IPv6: ADDRCONF(NETDEV_CHANGE): phy0-ap0: link becomes ready
Fri Jun  9 23:59:27 2023 kern.info kernel: [ 2491.243182] br-lan: port 4(phy0-ap0) entered blocking state
Fri Jun  9 23:59:27 2023 kern.info kernel: [ 2491.248950] br-lan: port 4(phy0-ap0) entered forwarding state
Fri Jun  9 23:59:27 2023 daemon.notice hostapd: phy0-ap0: interface state COUNTRY_UPDATE->ENABLED
Fri Jun  9 23:59:27 2023 daemon.notice hostapd: phy0-ap0: AP-ENABLED
Fri Jun  9 23:59:28 2023 daemon.notice netifd: Wireless device 'radio0' is now up
Fri Jun  9 23:59:28 2023 daemon.notice netifd: Network device 'phy0-ap0' link is up
Fri Jun  9 23:59:28 2023 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Fri Jun  9 23:59:28 2023 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 4 names
Fri Jun  9 23:59:28 2023 daemon.info dnsmasq[1]: read /tmp/hosts/odhcpd - 2 names
Fri Jun  9 23:59:28 2023 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Fri Jun  9 23:59:30 2023 daemon.warn odhcpd[1859]: A default route is present but there is no public prefix on lan thus we don't announce a default route by overriding ra_lifetime!
```

Djfe, 2023-06-10:
In-case anyone wants to take a look (I'm impressed at the huge amount of chatter for just disabling wifi):
`ubus monitor`
phy0: 
[testubus.log](https://github.com/openwrt/openwrt/files/11710752/testubus.log)

phy1
[testubus2.log](https://github.com/openwrt/openwrt/files/11710753/testubus2.log)

Djfe, 2023-06-10:
I did some further tests:
The Netgear WAX202 is also affected (I tested this one a few months ago)

Tested now:
ipq40xx is not affected by this bug (Linksys MR8300)
https://pastebin.com/pAsCMjkb
ath79 is not affected by this bug (TP-Link Archer C6 v2)
https://pastebin.com/mjxG0LfQ
another mt7621 device (but wifi5 not wifi6) affected by this bug (Cudy WR1300v1)
https://pastebin.com/YS0v5y50

Maker39, 2023-06-29:
Tasted on: Beeline SmartBox TURBO+ , SmartBox Pro. 
OpenWrt 23.05.0-rc1 (MediaTek MT7621)
There is the same problem.

mmortal03, 2023-06-29:
> Tested on: Beeline SmartBox TURBO+ , SmartBox Pro. OpenWrt 23.05.0-rc1 (MediaTek MT7621) There is the same problem.

rc2 is out, btw.

Djfe, 2023-06-29:
well yes, but this wasn't addressed, yet. (atleast I'm fairly certain it wasn't)
still ty for informing us about rc2. it has added new devices and fixed some stuff

Maker39, 2023-06-29:
23.05.0-rc2
Known issues

    lantiq/xrx200 target is not build because the DSA driver still shows some error messages. Fix is planned before the final OpenWrt 23.05 release.
    **Disabling Wifi in LuCI does not work (See https://github.com/openwrt/openwrt/issues/11725**)

LuckySpecialist, 2023-07-01:
I checked the problem on several devices.
TP-Link EC330-G5u v1 (ramips/mt7621):

1. The problem does not affect version 22.03 ([log](https://privatebin.io/?bb5c27bd90b679c4#4uGiMP6xBHfbZCFj1jNiqt6tc6yY4GfcBxn1bGnjAh57))
2. The problem repeats on 23.05 rc2 ([log](https://privatebin.io/?d257889218b65a7d#GQiLcJqmmz6x6QPvU1kh9W8nxeANdQcvUfU9AJyKHNti))

I also confirm the problem on 23.05 rc2 for devices:

1. Beeline SmartBox PRO (ramips/mt7621) ([log](https://bin.tiekoetter.com/?0231b243e6d01543#Bh2acT39FzKxGs3a8w2v9N82LxRqa3LdXCoqKLwn2XeH))
2. NETGEAR WNDR3800 (ath79/generic)

asamahy, 2023-08-04:
tested on Linksys EA8300 v1.0
device has 3 radios
the issue affects only the 5Ghz radios (QCA9886 and IPQ4019) but not the 2Ghz (IPQ4019)

ishanjain28, 2023-08-06:
This has been mentioned earlier in this thread and I am adding another data point. 5Ghz on my Archer A6 is also affected by this. Here are the logs,

```
NAME="OpenWrt"
VERSION="22.03.4"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 22.03.4"
VERSION_ID="22.03.4"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r20123-38ccc47687"
OPENWRT_BOARD="ramips/mt7621"
OPENWRT_ARCH="mipsel_24kc"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 22.03.4 r20123-38ccc47687"

Sun Aug  6 22:10:25 2023 daemon.notice hostapd: Remove interface 'wlan1'
Sun Aug  6 22:10:25 2023 daemon.notice hostapd: wlan1: interface state DISABLED->DISABLED
Sun Aug  6 22:10:25 2023 daemon.notice hostapd: wlan1-3: AP-DISABLED
Sun Aug  6 22:10:25 2023 daemon.notice hostapd: wlan1-3: CTRL-EVENT-TERMINATING
Sun Aug  6 22:10:25 2023 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Sun Aug  6 22:10:25 2023 daemon.err hostapd: hostapd_free_hapd_data: Interface wlan1-3 wasn't started
Sun Aug  6 22:10:25 2023 daemon.notice hostapd: wlan1-2: AP-DISABLED
Sun Aug  6 22:10:25 2023 daemon.notice hostapd: wlan1-2: CTRL-EVENT-TERMINATING
Sun Aug  6 22:10:25 2023 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Sun Aug  6 22:10:25 2023 daemon.err hostapd: hostapd_free_hapd_data: Interface wlan1-2 wasn't started
Sun Aug  6 22:10:25 2023 daemon.notice hostapd: wlan1-1: AP-DISABLED
Sun Aug  6 22:10:25 2023 daemon.notice hostapd: wlan1-1: CTRL-EVENT-TERMINATING
Sun Aug  6 22:10:25 2023 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Sun Aug  6 22:10:25 2023 daemon.err hostapd: hostapd_free_hapd_data: Interface wlan1-1 wasn't started
Sun Aug  6 22:10:25 2023 daemon.notice hostapd: wlan1: AP-DISABLED
Sun Aug  6 22:10:25 2023 daemon.notice hostapd: wlan1: CTRL-EVENT-TERMINATING
Sun Aug  6 22:10:25 2023 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Sun Aug  6 22:10:25 2023 daemon.err hostapd: hostapd_free_hapd_data: Interface wlan1 wasn't started
Sun Aug  6 22:10:25 2023 daemon.notice hostapd: nl80211: deinit ifname=wlan1 disabled_11b_rates=0
Sun Aug  6 22:10:25 2023 kern.info kernel: [735120.265048] device wlan1 left promiscuous mode
Sun Aug  6 22:10:25 2023 kern.info kernel: [735120.270247] br-lan: port 6(wlan1) entered disabled state
Sun Aug  6 22:10:25 2023 daemon.notice netifd: Wireless device 'radio1' is now down
Sun Aug  6 22:10:26 2023 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy1.conf (phy wlan1) --> new PHY
Sun Aug  6 22:10:27 2023 kern.info kernel: [735122.289052] br-lan: port 6(wlan1) entered blocking state
Sun Aug  6 22:10:27 2023 kern.info kernel: [735122.294546] br-lan: port 6(wlan1) entered disabled state
Sun Aug  6 22:10:27 2023 kern.info kernel: [735122.300743] device wlan1 entered promiscuous mode
Sun Aug  6 22:10:27 2023 daemon.notice hostapd: wlan1: interface state UNINITIALIZED->COUNTRY_UPDATE
Sun Aug  6 22:10:27 2023 daemon.notice hostapd: ACS: Automatic channel selection started, this may take a bit
Sun Aug  6 22:10:27 2023 daemon.notice hostapd: wlan1: interface state COUNTRY_UPDATE->ACS
Sun Aug  6 22:10:27 2023 daemon.notice hostapd: wlan1: ACS-STARTED
Sun Aug  6 22:10:28 2023 daemon.notice netifd: Wireless device 'radio1' is now up
Sun Aug  6 22:10:47 2023 daemon.notice hostapd: wlan1: ACS-COMPLETED freq=5260 channel=52
Sun Aug  6 22:10:47 2023 daemon.notice hostapd: wlan1: interface state ACS->HT_SCAN
Sun Aug  6 22:10:48 2023 daemon.notice hostapd: wlan1: interface state HT_SCAN->DFS
Sun Aug  6 22:10:48 2023 daemon.notice hostapd: wlan1: DFS-CAC-START freq=5260 chan=52 sec_chan=1, width=1, seg0=58, seg1=0, cac_time=60s
Sun Aug  6 22:10:48 2023 daemon.err hostapd: DFS start_dfs_cac() failed, -1
Sun Aug  6 22:10:48 2023 daemon.err hostapd: Interface initialization failed
Sun Aug  6 22:10:48 2023 daemon.notice hostapd: wlan1: interface state DFS->DISABLED
Sun Aug  6 22:10:48 2023 daemon.notice hostapd: wlan1: AP-DISABLED
```

raenye, 2023-09-04:
I seem to have the same problem on Redmi AX6S 2.4GHz radio.
```
# cat /etc/openwrt_release
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r23682+8-1eb67cb070'
DISTRIB_TARGET='mediatek/mt7622'
DISTRIB_ARCH='aarch64_cortex-a53'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r23682+8-1eb67cb070'
DISTRIB_TAINTS='no-all'
```

Oozil, 2023-09-05:
Unfortunately, the problem still exists and there is no solution

The 2.4GHz & 5GHz Wi-Fi networks are still working even after they are turned off. They are only stopped if the device is restarted, or instead of restarting, the wpad is restarted.
[image]

bobc26, 2023-09-06:
In order to make progress on this bug, it might help if someone who has hardware which
is affected by the bug adds the following debugging printout to two shell-scripts:

**/lib/netifd/netifd-wireless.sh**:
```
--- netifd-wireless.sh
+++ netifd-wireless.sh
@@ -389,6 +389,7 @@
 }
 
 init_wireless_driver() {
+       echo "$*" # debug printout
        name="$1"; shift
        cmd="$1"; shift
 
```
**/lib/netifd/wireless/mac80211.sh**:
```
--- mac80211.sh
+++ mac80211.sh
@@ -612,6 +612,7 @@
        local new_phy="$(get_board_phy_name "$path")"
        [ -z "$new_phy" -o "$new_phy" = "$phy" ] && return
 
+       echo "debug: rename_board_phy_by_path: iw $phy set name $new_phy"
        iw "$phy" set name "$new_phy" && phy="$new_phy"
 }
 
@@ -630,6 +631,7 @@
 
        [ "$prev_phy" = "$phy" ] && return 0
 
+       echo "debug: rename_board_phy_by_name: iw $prev_phy set name $phy"
        iw "$prev_phy" set name "$phy"
 )

```
After editing the shell-scripts, run,
 ` logread -f | egrep -i '}|bug'`
and then enable/disable wifi in LuCl.

My hardware, which is not affected by the bug, gives the following output when wifi is turned on:

> daemon.notice netifd: radio1 (xxxx): mac80211 **setup** radio1 {"config":{"path":"pci0000:00/0000:00:00.0","channel":"36","band":"5g","htmode":"HT20","cell_density":0},"interfaces":{"0":{"bridge":"br-lan","bridge-ifname":"br-lan","config":{"mode":"ap","ssid":"OpenWrt","encryption":"psk2+ccmp","isolate":true,**"key":"my_pa$$w0rd"**,"ieee80211w":2,"wpa_disable_eapol_key_retries":true,"network":["lan"],"mode":"ap","isolate":true},"vlans":{},"stas":{}}}}

and the following output when wifi is turned off:

> daemon.notice netifd: radio1 (xxxx): mac80211 **teardown** radio1 {"config: {"path":"pci0000:00/0000:00:00.0","channel":"36","band":"5g","htmode":"HT20","cell_density":0},**"data":{"phy":"phy1"}**,"interfaces":{"0":{"bridge":"br-lan","bridge-ifname":"br-lan","config":{"mode":"ap","ssid":"OpenWrt","encryption":"psk2+ccmp","isolate":true,**"key":"my_pa$$w0rd"**,"ieee80211w":2,"wpa_disable_eapol_key_retries":true,"network":["lan"],"mode":"ap","isolate":true},"vlans":{},"stas":{}}}}

The output includes the wifi password, so replace that with xxxxx.

Djfe mentioned that the bug output contains,

>  WARNING: Variable 'data' does not exist or is not an array/object

This suggests that the "teardown" data is missing  the following stanza,
> "data":{"phy":"phy1"}

That stanza should be created during wifi setup by function wireless_set_data in  /lib/netifd/wireless/mac80211.sh.

Djfe, 2023-09-06:
thank you! very helpful

raenye, 2023-09-09:
I edited `netifd` scripts as suggested by @bobc26; here is the output.

Enable and disable (with bug):
```
Sat Sep  9 20:37:30 2023 daemon.notice netifd: radio0 (10480): mac80211 setup    radio0 {"config":{"path":"platform/18000000.wmac","country":"ZZ","band":"2g","channel":"9","htmode":"HT40","noscan":"1","noscan":true,"txpower":1,"cell_density":0,"beacon_int":250},"interfaces":{"0":{"bridge":"br-lan","bridge-ifname":"br-lan","config":{"ifname":"phy0-ap0","mode":"ap","ssid":"ZZ","encryption":"psk2+ccmp","key":"ZZ","wpa_disable_eapol_key_retries":true,"ieee80211r":true,"ft_over_ds":false,"ft_psk_generate_local":true,"ieee80211k":true,"time_advertisement":2,"time_zone":"ZZ","wnm_sleep_mode":true,"wnm_sleep_mode_no_keys":true,"bss_transition":true,"proxy_arp":true,"network":["lan"],"mode":"ap","proxy_arp":true},"vlans":{},"stas":{}}}}
Sat Sep  9 20:37:45 2023 daemon.notice netifd: radio0 (11169): mac80211 teardown radio0 {"config":{"path":"platform/18000000.wmac","country":"ZZ","band":"2g","channel":"9","htmode":"HT40","noscan":"1","noscan":true,"txpower":1,"cell_density":0,"beacon_int":250},"interfaces":{"0":{"bridge":"br-lan","bridge-ifname":"br-lan","config":{"ifname":"phy0-ap0","mode":"ap","ssid":"ZZ","encryption":"psk2+ccmp","key":"ZZ","wpa_disable_eapol_key_retries":true,"ieee80211r":true,"ft_over_ds":false,"ft_psk_generate_local":true,"ieee80211k":true,"time_advertisement":2,"time_zone":"ZZ","wnm_sleep_mode":true,"wnm_sleep_mode_no_keys":true,"bss_transition":true,"proxy_arp":true,"network":["lan"],"mode":"ap","proxy_arp":true},"vlans":{},"stas":{}}}}
Sat Sep  9 20:37:45 2023 daemon.notice netifd: radio0 (11169): Bug: PHY is undefined for device 'radio0'
```

Setup and teardown JSONs are identical, with no `"data"` block.
No output from `rename_board_phy_by_xxx` either.

Maker39, 2023-09-10:
root@OpenWrt:~# cat /etc/openwrt_release
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r23885-703d667a0c'
DISTRIB_TARGET='ramips/mt7621'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r23885-703d667a0c'
DISTRIB_TAINTS=''

root@OpenWrt:~# logread -f | egrep -i '}|bug'
**DISABLE 0**
Sun Sep 10 13:46:38 2023 daemon.notice netifd: radio0 (24473): mac80211 teardown radio0 {"config":{"path":"1e140000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0","channel":"36","band":"5g","htmode":"VHT80","cell_density":0},"interfaces":{"0":{"bridge":"br-lan","bridge-ifname":"br-lan","config":{"mode":"ap","ssid":"OpenWrt","encryption":"psk2","key":"AP_PASSWD","network":["lan"],"mode":"ap"},"vlans":{},"stas":{}}}}
Sun Sep 10 13:46:38 2023 daemon.notice netifd: radio0 (24473): Bug: PHY is undefined for device 'radio0'

**ENABLE 0**
Sun Sep 10 13:46:53 2023 daemon.notice netifd: radio0 (24995): mac80211 setup radio0 {"config":{"path":"1e140000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0","channel":"36","band":"5g","htmode":"VHT80","cell_density":0},"interfaces":{"0":{"bridge":"br-lan","bridge-ifname":"br-lan","config":{"mode":"ap","ssid":"OpenWrt","encryption":"psk2","key":"AP_PASSWD","network":["lan"],"mode":"ap"},"vlans":{},"stas":{}}}}

**DISABLE 1**
Sun Sep 10 13:47:05 2023 daemon.notice netifd: radio1 (25604): mac80211 teardown radio1 {"config":{"path":"1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0","channel":"1","band":"2g","htmode":"HT20","cell_density":0},"interfaces":{"0":{"bridge":"br-lan","bridge-ifname":"br-lan","config":{"mode":"ap","ssid":"OpenWrt","encryption":"psk2","key":"AP_PASSWD","network":["lan"],"mode":"ap"},"vlans":{},"stas":{}}}}
Sun Sep 10 13:47:05 2023 daemon.notice netifd: radio1 (25604): Bug: PHY is undefined for device 'radio1'

**ENABLE 1**
Sun Sep 10 13:47:14 2023 daemon.notice netifd: radio1 (26126): mac80211 setup radio1 {"config":{"path":"1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0","channel":"1","band":"2g","htmode":"HT20","cell_density":0},"interfaces":{"0":{"bridge":"br-lan","bridge-ifname":"br-lan","config":{"mode":"ap","ssid":"OpenWrt","encryption":"psk2","key":"AP_PASSWD","network":["lan"],"mode":"ap"},"vlans":{},"stas":{}}}}

Maker39, 2023-09-11:
After reboot :

root@OpenWrt:~# logread -f | egrep -i '}|bug'
**step1-disable radio0**
Mon Sep 11 10:04:51 2023 daemon.notice netifd: radio0 (4181): mac80211 teardown radio0 {"config":{"path":"1e140000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0","channel":"36","band":"5g","htmode":"VHT80","cell_density":0},"data":{"phy":"phy0","txantenna":"0xffffffff","rxantenna":"0xffffffff"},"interfaces":{"0":{"bridge":"br-lan","bridge-ifname":"br-lan","config":{"mode":"ap","ssid":"OpenWrt","encryption":"psk2","key":"PASSWORD","network":["lan"],"mode":"ap"},"data":{"ifname":"phy0-ap0"},"vlans":{},"stas":{}}}}

**step2-enable radio0**
Mon Sep 11 10:16:40 2023 daemon.notice netifd: radio0 (5258): mac80211 setup radio0 {"config":{"path":"1e140000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0","channel":"36","band":"5g","htmode":"VHT80","cell_density":0},"interfaces":{"0":{"bridge":"br-lan","bridge-ifname":"br-lan","config":{"mode":"ap","ssid":"OpenWrt","encryption":"psk2","key":"PASSWORD","network":["lan"],"mode":"ap"},"vlans":{},"stas":{}}}}

**step3-disable radio0**
Mon Sep 11 10:18:28 2023 daemon.notice netifd: radio0 (5961): mac80211 teardown radio0 {"config":{"path":"1e140000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0","channel":"36","band":"5g","htmode":"VHT80","cell_density":0},"interfaces":{"0":{"bridge":"br-lan","bridge-ifname":"br-lan","config":{"mode":"ap","ssid":"OpenWrt","encryption":"psk2","key":"PASSWORD","network":["lan"],"mode":"ap"},"vlans":{},"stas":{}}}}
Mon Sep 11 10:18:29 2023 daemon.notice netifd: radio0 (5961): _Bug: PHY is undefined for device 'radio0'_

bobc26, 2023-09-12:
raenye and Maker39, thank you for contributing to the debugging effort!  Your output has helped me to make some progress, though I don't yet have a solution.

When wifi is turned on, function drv_mac80211_setup in /lib/netifd/wireless/mac80211.sh is called.  The first couple of steps taken by this function are:
```
drv_mac80211_setup() {
   ...
   find_phy 
   wireless_set_data phy="$phy" 
   ...
}
```
find_phy defines variable phy, while wireless_set_data makes a ubus call.

The bug is that, for some hardware, the ubus call made by wireless_set_data does not have any effect.

For debugging purposes I'm using a simplified version of drv_mac80211_setup that only works for radio0/phy0,
```
drv_mac80211_setup() {
   ...
   phy=phy0
   ubus call network.wireless notify '{ "command": 1, "device": "radio0", "data": { "phy": "phy0" } }'
   ...
}
```
My hardware is not affected by the bug, and I can turn the wifi on/off using this simplified drv_mac80211_setup.

But if the ubus call is delayed by 1 second,
```
drv_mac80211_setup() {
   ...
   phy=phy0
   sleep 1
   ubus call network.wireless notify '{ "command": 1, "device": "radio0", "data": { "phy": "phy0" } }'
   ...
}
```
then **all hardware is affected by the bug**. Openwrt versions 23.05-rc3 and 22.03.5 are both affected.

Evidently the bug is some kind of timeout which takes effect if there is a delay in making the ubus call.  This could be the race condition that nbd previously alluded to.

The commit that introduced the bug, [4d32330,](https://github.com/openwrt/openwrt/commit/4d323303e7e5743f541e3b41dfb2ac1627e8d96d) increased the time taken in  find_phy() and hence delayed the ubus call.  find_phy() now reads /etc/board.json.  The length of that file depends upon the hardware, and the longer that file is the more likely the hardware is to be affected by the bug.

The next step is to figure out how to increase the mysterious timeout.  Perhaps someone already knows the answer :smile:  

It appears that the data from the ubus call is handled by function netifd_handle_wdev_notify in package netifd/ubus.c, and passed on to function wireless_device_notify in netifd/wireless.c .  So I'm planning to poke around there.

raenye, 2023-09-12:
Thanks @bobc26.

My two affected devices:
1. Xiaomi Redmi AX6S (mt7622):
```
root@OpenWrt:~# wc /etc/board.json
       41        71       635 /etc/board.json
```
2. D-Link DAP-1620 rev B (mt7621):
```
root@OpenWrt:~# wc /etc/board.json
       63       115       994 /etc/board.json
```

bobc26, 2023-09-14:
I'm still examining netifd, but for now here's a patch that fixes the issue for me.  Hopefully it can be included in 23.05.0-rc4 to facilitate widespread testing.
```
--- a/netifd/wireless.c
+++ b/netifd/wireless.c
@@ -1010,7 +1010,7 @@ wireless_device_check_script_tasks(struct uloop_timeout *timeout)
        if (restart)
                wireless_device_retry_setup(wdev);
        else
-               uloop_timeout_set(&wdev->script_check, 1000);
+               uloop_timeout_set(&wdev->script_check, 3 * 1000);
 }
 
 /* creates a wireless device object. Called by config */
```
[bug_11725.patch](https://github.com/openwrt/openwrt/files/12603446/bug_11725.patch.txt)

raenye, 2023-09-14:
Thanks @bobc26.
Where should I put it to get it to compile?
Placing it in `package/network/config/netifd/patches/bug_11725.patch`, I get 
```
No file to patch.  Skipping patch.
1 out of 1 hunk ignored
Patch failed!  Please fix ./patches/bug_11725.patch!
```

bobc26, 2023-09-14:
**Retraction: the patch in the earlier post does not fix the bug**.  My apologies.  (Yesterday the bug did seem to have disappeared after applying the patch, but today after reflashing the router and reapplying the patch the bug has reappeared, much to my chagrin.)

bobc26, 2023-09-19:
@nbd has fixed this bug with commit **wireless: update prev_config on SET_DATA notify** [7a58b995](https://github.com/openwrt/netifd/commit/7a58b995fdbecd9beed57e4d66d42cb3cf66aee2)

raenye, 2023-09-19:
Post-patch builds work fine on my mt7621 & mt7622 devices. Thanks!

raenye, 2023-09-28:
@Djfe perhaps it's ok to close the issue?

Djfe, 2023-10-02:
Thank you @bobc26 and @jow- for helping with debugging this issue and thanks @nbd168 for the solution :)
I just verified for myself that this is fixed in rc4 as well
https://github.com/openwrt/openwrt/commit/829196e1b16bf20062d1d839ad38042290bb016c

I'll notify Hauke to remove this from "Known issues" now :)

closed 2023-10-02 by Djfe