openwrt/openwrt #11726: r8125 2.5GbE auto negotiate can not get 2.5G speed

xchgeaxeax, 2023-01-08:
### Describe the bug

I use r8125b pcie netcard passthrough to my OpenWrt in ESXi, but I can not get 2.5G speed but 1G.
I tried ethtool to find the problem, the 'Advertised link modes' max speed is 1000baseT/Full, but both 'Supported link modes' and 'Link partner advertised link modes' have 2500baseT/Full option.

### OpenWrt version

r24774-3ef187c32f

### OpenWrt target/subtarget

x86/64

### Device

VMware, Inc. VMware20,1

### Image kind

Self-built image

### Additional info

I find a similar issue in [OpenWrt forum ](https://forum.openwrt.org/t/8125b-drive-works-abnormally/147433/4) 
this command works.
`ethtool -s eth(*) autoneg on advertise 0x80000000002f`
so I edit /etc/rc.local, add these commands as a temporary solution

\# r8125 2500mbps autoneg
ethtool -s eth2 autoneg on advertise 0x80000000002f
ethtool -s eth3 autoneg on advertise 0x80000000002f
ethtool -s eth4 autoneg on advertise 0x80000000002f
ethtool -s eth5 autoneg on advertise 0x80000000002f

But if I use `service network restart` the card will drop to 1G.
Hope this issue can be solved

### Diffconfig

```text
CONFIG_TARGET_x86=y
CONFIG_TARGET_x86_64=y
CONFIG_TARGET_x86_64_DEVICE_generic=y
CONFIG_DEVEL=y
CONFIG_TOOLCHAINOPTS=y
CONFIG_ARIA2_BITTORRENT=y
CONFIG_ARIA2_NOXML=y
CONFIG_ARIA2_OPENSSL=y
CONFIG_ARIA2_WEBSOCKET=y
# CONFIG_BINUTILS_USE_VERSION_2_37 is not set
CONFIG_BINUTILS_USE_VERSION_2_39=y
CONFIG_BINUTILS_VERSION="2.39"
CONFIG_BINUTILS_VERSION_2_39=y
CONFIG_EXPERIMENTAL=y
# CONFIG_GCC_USE_VERSION_11 is not set
CONFIG_GCC_USE_VERSION_12=y
CONFIG_GCC_VERSION="12.2.0"
CONFIG_GCC_VERSION_12=y
# CONFIG_GDB is not set
CONFIG_GRUB_TIMEOUT="1"
CONFIG_ISO_IMAGES=y
CONFIG_LIBCURL_COOKIES=y
CONFIG_LIBCURL_CRYPTO_AUTH=y
CONFIG_LIBCURL_FILE=y
CONFIG_LIBCURL_FTP=y
CONFIG_LIBCURL_HTTP=y
CONFIG_LIBCURL_NGHTTP2=y
CONFIG_LIBCURL_OPENSSL=y
CONFIG_LIBCURL_PROXY=y
CONFIG_LIBCURL_TFTP=y
[... 329 lines cut ...]
CONFIG_PCRE_JIT_ENABLED=y
CONFIG_SQLITE3_COLUMN_METADATA=y
CONFIG_SQLITE3_DYNAMIC_EXTENSIONS=y
CONFIG_SQLITE3_FTS3=y
CONFIG_SQLITE3_FTS4=y
CONFIG_SQLITE3_FTS5=y
CONFIG_SQLITE3_JSON1=y
CONFIG_SQLITE3_RTREE=y
CONFIG_TARGET_EXT4_JOURNAL=y
CONFIG_TARGET_KERNEL_PARTSIZE=64
CONFIG_TARGET_OPTIMIZATION="-O3 -pipe"
CONFIG_TARGET_OPTIONS=y
CONFIG_TARGET_ROOTFS_PARTSIZE=512
# CONFIG_TARGET_ROOTFS_SQUASHFS is not set
CONFIG_VHDX_IMAGES=y
CONFIG_VMDK_IMAGES=y
CONFIG_boost-compile-visibility-hidden=y
CONFIG_boost-runtime-shared=y
CONFIG_boost-static-and-shared-libs=y
CONFIG_boost-variant-release=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

xchgeaxeax, 2023-01-08:
If anyone knows how to edit netcard auto negotiate speed permanently, please tell me.
Thanks

sjkhsl, 2023-01-16:
same question. Hope that can be resolved!

sjkhsl, 2023-01-17:
When I checked the information, I found that kmod-r8169 and kmod-phy-realtek conflict with r8125. You can try it. I am not convenient to test it now.

Noltari, 2025-01-14:
This is working on main and 24.10.

Tested on RPi 5B with r8126 (kmod-r8169 driver):
```
Settings for eth1:
        Supported ports: [ TP MII ]
        Supported link modes:   10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
                                2500baseT/Full
                                5000baseT/Full
        Supported pause frame use: Symmetric Receive-only
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
                                2500baseT/Full
                                5000baseT/Full
        Advertised pause frame use: No
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Link partner advertised link modes:  2500baseT/Full
        Link partner advertised pause frame use: Symmetric
        Link partner advertised auto-negotiation: Yes
        Link partner advertised FEC modes: Not reported
        Speed: 2500Mb/s
        Duplex: Full
        Port: Twisted Pair
        PHYAD: 0
        Transceiver: internal
        Auto-negotiation: on
        MDI-X: Unknown
        Supports Wake-on: pumbg
        Wake-on: d
        Link detected: yes
```

closed 2025-01-14 by Noltari