openwrt/openwrt #11727: VLAN tagging doesn't work on Zyxel GS1900-48

joe2xyz, 2023-01-08:
### Describe the bug

When two  Zyxel GS1900-48 switches are interconnected by a single cable with multiple VLANs set up as "t" (tagged), a PC on the first switch connected to "u" (untagged) port on VLAN 100 can ping a PC connected to the second switch "u" port on VLAN 101.

Thank you very much OpenWrt developers and maintainers. OpenWrt is fantastic project.

### OpenWrt version

r19803-9a599fee93

### OpenWrt target/subtarget

realtek/rtl839x

### Device

Zyxel GS1900-48

### Image kind

Official downloaded image

### Steps to reproduce

* Create multiple VLANs with the same numbers on two Zyxel GS1900-48 switches (for example VLAN 100, 101, 102)
* On both switches set LAN1 port as "t" (tagged) on all VLANs (for example "t" on VLAN 100, 101 and 102)
* On the first switch set LAN2 port as "u" on a single VLAN (for example "u" only on VLAN 100)
* On the second switch set LAN2 port as "u" on a single VLAN that is different (for example "u" only on VLAN 101)
* Connect one PC to LAN2 on the first switch and connect other PC to LAN2 on the second switch.
* Ping from one PC to the other

### Actual behaviour

PCs on different VLANs can ping each other.

### Expected behaviour

PCs on different VLANs should not be able to ping each other.

### Additional info

I tested the latest stable version:
openwrt-22.03.2-realtek-rtl839x-zyxel_gs1900-48-squashfs-sysupgrade.bin

And the version before:
openwrt-22.03.1-realtek-rtl839x-zyxel_gs1900-48-squashfs-sysupgrade.bin

And the result is the same - VLAN tagging doesn't work. I also tried the latest snapshot but unfortunately it kernel panics during the boot so I can't test it.

Also sometime I see the following kernel messages on the serial console:
`[230498.762516] rtl838x-eth 1b00a300.ethernet eth0: Ring contention: r: 0, last a40002dc, cur a4000400`
Not sure if that is related...

Thank you very much OpenWrt developers and maintainers. OpenWrt is amazing. Feel free to ask questions.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-01-08:
Can you connect a laptop with wireshark to the port where other switch normally should be and check if outgoing ping is properly tagged as per configuration?

joe2xyz, 2023-01-08:
I haven't used Wireshark but I had the same question and I connected a dumb switch that I know can't pass-through VLAN tags and it worked. So packets on tagged (trunk) port are not tagged.

brada4, 2023-01-08:
Then it is pretty obvious other switch cannot figure out to which partition packet belongs even if it wanted to.
Normally dumb switches pass through VLAN header and it is filtered away in client network card. Usually they do something like 4KB jumbo frames, useless in normal world, but providing enough space for those extra tags, and they are unlikely to alter packets thanks to dumbness.
Wireshark would give extra assurance, but your result seems trustworthy.

brada4, 2023-01-09:
Your switch combination looks like a bridge , i.e. disregards all configuration.
yes/no question -is config loaded in switch hardware:
```
 swconfig list
 swconfig dev switch0 show <- shall display all vlans and ports as web
```
I dont have/know DSA, should have something similar. Seems discrepancy between what you did in LUCI and dumped with UCI and actual hardware behaviour.

brada4, 2023-01-09:
I just invented simpler test case for lack of isolation part:
Connect device to port A, other to port B, configure them on different untagged VLAN-s and ping happens when it should not.

joe2xyz, 2023-01-10:
> discrepancy between what you did in LUCI and dumped with UCI

Screenshots might have different VLAN numbers / ports than `uci export` since I made them in different time. But the point is the same - if I connect two switches via tagged ports (tagged for all VLANs), the isolation doesn't work.

I'll post result of A / B test and  `swconfig list` and `swconfig dev switch0 show` later today. Thank you.

brada4, 2023-01-10:
For DSA (if no swconfig) you need `ip link show` and `bridge vlan show`

A/B test is to confirm that switch runs with blank configuration or not
Config readback is to understand if (l)uci configuration you entered is actually reaching kernel drivers or they somehow mismatch hardware.

brada4, 2023-01-10:
New release is out, though nothing explicitly around your configuration, try it anyway to start on same line (keep configuration over upgrade or use auc to do it for you)
 OpenWrt 22.03.3, r20028-43d71ad93e

joe2xyz, 2023-01-11:
Hi Andrew,

Sorry for the delay, it took me some time to finish all testing... I'll try this evening again with the new release. Meanwhile, this is for 22.03.2:

## Diagram

[image]

## Config

```
ssh root@10.0.0.11 uci export        | sed -E 's/(ula_prefix|macaddr) .*/\1 REMOVED/' > first-uci-export.txt
ssh root@10.0.0.13 uci export        | sed -E 's/(ula_prefix|macaddr) .*/\1 REMOVED/' > second-uci-export.txt
```

[first-uci-export.txt](https://github.com/openwrt/openwrt/files/10389832/first-uci-export.txt)
[second-uci-export.txt](https://github.com/openwrt/openwrt/files/10389835/second-uci-export.txt)

## bridge

```
ssh root@10.0.0.11 bridge vlan show  | sed 's|b8:ec:..:..:..:..|-----REMOVED-----|'  > first-bridge.txt
ssh root@10.0.0.13 bridge vlan show  | sed 's|b8:ec:..:..:..:..|-----REMOVED-----|'  > second-bridge.txt
```

[first-bridge.txt](https://github.com/openwrt/openwrt/files/10389838/first-bridge.txt)
[second-bridge.txt](https://github.com/openwrt/openwrt/files/10389842/second-bridge.txt)

## ip link

```
ssh root@10.0.0.11 ip l | sed -E 's|..:..:..:..:..:..|-------------|g' > first-ip-l.txt
ssh root@10.0.0.13 ip l | sed -E 's|..:..:..:..:..:..|-------------|g' > second-ip-l.txt
```

[first-ip-l.txt](https://github.com/openwrt/openwrt/files/10389849/first-ip-l.txt)
[second-ip-l.txt](https://github.com/openwrt/openwrt/files/10389854/second-ip-l.txt)

## ip address

```
ssh root@10.0.0.11 ip a | sed -E -e 's|..:..:..:..:..|-------------|g' -e 's|fe80:[^ ]+|fe80:--removed--|' > first-ip-a.txt
ssh root@10.0.0.13 ip a | sed -E -e 's|..:..:..:..:..|-------------|g' -e 's|fe80:[^ ]+|fe80:--removed--|' > second-ip-a.txt
```

[first-ip-a.txt](https://github.com/openwrt/openwrt/files/10389846/first-ip-a.txt)
[second-ip-a.txt](https://github.com/openwrt/openwrt/files/10389848/second-ip-a.txt)

## results

* PC1 can connect both PC2 and PC3
* PC2 cannot connect PC3 (very strange that this isolation works OK)

brada4, 2023-01-11:
Your visual config is OK
So isolation and VLANs on ASIC do work as intended.
VLAN list is either abbreviated without matching ASIC vlan ID-s to wire VLAN IDs, or by bug they never got programmed into switch hardware.

i.e.
The lan01 port has five vlan-s . Not sure how it works, if you repeat that configuration matching ASIC internal vid 10x with pvid 10x ? Or it might be assumed by default - i dunno.

It goes something like bridge vlan del dev lan0 vlan 105, , then add it back adding pvid parameter. Just check bridge vlan output after first if it adds pvid or not. as I said I am not sure how it should look normally. 

No need to post IP(v6) addressing, VLAN-s happen at much lower level.

joe2xyz, 2023-01-11:
Thank you Andrew.

To summarize you want me to run:
```
bridge vlan del dev lan01 vid 105
bridge vlan del dev lan01 vid 104
bridge vlan del dev lan01 vid 103
bridge vlan del dev lan01 vid 102
bridge vlan del dev lan01 vid 101
bridge vlan del dev lan01 vid 100

bridge vlan add dev lan01 vid 100 pvid 100
```
and see what happens....

brada4, 2023-01-11:
Yes,one is enough as poc

joe2xyz, 2023-01-11:
After fresh reboot - on the first switch:

```
bridge vlan show
```
~~~
port              vlan-id
lan01             100
                  101
                  102
                  103
                  104
                  105
lan02             100
                  101
                  102
                  103
                  104
                  105
lan03             102 PVID Egress Untagged
lan04             102 PVID Egress Untagged
lan05             102 PVID Egress Untagged
lan06             103 PVID Egress Untagged
lan07             103 PVID Egress Untagged
lan08             103 PVID Egress Untagged
lan48             102 PVID Egress Untagged
switch            100
                  101
                  102
                  103
                  104
                  105
~~~
```
bridge vlan del dev lan01 vid 105
bridge vlan show
```
~~~
port              vlan-id
lan01             100
                  101
                  102
                  103
                  104
lan02             100
                  101
                  102
                  103
                  104
                  105
lan03             102 PVID Egress Untagged
lan04             102 PVID Egress Untagged
lan05             102 PVID Egress Untagged
lan06             103 PVID Egress Untagged
lan07             103 PVID Egress Untagged
lan08             103 PVID Egress Untagged
lan48             102 PVID Egress Untagged
switch            100
                  101
                  102
                  103
                  104
                  105
~~~
```
bridge vlan del dev lan01 vid 104
bridge vlan del dev lan01 vid 103
bridge vlan del dev lan01 vid 102
bridge vlan del dev lan01 vid 101
bridge vlan del dev lan01 vid 100
```
* PC1 stopped being able to connect PC2 and PC3
```
bridge vlan show
```
```
port              vlan-id
lan02             100
                  101
                  102
                  103
                  104
                  105
lan03             102 PVID Egress Untagged
lan04             102 PVID Egress Untagged
lan05             102 PVID Egress Untagged
lan06             103 PVID Egress Untagged
lan07             103 PVID Egress Untagged
lan08             103 PVID Egress Untagged
lan48             102 PVID Egress Untagged
switch            100
                  101
                  102
                  103
                  104
                  105
```
```
bridge vlan add dev lan01 vid 102 pvid 102
```
* PC1 can again connect both PC2 and PC3
```
bridge vlan show
```
~~~
port              vlan-id
lan01             102 PVID
lan02             100
                  101
                  102
                  103
                  104
                  105
lan03             102 PVID Egress Untagged
lan04             102 PVID Egress Untagged
lan05             102 PVID Egress Untagged
lan06             103 PVID Egress Untagged
lan07             103 PVID Egress Untagged
lan08             103 PVID Egress Untagged
lan48             102 PVID Egress Untagged
switch            100
                  101
                  102
                  103
                  104
                  105
~~~

brada4, 2023-01-11:
Now fix up one vlan with PVID on both switches and check that it is isolated from other vlan on other switch.
If yes - uci config is not rendered correctly into `bridge add ????` configs

joe2xyz, 2023-01-12:
After fresh reboot of both switches:

* On the first switch only
```
bridge vlan del dev lan01 vid 105
bridge vlan del dev lan01 vid 104
bridge vlan del dev lan01 vid 103
bridge vlan del dev lan01 vid 102
bridge vlan del dev lan01 vid 101
bridge vlan del dev lan01 vid 100
bridge vlan add dev lan01 vid 102 pvid 102
bridge vlan show
```
```
~~~
port              vlan-id
lan01             102 PVID
lan02             100
                  101
                  102
                  103
                  104
                  105
lan03             102 PVID Egress Untagged
lan04             102 PVID Egress Untagged
lan05             102 PVID Egress Untagged
lan06             103 PVID Egress Untagged
lan07             103 PVID Egress Untagged
lan08             103 PVID Egress Untagged
lan48             102 PVID Egress Untagged
switch            100
                  101
                  102
                  103
                  104
                  105
~~~
* PC1 can speak both to PC2 and PC3
* Now the same commands and the same output on second switch
* PC1 can speak only to PC2, it is not possible to ping PC3 - Isolation works ;-)

brada4, 2023-01-12:
Confirm that 2 different VLAN-s work  (i.e isolated from eachother and bridged to other side) and peek at wire with wireshark that packets in 2 VLAN-s have distinct 8021q tags attached. Just run ping in one side to unused IP in same subnet, that would broadcast ARP requests, which should appear tagged on the wire.

joe2xyz, 2023-01-12:
> Confirm that 2 different VLAN-s work 

So PC1 can speak to PC2 (VLAN 102 works) but cannot speak to PC3 (VLAN 102 can't connect VLAN 103). Would you like me to configure both 102 and 103 on LAN01 and see what is going to happen?

brada4, 2023-01-12:
Exactly. To confirm wrong conf is applied to switch via web and have a fix. Then link all similar DSA issues together and hope for update between releases.

joe2xyz, 2023-01-12:
After fresh reboot of both switches:

```
bridge vlan del dev lan01 vid 105
bridge vlan del dev lan01 vid 104
bridge vlan del dev lan01 vid 103
bridge vlan del dev lan01 vid 102
bridge vlan del dev lan01 vid 101
bridge vlan del dev lan01 vid 100
bridge vlan add dev lan01 vid 102
bridge vlan add dev lan01 vid 103
bridge vlan show
```
~~~
port              vlan-id
lan01             102
                  103
lan02             100
                  101
                  102
                  103
                  104
                  105
lan03             102 PVID Egress Untagged
lan04             102 PVID Egress Untagged
lan05             102 PVID Egress Untagged
lan06             103 PVID Egress Untagged
lan07             103 PVID Egress Untagged
lan08             103 PVID Egress Untagged
lan48             102 PVID Egress Untagged
switch            100
                  101
                  102
                  103
                  104
                  105
~~~

* Isolation broken again - PC1 can speak both to PC2 and 3

joe2xyz, 2023-01-13:
I have a confirmation from OpenWrt forum that RTL8380 and therefore my assumption is that this is related to RTL839x. There is only one device with Realtek RTL839x so that's probably the reason nobody has reported this bug:
https://downloads.openwrt.org/releases/22.03.3/targets/realtek/rtl839x/

Also bmork on OpenWrt forum mentioned that `Ring contention` error is related to sending too much traffic to CPU which shouldn't happen on a L2 switch.
https://forum.openwrt.org/t/bridge-switch-on-zyxel-switch-requires-unmanaged-interface-to-work/147241/20?u=joe2

So it seems that RTL839x is missing something from RTL838x driver implementation.

brada4, 2023-01-13:
There is no way manual commands persist over reboot.
My idea of testing was to make 2 vlans with PVID manually with commands and check that they are properly beamed to other switch with adequate isolation all ways.
No traffic should hit management interface if it is not configured in VLAN-s with massive broadcast traffic. Lacking isolation that might be the pitfall indeed. Normally no problem dropping little traffic, TCP slows down, it is designed to do so, UDP retries after timeout IF feedback is provided.

joe2xyz, 2023-01-13:
> My idea of testing was to make 2 vlans with PVID manually with commands and check that they are properly beamed to other switch with adequate isolation all ways.

I'm sorry, but I don't understand, because I thought I tested exactly what you asked for. If you check my previous post you can see what commands I run. Do you want me to run different commands?

brada4, 2023-01-13:
add pvids to both vlans that need to be tagged. DSA-s cool feature is that you can change them between ASIC and wire.

joe2xyz, 2023-01-16:
Hi Andrew,

I've just finished testing on new release 22.03.3 and I still have the same problems. I tried to add PVID to both tagged VLANs and when I added PVID for the second time, it replaced PVID that was set for the first time (PVID can't be on two tagged VLANs).

I also tried PVID / no PVIDs for untagged VLANs and no effect. Isolation still doesn't work.

brada4, 2023-01-16:
So there is a bug in drivers that VLAN config does not reach wire.
 PVID vlan catches all, my doubt was if it catches all for that VLAN or all on wire.
Sorry to say but for time of being best option is vendor firmware to have LACP+VLAN on interconnect wires working properly. Its not peanuts paid to keep ninety users waiting for longer.

joe2xyz, 2023-01-16:
Yes, it's a bug only in RTL839x since other people report that RTL838x works OK.

Ninety users? This is for my home ;-). If this is not OpenWrt issue, it is possible to report the bug to upstream?

joe2xyz, 2023-01-16:
I'm happy to open an upstream bug report about this. I just need directions where / what to do. Thank you.

brada4, 2023-01-16:
Some more knowledgeable in the matter should tell if it is kernel.org or openwrt backports.

You mentioned [48-port SMB](https://www.zyxel.com/global/en/products/switch/8-10-16-24-48-port-gbe-smart-managed-switch-gs1900-series) model in title

joe2xyz, 2023-01-17:
> You mentioned [48-port SMB](https://www.zyxel.com/global/en/products/switch/8-10-16-24-48-port-gbe-smart-managed-switch-gs1900-series) model in title

Yes, it is 48 port version of ZyXEL GS1900 with `realtek/rtl839x`

> Some more knowledgeable in the matter should tell if it is kernel.org or openwrt backports.

Do you know who do we need to ask?

svanheule, 2023-01-18:
> If this is not OpenWrt issue, it is possible to report the bug to upstream?

No, not possible to report upstream since the relevant drivers are only in OpenWrt.

It is possible this issue was fixed with commit ed9bd9824a47 ("realtek: refactor keep vlan tag setup, fix tagged forwarding") by @luizluca, but 22.03 doesn't have this change (due to the refactoring). Have you tried running snapshot images? Note that these don't include LuCI by default.

luizluca, 2023-01-19:
Yeah, it looks like the bug I fixed. Indeed, we could backport a hacky fix that only set the correct register value for that family. We just need to add a new if to the rtl83xx_port_enable setting the correct hex value to the correct register. It is not pretty, not maintainable but it does fix a broken device for a released version. I do not have that device to test but it should be something like:

else if (priv->family_id == RTL839X_FAMILY_ID)
        sw_w32(BIT(5), RTL839X_VLAN_PORT_TAG_STS_CTRL);
else ...

(Wrote inside github mobile app, not tested at all)

joe2xyz, 2023-01-19:
> Have you tried running snapshot images? Note that these don't include LuCI by default.

Thank you @svanheule , I tried a snapshot approx. 1/2 weeks ago and it didn't boot. It kernel panicked during the boot.

> Indeed, we could backport

Thank you @luizluca , I don't want to ask to backport something that is difficult to maintain. Is it possible to test the commit on the stable release by for example copying some files from the snapshot? I'll test the snapshot again this evening (it's morning for me now) and I can drop here info about the kernel panic I get during the boot.

> I do not have that device to test

I have more than one `RTL839X` connected together via tagged port so feel free to use me as your testing resource ;-)

svanheule, 2023-01-19:
> It kernel panicked during the boot.

That also sounds bad. Could be due to another issue.

> Yeah, it looks like the bug I fixed. Indeed, we could backport a hacky fix that only set the correct register value for that family.

I tried applying the VLAN patch to the openwrt-22.03 branch, and it did so without complaining. But I would have to check if there were any commits in between that it might rely on.

joe2xyz, 2023-01-19:
> That also sounds bad. Could be due to another issue.

I've just downloaded a snapshot built on 18th so I'll check in a few hours if it is still panicking or not.
```
zyxel_gs1900-48-initramfs-kernel.bin	dece0c98717075ee1168801c444316f0b09d261b78460e70a00e37b696a107e2	6439.4 KB	Wed Jan 18 18:03:52 2023
zyxel_gs1900-48-squashfs-sysupgrade.bin	2efdcd72804bded4f77c443c1c03fbda99917689afa08e7b99e2627ffd871232	6144.3 KB	Wed Jan 18 18:03:52 2023 
```

svanheule, 2023-01-19:
OK, that's an issue with SMP. If I'm not mistaken, it's due to an issue with the new timer, which wasn't present with the MIPS timer (so 22.03 is not affected by this problem)

joe2xyz, 2023-01-20:
Good news that it's not related to RTL839X.

But the problem is that with that issue I can't test your commit https://github.com/openwrt/openwrt/commit/ed9bd9824a477b2cca0887867155a73b38775d80 since it doesn't boot ;-)

svanheule, 2023-01-21:
Pushed a fix for the RCU stalls to master, so the next snapshot build should have it included.

joe2xyz, 2023-01-23:
I quickly `sysupgrade`ed to:

```
zyxel_gs1900-48-initramfs-kernel.bin	29cacba63ad005a5927cbbae69223f1c9d64e12833def5f8929454bed18bd0b8	6438.9 KB	Sun Jan 22 11:58:33 2023
zyxel_gs1900-48-squashfs-sysupgrade.bin	0e6f4f251e80c1a0f9a3f6ba4089e48f9e2b6722d36d184e2f2a960e8bcc2365	6144.3 KB	Sun Jan 22 11:58:31 2023
```

and it boots. Thank you.

I'll test VLANs later this evening.

joe2xyz, 2023-01-24:
I didn't have time to test the VLANs yesterday, I'm going to do that ASAP.

But I noticed that I'm not getting any `rtl838x-eth 1b00a300.ethernet eth0: Ring contention: r: 0, last a40002dc, cur a4000400` messages so that's the first huge improvement I noticed. Thank you.

joe2xyz, 2023-01-25:
I'm happy to say that VLANs on the current snapshot work as they should.

Thank you very much everybody and thank you @svanheule and @luizluca for writing the Realtek driver. It's fantastic to have open-source powering both my AP and my switch at home ;-)

Next task will be testing STP after I connect all switches into a loop.

In case you want me to test something, pls let me know. If I don't hear from anyone I'll close the bug on Friday.

brada4, 2023-01-25:
I would recommend testing LACP instead....

joe2xyz, 2023-01-25:
Thank @brada4 for the tip. I think I need STP and I can't use LACP for something like this:

[image]

joe2xyz, 2023-01-27:
I'm happy to say that VLANs on the current snapshot work as they should. Thank you very much everybody and thank you @svanheule and @luizluca for writing the Realtek driver.

closed 2023-01-27 by joe2xyz

Neustradamus, 2024-12-22:
Good news, there is a new merged PR linked to Zyxel GS1900:
- https://github.com/openwrt/openwrt/pull/16442