openwrt/openwrt #11730: Firmware files and some kernel modules are missing from the x86-64-rootfs.tar.gz

laszlo-t, 2023-01-08:
### Describe the bug

The /lib/firmware directory is empty in [rootfs.tar.gz](https://downloads.openwrt.org/releases/22.03.3/targets/x86/64/openwrt-22.03.3-x86-64-rootfs.tar.gz) and it contains lesser kernel modules than the [generic-ext4-combined-efi.img.gz](https://downloads.openwrt.org/releases/22.03.3/targets/x86/64/openwrt-22.03.3-x86-64-generic-ext4-combined-efi.img.gz)

### OpenWrt version

r20028-43d71ad93e

### OpenWrt target/subtarget

x86/64

### Device

just a PC

### Image kind

Official downloaded image

### Actual behaviour

The rootfs is different between in the generic-ext4-combined-efi.img.gz and rootfs.tar.gz on x86 targets.

### Expected behaviour

I think the rootfs.tar.gz should contain the same files as the rootfs partition in the generic-ext4-combined-efi.img.gz on x86 targets.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-01-09:
You can make your own images content in new firmwaee selecor
https://firmware-selector.openwrt.org/?version=22.03.3&target=x86%2F64&id=generic

laszlo-t, 2023-01-11:
Wow, I didn't know that, but I could extracted the files from original image as well.

I opened this bug because I think it is misleading that same builds have different rootfs. I lost some hours after I upgraded to 22.03.3 with the above mentioned rootfs.tar.gz. I had to investigate and find the root cause why there is no network connection at all.

brada4, 2023-01-12:
Kernel typically loads modules for exactly same version.
[image]
```
ext-root/lib/modules> ls
5.10.161
ext-root/lib/modules> ls 5.10.161/
amd-xgbe.ko        e1000.ko      igb.ko        mii.ko                 nf_flow_table_ipv4.ko  nf_log_ipv6.ko     nf_tables.ko      nft_fib_ipv4.ko      nft_limit.ko   nft_objref.ko       nft_reject_ipv6.ko  ppp_async.ko    ptp.ko      vfat.ko
bnx2.ko            ena.ko        igc.ko        nf_conntrack.ko        nf_flow_table_ipv6.ko  nf_nat.ko          nft_chain_nat.ko  nft_fib_ipv6.ko      nft_log.ko     nft_quota.ko        nft_reject.ko       ppp_generic.ko  r8169.ko
button-hotplug.ko  fat.ko        ixgbe.ko      nf_defrag_ipv4.ko      nf_flow_table.ko       nfnetlink.ko       nft_counter.ko    nft_fib.ko           nft_masq.ko    nft_redir.ko        nls_cp437.ko        pppoe.ko        realtek.ko
crc-ccitt.ko       forcedeth.ko  libcrc32c.ko  nf_defrag_ipv6.ko      nf_log_common.ko       nf_reject_ipv4.ko  nft_ct.ko         nft_flow_offload.ko  nft_nat.ko     nft_reject_inet.ko  nls_iso8859-1.ko    pppox.ko        slhc.ko
e1000e.ko          i2c-dev.ko    mdio.ko       nf_flow_table_inet.ko  nf_log_ipv4.ko         nf_reject_ipv6.ko  nft_fib_inet.ko   nft_hash.ko          nft_numgen.ko  nft_reject_ipv4.ko  nls_utf8.ko         pps_core.ko  
```

I would guess that you installed rootfs from different kernel. You need to do both at once, a bit painful lesson in the moment you lose network connectivity.
Do some practice in a virtual machine 21->22..0->22..3 
Normally one does full sysupgrade with next release, or attended upgrades that maintain previous packages moving them to squashfs aka auc.

laszlo-t, 2023-01-12:
I didn't forget updating the kernel just didn't want to go into details.

But here they are:
a GPT disk with 3 partitions: EFI, rootfs, data
MSI Z87I motherboard with Realtek networks

I installed the openwrt originally by the generic-ext4-combined-efi.img.gz. It contains every files that I needed and had network without any additional packages.

If you check the files here, you can see there is a generic-kernel.bin as well.
https://downloads.openwrt.org/releases/22.03.3/targets/x86/64/
So you could think you have everything for a successful upgrade (generic-kernel.bin, rootfs.tar.gz, except the bootloader).

I made some backup, booted from systemrescuecd, replaced the kernel with new one on EFI, deleted all files on rootfs  partition and copied the new files from the rootfs.tar.gz.
It should have worked but important files are missing from the rootfs.tar.gz (the Realtek modules for me). But they are in the generic-ext4-combined-efi.img.gz.

I still think the rootfs should be consistent in all type of x86 images.

brada4, 2023-01-12:
Hmmm... apparenly part of guide about extracting tarball is wrong, you need to dd rootfs-es instead.
This tarball seems like a fit for containers like docker, no drivers, external kernel, no real hardware support.

laszlo-t, 2023-01-12:
Yes, the referred [documentation](https://openwrt.org/docs/guide-user/installation/openwrt_x86) says this:

> **rootfs.tar.gz** This contains all the files from the root partition. It can be extracted onto a root filesystem without the need of overwriting the partition. To avoid conflicts, it is highly recommended you backup any older files and extract this file onto an empty filesystem.

Anyway I used this method to get all files:
```
losetup -P -f --show openwrt-22.03.3-x86-64-generic-ext4-combined-efi.img
mount /dev/loop0p1 /mnt/EFI/
mount /dev/loop0p2 /mnt/rootfs/
```

jiggol, 2023-05-02:
I also experienced the same issue as the author, and I believe that the contents of the img file should be identical.

brada4, 2023-05-02:
There should be 2 files
 - one tarball of contents of img per device.
 - and another for container without any kmod-s per architecture for containers.
I am quite happy to find 2nd image in 1st place, but I understand dissatisfaction w uneasy upgrade.