openwrt/openwrt #11736: 2.4ghz wireless interface can't start after after 22.03.1 upgrade MTC Wireless Router WR1201

F4HTB, 2023-01-08:
### Describe the bug

Hello dear All, 

Without exposing all these adventures, I wanted to point out that the 2.4ghz wifi interface no longer start after an upgrade to version 22.03.1 on my WR1201.

I have at home an internet box followed by a WR1201 (called R1) with interfaces in bridge mode and on the remote site the reception is done with a bullet M2 ubiquiti with relayd (called R2) and an ipv6 dhcp interface which itself is connected to a last WR1201 (called R3) in wifi access point mode with bridge interfaces for the internal wifi distribution of the remote site.

I had to migrate to 22 because from 21 because there seemed to be another bug on 21.
As said above, I use to make a wifi bridge an M2 bullet and a second WR1201 router on the remote site.
When I migrated my bullet M2 from the original firmware to openWRT (latest 19.07.9 which works), the lan interface on the Bullet or WR1201 side was constantly switching to blocking and forwarding then blocking etc... I tried the Bullet (R2) on a classic switch, it works.
After having migrated my WR1201 (called R2), in 22, the problem was solved but this time it is the wifi interface of the WR1201 in 2.4ghz which does not start any more. Same problem found on my first WR1201 (called R1).

I therefore concluded that there is a problem of negotiation on the ethernet ports of the WR1201 in 21 solved in 22.
I therefore concluded that there is a problem on the 2.4ghz wifi interface on the WR1201 in 22 since noted twice.
I have test all available 22.X version, alwais same bug.

Thank you for your efforts, if anyone has an idea, I'm interested.

I can provide any additional log if you ever tell me where to look.

Cordially,
PS: Thanks to the creators of openWRT.

WR1201 :

Model	MTC Wireless Router WR1201
Architecture	MediaTek MT7621 ver:1 eco:3
Target Platform	ramips/mt7621
Firmware Version	OpenWrt 22.03.1 r19777-2853b6d652 / LuCI openwrt-22.03 branch git-22.245.77528-487e58a
Kernel Version	5.10.146
Here is log from WR1201 :

**System Log**

Sun Jan  8 21:16:21 2023 kern.err kernel: [   57.999404] mt76x2e 0000:02:00.0: MCU message 02 (seq 11) timed out
Sun Jan  8 21:16:21 2023 kern.info kernel: [   58.180732] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
Sun Jan  8 21:16:21 2023 kern.info kernel: [   58.191725] mt76x2e 0000:02:00.0: Build: 1
Sun Jan  8 21:16:21 2023 kern.info kernel: [   58.200245] mt76x2e 0000:02:00.0: Build Time: 201607111443____
Sun Jan  8 21:16:21 2023 kern.info kernel: [   58.215283] mt76x2e 0000:02:00.0: Firmware running!
Sun Jan  8 21:16:21 2023 kern.info kernel: [   58.229705] ieee80211 phy1: Hardware restart was requested
Sun Jan  8 21:16:22 2023 kern.err kernel: [   59.279391] mt76x2e 0000:02:00.0: MCU message 02 (seq 14) timed out
Sun Jan  8 21:16:22 2023 kern.info kernel: [   59.390692] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
Sun Jan  8 21:16:22 2023 kern.info kernel: [   59.402334] mt76x2e 0000:02:00.0: Build: 1
Sun Jan  8 21:16:22 2023 kern.info kernel: [   59.410706] mt76x2e 0000:02:00.0: Build Time: 201607111443____
Sun Jan  8 21:16:22 2023 kern.info kernel: [   59.425905] mt76x2e 0000:02:00.0: Firmware running!
Sun Jan  8 21:16:22 2023 kern.info kernel: [   59.439745] ieee80211 phy1: Hardware restart was requested
Sun Jan  8 21:16:23 2023 kern.err kernel: [   60.479508] mt76x2e 0000:02:00.0: MCU message 02 (seq 2) timed out
Sun Jan  8 21:16:23 2023 kern.info kernel: [   60.611017] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
Sun Jan  8 21:16:23 2023 kern.info kernel: [   60.622128] mt76x2e 0000:02:00.0: Build: 1
Sun Jan  8 21:16:23 2023 kern.info kernel: [   60.630377] mt76x2e 0000:02:00.0: Build Time: 201607111443____
Sun Jan  8 21:16:23 2023 kern.info kernel: [   60.645490] mt76x2e 0000:02:00.0: Firmware running!
Sun Jan  8 21:16:23 2023 kern.info kernel: [   60.659721] ieee80211 phy1: Hardware restart was requested
Sun Jan  8 21:16:24 2023 kern.err kernel: [   61.759738] mt76x2e 0000:02:00.0: MCU message 02 (seq 5) timed out
Sun Jan  8 21:16:25 2023 kern.info kernel: [   61.930730] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
Sun Jan  8 21:16:25 2023 kern.info kernel: [   61.942111] mt76x2e 0000:02:00.0: Build: 1
Sun Jan  8 21:16:25 2023 kern.info kernel: [   61.950369] mt76x2e 0000:02:00.0: Build Time: 201607111443____
Sun Jan  8 21:16:25 2023 kern.info kernel: [   61.965605] mt76x2e 0000:02:00.0: Firmware running!
Sun Jan  8 21:16:25 2023 kern.info kernel: [   61.979785] ieee80211 phy1: Hardware restart was requested
Sun Jan  8 21:16:26 2023 kern.err kernel: [   63.039395] mt76x2e 0000:02:00.0: MCU message 02 (seq 8) timed out
Sun Jan  8 21:16:26 2023 kern.info kernel: [   63.141014] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
Sun Jan  8 21:16:26 2023 kern.info kernel: [   63.152086] mt76x2e 0000:02:00.0: Build: 1
Sun Jan  8 21:16:26 2023 kern.info kernel: [   63.160344] mt76x2e 0000:02:00.0: Build Time: 201607111443____
Sun Jan  8 21:16:26 2023 kern.info kernel: [   63.175406] mt76x2e 0000:02:00.0: Firmware running!
Sun Jan  8 21:16:26 2023 kern.info kernel: [   63.189716] ieee80211 phy1: Hardware restart was requested

**Kernel Log**

[  176.639140] ieee80211 phy1: Hardware restart was requested
[  177.678915] mt76x2e 0000:02:00.0: MCU message 02 (seq 2) timed out
[  177.800187] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[  177.811168] mt76x2e 0000:02:00.0: Build: 1
[  177.820093] mt76x2e 0000:02:00.0: Build Time: 201607111443____
[  177.835154] mt76x2e 0000:02:00.0: Firmware running!
[  177.849111] ieee80211 phy1: Hardware restart was requested
[  178.878912] mt76x2e 0000:02:00.0: MCU message 02 (seq 5) timed out
[  179.010211] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[  179.021188] mt76x2e 0000:02:00.0: Build: 1
[  179.030109] mt76x2e 0000:02:00.0: Build Time: 201607111443____
[  179.045164] mt76x2e 0000:02:00.0: Firmware running!
[  179.059113] ieee80211 phy1: Hardware restart was requested
[  180.158955] mt76x2e 0000:02:00.0: MCU message 02 (seq 8) timed out
[  180.330225] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[  180.341209] mt76x2e 0000:02:00.0: Build: 1
[  180.350140] mt76x2e 0000:02:00.0: Build Time: 201607111443____
[  180.365196] mt76x2e 0000:02:00.0: Firmware running!
[  180.379109] ieee80211 phy1: Hardware restart was requested
[  181.438917] mt76x2e 0000:02:00.0: MCU message 02 (seq 11) timed out
[  181.540211] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[  181.551208] mt76x2e 0000:02:00.0: Build: 1
[  181.560136] mt76x2e 0000:02:00.0: Build Time: 201607111443____
[  181.575194] mt76x2e 0000:02:00.0: Firmware running!
[  181.589096] ieee80211 phy1: Hardware restart was requested
[  182.638911] mt76x2e 0000:02:00.0: MCU message 02 (seq 14) timed out
[  182.750237] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[  182.761263] mt76x2e 0000:02:00.0: Build: 1
[  182.770180] mt76x2e 0000:02:00.0: Build Time: 201607111443____
[  182.785240] mt76x2e 0000:02:00.0: Firmware running!
[  182.799095] ieee80211 phy1: Hardware restart was requested
[  183.838915] mt76x2e 0000:02:00.0: MCU message 02 (seq 2) timed out
[  183.960194] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[  183.971170] mt76x2e 0000:02:00.0: Build: 1
[  183.980114] mt76x2e 0000:02:00.0: Build Time: 201607111443____
[  183.995201] mt76x2e 0000:02:00.0: Firmware running!
[  184.009099] ieee80211 phy1: Hardware restart was requested
[  185.038901] mt76x2e 0000:02:00.0: MCU message 02 (seq 5) timed out
[  185.170208] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[  185.181191] mt76x2e 0000:02:00.0: Build: 1
[  185.190132] mt76x2e 0000:02:00.0: Build Time: 201607111443____
[  185.205222] mt76x2e 0000:02:00.0: Firmware running!
[  185.219095] ieee80211 phy1: Hardware restart was requested
[  186.318900] mt76x2e 0000:02:00.0: MCU message 02 (seq 8) timed out
[  186.490193] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[  186.501170] mt76x2e 0000:02:00.0: Build: 1
[  186.510118] mt76x2e 0000:02:00.0: Build Time: 201607111443____
[  186.525176] mt76x2e 0000:02:00.0: Firmware running!
[  186.539120] ieee80211 phy1: Hardware restart was requested
[  187.598904] mt76x2e 0000:02:00.0: MCU message 02 (seq 11) timed out
[  187.700200] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[  187.711183] mt76x2e 0000:02:00.0: Build: 1
[  187.720130] mt76x2e 0000:02:00.0: Build Time: 201607111443____
[  187.735188] mt76x2e 0000:02:00.0: Firmware running!
[  187.749090] ieee80211 phy1: Hardware restart was requested

### OpenWrt version

r19777-2853b6d652, r16688-fa9a932fdb

### OpenWrt target/subtarget

ramips/mt7621

### Device

MTC Wireless Router WR1201

### Image kind

Official downloaded image

### Steps to reproduce

Migrate MTC Wireless Router WR1201 from 21.x to 22.x

### Actual behaviour

Wifi interface 2.4ghz dont start.

### Expected behaviour

That this interface work as normal

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

F4HTB, 2023-01-08:
cat /etc/config/wireless:
config wifi-device 'radio0'
        option type 'mac80211'
        option hwmode '11a'
        option path '1e140000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
        option htmode 'VHT80'
        option channel '40'
        option cell_density '0'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'F4HTB'
        option encryption 'psk-mixed'
        option key 'cce6f556c44a13c4c3f92dee67'
        option disassoc_low_ack '0'

config wifi-device 'radio1'
        option type 'mac80211'
        option channel '11'
        option hwmode '11g'
        option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0'
        option htmode 'HT40'
        option cell_density '0'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option network 'lan'
        option mode 'ap'
        option ssid 'F4HTB'
        option encryption 'psk-mixed'
        option key 'cce6f556c44a13c4c3f92dee67'
        option disassoc_low_ack '0'

DruvsEgo, 2023-01-08:
Turn off and on from electricity or
do a reset and set up again, sometimes there are errors when saving or restoring settings.

DragonBluep, 2023-01-09:
There is a conflict between `option hwmode '11g'` and `option htmode 'HT40'`. Please try to reset and reconfigure it in LuCI.

F4HTB, 2023-01-09:
Dear all, Thanks for your replies.
I hav try a full reset and active the wireless interface.
Nothing nothing worked in your proposals.
After the reset it use this configuration:

config wifi-device 'radio0'
        option type 'mac80211'
        option path '1e140000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
        option channel '36'
        option band '5g'
        option htmode 'VHT80'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'none'

config wifi-device 'radio1'
        option type 'mac80211'
        option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0'
        option channel '1'
        option band '2g'
        option htmode 'HT20'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'none'

But does not work. I can't see OpenWrt in 2.4ghz and alwais same report in log:
Sun Jan 8 21:16:23 2023 kern.err kernel: [ 60.479508] mt76x2e 0000:02:00.0: MCU message 02 (seq 2) timed out

I think 0000:02:00.0 is option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0'

Maybe another idea?

Many thanks for the help

khanjui, 2023-01-09:
Latest snapshot should work fine with that chipset.

F4HTB, 2023-01-10:
Thanks for reply.

I will try the snapshot asap and give you a repport, but for the moment i need something stable.
Do you think that I should write another ticket for the BULETT M2 about the lan interface problem in version after 19.7?

rwalli, 2023-02-02:
> Latest snapshot should work fine with that chipset.

I have a similar problem as @F4HTB but my router will start and lose 2.4ghz wireless with the same log entries after a few days.

what's your uptime with the snapshot?

rwalli, 2023-03-13:
> only reboot helps to start

reboot did not help for me, only disconnecting power helps.

DragonBluep, 2023-03-13:
Recently I suffered the same problem on a device. This is because the reset pin of PCIe NIC is not properly controlled. You can try to find it.
ref: https://github.com/openwrt/openwrt/commit/6f2c95f0cf0519631adb5100ab71f908a2968cca

DragonBluep, 2023-03-13:
1. Try to boot your device until wireless is turned on.
2. Run this script to see which GPIO crash your wireless:
```
#!/bin/sh
GPIOCHIP=480
BASE=$(cat /sys/class/gpio/gpiochip${GPIOCHIP}/base)
NGPIO=$(cat /sys/class/gpio/gpiochip${GPIOCHIP}/ngpio)
max=$(($BASE+$NGPIO))
gpio=$BASE
while [ $gpio -lt $max ] ; do
	echo $gpio > /sys/class/gpio/export
	[ -d /sys/class/gpio/gpio${gpio} ] && {
		echo out > /sys/class/gpio/gpio$gpio/direction
 
		echo "[GPIO$gpio] Trying value 0"
		echo 0 > /sys/class/gpio/gpio$gpio/value
		sleep 5s

		echo $gpio > /sys/class/gpio/unexport
	}
	gpio=$((gpio+1))
done
```
3. If the script doesn't work, try to change the first line `GPIOCHIP=448` or `GPIOCHIP=416` and run again.

DragonBluep, 2023-10-24:
Please test this one. [openwrt-mt76x2.tar.xz.zip](https://github.com/openwrt/mt76/files/13116719/openwrt-mt76x2.tar.xz.zip)

frollic, 2023-12-08:
upgraded to 23.05.2, still working.

vlouis, 2024-06-29:
Hi, I have this MTC Wireless Router WR1201 router (Strong 1200) and got the same "doesn't detect the 2.4GHz interface" issue. The GPIO reset issue could be the root cause, I seen on some similar routers was resolved by patching the ".dts" file. Could lead me somebody how to identify the "reset gpio" for this device?

vlouis, 2024-06-29:
Using the @DragonBluep "openwrt-mt76x2.tar.xz.zip" and his bash script, I didn't found the gpio what reset the wireless chips. I ran the script several times, increased the delay to 10s... didn't seen at my phone to disconnect from wifi.
root@OpenWrt:/tmp# ./gpiotest.sh
[GPIO480] Trying value 0
[GPIO481] Trying value 0
[GPIO482] Trying value 0
[GPIO483] Trying value 0
[GPIO484] Trying value 0
[GPIO485] Trying value 0
[GPIO486] Trying value 0
[GPIO487] Trying value 0
sh: write error: Resource busy
[GPIO489] Trying value 0
[GPIO490] Trying value 0
[GPIO491] Trying value 0
[GPIO492] Trying value 0
[GPIO493] Trying value 0
[GPIO494] Trying value 0
[GPIO495] Trying value 0
[GPIO496] Trying value 0
[GPIO497] Trying value 0
[GPIO498] Trying value 0
sh: write error: Resource busy
[GPIO500] Trying value 0
[GPIO501] Trying value 0
sh: write error: Resource busy
sh: write error: Resource busy
sh: write error: Resource busy
sh: write error: Resource busy
sh: write error: Resource busy
[GPIO507] Trying value 0
[GPIO508] Trying value 0
[GPIO509] Trying value 0
[GPIO510] Trying value 0
[GPIO511] Trying value 0

root@OpenWrt:/tmp# ./gpiotest.sh
[GPIO448] Trying value 0
[GPIO449] Trying value 0
[GPIO450] Trying value 0
[GPIO451] Trying value 0
[GPIO452] Trying value 0
[GPIO453] Trying value 0
[GPIO454] Trying value 0
[GPIO455] Trying value 0
[GPIO456] Trying value 0
[GPIO457] Trying value 0
[GPIO458] Trying value 0
[GPIO459] Trying value 0
[GPIO460] Trying value 0
[GPIO461] Trying value 0
[GPIO462] Trying value 0
[GPIO463] Trying value 0
[GPIO464] Trying value 0
[GPIO465] Trying value 0
[GPIO466] Trying value 0
[GPIO467] Trying value 0
[GPIO468] Trying value 0
[GPIO469] Trying value 0
[GPIO470] Trying value 0
[GPIO471] Trying value 0
[GPIO472] Trying value 0
[GPIO473] Trying value 0
[GPIO474] Trying value 0
[GPIO475] Trying value 0
[GPIO476] Trying value 0
[GPIO477] Trying value 0
[GPIO478] Trying value 0
[GPIO479] Trying value 0

root@OpenWrt:/tmp# ./gpiotest.sh
[GPIO416] Trying value 0
[GPIO417] Trying value 0
[GPIO418] Trying value 0
[GPIO419] Trying value 0
[GPIO420] Trying value 0
[GPIO421] Trying value 0
[GPIO422] Trying value 0
[GPIO423] Trying value 0
[GPIO424] Trying value 0
[GPIO425] Trying value 0
[GPIO426] Trying value 0
[GPIO427] Trying value 0
[GPIO428] Trying value 0
[GPIO429] Trying value 0
[GPIO430] Trying value 0
[GPIO431] Trying value 0
[GPIO432] Trying value 0
[GPIO433] Trying value 0
[GPIO434] Trying value 0
[GPIO435] Trying value 0
[GPIO436] Trying value 0
[GPIO437] Trying value 0
[GPIO438] Trying value 0
[GPIO439] Trying value 0
[GPIO440] Trying value 0
[GPIO441] Trying value 0
[GPIO442] Trying value 0
[GPIO443] Trying value 0
[GPIO444] Trying value 0
[GPIO445] Trying value 0
[GPIO446] Trying value 0
[GPIO447] Trying value 0
Could be this board to don't use a gpio line to wireless chips reset pin?

DragonBluep, 2024-06-29:
@vlouis So 2.4G works correctly now?

vlouis, 2024-06-30:
@DragonBluep Is better with your version. The strange is, this router worked well for several months with openwrt, started recently to appear this 2.4 wireless bug. Even after several restart remains without the 2.4 part. Now, with your openwrt, after restart sometimes didn't worked (played with settings), but usually was enough a single restart to came back again.
If I could have a question... that script, how should work? The "sh: write error: Resource busy" means that gpio is already used for something? If you could enlighten me what I should do, I could try to build and test few things, probably isn't much harder to build it than an AOSP rom. If I'm not wrong, the gpio lines that command the wireless chips reset, should be included in the "dts" file... just need to find it that/those gpio's.

frollic, 2024-06-30:
I still can't reproduce it, it works just fine with the 23.05.2 release.

I have 10+ 2.4GHz IoT devices connected through it.

DragonBluep, 2024-06-30:
> If I could have a question... that script, how should work?

See wiki page https://openwrt.org/docs/guide-developer/add.new.device.

> The "sh: write error: Resource busy" means that gpio is already used for something?

yes.

> If I'm not wrong, the gpio lines that command the wireless chips reset, should be included in the "dts" file... just need to find it that/those gpio's.

Yes. You can have a look on `target/linux/ramips/dts/mt7621_d-team_pbr-m1.dts`. Usually, pcie reset gpios are GPIO7, GPIO8 and GPIO19.

vlouis, 2024-07-23:
Hi, I didn't give up, just lack of time... 
I needed some time to figure out how to build openwrt FW for my router and to prepare few versions... 
So, I used the new 23.05.4 sources and built few versions:
Without any modification - at the start, and after few soft/hard reset had issue with wifi (both with 5GHz and 2.4). Tried with the official 23.05.4 build too, same things.

With the _reset-gpios = <&gpio 8 GPIO_ACTIVE_LOW>, <&gpio 19 GPIO_ACTIVE_LOW>;_ (and nothing else) added to the dts file - still wifi issues. If I run the "cat /sys/kernel/debug/gpio" appears the new _gpio-488 (                    |reset               ) out hi ACTIVE LOW_ - (means the gpio-488 is the "gpio 8" and gpio-499 is the "gpio 19" - the gpio 19 (499) is always used for reset?). Almost always the 2.4 wifi worked, the 5GHz wifi detected (MediaTek MT76x2E 802.11ac/n) but not functional.

With the _groups = "i2c", "uart3", "uart2", "jtag", "wdt", "rgmii2", "sdhci";_  in the mt7621_mtc_wr1201.dts - when I ran the gpio test script, at the gpio 488 the router halted then rebooted...

I read something about the longer pci reset, I found that already present (set to 100ms) in the _/target/linux/ramips/patches-5.15/108-v6.3-PCI-mt7621-Delay-phy-ports-initialization_, tried a build with a delay increased to 500ms, without any improvement.

I start to think my router hardware to have something, but in this case why it's working almost perfect with DragonBluep test build? _What the special magic were added DragonBluep to that test build..._ :-)

Next... I want to try the same things with latest snapshot... or any advice welcomed...

DragonBluep, 2024-07-23:
@vlouis I don't remember what changes I‘ve made on the test image, but it's obvious GPIO 8 is one of the reset GPIOs from your test result. So please try this patch and run the test script again to ensure we find all reset pins.
```
--- a/target/linux/ramips/dts/mt7621_mtc_wr1201.dts
+++ b/target/linux/ramips/dts/mt7621_mtc_wr1201.dts
@@ -166,8 +166,18 @@
 	status = "okay";
 };
 
+&pcie_pins {
+	pcie {
+		groups = "uart3", "pcie";
+		function = "gpio";
+	};
+};
+
 &pcie {
 	status = "okay";
+
+	reset-gpios = <&gpio 8 GPIO_ACTIVE_LOW>,
+		      <&gpio 19 GPIO_ACTIVE_LOW>;
 };
 
 &pcie0 {
@@ -200,7 +210,7 @@
 
 &state_default {
 	gpio {
-		groups = "rgmii2";
+		groups = "i2c", "uart2", "jtag", "wdt", "rgmii2", "sdhci";
 		function = "gpio";
 	};
 };
```

vlouis, 2024-07-26:
@DragonBluep Thank you for your support! The "magic", I think, is the using the uart3 as GPIO (MT7621's "UART3_MODE" register), using for pcie reset the GPIO 8 and 19, and the snapshot image.
I built two images, 23.05.4 and a snapshot. At the snapshot, the range of GPIO's is shifted to 512-543:
_root@OpenWrt:~# cat /sys/kernel/debug/gpio
gpiochip0: GPIOs 512-543, parent: platform/1e000600.gpio, 1e000600.gpio-bank0:
 gpio-520 (                    |reset               ) out hi ACTIVE LOW
 gpio-531 (                    |reset               ) out hi ACTIVE LOW
 gpio-534 (                    |green:usb           ) out hi ACTIVE LOW
 gpio-535 (                    |green:wps           ) out hi ACTIVE LOW
 gpio-536 (                    |green:power         ) out lo ACTIVE LOW
 gpio-537 (                    |reset               ) in  hi IRQ ACTIVE LOW
 gpio-538 (                    |green:eth_link      ) out lo ACTIVE LOW
gpiochip1: GPIOs 544-575, parent: platform/1e000600.gpio, 1e000600.gpio-bank1:
gpiochip2: GPIOs 576-607, parent: platform/1e000600.gpio, 1e000600.gpio-bank2:_

I think no other reset GPIO found (I didn't noticed any wifi or router restart):
_root@OpenWrt:~# /tmp/gpiotest.sh
[GPIO512] Trying value 0
[GPIO513] Trying value 0
[GPIO514] Trying value 0
[GPIO515] Trying value 0
[GPIO516] Trying value 0
[GPIO517] Trying value 0
[GPIO518] Trying value 0
[GPIO519] Trying value 0
sh: write error: Resource busy
[GPIO521] Trying value 0
[GPIO522] Trying value 0
[GPIO523] Trying value 0
[GPIO524] Trying value 0
[GPIO525] Trying value 0
[GPIO526] Trying value 0
[GPIO527] Trying value 0
[GPIO528] Trying value 0
[GPIO529] Trying value 0
[GPIO530] Trying value 0
sh: write error: Resource busy
[GPIO532] Trying value 0
[GPIO533] Trying value 0
sh: write error: Resource busy
sh: write error: Resource busy
sh: write error: Resource busy
sh: write error: Resource busy
sh: write error: Resource busy
[GPIO539] Trying value 0
[GPIO540] Trying value 0
[GPIO541] Trying value 0
[GPIO542] Trying value 0
[GPIO543] Trying value 0_ 
With the 23.05.4, the GPIO range is 480-511, the result is the same.
About the wifi stability, I tried to soft restart few times, the snapshot looks like more stable. With the 23.05.4, after first restart got the "Generic unknown" for the 2.4GHz wifi, came back after another restart... But with the snapshot, restarted 4 times without issue. I built a snapshot image without the _groups = "i2c", "uart2", "jtag", "wdt", "rgmii2", "sdhci";_ - for now is working very well, I will see it if after few weeks remain same stable.

DragonBluep, 2024-07-26:
Please edit your script and also try gpiochip1: GPIOs 544-575 in snapshot version. Some devices have 3 or 4 reset pins.

vlouis, 2024-07-28:
Hi @DragonBluep - I tested the remains GPIO ranges, I think isn't exist other gpio-reset pin:
_root@OpenWrt:~# /tmp/gpiotest.sh
[GPIO544] Trying value 0
[GPIO545] Trying value 0
[GPIO546] Trying value 0
[GPIO547] Trying value 0
[GPIO548] Trying value 0
[GPIO549] Trying value 0
[GPIO550] Trying value 0
[GPIO551] Trying value 0
[GPIO552] Trying value 0
[GPIO553] Trying value 0
[GPIO554] Trying value 0
[GPIO555] Trying value 0
[GPIO556] Trying value 0
[GPIO557] Trying value 0
[GPIO558] Trying value 0
[GPIO559] Trying value 0
[GPIO560] Trying value 0
[GPIO561] Trying value 0
[GPIO562] Trying value 0
[GPIO563] Trying value 0
[GPIO564] Trying value 0
[GPIO565] Trying value 0
[GPIO566] Trying value 0
[GPIO567] Trying value 0
[GPIO568] Trying value 0
[GPIO569] Trying value 0
[GPIO570] Trying value 0
[GPIO571] Trying value 0
[GPIO572] Trying value 0
[GPIO573] Trying value 0
[GPIO574] Trying value 0
[GPIO575] Trying value 0_
The exactly same with the range 576-607. I will continue testing the my snapshot image (with GPIO 8 & 19 as pcie reset gpio) and will come back if any issue will appear. 
Thank you again for your assistance!

closed 2024-07-28 by openwrt-bot

F4HTB, 2024-09-17:
Hello all, 
Can someone provide the stable version OpenWrt 23.05.4 With this modification?
If it's possible many thanks in advance.