openwrt/openwrt #11742: br-lan acts as a hub not a switch on ea4500 on v22.03.3

dandudikof, 2023-01-09:
### Describe the bug

After a hard upgrade from 21.02.5
Was unable to flash 22.03.2 nor 22.03.3 nor factory linksys firmware(3 attempts each).
from a primary active partition to secondary (med flashing lights no full boot),
only when switched to secondary active 
and flashed to primary did it work.

But unlike in 21.02.5 where br-lan acts as a switch
in 22.03.3  br-lan seems to act as a hub or mirrored ports  
Where every lan port can see what other ports are doing 
(default config after resset)

tcpdump on a port echos all the trafic from other ports.

### OpenWrt version

r20028-43d71ad93e

### OpenWrt target/subtarget

kirkwood/generic

### Device

Linksys EA4500 (Viper)

### Image kind

Official downloaded image

### Steps to reproduce

tcpdump any lan port and see trafic from other ports 

### Expected behaviour

expect to only see broadcast when tcpdumping lan ports on the switch

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

mrkiko, 2023-01-09:
Just to be sure - you're executing tcpdump from outside the device, on a device connected to to one of the ports involved in br-lan bridges, right?
Thanks!

dandudikof, 2023-01-09:
yes , correct

champtar, 2023-01-09:
See https://github.com/openwrt/openwrt/commit/a0bae2fef8bdd4f76767e1b29deb1adf279403e9, the switch is know broken in 22.03 (with default configuration at least, might work with vlan), your device (https://openwrt.org/toh/linksys/ea4500) should have been disabled but was missed