openwrt/openwrt #11753: /sbin/validate_data fails silently if option or list keyword is missing

tcuje, 2023-01-10:
### Describe the bug

While configuring my system, I missed adding the `option` keyword in front of one config entry. Afterwards all `/sbin/validate_data` tries for the config file returned error code 255, but did not provide output or otherwise hint what kind of error it is. All other (correct) options in the config file were not evaluated either anymore.

I think this issue may happen to a lot of people not used to working with these config files and is a common typo even for experienced users. Even worse is that the provided error code without any output does not help in finding the cause for the issue.

### OpenWrt version

r19803-9a599fee93

### OpenWrt target/subtarget

rockchip/armv8

### Device

FriendlyElec NanoPi R2S

### Image kind

Official downloaded image

### Steps to reproduce

Create an invalid config file with a value that misses the `option` and `list` keywords, like:

```echo -e "config testsection 'test'\n\tmissing_option_keyword\t\t'some-value\n\toption second_option\t\t'yes''" >> /etc/config/some-testpkg```

which creates

```
config testsection 'test'
	missing_option_keyword		'some-value'
	option second_option		'yes'
```

Afterwards run validation check for this file and values like:
```
/sbin/validate_data some-testpkg testsection test 'missing_option_keyword:string' 'second_option:string'
```

Even the second correct entry is not evaluated and no output given either.

Having the broken value entry in another section leads to the same experience.

### Actual behaviour

validate_data returns code 255 and provides no output at all

### Expected behaviour

validate_data states that the validation failed for config option 'missing_option_keyword' and provides output with validation results for all other options

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Spudz76, 2023-01-10:
seems like exit code 255 means corrupt file, which is what it is when you break format rules like forgetting a keyword

probably the way it parses it doesn't know where, only that it can't be parsed (I assume it asks uci for the sections, so then uci would need to know how to point to errors and also hand that back to validate_data, somehow)

Spudz76, 2023-01-10:
also you should be using uci anyway not blasting files directly with corrupt content...

jow-, 2023-01-10:
The `validate_data` applet is meant for semantic validation after uci parsing. It does not and cannot check the correctness of the uci syntax itself since such error details are not exposed by the libuci api. Also loading a uci file is all-or-nothing, there is no way for `validate_data` to obtain partial parsing results to validate.

Use `uci -q show some-testpkg && validate-data some-testpkg testsection test ...` to first check syntactic and then semantic correctness.

closed 2023-01-10 by jow-