openwrt/openwrt #11756: ramips/mt7621: panic when reading /sys/kernel/debug/ppe0/entries with 5.15 kernel

stintel, 2023-01-10:
### Describe the bug

Reading /sys/kernel/debug/ppe0/entries causes a kernel panic on ramips.

### OpenWrt version

r21740-0cc1c302b1

### OpenWrt target/subtarget

ramips/mt7621

### Device

TP-Link EAP615-Wall v1

### Image kind

Self-built image

### Steps to reproduce

Build and flash ramips/mt7621 image with CONFIG_TESTING_KERNEL=y
cat /sys/kernel/debug/ppe0/entries

### Actual behaviour

```
[70446.485985] CPU 0 Unable to handle kernel paging request at virtual address 00000000, epc == 804e0ed4, ra == 804e0ea4
[70446.496766] Oops[#1]:
[70446.499068] CPU: 0 PID: 5451 Comm: cat Not tainted 5.15.86 #0
[70446.504814] $ 0   : 00000000 00000001 a1a03ccc 00000000
[70446.510048] $ 4   : 8f266d8b 00000003 00000000 00000000
[70446.515271] $ 8   : 00000001 00000001 00000000 8201fb9d
[70446.520490] $12   : a1a03cb8 84c1e061 84c1f000 fffffffa
[70446.525714] $16   : a1a03ca0 000000c2 00000001 81968080
[70446.530934] $20   : 8200dea0 8082e1c0 00000000 00000000
[70446.536156] $24   : 00000002 00000020
[70446.541376] $28   : 8201e000 8201fcd8 8082e228 804e0ea4
[70446.546598] Hi    : 00000005
[70446.549461] Lo    : 0a3d70b8
[70446.552324] epc   : 804e0ed4 mtk_ppe_debugfs_foe_show.constprop.0+0x1d4/0x388
[70446.559460] ra    : 804e0ea4 mtk_ppe_debugfs_foe_show.constprop.0+0x1a4/0x388
[70446.566567] Status: 11000403 KERNEL EXL IE
[70446.570745] Cause : 40800008 (ExcCode 02)
[70446.574735] BadVA : 00000000
[70446.577597] PrId  : 0001992f (MIPS 1004Kc)
[70446.581671] Modules linked in: mt7915e mt76_connac_lib mt76 mac80211 cfg80211 sch_cake hwmon compat cls_bpf act_bpf sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact ifb netlink_diag sha256_generic libsha256 seqiv jitterentropy_rng drbg hmac cmac leds_gpio ramoops reed_solomon pstore gpio_button_hotplug crc32c_generic
[70446.618105] Process cat (pid: 5451, threadinfo=ab181127, task=9f6bd6fe, tls=77e8ddf0)
[70446.625913] Stack : 83ad7540 00000000 000001f8 ffffff05 8082e1f0 00000000 00000000 81405f68
[70446.634271]         814416f8 801f7f08 81405f68 8201fd04 8201fd04 8f266d8b 8201fdf0 80830000
[70446.642627]         80830000 8082e1c0 807d49c8 808a0000 a1a03cb4 a1a03cb8 a1a03cbe a1a03cbc
[70446.650984]         00000000 8201fec0 82010000 8201fdf0 00002000 8f266d8b 8200dea0 00000001
[70446.659340]         7f850244 8201fdf8 81405f68 8201fde0 00000000 7ffff001 8200deb8 8021a304
[70446.667693]         ...
[70446.670135] Call Trace:
[70446.672566] [<804e0ed4>] mtk_ppe_debugfs_foe_show.constprop.0+0x1d4/0x388
[70446.679337] [<8021a304>] seq_read_iter+0x1c0/0x5cc
[70446.684118] [<8021a824>] seq_read+0x114/0x188
[70446.688460] [<802dd008>] full_proxy_read+0x74/0xec
[70446.693238] [<801e64f4>] vfs_read+0xb8/0x34c
[70446.697497] [<801e6ce4>] ksys_read+0x80/0x134
[70446.701837] [<80014550>] syscall_common+0x34/0x58
[70446.706537]
[70446.708016] Code: 9447000a  7c0630a0  7c0318a0 <8ec40000> 8ec50004  00263402  00231c02  7c0738a0  afa60064
[70446.717765]
[70446.719561] ---[ end trace 5c7b2140c1c4b43c ]---
[70446.724311] Kernel panic - not syncing: Fatal exception
[70446.729583] Rebooting in 3 seconds..
```
[... 212 lines cut ...]
# CONFIG_STRACE_NONE is not set
CONFIG_TESTING_KERNEL=y
CONFIG_USE_LLVM_HOST=y
# CONFIG_WOLFSSL_HAS_CERTGEN is not set
# CONFIG_WOLFSSL_HAS_CHACHA_POLY is not set
# CONFIG_WOLFSSL_HAS_ECC25519 is not set
# CONFIG_WOLFSSL_HAS_OPENVPN is not set
# CONFIG_WOLFSSL_HAS_TLSV10 is not set
# CONFIG_WOLFSSL_HAS_TLSV13 is not set
CONFIG_WPA_MSG_MIN_PRIORITY=0
# CONFIG_WPA_WOLFSSL is not set
CONFIG_ZLIB_OPTIMIZE_SPEED=y
CONFIG_uhttpd_lua=y
CONFIG_uhttpd_ucode=y
# CONFIG_LLDPD_WITH_CDP is not set
# CONFIG_LLDPD_WITH_EDP is not set
# CONFIG_LLDPD_WITH_FDP is not set
# CONFIG_LLDPD_WITH_SONMP is not set
CONFIG_PACKAGE_getrandom=y
CONFIG_PACKAGE_kmod-crypto-crc32c=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

stintel, 2023-01-10:
The problem is most likely introduced by 9721a42a272aab941ded476a4baa4a85031c3f07.

Oozil, 2023-01-10:
> The problem is most likely introduced by [9721a42](https://github.com/openwrt/openwrt/commit/9721a42a272aab941ded476a4baa4a85031c3f07).

please i need help for compile ath10k-4.19 in snapshot source code

stintel, 2023-01-10:
@dangowrt can I have your feedback on fcec245d91ed5f469a02199ee0c16769ff2643f1? Not sure editing the patch in place is the right way to go. But I guess since it's still pending, it probably is.

DragonBluep, 2023-01-11:
Confirm that https://github.com/stintel/openwrt/commit/b055c62ca7791bd7d989fd68edff0ccc63e3d624 can fix it.
```
root@OpenWrt:~# cat /sys/kernel/debug/ppe0/entries
01996 UNB IPv4 5T orig=192.168.1.149:5512->192.168.1.1:22 new=0.0.0.0:0->0.0.0.0:0 eth=00:00:00:00:00:00->00:00:00:00:00:00 etype=0000 vlan=0,0 ib1=10000134 ib2=00000000
```

closed 2023-01-16 by openwrt-bot