openwrt/openwrt #11760: bananapi_bpi-r3 bash: 8: Bad file descriptor

nejtr0n, 2023-01-10:
### Describe the bug

Hello.
Could not build image with some custom packages. It seems to be misconfiguration of kernel or image size.
Error:
```
........
FIT description: ARM64 OpenWrt FIT (Flattened Image Tree)
Created:         Mon Jan  9 10:35:20 2023
 Image 0 (kernel-1)
  Description:  ARM64 OpenWrt Linux-5.15.86
  Created:      Mon Jan  9 10:35:20 2023
  Type:         Kernel Image
  Compression:  gzip compressed
  Data Size:    5368928 Bytes = 5243.09 KiB = 5.12 MiB
  Architecture: AArch64
  OS:           Linux
  Load Address: 0x44000000
  Entry Point:  0x44000000
  Hash algo:    crc32
  Hash value:   a490238b
  Hash algo:    sha1
  Hash value:   8f3f53269f5efef723dd71b3e1998e68cc285c21
 Image 1 (fdt-1)
  Description:  ARM64 OpenWrt bananapi_bpi-r3 device tree blob
  Created:      Mon Jan  9 10:35:20 2023
  Type:         Flat Device Tree
  Compression:  uncompressed
  Data Size:    32768 Bytes = 32.00 KiB = 0.03 MiB
  Architecture: AArch64
  Load Address: 0x43ff8000
  Hash algo:    crc32
  Hash value:   63127f08
  Hash algo:    sha1
  Hash value:   a00d33951f62b33677cc0b5172fa81be816d55e5
 Image 2 (fdt-mt7986a-bananapi-bpi-r3-nor)
  Description:  ARM64 OpenWrt bananapi_bpi-r3 device tree overlay mt7986a-bananapi-bpi-r3-nor
  Created:      Mon Jan  9 10:35:20 2023
  Type:         Flat Device Tree
  Compression:  uncompressed
  Data Size:    32768 Bytes = 32.00 KiB = 0.03 MiB
  Architecture: AArch64
  Load Address: 0x43ff0000
  Hash algo:    crc32
  Hash value:   95d48956
  Hash algo:    sha1
  Hash value:   1c93f277ffa3aecc5aef781cd3d6c72297d961d5
 Image 3 (fdt-mt7986a-bananapi-bpi-r3-emmc-nor)
  Description:  ARM64 OpenWrt bananapi_bpi-r3 device tree overlay mt7986a-bananapi-bpi-r3-emmc-nor
  Created:      Mon Jan  9 10:35:20 2023
  Type:         Flat Device Tree
  Compression:  uncompressed
  Data Size:    32768 Bytes = 32.00 KiB = 0.03 MiB
  Architecture: AArch64
  Load Address: 0x43fe8000
  Hash algo:    crc32
  Hash value:   9310ba5d
  Hash algo:    sha1
  Hash value:   b6e94c2825be3781ded07624307ce50d34ad99b0
 Image 4 (fdt-mt7986a-bananapi-bpi-r3-emmc-snand)
  Description:  ARM64 OpenWrt bananapi_bpi-r3 device tree overlay mt7986a-bananapi-bpi-r3-emmc-snand
  Created:      Mon Jan  9 10:35:20 2023
  Type:         Flat Device Tree
  Compression:  uncompressed
  Data Size:    32768 Bytes = 32.00 KiB = 0.03 MiB
  Architecture: AArch64
  Load Address: 0x43fe0000
  Hash algo:    crc32
  Hash value:   da7d197c
  Hash algo:    sha1
[... 439 lines cut ...]
CONFIG_PACKAGE_mc=y
CONFIG_PACKAGE_more=y
CONFIG_PACKAGE_nano=y
CONFIG_PACKAGE_qosify=m
CONFIG_PACKAGE_runc=y
CONFIG_PACKAGE_tc-tiny=m
CONFIG_PACKAGE_terminfo=y
CONFIG_PACKAGE_tini=y
CONFIG_PACKAGE_usb-modeswitch=y
CONFIG_PACKAGE_usbutils=y
CONFIG_PACKAGE_vim-full=y
CONFIG_PACKAGE_xdp-filter=m
CONFIG_PACKAGE_xdp-loader=m
CONFIG_PACKAGE_xdpdump=m
CONFIG_PACKAGE_xtables-nft=y
CONFIG_PACKAGE_zlib=y
CONFIG_REPRODUCIBLE_DEBUG_INFO=y
CONFIG_SDK=y
CONFIG_SDK_LLVM_BPF=y
CONFIG_USE_LLVM_BUILD=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

polskafan, 2023-01-10:
I experience the same issue when trying to build the sdcard image. I already looked at target/linux/mediatek/image/filogic.mk where the artifact gets defined, but couldn't quite figure it out.

https://github.com/openwrt/openwrt/blob/a96382c1bb204698cd43e82193877c10e4b63027/target/linux/mediatek/image/filogic.mk (initial support)
https://github.com/openwrt/openwrt/blob/master/target/linux/mediatek/image/filogic.mk (master)

Seems there were some changes and it now checks for some config settings.

My config has both settings enabled:
CONFIG_TARGET_ROOTFS_INITRAMFS=y
CONFIG_TARGET_ROOTFS_SQUASHFS=y

Not sure, where it goes wrong.

polskafan, 2023-01-10:
The image might be too big. "bash: 8: Bad file descriptor" seems to be a failed call to an error message in the check-size routine:
https://github.com/openwrt/openwrt/issues/11301#issuecomment-1321453060

So, if I get it right, the trick might be to build with CONFIG_TARGET_ROOTFS_INITRAMFS=n. According to https://github.com/openwrt/openwrt/commit/1a6f6a1e8cbc29042f63771d5f05bdbe622d9078 the image size can be bigger then.

nejtr0n, 2023-01-11:
It seems to my image compiled too with CONFIG_TARGET_ROOTFS_INITRAMFS=n.

I've tried to increase partitions, but no succeed
https://github.com/openwrt/openwrt/blob/master/target/linux/mediatek/image/filogic.mk
```
.....
				$(if $(CONFIG_TARGET_ROOTFS_INITRAMFS),\
				   pad-to 12M | append-image-stage initramfs-recovery.itb | check-size 64m |\
				) \
				   pad-to 64M | bl2 spim-nand-ddr4 |\
				   pad-to 65M | bl31-uboot bananapi_bpi-r3-snand |\
				   pad-to 69M | bl2 nor-ddr4 |\
				   pad-to 70M | bl31-uboot bananapi_bpi-r3-nor |\
				   pad-to 71M | bl2 emmc-ddr4 |\
				   pad-to 72M | bl31-uboot bananapi_bpi-r3-emmc |\
				   pad-to 76M | mt7986-gpt emmc |\
				$(if $(CONFIG_TARGET_ROOTFS_SQUASHFS),\
				   pad-to 84M | append-image squashfs-sysupgrade.itb | check-size | gzip \
				)
  IMAGE_SIZE := $$(shell expr 84 + $$(CONFIG_TARGET_ROOTFS_PARTSIZE))m
....
  ```

nejtr0n, 2023-01-11:
May be there are some other places, where we need to change initramfs size ....

polskafan, 2023-01-11:
Partition table is defined at line 19-39. Guess that needs to be adjusted too.

polskafan, 2023-01-11:
I looked at the image that is provided by Sinovoip. If I understand correctly, their kernel partition has a size of 64M:
https://github.com/BPI-SINOVOIP/BPI-R3-OPENWRT-V21.02.3/blob/e17fcb9e479b481222798dffe9ff186e3527db9c/target/linux/mediatek/image/make_bpi-r3_bundle_image.sh

So it should be possible to adjust the partition size.

nejtr0n, 2023-01-11:
I need Initramfs to boot from lvm in the future. 
And there is no way to burn image to emmc. 
Seems,  that initramfs is used in uboot too. 
https://github.com/openwrt/openwrt/blob/master/package/boot/uboot-mediatek/patches/430-add-bpi-r3.patch

From my point of view exclude it is not right way.
I think partion sizes should be calculated from compiled parts like initramfs dynamically.
But I need more info about boot process of bpir3

polskafan, 2023-01-12:
I successfully installed the resulting image without the initramfs using the SD card -> NAND -> eMMC way, that is described here:
https://forum.openwrt.org/t/how-to-flash-emmc-on-banana-pi-r3/137418/35

If you have access to the serial console, you can also do a TFTP boot from the bootloader. But yeah, expanding the partition would just make that issue go away completely.

polskafan, 2023-01-12:
I first tried this, until I realized the official openwrt doesn't build an image for NAND:
https://wiki.banana-pi.org/Getting_Started_with_BPI-R3

The command to erase flash is ```mtd erase /dev/mtd0```

So I had to stop after the NAND erase and follow the other set of instructions.

nejtr0n, 2023-01-14:
@polskafan , yeah, after erasing NAND everything seems to be working.

Flased fresh image to emmc with uart from boot menu.

Thank you very much bro!! I think we could close issue?

nejtr0n, 2023-01-14:
But all settings are wiped after reboot =)

root@OpenWrt:~# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                70.5M     70.5M         0 100% /rom
tmpfs                   997.9M    100.0K    997.8M   0% /tmp
tmpfs                   997.9M    224.0K    997.7M   0% /tmp/root
overlayfs:/tmp/root     997.9M    224.0K    997.7M   0% /
tmpfs                   512.0K         0    512.0K   0% /dev
overlayfs:/tmp/root     997.9M    224.0K    997.7M   0% /opt/docker

polskafan, 2023-01-14:
I have read that this can happen, when using the eMMC. But settings are saved for me. I build my image with the script from the OpenMPTCPRouter repository, maybe something is different there. So I don't know whats going on there.

I guess this issue could be closed for me, even though the "Bad file descriptor" message is clearly a bug.

tidalf, 2023-01-17:
it affects the bpi-r3 updates through auc or attendedsysupgrade too (for some days) : 
```
bash: line 1: 8: Bad file descriptor
gzip: /home/aparcar/asu/worker1/cache/SNAPSHOT/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-e8d66c515999-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz: No such file or directory
make[3]: *** [Makefile:31: /home/aparcar/asu/worker1/cache/SNAPSHOT/mediatek/filogic/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-e8d66c515999-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz] Error 1
make[2]: *** [Makefile:204: build_image] Error 2
make[1]: *** [Makefile:142: _call_image] Error 2
make: *** [Makefile:258: image] Error 2
Error: Error while building firmware. See stdout/stderr
Bad message (74)
```

nejtr0n, 2023-03-02:
I've noticed, that it is **dockerd** related issue.

When I include dockerd in firmware, root mounting on ram
```
root@OpenWrt:~# df -h
Filesystem Size Used Available Use% Mounted on
/dev/root 70.5M 70.5M 0 100% /rom
tmpfs 997.9M 100.0K 997.8M 0% /tmp
tmpfs 997.9M 224.0K 997.7M 0% /tmp/root
overlayfs:/tmp/root 997.9M 224.0K 997.7M 0% /
tmpfs 512.0K 0 512.0K 0% /dev
overlayfs:/tmp/root 997.9M 224.0K 997.7M 0% /opt/docker
```

So I compiled **dockerd** with module option, and install it after flashing openwrt to sdcard.
Everything seems to be good except docker data-dir (**/opt/docker**) have to be mounted on **ext4** partition.